wmbusmeters/wmbusmeters #2: Valid JSON

shirakabaprojects, 2018-10-05:
In the JSON output the json keys "media" and "meter" has been left unquoted by double quotation marks. Quotation of strings is required by the JSON standard and could be easily fixed by the following patch.

```
--- a/meter_multical21.cc
+++ b/meter_multical21.cc
@@ -453,7 +453,9 @@ void MeterMultical21::printMeterFields(FILE *output, char separator)
 
 void MeterMultical21::printMeterJSON(FILE *output)
 {
-    fprintf(output, "{media:\"%s\",meter:\"multical21\","
+    fprintf(output, "{"
+           QS(media,%s)
+           QS(meter,multical21)
            QS(name,%s)
            QS(id,%s)
            Q(total_m3,%f)
```

weetmuts, 2018-10-05:
You are right, thank you! I committed a fix to the three meter files and test.

weetmuts, 2018-10-16:
Fixed.

closed 2018-10-16 by weetmuts