wmbusmeters/wmbusmeters #7: warning: Unknown format signature hash 0x12c4! Cannot decode telegram.

pulsejets, 2019-01-27:
Hi 

I have  a multical21  , I get soem data  , but I keep getting:
./wmbusmeters --c1 /dev/ttyUSB0 john_maaler multical21 57722865 FBF8D9E934BB81410736F0F97107xxxx
john_maaler	57722865	 20.575 m3	 0.000 m3	-°C	-°C	OK	2019-01-27 14:52.22
(multical21) warning: Unknown format signature hash 0x12c4! Cannot decode telegram.
john_maaler	57722865	 20.575 m3	 0.000 m3	-°C	-°C	OK	2019-01-27 14:54.00
(multical21) warning: Unknown format signature hash 0x12c4! Cannot decode telegram.
john_maaler	57722865	 20.575 m3	 0.000 m3	-°C	-°C	OK	2019-01-27 14:55.38

warning: Unknown format signature hash 0x12c4! Cannot decode telegram.

weetmuts, 2019-01-27:
Yep, I can fix that. Its because your meter has a different configuration than
my meter.  

The Multical21 sends 7 compressed messages, with data but no format info.
There is a hash value in the compressed messages that identifies the format signature.
Since wmbusmeters does not recognize the hash, it cannot decode the data.

The 8th message is a full length message with both format and data.
This is the data that you can see. 

Can you run with --debug for a few minutes and then email me the log?
(The log will only contain the decrypted data and no keys, and no need to send me any keys.)

With that information I can definitely add your meters hash, to 
get the short messages working, and perhaps
even figure out how the hash is calculated from the long message.
Which is really how it should be done. :-)

pulsejets, 2019-01-27:
(serial /dev/ttyUSB1) received "A5E20321442D2C922872571B168D2079CC8C3A207FDB39C07D35F45926A9D63CEBE7DE0A4D778C000000F04C"
(im871a) timestamp 00008c77
(im871a) rssi 00
(im871a) got crc16 4cf0 expected 4cf0
(wmbus) received telegram 57722892 C-field=44 M-field=2c2d (KAM) A-field-version=1b A-field-dev-type=16 (Cold water meter) Ci-field=8d (Extended Link Layer II (8 Byte)) CC-field=20 (slow_resp sync) ACC=79 SN=203a8ccc
(wmbus) frame "21442D2C922872571B168D2079CC8C3A207FDB39C07D35F45926A9D63CEBE7DE0A4D"
(wmbus) payload "7FDB39C07D35F45926A9D63CEBE7DE0A4D"
(wmbus) 00: 21 length (33 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 2d2c m-field (2c2d=KAM)
(wmbus) 04: 92287257 a-field-addr (57722892)
(wmbus) 08: 1b a-field-version
(wmbus) 09: 16 a-field-type (Cold water meter)
(wmbus) 0a: 8d ci-field (Extended Link Layer II (8 Byte))
(wmbus) 0b: 20 cc-field (slow_resp sync)
(wmbus) 0c: 79 acc
(wmbus) 0d: cc8c3a20 sn
(wmbus) 21442D2C922872571B168D2079CC8C3A20
(multical21) telegram for vandmaaler 57722892
(Mode1) decrypting "7FDB39C07D35F45926A9D63CEBE7DE0A4D"
(Mode1) IV 2D2C922872571B1620CC8C3A20000000
(Mode1) decrypted first block "F4307912C40DFF00002F4E00003D0102"
(Mode1) IV+1 2D2C922872571B1620CC8C3A20000001
(Mode1) decrypted second block "03"
(Mode1) decrypted "F4307912C40DFF00002F4E00003D010203"
(multical21) log "telegram=|21442D2C922872571B168D2079CC8C3A20|F4307912C40DFF00002F4E00003D010203|+880"
(multical21) warning: Unknown format signature hash 0x12c4! Cannot decode telegram.
(multical21) 11: f430 payload crc
(multical21) 13: 79 frame type (short frame)
(multical21) 14: 12c4 format signature
(multical21) 21442D2C922872571B168D2079CC8C3A20F4307912C4
vandmaaler	57722892	 0.000 m3	 0.000 m3	-°C	-°C	OK	2019-01-27 19:01.53

pulsejets, 2019-01-27:
./wmbusmeters --c1 /dev/ttyUSB1 vandmaaler  multical21 57722892 1B83B875A45F97A35916C75919C075E5

(multical21) warning: Unknown format signature hash 0x12c4! Cannot decode telegram.
vandmaaler	57722892	 0.000 m3	 0.000 m3	-°C	-°C	OK	2019-01-27 18:19.35
(multical21) warning: Unknown format signature hash 0x12c4! Cannot decode telegram.
vandmaaler	57722892	 0.000 m3	 0.000 m3	-°C	-°C	OK	2019-01-27 18:22.44
(multical21) warning: Unknown format signature hash 0x12c4! Cannot decode telegram.
vandmaaler	57722892	 0.000 m3	 0.000 m3	-°C	-°C	OK	2019-01-27 18:24.19
(multical21) warning: Unknown format signature hash 0x12c4! Cannot decode telegram.
vandmaaler	57722892	 0.000 m3	 0.000 m3	-°C	-°C	OK	2019-01-27 18:25.53
(multical21) warning: Unknown format signature hash 0x12c4! Cannot decode telegram.
vandmaaler	57722892	 0.000 m3	 0.000 m3	-°C	-°C	OK	2019-01-27 18:29.02
(multical21) warning: Unknown format signature hash 0x12c4! Cannot decode telegram.
vandmaaler	57722892	 0.000 m3	 0.000 m3	-°C	-°C	OK	2019-01-27 18:30.36
(multical21) warning: Unknown format signature hash 0x12c4! Cannot decode telegram.
vandmaaler	57722892	 0.000 m3	 0.000 m3	-°C	-°C	OK	2019-01-27 18:32.11
(multical21) warning: Unknown format signature hash 0x12c4! Cannot decode telegram.
vandmaaler	57722892	 0.000 m3	 0.000 m3	-°C	-°C	OK	2019-01-27 18:33.45
(multical21) warning: Unknown format signature hash 0x12c4! Cannot decode telegram.
vandmaaler	57722892	 0.000 m3	 0.000 m3	-°C	-°C	OK	2019-01-27 18:35.19
(multical21) warning: Unknown format signature hash 0x12c4! Cannot decode telegram.
vandmaaler	57722892	 0.000 m3	 0.000 m3	-°C	-°C	OK	2019-01-27 18:36.53
(multical21) warning: Unknown format signature hash 0x12c4! Cannot decode telegram.
vandmaaler	57722892	 0.000 m3	 0.000 m3	-°C	-°C	OK	2019-01-27 18:38.27
vandmaaler	57722892	 20.015 m3	 0.000 m3	-°C	-°C	OK	2019-01-27 18:40.01
(multical21) warning: Unknown format signature hash 0x12c4! Cannot decode telegram.
vandmaaler	57722892	 20.015 m3	 0.000 m3	-°C	-°C	OK	2019-01-27 18:41.35
(multical21) warning: Unknown format signature hash 0x12c4! Cannot decode telegram.
vandmaaler	57722892	 20.015 m3	 0.000 m3	-°C	-°C	OK	2019-01-27 18:43.09
(multical21) warning: Unknown format signature hash 0x12c4! Cannot decode telegram.
vandmaaler	57722892	 20.015 m3	 0.000 m3	-°C	-°C	OK	2019-01-27 18:44.43
(multical21) warning: Unknown format signature hash 0x12c4! Cannot decode telegram.
vandmaaler	57722892	 20.015 m3	 0.000 m3	-°C	-°C	OK	2019-01-27 18:46.17
root@kenneth-ThinkPad-X1-Carbon:~/wmbusmeters/build# uild# ./wmbusmeters --debug  dev/ttyUSB1 vandmaaler  multical21 57722892 1B83B875A45F97A35916C75919C07BE5
For each meter you must supply a: name,type,id and key.
root@kenneth-ThinkPad-X1-Carbon:~/wmbusmeters/build# ./wmbusmeters --c1 --debug  /dev/ttyUSB1 vandmaaler  multical21 57722892 1B83B875A45F97A35916C75919C07BE5
(cmdline) using usb device: /dev/ttyUSB1
(cmdline) number of meters: 1
(serial) opened /dev/ttyUSB1
(amb8465) are you there?
(serial /dev/ttyUSB1) sent "FF0B00F4"
(serial) closed /dev/ttyUSB1
(serial) closed /dev/ttyUSB1
(serial) opened /dev/ttyUSB1
(im871a) are you there?
(serial /dev/ttyUSB1) sent "A5010100"
(serial /dev/ttyUSB1) received "A58102004CA3"
(serial) closed /dev/ttyUSB1
(im871a) got crc16 a34c expected a34c
(serial) closed /dev/ttyUSB1
(im871a) detected on /dev/ttyUSB1
(serial) opened /dev/ttyUSB1
(im871a) set link mode 06
(serial /dev/ttyUSB1) sent "A5010306000206300101"
(serial /dev/ttyUSB1) received "A58104009CF7"
(im871a) got crc16 f79c expected f79c
(im871a) set config completed
(im871a) get config
(serial /dev/ttyUSB1) sent "A5010500"
(serial /dev/ttyUSB1) received "A5810617FF000600B32587141000010001FF07023200010100007939D6"
(im871a) got crc16 d639 expected d639
(im871a) get config completed
(im871a) config: device mode 00
(im871a) config: link mode 06
(im871a) config: wmbus c-field 00
(im871a) config: wmbus mfg id 25b3 (IMS)
(im871a) config: wmbus device id 00101487
(im871a) config: wmbus version 01
(im871a) config: wmbus device type 00
(im871a) config: radio channel 01
[... 112 lines cut ...]
(wmbus) 0a: 8d ci-field (Extended Link Layer II (8 Byte))
(wmbus) 0b: 20 cc-field (slow_resp sync)
(wmbus) 0c: 06 acc
(wmbus) 0d: 388c3a20 sn
(wmbus) 21442D2C332872571B168D2006388C3A20
(im871a) telegram ignored by all configured meters!
(serial /dev/ttyUSB1) received "A5E20321442D2C932872571B168D2064108C3A2019D6D6FCCD837B4E4572588157113BDC99E28A000000F46C"
(im871a) timestamp 00008ae2
(im871a) rssi 00
(im871a) got crc16 6cf4 expected 6cf4
(wmbus) received telegram 57722893 C-field=44 M-field=2c2d (KAM) A-field-version=1b A-field-dev-type=16 (Cold water meter) Ci-field=8d (Extended Link Layer II (8 Byte)) CC-field=20 (slow_resp sync) ACC=64 SN=203a8c10
(wmbus) frame "21442D2C932872571B168D2064108C3A2019D6D6FCCD837B4E4572588157113BDC99"
(wmbus) payload "19D6D6FCCD837B4E4572588157113BDC99"
(wmbus) 00: 21 length (33 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 2d2c m-field (2c2d=KAM)
(wmbus) 04: 93287257 a-field-addr (57722893)
(wmbus) 08: 1b a-field-version
(wmbus) 09: 16 a-field-type (Cold water meter)
(wmbus) 0a: 8d ci-field (Extended Link Layer II (8 Byte))
(wmbus) 0b: 20 cc-field (slow_resp sync)
(wmbus) 0c: 64 acc
(wmbus) 0d: 108c3a20 sn
(wmbus) 21442D2C932872571B168D2064108C3A20
(im871a) telegram ignored by all configured meters!

pulsejets, 2019-01-27:
Hi 

I hope it is enough, I see alot of meters  :)

weetmuts, 2019-01-27:
Thanks, but you need to capture a log until you see 
(multical21) 13: 78 frame type (long frame)

The long frame is needed to understand the short frame.

pulsejets, 2019-01-27:
Was it enough ?

weetmuts, 2019-01-27:
Yep! Thanks, looking at it right now.

pulsejets, 2019-01-27:
Super , thanks :)

weetmuts, 2019-01-27:
Fix pushed. Please try it out!

Thanks for providing the log, I could fix a lot of small things.
In particular proper decoding of short frames and how the 
hash is calculated from the long message.

pulsejets, 2019-01-28:
vandmaaler	57722892	 20.026 m3	 0.000 m3	 0.278 m3/h	 2°C	 3°C	OK	2019-01-28 01:29.37

it works great :)

I nice feature could be that wmbusmeters could take the KEM fil as argument ad passwd , som it automatic would get mbus id , type , and key from the KEM file

you have all info in kem file .. 

 ./wmbusmeters --c1  /dev/ttyUSB0 <name> <kem file> <passwd>

closed 2019-01-28 by pulsejets

weetmuts, 2019-01-28:
@pulsejets You can create an issue for having built in support for KEM-files.