wmbusmeters/wmbusmeters #8: Compile under MacOSX

kidpixo, 2019-02-17:
Hi,
sorry for the dumb question: I compiled wmbusmeters in a docker/debian container , but docker under mac cannot pass usb device to the hypervisor.

Under mac I cannot get it to compile. Here my g++ compiler specs and the output below.

I see in the Travis tests here that you are using C++ 4.8.something, maybe my problem is there?

A side question: 
I cannot find a driver for mac to expose my cheap usb SDR (a chinese Realtek RTL28380, confirmed working with rtl_sdr and rtl-wmbus) under /dev/. I guess the rtl_sdr uses directly libusb to search for the device. There is a way to pass the data from stdin? Or any suggestion for running under mac?
In the future I'll switch this to a raspberry.

Thanks!

```
$ g++ -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/opt/local/libexec/gcc/x86_64-apple-darwin16/8.2.0/lto-wrapper
Target: x86_64-apple-darwin16
Configured with: /opt/local/var/macports/build/_opt_bblocal_var_buildworker_ports_build_ports_lang_gcc8/gcc8/work/gcc-8.2.0/configure --prefix=/opt/local --build=x86_64-apple-darwin16 --enable-languages=c,c++,objc,obj-c++,lto,fortran --libdir=/opt/local/lib/gcc8 --includedir=/opt/local/include/gcc8 --infodir=/opt/local/share/info --mandir=/opt/local/share/man --datarootdir=/opt/local/share/gcc-8 --with-local-prefix=/opt/local --with-system-zlib --disable-nls --program-suffix=-mp-8 --with-gxx-include-dir=/opt/local/include/gcc8/c++/ --with-gmp=/opt/local --with-mpfr=/opt/local --with-mpc=/opt/local --with-isl=/opt/local --enable-stage1-checking --disable-multilib --enable-lto --enable-libstdcxx-time --with-build-config=bootstrap-debug --with-as=/opt/local/bin/as --with-ld=/opt/local/bin/ld --with-ar=/opt/local/bin/ar --with-bugurl=https://trac.macports.org/newticket --disable-tls --with-pkgversion='MacPorts gcc8 8.2.0_3'
Thread model: posix
gcc version 8.2.0 (MacPorts gcc8 8.2.0_3)
```

```
 make
g++ -Os -fPIC -fmessage-length=0 -std=c++11 -Wall -Wno-maybe-uninitialized -Wno-unused-function "-DWMBUSMETERS_VERSION=\"0.8.1\"" aes.cc -c -o build/aes.o
g++ -Os -fPIC -fmessage-length=0 -std=c++11 -Wall -Wno-maybe-uninitialized -Wno-unused-function "-DWMBUSMETERS_VERSION=\"0.8.1\"" cmdline.cc -c -o build/cmdline.o
In file included from /usr/include/_wctype.h:47,
                 from /usr/include/wchar.h:92,
                 from /opt/local/include/gcc8/c++/cwchar:44,
                 from /opt/local/include/gcc8/c++/bits/postypes.h:40,
                 from /opt/local/include/gcc8/c++/bits/char_traits.h:40,
                 from /opt/local/include/gcc8/c++/string:40,
                 from /opt/local/include/gcc8/c++/stdexcept:39,
                 from /opt/local/include/gcc8/c++/array:39,
                 from /opt/local/include/gcc8/c++/tuple:39,
                 from /opt/local/include/gcc8/c++/functional:54,
                 from util.h:23,
                 from meters.h:21,
                 from cmdline.h:21,
                 from cmdline.cc:18:
/usr/include/_types/_wctype_t.h:31:9: error: '__darwin_wctype_t' does not name a type; did you mean '__darwin_time_t'?
 typedef __darwin_wctype_t wctype_t;
         ^~~~~~~~~~~~~~~~~
         __darwin_time_t
In file included from /usr/include/wchar.h:92,
                 from /opt/local/include/gcc8/c++/cwchar:44,
                 from /opt/local/include/gcc8/c++/bits/postypes.h:40,
                 from /opt/local/include/gcc8/c++/bits/char_traits.h:40,
                 from /opt/local/include/gcc8/c++/string:40,
                 from /opt/local/include/gcc8/c++/stdexcept:39,
                 from /opt/local/include/gcc8/c++/array:39,
                 from /opt/local/include/gcc8/c++/tuple:39,
                 from /opt/local/include/gcc8/c++/functional:54,
                 from util.h:23,
                 from meters.h:21,
                 from cmdline.h:21,
                 from cmdline.cc:18:
/usr/include/_wctype.h:84:22: error: 'wctype_t' has not been declared
 iswctype(wint_t _wc, wctype_t _charclass)
                      ^~~~~~~~
/usr/include/_wctype.h:171:1: error: 'wctype_t' does not name a type; did you mean 'iswctype'?
 wctype_t
 ^~~~~~~~
 iswctype
In file included from /opt/local/include/gcc8/c++/cwchar:44,
                 from /opt/local/include/gcc8/c++/bits/postypes.h:40,
                 from /opt/local/include/gcc8/c++/bits/char_traits.h:40,
                 from /opt/local/include/gcc8/c++/string:40,
[... 388 lines cut ...]
/opt/local/include/gcc8/c++/bits/basic_string.h:6482:50: error: 'vsnprintf' is not a member of 'std'
     return __gnu_cxx::__to_xstring<string>(&std::vsnprintf, __n,
                                                  ^~~~~~~~~
/opt/local/include/gcc8/c++/bits/basic_string.h:6482:50: note: suggested alternative: 'isprint'
     return __gnu_cxx::__to_xstring<string>(&std::vsnprintf, __n,
                                                  ^~~~~~~~~
                                                  isprint
/opt/local/include/gcc8/c++/bits/basic_string.h: In function 'std::__cxx11::string std::__cxx11::to_string(double)':
/opt/local/include/gcc8/c++/bits/basic_string.h:6491:50: error: 'vsnprintf' is not a member of 'std'
     return __gnu_cxx::__to_xstring<string>(&std::vsnprintf, __n,
                                                  ^~~~~~~~~
/opt/local/include/gcc8/c++/bits/basic_string.h:6491:50: note: suggested alternative: 'isprint'
     return __gnu_cxx::__to_xstring<string>(&std::vsnprintf, __n,
                                                  ^~~~~~~~~
                                                  isprint
/opt/local/include/gcc8/c++/bits/basic_string.h: In function 'std::__cxx11::string std::__cxx11::to_string(long double)':
/opt/local/include/gcc8/c++/bits/basic_string.h:6500:50: error: 'vsnprintf' is not a member of 'std'
     return __gnu_cxx::__to_xstring<string>(&std::vsnprintf, __n,
                                                  ^~~~~~~~~
/opt/local/include/gcc8/c++/bits/basic_string.h:6500:50: note: suggested alternative: 'isprint'
     return __gnu_cxx::__to_xstring<string>(&std::vsnprintf, __n,
                                                  ^~~~~~~~~
                                                  isprint
make: *** [build/cmdline.o] Error 1
```

weetmuts, 2019-02-17:
This looks fun! From what you write, it seems that it should be possible to receive the meter wmbus messages using a generic RTL-SDR usb dongle and the software rtl-wmbus. (wmbusmeters currently expects you to have either an im871A or an amber usb stick.)

If you can send me some example output from running the "rtl_sdr | rtl-wmbus" software when it captures your meter using the RTL-SDR dongle, then I can add a generic way of piping the output from rtl-wmbus into wmbusmeters. Please also send the command line you use to capture the wmbus data.
The /dev/ttyUSB0 device is used to talk to the radiochip in the im871A/amber dongles, so
there is no need to simulate a whole /dev/tty device for this new functionality.

I will take a look at compiling for macosx as well.

kidpixo, 2019-02-17:
Wow, your are super quick! Thanks!

I'm not sure if I understand the `/dev/ttyUSB0` part, under mac seems to be slightly different from *NIX

First, my exact HW is :
```$ rtl_test -t
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Fitipower FC0012 tuner
Supported gain values (5): -9.9 -4.0 7.1 17.9 19.2
Sampling at 2048000 S/s.No E4000 tuner found, aborting.
```

My output (truncated and filtered only lines with `^[TC]1;1;1`):

```
$ rtl_sdr -f 868.9M -s 1600000 - 2>/dev/null | ./rtl_wmbus

C1;1;1;2019-02-09 07:14:18.000;117;102;94740459;0x49449344590474943508780dff5f3500827f0000f10007b06effff530100005f2c620100007f2118010000008000800080008000000000000000000e003f005500d4ff2f046d10086922
C1;1;1;2019-02-09 07:14:39.000;127;116;94740258;0x49449344580274943508780dff5f350082190000ef0007b06effff070100005f2c450100007f2179000000008000800080008000000000000000000c002e005700beff2f046d10086922
C1;1;1;2019-02-09 07:14:48.000;141;128;94740406;0x49449344060474943508780dff5f350082240000810007b06effff160100005f2c630100007f219100000000800080008000800000000000000000120040005100b8ff2f046d10086922
C1;1;1;2019-02-09 07:14:48.000;32;31;94740460;0x49449344600474943508780dff5f350082880000100007b06effff180100005f2c240000007f2178000000008000800080008000000000000006000b000000070036002f046d10086922
C1;1;1;2019-02-09 07:15:04.000;32;42;94740460;0x314493446004749435087a890000200b6e1801004b6e240000426c5f2ccb086e780000c2086c7f21326cffff046d11086922
C1;1;1;2019-02-09 07:15:07.000;143;133;94740461;0x49449344610474943508780dff5f3500828d00000e0007b06effff100200005f2c480300007f2177010000008000800080008000000000000010003e006c00a20055ff2f046d10086922
C1;1;1;2019-02-09 07:15:21.000;150;116;94740461;0x314493446104749435087a8e0000200b6e1002004b6e480300426c5f2ccb086e770100c2086c7f21326cffff046d11086922
C1;1;1;2019-02-09 07:15:53.000;72;41;94740402;0x49449344020474943508780dff5f350082da0000ee0007b06effff240000005f2c090000007f2124000000008000800080008000000000000000000200070000000f002f046d11086922
C1;1;1;2019-02-09 07:15:57.000;129;131;94740408;0x314493440804749435087a330000200b6e3000004b6e370000426c5f2ccb086e230000c2086c7f21326cffff046d12086922
C1;1;1;2019-02-09 07:16:11.000;125;112;94740459;0x49449344590474943508780dff5f3500827f00007f0007b06effff530100005f2c620100007f2118010000008000800080008000000000000000000e003f005500d4ff2f046d12086922
C1;1;1;2019-02-09 07:16:41.000;128;119;94740406;0x49449344060474943508780dff5f3500822400000f0007b06effff160100005f2c630100007f219100000000800080008000800000000000000000120040005100b8ff2f046d12086922
C1;1;1;2019-02-09 07:16:50.000;145;129;94740461;0x49449344610474943508780dff5f3500828e00005e0107b06effff100200005f2c480300007f2177010000008000800080008000000000000010003e006c00a20055ff2f046d12086922
C1;1;1;2019-02-09 07:16:57.000;163;126;94740406;0x314493440604749435087a250000200b6e1601004b6e630100426c5f2ccb086e910000c2086c7f21326cffff046d13086922
C1;1;1;2019-02-09 07:17:31.000;118;107;94740401;0x49449344010474943508780dff5f350082d30000f10007b06effff250000005f2c210000007f21250000000080008000800080000000000000000002001300000004002f046d13086922
C1;1;1;2019-02-09 07:17:38.000;42;50;94740448;0x314493444804749435087a360000200b6e3001004b6e600100426c5f2ccb086e720000c2086c7f21326cffff046d13086922
C1;1;1;2019-02-09 07:17:53.000;109;99;94740405;0x314493440504749435087a1b0000200b6e4500004b6e290000426c5f2ccb086e210000c2086c7f21326cffff046d13086922
C1;1;1;2019-02-09 07:18:01.000;122;114;94740459;0x49449344590474943508780dff5f3500827f0000110007b06effff530100005f2c620100007f2118010000008000800080008000000000000000000e003f005500d4ff2f046d14086922
C1;1;1;2019-02-09 07:18:07.000;30;52;94740431;0x49449344310474943508780dff5f350082310000610107b06effff280100005f2c450000007f21770000000080008000800080000000000000000008002000050020002f046d14086922
C1;1;1;2019-02-09 07:18:19.000;62;58;94740459;0x244493445904749435087a8000082004ed392f094a2901fd0c10046d1408692202fd3cc201
C1;1;1;2019-02-09 07:18:23.000;140;122;94740258;0x49449344580274943508780dff5f3500821900000f0007b06effff070100005f2c450100007f2179000000008000800080008000000000000000000c002e005700beff2f046d13086922
C1;1;1;2019-02-13 21:22:16.000;142;136;94740410;0x314493441004749435087aab0000200b6e1703004b6e500200426c5f2ccb086e320200c2086c7f21326cffff046d17166d22
C1;1;1;2019-02-13 21:22:22.000;88;92;94740453;0x314493445304749435087ad50000200b6e3001004b6e090200426c5f2ccb086e960000c2086c7f21326cffff046d17166d22
T1;1;1;2019-02-13 21:22:26.000;143;109;40760654;0x32446850540676406980a0119f255905d024f001a008210b71c4008f5668011d00000000000000000000000119adaeaf939d9a
T1;1;1;2019-02-13 21:22:26.000;97;98;41117494;0x32446850947411416980a0119f252800d0242400e106cf06071900050602000100000000000000000000000000000e02000703
T1;1;1;2019-02-13 21:22:28.000;63;51;41115660;0x32446850605611416980a0119f258205d024b2011609c70b89a600676eb3251f22020000000000000000000113bc9e86859b7e
C1;1;1;2019-02-13 21:22:30.000;152;131;94740255;0x314493445502749435087a490000200b6e8501004b6e030300426c5f2ccb086e290100c2086c7f21326cffff046d18166d22
C1;1;1;2019-02-13 21:22:36.000;143;125;94740254;0x314493445402749435087a420000200b6e9500004b6e570000426c5f2ccb086e630000c2086c7f21326cffff046d18166d22
T1;1;1;2019-02-13 21:22:39.000;44;52;41021357;0x32446850571302416980a0119f250000d024000021083d08000000000000000000000000000000000000000000000000000000
[... cut ...]
T1;1;1;2019-02-13 21:23:11.000;114;142;40760697;0x32446850970676406980a0119f25d103d02493017307ac0767cd007844762b180200000000000000000000002262a641576236
T1;1;1;2019-02-13 21:23:14.000;127;155;40760696;0x32446850960676406980a0119f250d04d024a30002084a09284000586770362f13000000000000000000000117736368695453
T1;1;1;2019-02-13 21:23:15.000;122;106;40760652;0x32446850520676406980a0119f255002d024e6003009940b3b5900361c2b0105010000000000000000000000083b60481d4c78
C1;1;1;2019-02-13 21:23:15.000;124;108;94740261;0x314493446102749435087a770000200b6e7700004b6e560000426c5f2ccb086e540000c2086c7f21326cffff046d18166d22

```

weetmuts, 2019-02-17:
Thanks, wmbusmeters seems to parse (at least part of) that output. I added a work-in-progress
meter qcaloric (which I guess that you are reading?) to wmbusmeters and pushed to github.

Take the bytes you posted:
C1;1;1;2019-02-09 07:14:18.000;117;102;94740459;0x49449344590474943508780dff5f3500827f0000f10007b06effff530100005f2c620100007f2118010000008000800080008000000000000000000e003f005500d4ff2f046d10086922
and store only the data bytes in a file simulation_kidpixo.txt like this:
telegram=|49449344590474943508780dff5f3500827f0000f10007b06effff530100005f2c620100007f2118010000008000800080008000000000000000000e003f005500d4ff2f046d10086922|

Now run: ./build/wmbusmeters --c1 --debug simulation_kidpixo.txt HeatMe qcaloric 94740459 ""
and you will get the following output:
(cmdline) using usb device: simulation_kidpixo.txt
(cmdline) number of meters: 1
(simulator) found simulation_kidpixo.txt
(simulator) set link mode 00
(simulator) set link mode completed
(simulator) get link mode
(simulator) config: link mode 00
(cmdline) using link mode: C1
(qcaloric) configured "HeatMe" "qcaloric" "94740459" not-encrypted
(simulator) from file "49449344590474943508780dff5f3500827f0000f10007b06effff530100005f2c620100007f2118010000008000800080008000000000000000000e003f005500d4ff2f046d10086922"
(wmbus) received telegram 94740459 C-field=44 M-field=4493 (QDS) A-field-version=35 A-field-dev-type=08 (Heat Cost Allocator) Ci-field=78 (EN 13757-3 Application Layer without Transport Layer (to be defined))
(wmbus) frame "49449344590474943508780DFF5F3500827F0000F10007B06EFFFF530100005F2C620100007F2118010000008000800080008000000000000000000E003F005500D4FF2F046D10086922"
(wmbus) payload "0DFF5F3500827F0000F10007B06EFFFF530100005F2C620100007F2118010000008000800080008000000000000000000E003F005500D4FF2F046D10086922"
(wmbus) 00: 49 length (73 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 9344 m-field (4493=QDS)
(wmbus) 04: 59047494 a-field-addr (94740459)
(wmbus) 08: 35 a-field-version
(wmbus) 09: 08 a-field-type (Heat Cost Allocator)
(wmbus) 0a: 78 ci-field (EN 13757-3 Application Layer without Transport Layer (to be defined))
(wmbus) 4944934459047494350878
(qcaloric) HeatMe 94740459 (qcaloric) log "telegram=|4944934459047494350878|0DFF5F3500827F0000F10007B06EFFFF530100005F2C620100007F2118010000008000800080008000000000000000000E003F005500D4FF2F046D10086922|+0"
(dvparser) found new format "0DFF5F046D" with hash 3cd3, remembering!
(qcaloric) 0b: 0D dif (variable length Instantaneous value)
(qcaloric) 0c: FF vif (Vendor extension)
(qcaloric) 0d: 5F vife (?)
(qcaloric) 0e: 00827F0000F10007B06EFFFF530100005F2C620100007F2118010000008000800080008000000000000000000E003F005500D4FF2F
(qcaloric) 43: 04 dif (32 Bit Integer/Binary Instantaneous value)
(qcaloric) 44: 6D vif (Date and time type)
(qcaloric) 45: 10086922
(qcaloric) 49449344590474943508780DFF5F00827F0000F10007B06EFFFF530100005F2C620100007F2118010000008000800080008000000000000000000E003F005500D4FF2F046D10086922
HeatMe  94740459         0.000 kwh      2019-02-17 21:54.19
(serial) event loop stopped!

weetmuts, 2019-02-17:
Assuming that my mbus parser works correctly, it looks like the meter produces a large chunk
of variable length instantaneous data, followed by a datetime.
If this is correct, then the meter sends most of its data as a binary blob, that we need to decode somehow.

If you have some known readings from the meter, that would help in decoding...

kidpixo, 2019-02-17:
Wow! I'm impressed!  Thanks!

You're right, it is indeed the Qcaloric5.5 [Heat cost allocators – QUNDIS EN](https://qundis.com/products/heat-cost-allocators/) . As far I understood, the data could be send AES-128 encrypted, the binary blob could indicate that it is the case.

I can try to get some reading from the cost allocator display directly, but I cannot sync it directly with a telegram.

I must ask the producer for the AES key I guess...

kidpixo, 2019-02-17:
I found a reference to their wmbus protocol also in OpenHub (search THING_TYPE_NAME_QUNDIS_QCALORIC_5_5 ) > [openhab-binding-wmbus/WMBusHKVDiscoveryService.java at 4cbc7257285e815bb4b837b9814e6f07603615f5 · pokerazor/openhab-binding-wmbus](https://github.com/pokerazor/openhab-binding-wmbus/blob/4cbc7257285e815bb4b837b9814e6f07603615f5/src/de.unidue.stud.sehawagn.openhab.binding.wmbus/src/main/java/de/unidue/stud/sehawagn/openhab/binding/wmbus/internal/discovery/WMBusHKVDiscoveryService.java#L34)

weetmuts, 2019-02-17:
I do not think it is encrypted, as you can see the data here:
00827F0000F10007B06EFFFF530100005F2C620100007F2118010000008000800080008000000000000000000E003F005500D4FF2F
encrypted data would be much more random.

Thanks for the link, I will take a look, and also take a look at the macosx build problem.

weetmuts, 2019-02-17:
>I can try to get some reading from the cost allocator display directly, but I cannot sync it directly with a telegram.

That is ok, as long as it is relatively close, it would help a lot. I am also interested in which
unit (kw/calories/...) the qcaloric reports the values.

kidpixo, 2019-02-19:
Sorry for the delay, here some data I got from the sensor and from the datasheet [Data-sheet_Q-caloric-5.5_DST2-QCAL-GB0-P2P3_Qcal_55_070518_V260_FINAL.pdf](https://qundis.de/wp-content/uploads/2018/06/Data-sheet_Q-caloric-5.5_DST2-QCAL-GB0-P2P3_Qcal_55_070518_V260_FINAL.pdf)

Those the data I got from reading the display directly : 

| Key                          | Value (end Jan 2019) | Value (18. Feb 2019) |
| -----                        | -------------------- | -------------------- |
| Current consumption          | 00098                | 00126                |
| Display test: Everything on  |                      |                      |
| Display test: Everything off |                      |                      |
| Due date                     | MD 31.12             | MD 31.12             |
| Due date value               | M 00145              | M 00145              |
| Checksum                     | M c03250P            | M c03250P            |
| Evaluation level K-level     | k 060                | k 060                |
| Variable display             | FC-22s               | FC-22s               |

`Current consumption` could be Watts or Calories?

Due to the readings, assuming a rate of 1.4 consumption/day ( 126-98 / 20 days ) the actual reading in the telegram you took should be around 100-ish.

Variable display shows the actual sensor configuration:

- FC = Code for Q walk-by & Q AMR, C-mode
- 2 =  Code for the 20x algorithm
- 2 = Code for 2-sensor measuring system

Measuring system table from the datasheet:

| Scaling               | Product or unit scale                                    |
| :-----                | :--------------------                                    |
| Radiator output range | Algorithm 2: 21 - 9999 Watts                             |
| Temperature sensor    | NTC (pre-aged)                                           |
| 2 sensors             | 1 sensor for radiator temperature and                    |
|                       | 1 sensor for room air temperature                        |
| Sensor                | temperature range 0 °C - 105 °C                          |
| tm-max                | 105 °C                                                   |
| tm-min(*)             | 35 °C (2-sensor system)                                  |
| Algorithm 2           | 2-sensor measuring system 999 values (basis: K-level 60) |

Seems that we should find the `Current consumption` value + 2 temperatures. 
Searching around seems that the values from the sensor should be adjusted with some "magic" factor from the producer (this take in account the temperature gradient between the sensor and the heater), that obviously behind a customer login for Qundis. 
This is German, the English version is not working > [Kc-Werte-Datenbank für QUNDIS-Heizkostenverteiler erweitert - QUNDIS](https://qundis.de/kc-werte-datenbank-fuer-qundis-heizkostenverteiler-erweitert/#)

How did you find out the Manufacturer ID ? I was searching for the data but I found nothing!

The last thing, I'm not sure if the reading is coming from exactly this sensor, because the whole building has the same. I analysing a couple of thousand telegrams to see if I can filter the one with highest rssi .

kidpixo, 2019-02-19:
I extracted only some the telegrams for the sensor ID I found above:

```C1;1;1;2019-02-08 21:03:06.000;44;49;94740407;0x314493440704749435087adb0000200b6e1700004b6e280000426c5f2ccb086e140000c2086c7f21326cffff046d05166822
C1;1;1;2019-02-08 21:10:42.000;47;34;94740407;0x314493440704749435087adc0000200b6e1700004b6e280000426c5f2ccb086e140000c2086c7f21326cffff046d0c166822
C1;0;1;2019-02-15 08:54:19.000;135;137;94740407;0x49449344070474943508780dff5f350082b90000100007b06effff200000005f2c280000007fa11400000000800080008000800000000000000080000005800880797f9782369c04b791
C1;1;1;2019-02-15 08:54:36.000;148;108;94740407;0x314493440704749435087aba0000200b6e2000004b6e280000426c5f2ccb086e140000c2086c7f21326cffff046d38096f22```

weetmuts, 2019-02-22:
Thanks Kidpixo! I got myself a rtl-sdr dongle and I can listen to FM radio using gqrx. But I have trouble receiving wmbus messages. Do I need to do something special or have a special antenna for it to work?

weetmuts, 2019-02-22:
>How did you find out the Manufacturer ID ? I was searching for the data but I found nothing!

The decode of the telegram reported the QDS as the triplet that identifies the manufacturer, this was not in my list, but simply googling for heat cost allocators and qnd will give you Qundis, which clearly is the only manufacturer that would choose the abbreviation QDS. :-)

kidpixo, 2019-02-22:
> Thanks Kidpixo! I got myself a rtl-sdr dongle and I can listen to FM radio using gqrx. But I have trouble receiving wmbus messages. Do I need to do something special or have a special antenna for it to work?

Uhm I did nothing special: on a Mac, just plugged in and rtl_sdr or rtl_test could use it. 
For wmbus I just use the command above `rtl_sdr -f 868.9M -s 1600000 - 2>/dev/null | rtl_wmbus`.

The same cheap antenna I got with the stick is working fine, I receive also a loto of AMR readings via https://github.com/bemasher/rtlamr
I cannot test wmbusmeters because mac don't expose it as device under /dev/ and standard docker cannot handle devices. 
I'm using docker for wmbusmeters and pass the data from mac rtl_sdr in the container.

kidpixo, 2019-02-22:
> > How did you find out the Manufacturer ID ? I was searching for the data but I found nothing!
> 
> The decode of the telegram reported the QDS as the triplet that identifies the manufacturer, this was not in my list, but simply googling for heat cost allocators and qds will give you Qundis, which clearly is the only manufacturer that would choose the abbreviation QDS. :-)

Clever :-D 👍

kidpixo, 2019-02-22:
About my data, I do think they are encoded, and I have a crazy idea about the encryption.

Above I posted the [photo of one sensor](https://photos.app.goo.gl/jXwH3mgezW9C31AV9) and the ID printed on the cover is 94740407.
I was curious about the bar code and scan it gives me 0947404077 ( or 0[SensorID]7 ).
The same happens for all other 6 sensor in my flat.
Reading  documentation on  WMbus, I read that `Manufacturer (4 Hex)+Sensor ID (8Hex)+Version (2Hex)`  is an unambiguous address for a particular sensor.
Those are all data anyone can read from the data packet on air, BUT those 2 character extra in the bar code are available only to people with physical access to the device: the technician that install them.

Would make sense that each device use a different encryption key? 
Something like a combination of [Manufacturer (4 Hex), Barcode Sensor ID (10Hex) , Version (2Hex)  ] = 16 Hex ?

I figure something like this : the technician install the device, scan the code and add is to the client pool with a dedicated encryption key unique to the device.

Maybe I'm just in a scifi movie and the real key is `0000000000000000` ...

weetmuts, 2019-02-23:
It now compiles under MacOSX 10.14.3 with a recent XCode. (clang-1000.10.44.4)
But I have compiled on a virtual mac, so I have no access to the physical usb ports.
Can you try if it works?

kidpixo, 2019-02-23:
it works!!! Thank you!

kidpixo, 2019-02-23:
Another thing I notice: 
```$ ./build_debug/wmbusmeters --c1 --debug simulation_kidpixo.txt HeatMe qcaloric 94740459 ""```
is fine , but  
```$ cd build_debug
$ ./wmbusmeters --c1 --debug ../simulation_kidpixo.txt HeatMe qcaloric 94740459 ""
(cmdline) using usb device: ../simulation_kidpixo.txt
(cmdline) number of meters: 1
Device ../simulation_kidpixo.txt is not a character device.
```

That comes from this check `if (!S_ISCHR(info.st_mode))` in [util.cc](https://github.com/weetmuts/wmbusmeters/blob/205d360e29217c75447fabae00c19d55b917f01a/src/util.cc#L277) .

Shouldn't it work anyway?

kidpixo, 2019-02-23:
A last question, do you think the messages are encrypted now? Or you think it is possible to decode them?

weetmuts, 2019-02-23:
Just committed a fix, so that the file simulation_yyy.txt can be located in some other directory than cwd.
Please test.

I have not yet had time to examine the messages. But I still do not believe that they are encrypted per se. But lets see! :-)

weetmuts, 2019-02-25:
Now there is support for receiving data from rtlwmbus. Try this command line,
and make sure rtl_srd and rtl_wmbus is in /usr/bin
wmbusmeters --debug  rtlwmbus Heater qcaloric 94740459 ""
rtlwmbus defaults to "/usr/bin/rtl_sdr -f 868.9M -s 16000000 | /usr/bin/rtl_wmbus"
If you want a different command, do:
wmbusmeters --debug  'rtlwmbus:rtl_sdr f 12345 | rtl_wmbus' Heater qcaloric 94740459 ""

Please test! It should work for you on macosx.

kidpixo, 2019-02-26:
Thanks! You're surely really active!
I'll test it later this week, I have a huge pile of crumbling work to deal with. 

I think this latest push could be useful to a lot of users.

> Now there is support for receiving data from rtlwmbus. Try this command line,
> and make sure rtl_srd and rtl_wmbus is in /usr/bin
> wmbusmeters --debug rtlwmbus Heater qcaloric 94740459 ""
> rtlwmbus [...]

I have some problem with installing stuff system wide (laptop provided from my work, nice IT , the usual) and normally symlink the self-compiled stuff in ~/.local/bin/ or I just leave them where they are built.

Would be a huge deal to :

- accept the executable path (I look in your implementation, seems impractical) OR
- call trl_sdr/rtl_wmbus without complete path, praying the Old Ones that they are in user's PATH OR
- pipe from stdin ?

weetmuts, 2019-02-26:
Ok, I made the following changes and pushed the code to github.

wmbusmeters rtlwmbus Heater qcaloric 12345678 ""
will use no prefix for rtl_sdr and rtl_wmbus.
but using config files in daemon mode will prefix with /usr/bin/
it will also default to 868.95M for the frequency.

You can also specify
wmbusmeters rtlwmbus:868.9M Heater qcaloric 12345678 ""
and it will use this slightly off frequency instead.

Or you can specify the entire command line:
wmbusmeters 'rtlwmbus:/bar/rtl_sdr -special | /foo/rtl_wmbus -specials' Heater qcaloric 12345678 ""

Add --verbose to see the command line it uses.

pulsejets, 2019-02-27:
Hi Fredrik 

Would wmbusmeters rtlwmbus work with Kampstup watermeter ?

/Kenneth

weetmuts, 2019-02-27:
Yes, I am assuming and hoping it should work with with Kamstrup watermeters. 

But I have not yet got my rtl-sdr.com dongle to work with my multical21, but it could be a faulty antenna connecter. I have ordered another dvb-t dongle, compatible with rtl_sdr, to see if I can get that working. 
I'll let you know when/if I succeed.

pulsejets, 2019-02-27:
Hi 
I have a rtl-sdr (working)  , so if you want me to test  please say so!!

so rtlwmbus instead of usbdevice ?

weetmuts, 2019-02-27:
Cool! Yes, please test!

You build https://github.com/xaelsouth/rtl-wmbus and put the rtl_wmbus binary in your path.
(/usr/bin if you intend to run wmbusmeters as a daemon).

If you have such a water meter nearby, you can listen to it without having the key, simply
by running: "wmbusmeters rtlwmbus" or "wmbusmeters rtlwmbus:868.9M" if you need to 
adjust the freq depending on your dongle.

You will get output like:

No meters configured. Printing id:s of all telegrams heard!

Received telegram from: 12345678
          manufacturer: (KAM) Kamstrup Energie A/S
           device type: Cold water meter
Received telegram from: 12345678
          manufacturer: (KAM) Kamstrup Energie A/S
           device type: Cold water meter

pulsejets, 2019-02-27:
Hi
I have the key for my own meter, so 

wmbusmeters --c1 rtlwmbus Watermeter  multical21 57722892 1B83B875A45F97A35XXXXXXXXXXXXX

?

weetmuts, 2019-02-27:
Yep. You do not need --c1 when you know the meter. Also rtlwmbus can listen to both C1 and T1 traffic at the same time.

weetmuts, 2019-02-28:
Aha! @kidpixo your heat cost allocator sends two kinds of wmbus messages interleaved.
One is a couple of proprietary bytes, not yet understood.
But the other is plain wmbus data. 

So I just pushed code to github that makes it possible for you to do:
./build/wmbusmeters simulations/simulation_kidpixo.txt MyElement qcaloric 78563412 ""
MyElement	78563412	 127.000 hca	2019-02-28 19:39.27

./build/wmbusmeters --format=json simulations/simulation_kidpixo.txt MyElement qcaloric 78563412 ""
{"media":"heat_cost_allocation","meter":"qcaloric","name":"MyElement","id":"78563412","current_consumption_hca":127.000000,"timestamp":"2019-02-28T18:41:36Z"}

I invented the unit hca, since I do not know what heat cost allocation is actually reported as.
In fact, other wmbus implementations list it as "Units for H.C.A." so there.... :-)

If you run with --debug you see those telegrams decoded as:
(qcaloric) 0f: 0B dif (6 digit BCD Instantaneous value)
(qcaloric) 10: 6E vif (Units for H.C.A.)
(qcaloric) 11: 270100 current consumption (127.000000 hca)
(qcaloric) 14: 4B dif (6 digit BCD Instantaneous value storagenr=1)
(qcaloric) 15: 6E vif (Units for H.C.A.)
(qcaloric) 16: 450100
(qcaloric) 19: 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
(qcaloric) 1a: 6C vif (Date type G)
(qcaloric) 1b: 5F2C
(qcaloric) 1d: CB dif (6 digit BCD Instantaneous value storagenr=1)
(qcaloric) 1e: 08 dife (subunit=0 tariff=0 storagenr=17)
(qcaloric) 1f: 6E vif (Units for H.C.A.)
(qcaloric) 20: 790000
(qcaloric) 23: C2 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
(qcaloric) 24: 08 dife (subunit=0 tariff=0 storagenr=17)
(qcaloric) 25: 6C vif (Date type G)
(qcaloric) 26: 7F21
(qcaloric) 28: 32 dif (16 Bit Integer/Binary Value during error state)
(qcaloric) 29: 6C vif (Date type G)
(qcaloric) 2a: FFFF
(qcaloric) 2c: 04 dif (32 Bit Integer/Binary Instantaneous value)
(qcaloric) 2d: 6D vif (Date and time type)
(qcaloric) 2e: 200B7422 

I'll try to fix the date time extraction code. Please check if you 
are intersted in any of the other values reported.

weetmuts, 2019-02-28:
So in the above telegram we got:
127 hca  (currently shown by wmbusmeters as current consumption)
145 hca
17 hca 
79 hca
(but what are the others...)

kidpixo, 2019-03-01:
Thank you so much @weetmuts ! 
I let it running last night and it got the right reading in hca units from the Allocator : 145 Whatever, it is going up! 
:-) 

I'm less and less happy with non-standard standard and industry not making open their stuff based on open standard/projects....

I would really like to crack open those telegrams, thank a lot for your work!

I remember that project like [FHEM](https://www.fhem.de/) should already have some wmbus device integrated, maybe QDS is somewhere...

I found some reference to QDS heat allocator in : 

1. [TR-03109-1_Anlage_Feinspezifikation_Drahtlose_LMN-Schnittstelle.pdf](https://www.bsi.bund.de/SharedDocs/Downloads/DE/BSI/Publikationen/TechnischeRichtlinien/TR03109/TR-03109-1_Anlage_Feinspezifikation_Drahtlose_LMN-Schnittstelle.pdf?__blob=publicationFile&v=2)
2.  [WHE542-D291S – Heat Cost Allocation with W.M-Bus communication – Equobox](http://www.equobox.com/portfolio-articoli/whe542-d291s-heat-cost-allocation-with-wm-bus-communication/)

From 1. the telegram should contain:

- current cunsumption value 1234 HCA units
- set date 30.04.2007
- consumption at set date 23456 HCA units
- current temperature at sensor 25 °C
- another temperature ? My allocator (see above) are using  Algorithm 2:2-sensor measuring system.

Cheers

weetmuts, 2019-03-01:
Cool does this mean that you have succesfully run wmbusmeters on MacOSX using rtl_wmbus?

Thanks for the links to the docs. It seems like your meter outputs exactly the data in your second link.
I have now written code to extract all the data that was found in your telegrams, and pushed it to github. The json output looks like this (and is identical to the Equobox doc).

{"media":"heat_cost_allocation","meter":"qcaloric","name":"Element","id":"94740258","current_consumption_hca":127.000000,"set_date":"2018-12-31","consumption_at_set_date_hca":145.000000,"set_date_17":"2019-01-31","consumption_at_set_date_17_hca":79.000000,"error_date":"2127-15-31","device_date_time":"2019-02-20 11:45","timestamp":"2019-03-01T14:41:59Z"}

And then, there is the raw vendor specific string, but it is only shown in the debug output for the moment. But if you run with --debug and grep "vendor extension data", then you will see those.
However, if all the data you need is in the data above, I do not think we need to reverse engineer that stuff.

kidpixo, 2019-03-02:
Yes, I did it! Thanks a lot!
I feel we over used this github issue, this thread should have been cut in 2-3 different issues. If you want, I can cut this answer and start a new issue.

For the data itself, I would like to see if I can decode the private packets, I'll search around. Thanks a lot!!

My latest question is, now that I have a working chain to get the radio signal (rtl_sdr)  > extract wmbus packet (rtl-wmbus) > decode my HCA sensors (wmbusmeters) : 
is there a way to tell wmbusmeters to decode all sensors? something like specify * as sensor ID?

I would like to run a single instance of wmbusmeters to get all my sensor , normally I use a grep regex in the `rtl_sdr)  | rtl-wmbus | grep 'IDS'` chain.

xaelsouth, 2019-03-02:
Hi weetmuts and kidpixo,

really nice work that you both are doing here. :)

weetmuts, i have found your project more than a year ago, just after rtl-wmbus was on github, but have not had enough elan to participate in wmbusmeters.

Anyway, some of wireless MBus devices are reverse engineered by FHEM people. These devices use proprietary but unencrypted dataformat. Last time more and more manufacturers shift to Open Metering (OMS) and using encrypted datagrams only.

https://github.com/mhop/fhem-mirror/tree/master/fhem/FHEM

Another interesting link for you is (probably) https://github.com/rscada/libmbus , see for  wired M-Bus datagram examples. You can test your own software by converting these  into wireless M-Bus datagrams: first cut the M-Bus trailer and replace the M-Bus header with the wireless one.

Just in case, you will find windows port of libmbus https://github.com/xaelsouth/libmbus-winport

Xael

weetmuts, 2019-03-03:
Thanks @xaelsouth!

@kidpixo yep, I think we can safely close this issue and open up new ones. :-) Please do so!

kidpixo, 2019-03-03:
Hi @xaelsouth , thanks but I did absolutely nothing, apart more workload for @weetmuts :-D

I'm aware of FHEM, but personally using [HomeAssistant](https://www.home-assistant.io/), that's why I was searching for a more modular *NIX style solution : testing my sensors and maybe integrate in hass. 

They use a docker container instead of a plugin architecture, I find it really nice.

Thanks for the link!

@weetmuts doing it!

closed 2019-03-03 by kidpixo