wmbusmeters/wmbusmeters #9: (dvparser) warning: unexpected end of data

mikkel75, 2019-02-28:
Hi,

And 1st of all THANKS for you great work!
I see the following error when trying to parse iperl data:

Do you know what could cause this?
I started with no meter defined and checked ID and CRC was always OK.
Could this be caused by missing encryption-key?

Thanks,
M

`
hac@hac:~/wmbusmeters/build$ ./wmbusmeters --debug /dev/ttyUSB3 iPerl iperl 20222542 ""
(config) using device: /dev/ttyUSB3
(config) number of meters: 1
(serialtty) opened /dev/ttyUSB3
(amb8465) are you there?
(serial /dev/ttyUSB3) sent "FF0B00F4"
(serialtty) closed /dev/ttyUSB3
(serialtty) closed /dev/ttyUSB3
(serialtty) opened /dev/ttyUSB3
(im871a) are you there?
(serial /dev/ttyUSB3) sent "A5010100"
(serial /dev/ttyUSB3) received "A58102004CA3"
(serialtty) closed /dev/ttyUSB3
(im871a) got crc16 a34c expected a34c
(serialtty) closed /dev/ttyUSB3
(im871a) detected on /dev/ttyUSB3
(serialtty) opened /dev/ttyUSB3
(im871a) set link mode 03
(serial /dev/ttyUSB3) sent "A5010306000203300101"
(serial /dev/ttyUSB3) received "A58104009CF7"
(im871a) got crc16 f79c expected f79c
(im871a) set config completed
(im871a) get config
(serial /dev/ttyUSB3) sent "A5010500"
(serial /dev/ttyUSB3) received "A5810617FF000300B32536171000010001FF07023200010100017938F1"
(im871a) got crc16 f138 expected f138
(im871a) get config completed
(im871a) config: device mode 00
(im871a) config: link mode 03
(im871a) config: wmbus c-field 00
(im871a) config: wmbus mfg id 25b3 (IMS)
(im871a) config: wmbus device id 00101736
(im871a) config: wmbus version 01
(im871a) config: wmbus device type 00
(im871a) config: radio channel 01
(im871a) config: radio power level 07
(im871a) config: radio data rate 02
(im871a) config: radio rx window 32
(im871a) config: auto power saving 00
(im871a) config: auto RSSI attachment 01
(im871a) config: auto rx timestamp attachment 01
(im871a) config: led control 00
(im871a) config: rtc control 01
(config) using link mode: T1
(iperl) configured "iPerl" "iperl" "20222542" not-encrypted
(serial /dev/ttyUSB3) received "A5E2031E44AE4C4225222068077A473010059DB71B3A527B2B5524C17F5BDD82FA5AF38D945362BCE1"
(im871a) timestamp 53948df3
(im871a) rssi 62
(im871a) got crc16 e1bc expected e1bc
(wmbus) received telegram 20222542 C-field=44 M-field=4cae (SEN) A-field-version=68 A-field-dev-type=07 (Water meter) Ci-field=7a (EN 13757-3 Application Layer with short Transport Layer) CC-field=00 (slow_resp) ACC=47
(wmbus) frame "1E44AE4C4225222068077A473010059DB71B3A527B2B5524C17F5BDD82FA5A"
(wmbus) payload "9DB71B3A527B2B5524C17F5BDD82FA5A"
(wmbus) 00: 1e length (30 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: ae4c m-field (4cae=SEN)
(wmbus) 04: 42252220 a-field-addr (20222542)
(wmbus) 08: 68 a-field-version
(wmbus) 09: 07 a-field-type (Water meter)
(wmbus) 0a: 7a ci-field (EN 13757-3 Application Layer with short Transport Layer)
(wmbus) 0b: 47 acc
(wmbus) 0c: 30 status ()
(wmbus) 0d: 1005 configuration ()
(wmbus) 1E44AE4C4225222068077A47301005
(iperl) telegram for iPerl 20222542
(iperl) log "telegram=|1E44AE4C4225222068077A47301005|9DB71B3A527B2B5524C17F5BDD82FA5A|+730"
(dvparser) warning: unexpected end of data
wmbusmeters: src/dvparser.cc:261: bool parseDV(Telegram*, std::vector<unsigned char>&, std::vector<unsigned char>::iterator, size_t, std::map<std::__cxx11::basic_string<char>, std::pair<int, DVEntry> >*, std::vector<unsigned char>::iterator*, size_t, uint16_t*, std::function<int(int, int, int)>): Assertion `data+datalen <= databytes.end()' failed.
Aborted
`

weetmuts, 2019-02-28:
Yes I believe that the encryption key is missing when you run wmbusmeters for this meter.

Your telegrams content part is:  9DB71B3A527B2B5524C17F5BDD82FA5A
It ought to begin with 2F2F, like this other iperl meter telegram:
2F2F0413181E0000023B00002F2F2F2F
The 2F are no-op bytes inserted by the sender so that it is possible to verify
that the decryption went ok. I suppose I should add a test that actually checks 
this and prints a more helpful reply. :-)

weetmuts, 2019-02-28:
And thanks for the thanks! :-)

weetmuts, 2019-02-28:
Pushed the check for 2F2F to github.

closed 2019-03-05 by weetmuts