wmbusmeters/wmbusmeters #10: hi

marcomilazzo, 2019-03-03:
is there a way to see the raw string before it's processed ?
it would be very usefull.
in my case i want to see if my soc(cc1310) is receiving correctly
regards
marco

weetmuts, 2019-03-03:
Interesting, what is a soc(cc1310)?
You can see the raw strings if you run with --debug and look for lines like:
(wmbus) frame "2A442D2C.....F275", these show the full wmbus frame with header (before decryption)
and on the serial tty level:
(serial ....) received "A5...7703", these show what is received over the serial tty from the dongle.
(serialcmd) received "A5...7703", these show what is received over the pipe from rtl_wmbus.

marcomilazzo, 2019-03-03:
Fantastic :)
cc1310 is a texas radio 868MHz  with microcontroller iside
i justi tried to compile your code following 
make && make test ..............
and i get "Segmentation fault"
i'm not a programmer ! and i now little of linux ! 
but anyway  i got the wmbusmeters_0.9.0_armhf.deb 
tried to install the deb with no succes
also when i try to lounc the program from his directory i get the error
what shall i do?
thans a lot
marco

marcomilazzo, 2019-03-03:
bt the way
new meters will use 169 MHz with wm-bus narrow-band
i think mode n1 and n2 
the decoding is a little different .not manchester 
do you plan to do do also this ?
marco :)

weetmuts, 2019-03-03:
I would love to add new meters as well as new wmbus modes N1 and N2, but I need your help to test and verify the code, since I only have 868.95MHz meter.

It seems like you want to run it on a raspberry pi?
The deb support is not ready yet, so the deb is probably broken. Sorry about that.
Try just building the binary with "make HOST=arm" and copy the binary to the raspberry.

If you have a Gnu/Linux setup, that is much easier when developing the new features.
For the latest Ubuntu it should be trivial:
make; make test. 
You can see the latest Travis build here: https://travis-ci.org/weetmuts/wmbusmeters/jobs/500384190
to see what a proper build should look like.

Second, how do you communicate with cc1310 from Ubuntu/Raspbian?

marcomilazzo, 2019-03-03:
it would be a pleasure to work togheter
i use linux on wmware. i got the error there
but maybe it's the linux mate i'm using
i will try with another version . can you tell me wich version ?
no rasberry !
i can test the 169 MHz with you
ore you on mac ?
we must first resolve the problem on linux  or better on windows :)
all the compilers for microcontrollers i use  are under   windows
is the code made for raspberry?
no problem for testing .i have the 169MHz fully working
you shall guide me for making the software work under linux 
as i told you i am not a programmer  (only microcontrollers)
marco

marcomilazzo, 2019-03-03:
ah
i comumicate with the cc1310  with an edge router (another cc1310) wich is connected to internet with a 4g modem
i'm into iot
i'm using 6lowpan 
but i use a cloud only for messages
marco

weetmuts, 2019-03-04:
The code should compile under MacOSX and a recent Xcode.

However I do not quite understand your use case:

will you use cc1310 to send wbus telegrams that wmbusmeters should receive
or 
will you use cc1310 to receive telegrams and feed the received data into wmbusmeters?

marcomilazzo, 2019-03-04:
hi
i receive the telegrams from installed meters with the cc1310 
i don't need a pc ! it's like having a standalone usb rtl-sdr 
the micro does all the decoding 
then i send the telegram to the web
my friend has a Mac so i will ask him
can i compile your code to work under linux?
marco

marcomilazzo, 2019-03-04:
what i want to do is to receive the telegrams with  a cc1120 wich is a narrowband transreceiver
this is why i need to see the raw data
i will make a little repeater that receive in wide band and the sends in narrow band
this way i will have a good range 
this is on 868 MHz 
on 169 we already have narrow-band
marco

weetmuts, 2019-03-04:
Aha, ok!

And how do you intend the output from cc1310 to be sent to wmubsmeters for decoding?
1) Should wmbusmeters open a tcp connection to an ip address and receive realtime data from there.
2) Should wmbusmeters open a listen port and wait for the cc1310 to connect and push telegrams?
3) Should wmbusmeters perform a REST api call to the web/cloud to fetch the telegram for decoding?
4) Should wmubsmeters open a REST server to which the telegrams are pushed?

> can i compile your code to work under linux?
Yes, my default build environment is Ubuntu, so it should work out of the box.
If you do "sudo apt install build-essential" you probably get all necessary build dependencies.

Please sync the latest version of the git repository and do a: 
make > logg 2>&1
make test > testlogg 2>&1
and post logg and thestlogg here, or email them to me.

marcomilazzo, 2019-03-04:
>>And how do you intend the output from cc1310 to be sent to wmubsmeters for decoding?
no .  i will use eventualy your code for testing the cc1120 .it's not easy to set these radios.
the raw data is the most important thing for me right now. preample and sync ... post preamble
so your code will be very helpfull . i have already working chips but i like to see the low level
texas instruments have a complete wm-stack http://www.ti.com/tool/WMBUS
i will try as soon as possible your suggestion
and send you a log 
i believe it would by a joke for you the get the error under linux 
i have like you a im871a  dongle  for testing 
conclusion. your incredible code is for lab purposes . in real world i will use soc (micro with radio)
when we talk about thousand of telegrams  here comes in 6lowpan
have a look .it's amazing . it's like havung a real ethernet under radios
marco

marcomilazzo, 2019-03-04:
hi
i downloaded last ubuntu
compiled all
with rtl-sdr 
i get
(serialtty) internal error after select! errno=Bad file descriptor
if i use the im871 i get no output

marcomilazzo, 2019-03-04:
logg
empty
testlogg 
./build/testinternals
./test.sh build/wmbusmeters
Internal test OK
C1 OK
T1 OK
SHELL OK
Meterfiles OK
Meterfiles dir OK
Config1 OK
Config2 with logfile and meterfiles OK
OK: listen to all with logfile
OK: listen to all with stdout

marcomilazzo, 2019-03-04:
i got at least the primary screen
how do i read from the rtl-sdr?
the software installed a new user : wmbusmeters 
is it port S0?
i don't understand what is the name of the port
i tried
marco@ubuntu:~$ wmbusmeters /dev/ttyS0 MyTapWater multical21 12345678 00112233445566778899AABBCCDDEEFF 
No wmbus device found! Exiting!

"rtlwmbus:<commandline>" to have wmbusmeters spawn
that commandline instead, its output is expected to be like rtl_wmbus.

As meter quadruples you specify:
<meter_name> a mnemonic for this particular meter
<meter_type> one of the supported meters
<meter_id> an 8 digit mbus id, usually printed on the meter
<meter_key> an encryption key unique for the meter
    if the meter uses no encryption, then supply ""

Supported water meters:
Kamstrup Multical 21 (multical21)
Kamstrup flowIQ 3100 (flowiq3100)
Sontex Supercom 587 (supercom587)
Sensus iPERL (iperl)

Supported heat cost allocator:
Qundis Q caloric (qcaloric)

Work in progress:
Heat meter Kamstrup Multical 302 (multical302)
Electricity meter Kamstrup Omnipower (omnipower)

weetmuts, 2019-03-05:
Good to see that it is easy to build on Ubuntu!

Are you running Ubuntu inside a virtual machine on a mac? Make sure that the usb device is captured by the virtual machine.

Then run the command "dmesg" in Ubuntu, and you will see for the imst871a dongle:
[97176.997120] usb 1-3: new full-speed USB device number 51 using xhci_hcd
[97177.147751] usb 1-3: New USB device found, idVendor=10c4, idProduct=ea60
[97177.147760] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[97177.147767] usb 1-3: Product: WiMOD iM871A-usb
[97177.147773] usb 1-3: Manufacturer: Silicon Labs
[97177.147779] usb 1-3: SerialNumber: 00B94096
[97177.149977] cp210x 1-3:1.0: cp210x converter detected
[97177.151785] usb 1-3: cp210x converter now attached to ttyUSB0

Here you can see that the dongle is now connected to /dev/ttyUSB0
And you can do
wmbusmeters --debug --c1 /dev/ttyUSB0
Now you will get nice info printouts of every c1 telegram that the dongle receives.
You can try --t1 as well.

I was just about to add --n1a for you, but it turned out that according to the
ims871a manual, that dongle cannot listen to N traffic....
read page 15 in WMBus_HCI_Spec_V1_6.pdf, perhaps you need a
im170A dongle for that?

weetmuts, 2019-03-05:
@xaelsouth Are you planning to add support for N1 and N2 radio messages (169MHz) to rtl_wmbus?

weetmuts, 2019-03-05:
I just added support for --n1a to --n1f so that it might be possible to listen to channel n1a to n1f
using an imst dongle.

Try this, with the narrow band channel a-f that your meter sends on:
wmbusmeters --debug --n1a /dev/ttyUSB0

It does set the proper n1a link mode in the imst871a dongle, but I cannot guarantee
that it can actually hear anything, even though the dongle happily complies.

weetmuts, 2019-03-05:
Wait with "sudo make install" until you have got it working from the command line.

marcomilazzo, 2019-03-06:
i'm running ubuntu an vitual machine
the rtl-sdr dongle works fine ( i tried  with cubic sdr and rtl-wmbus )
what is the command for reading the imst871a dongle ?
i tried 
wmbusmeters --t1 --debug  /dev/im871a
got 
marco@ubuntu:~/Desktop/wmbusmeters-master$ wmbusmeters --t1 --debug  /dev/im871a(wmbusmeters) version: 0.9.0
(config) using device: /dev/im871a
(config) number of meters: 0
(serialtty) opened /dev/im871a
(serial /dev/im871a) received "A5E2034944B4095641071505077AA21806101DAE1CA92C3B0F1F0E182A30030000003255120603197F96980000000000000000000000000000000000000000000000000000000000007F96980100000000ABB62CA5E2031E44AE4C8730702068077A1F001005EF666A497CBAD4564A80CA41D7F4144100000000A553E8"
(amb8465) are you there?
(serial /dev/im871a) sent "FF0B00F4"
(serialtty) closed /dev/im871a
(serialtty) closed /dev/im871a
(serialtty) opened /dev/im871a
(im871a) are you there?
(serial /dev/im871a) sent "A5010100"
(serial /dev/im871a) received "A58102004CA3"
(serialtty) internal error after select! errno=Bad file descriptor
(serialtty) internal error after select! errno=Bad file descriptor
(serialtty) internal error after select! errno=Bad file descriptor
(serialtty) internal error after select! errno=Bad file descriptor
(serialtty) internal error after select! errno=Bad file descriptor
.........
(serialtty) internal error after select! errno=Bad file descriptor
(serialtty) internal error after select! errno=Bad file descriptor
(serialtty) internal error after select! errno=Bad file descriptor
(serialtty) closed /dev/im871a
(im871a) got crc16 a34c expected a34c
(serialtty) closed /dev/im871a
(im871a) detected on /dev/im871a
(serialtty) opened /dev/im871a
(im871a) set link mode 03
(serial /dev/im871a) sent "A5010306000203300101"
(serial /dev/im871a) received "A58104009CF7"
(im871a) got crc16 f79c expected f79c
(im871a) set config completed
(im871a) get config
(serial /dev/im871a) sent "A5010500"
(serial /dev/im871a) received "A5810617FF000300B32591141000010001FF07023200010100000E1E53"
(im871a) got crc16 531e expected 531e
(im871a) get config completed
(im871a) config: device mode 00
(im871a) config: link mode 03
(im871a) config: wmbus c-field 00
(im871a) config: wmbus mfg id 25b3 (IMS)
(im871a) config: wmbus device id 00101491
(im871a) config: wmbus version 01
(im871a) config: wmbus device type 00
(im871a) config: radio channel 01
(im871a) config: radio power level 07
(im871a) config: radio data rate 02
(im871a) config: radio rx window 32
(im871a) config: auto power saving 00
(im871a) config: auto RSSI attachment 01
(im871a) config: auto rx timestamp attachment 01
(im871a) config: led control 00
(im871a) config: rtc control 00
(config) using link mode: T1
No meters configured. Printing id:s of all telegrams heard!

(serial /dev/im871a) received "A5E2031E44AE4C8730702068077A22001005944D8A3A1E34F1DA565891792F26F5B300000000A6132B"
(im871a) timestamp 00000000
(im871a) rssi a6
(im871a) got crc16 2b13 expected 2b13
(wmbus) received telegram 20703087 C-field=44 M-field=4cae (SEN) A-field-version=68 A-field-dev-type=07 (Water meter) Ci-field=7a (EN 13757-3 Application Layer with short Transport Layer) CC-field=00 (slow_resp) ACC=22 
(wmbus) frame "1E44AE4C8730702068077A22001005944D8A3A1E34F1DA565891792F26F5B3"
(wmbus) payload "944D8A3A1E34F1DA565891792F26F5B3"
(wmbus) 00: 1e length (30 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: ae4c m-field (4cae=SEN)
(wmbus) 04: 87307020 a-field-addr (20703087)
(wmbus) 08: 68 a-field-version
(wmbus) 09: 07 a-field-type (Water meter)
(wmbus) 0a: 7a ci-field (EN 13757-3 Application Layer with short Transport Layer)
(wmbus) 0b: 22 acc
(wmbus) 0c: 00 status ()
(wmbus) 0d: 1005 configuration ()
(wmbus) 1E44AE4C8730702068077A22001005
Received telegram from: 20703087
          manufacturer: (SEN) Sensus GmbH
           device type: Water meter
(im871a) telegram ignored by all configured meters!
^C(serialtty) event loop stopped!
(serialtty) closed /dev/im871a
(serialtty) closed /dev/im871a
marco@ubuntu:~/Desktop/wmbusmeters-master$ 
so it working :)
stll some errors to resolve ((serialtty) internal error after select! errno=Bad file descriptor)
marco

marcomilazzo, 2019-03-06:
is it possible to use the rtl-sdr dongle?
maybe there will be no problem for decoding 160MHz band

marcomilazzo, 2019-03-06:
i can send wm-bus 169MHz telegrams to test all

weetmuts, 2019-03-06:
Cool, you got it working and apparently you have a Sensus iPerl water meter nearby, sending T1 telegrams.

Now try with --n1a instead of --t1 (or whatever N channel your device transmits on)
and see if you can receive anything. Perhaps it will work, but probably not....

rtl_sdr does not actually decode anything, it is rtl_wmbus (https://github.com/xaelsouth/rtl-wmbus)
that does the real decoding. rtl_wmbus currently only listens to 868.95MHz for C1 and T1 telegrams.
So you could perhaps create an issue there to support N telegrams at 169MHz.

marcomilazzo, 2019-03-06:
yes a was meaning rtl_wmbus 
as soon as possible i will try the n1
i must program the soc for this
i was asking you to implement the rtl-sdr . it's cheap works well .has external antenna connector
this would make the code for a more flexible use
anyway can you fix that error ?  internal error after select
it happens only at the beginning
anyway you did a fantastic work
what is the command to ad AES ?
wmbusmeters --t1 --debug /dev/im871a and what ?
marco

marcomilazzo, 2019-03-06:
where are you from?
are you an engineer?

marcomilazzo, 2019-03-06:
if the im871a doesn't receive on 169 wich is normal(it's made just for 868) how can i make the test?
this is why rtl-sdr is usefull . it goes on all frequencies

weetmuts, 2019-03-06:
Sorry, I cant help you with the decoding of the actual radio signals for 169MHz, I have neither the skill to write the radio code, nor an actual meter to listen to.

Check the README for how to supply actual meter information and the AES key on the commandline.

I will take a look if I can reproduce the select error.

weetmuts, 2019-03-20:
Did you receive any 169 MHz telegrams using the imst dongle?

marcomilazzo, 2019-03-20:
Hi Fredrik

As long as i now the imst dongle is only for 868 !!

I receive 169 with rtl-sdr 

For making the telegram I use the kit from silabs 

When I will make a board with this radio I can send one to you

 

marco

A: weetmuts/wmbusmeters
Cc: marco milazzo; Author

 

Questa e-mail è stata controllata per individuare virus con Avast antivirus.
https://www.avast.com/antivirus

weetmuts, 2019-05-21:
Thanks Marco! Closing this for now.

closed 2019-05-21 by weetmuts

nestorayuso, 2019-05-21:
to receive 169MHz you can use Radiocrafts modules.
there is a new wize alliance protocol based on wireless mbus N-mode

weetmuts, 2019-05-21:
@nestorayuso , interesting, if I want to add support for radiocraft receiver modules,
which product do you recommend that I buy?

nestorayuso, 2019-05-21:
Radiocrafts RC1701HP-MBUS4 or RC1701VHP-MBUS4 for wireless mbus N-mode
Radiocrafts RC1701HP-WIZE for Wize protocol