wmbusmeters/wmbusmeters #11: Decode multiple sensors at once.

kidpixo, 2019-03-03:
Thanks to #8  now I have a working chain to get wmbus data: Thanks @weetmuts , great work!

My work chain is:

1. get the radio signal [osmocom/rtl-sdr](https://github.com/osmocom/rtl-sdr)
2. extract wmbus packet [xaelsouth/rtl-wmbus: Software defined receiver for wireless M-Bus with RTL-SDR](https://github.com/xaelsouth/rtl-wmbus)
3. decode my HCA sensors [weetmuts/wmbusmeters](https://github.com/weetmuts/wmbusmeters)

My problem is how tell `wmbusmeters` to decode a series/all sensors, not a single one. 
Something like specify `*` as sensor ID?

Thanks to the new input, I filter the incoming data using a command like `rtl_sdr  | rtl-wmbus | grep [IDs]` .

The use case is to get with a single running instance all the data from my flat (6 Heating Cost Allocators).

weetmuts, 2019-03-03:
Yes, it would be trivial to add wildcard support to the meter specification (like *), 
so that all meters of that particular type are reported by wmbus. 
This assumes that they are not encrypted, or they all have the same encryption key.

Perhaps in addition to *, one could specify multiple numbers and ranges?
id=12345678,12345679,12347777-12348888 

What does the distribution of the ids in your use case look like?

kidpixo, 2019-03-03:
In my specific case I end up with 

```rtl_sdr | rtl-wmbus | grep '94740[24][05][4-8]'```

All my sensors (and some neighbours friends of mine) start with `94740` : this could be a producer way-of-doing thing OR the building management bought them in block and they end up being sequential.

The list if ids would be great, but don't you use the user defined sensor id in the output? 

For my specific case (refer to #8 for more details), they all have the same manufacturer and configuration, that specify at once if they use encryption or not.  The sensors produce 2 telegrams, one not encrypted for sure.

weetmuts, 2019-03-05:
Ok, I added support for multiple comma separated ids: 12345678,12345679,123456791 
or a single * wildcard to match any ids.

The multiple ids and the wildcard can be used both in config files and on the command line.
But protect the wildcard on the command line '*' as usual.

The output will have the same meter name for all meters, but the id will of course change.

Please test to see if this is what you need.

weetmuts, 2019-03-20:
Did it work?

kidpixo, 2019-03-20:
Hi, sorry for the delay! 
I'm currently traveling in USA and don't have the usb adapter with me, I did some quick test  at home but didn't get to work! I'll try next week again.

kidpixo, 2019-04-11:
yes , it works!

weetmuts, 2019-04-11:
Woho! Great!

kidpixo, 2019-04-12:
Thank you for your support! 
Now, I have 2 thing in my plan : 

- add the output to [Home Assistant dev docs](https://developers.home-assistant.io/) via mqtt messaging
- get the encryption key for my Heaters.

weetmuts, 2019-04-26:
Nice! I will close this issue then. Bibo added this info on hass integration. Perhaps useful to you.
https://github.com/BIBOLV/wmbusmeters/wiki/Home-automation

closed 2019-04-26 by weetmuts