wmbusmeters/wmbusmeters #12: Decode of iperl wmbus data

mikkel75, 2019-03-11:
Hi,

I now have my iPerl water meter unencrypted, but I still see some strange warning by wmbusmeters.

> 
(serial /dev/ttyUSB0) received "A5E2031844AE4C4225222068077A55000000041389E20100023B00000000000091B3FB"
(im871a) timestamp 00000000
(im871a) rssi 91
(im871a) got crc16 fbb3 expected fbb3
(wmbus) received telegram 20222542 C-field=44 M-field=4cae (SEN) A-field-version=68 A-field-dev-type=07 (Water meter) Ci-field=7a (EN 13757-3 Application Layer with short Transport Layer) CC-field=00 (slow_resp) ACC=55
(wmbus) frame "1844AE4C4225222068077A55000000041389E20100023B0000"
(wmbus) payload "041389E20100023B0000"
(wmbus) 00: 18 length (24 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: ae4c m-field (4cae=SEN)
(wmbus) 04: 42252220 a-field-addr (20222542)
(wmbus) 08: 68 a-field-version
(wmbus) 09: 07 a-field-type (Water meter)
(wmbus) 0a: 7a ci-field (EN 13757-3 Application Layer with short Transport Layer)
(wmbus) 0b: 55 acc
(wmbus) 0c: 00 status ()
(wmbus) 0d: 0000 configuration ()
(wmbus) 1844AE4C4225222068077A55000000
(meter) iperl: for me? 20222542
(meter) iperl: not for me: not my id
(meter) iperl2: for me? 20222542
(meter) iperl2: yes for me
(iperl) telegram for iperl2 20222542
(meter_iperl) warning: telegram payload does not start with 2F2F (did you use the correct encryption key?)
(iperl) log "telegram=|1844AE4C4225222068077A55000000|041389E20100023B0000|+7045"
(iperl) 0f: 04 dif (32 Bit Integer/Binary Instantaneous value)
(iperl) 10: 13 vif (Volume l)
(iperl) 11: * 89E20100 total consumption (123.529000 m3)
(iperl) 15: 02 dif (16 Bit Integer/Binary Instantaneous value)
(iperl) 16: 3B vif (Volume flow l/h)
(iperl) 17: 0000
(iperl) 1844AE4C4225222068077A55000000041389E20100023B0000
iperl2  20222542         123.529 m3     2019-03-11 13:42.26
> 

It seems to decode the m3 correct, but why the warning?

Thanks!

weetmuts, 2019-03-11:
Aha! Apparently the meter only adds the 2f2f markers when the data is actually encrypted.
Fixed that and pushed it. Try again, and you should see no warnings.

mikkel75, 2019-03-12:
Thanks - much better now!

`
(serial /dev/ttyUSB0) received "A5E2031844AE4C4225222068077A670000000413CFE20100023B0000000000008AC681"
(im871a) timestamp 00000000
(im871a) rssi 8a
(im871a) got crc16 81c6 expected 81c6
(wmbus) received telegram 20222542 C-field=44 M-field=4cae (SEN) A-field-version=68 A-field-dev-type=07 (Water meter) Ci-field=7a (EN 13757-3 Application Layer with short Transport Layer) CC-field=00 (slow_resp) ACC=67
(wmbus) frame "1844AE4C4225222068077A670000000413CFE20100023B0000"
(wmbus) payload "0413CFE20100023B0000"
(wmbus) 00: 18 length (24 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: ae4c m-field (4cae=SEN)
(wmbus) 04: 42252220 a-field-addr (20222542)
(wmbus) 08: 68 a-field-version
(wmbus) 09: 07 a-field-type (Water meter)
(wmbus) 0a: 7a ci-field (EN 13757-3 Application Layer with short Transport Layer)
(wmbus) 0b: 67 acc
(wmbus) 0c: 00 status ()
(wmbus) 0d: 0000 configuration ()
(wmbus) 1844AE4C4225222068077A67000000
(meter) iperl: for me? 20222542
(meter) iperl: yes for me
(iperl) telegram for iperl 20222542
(iperl) log "telegram=|1844AE4C4225222068077A67000000|0413CFE20100023B0000|+300"
(dvparser) found new format "0413023B" with hash c944, remembering!
(iperl) 0f: 04 dif (32 Bit Integer/Binary Instantaneous value)
(iperl) 10: 13 vif (Volume l)
(iperl) 11: * CFE20100 total consumption (123.599000 m3)
(iperl) 15: 02 dif (16 Bit Integer/Binary Instantaneous value)
(iperl) 16: 3B vif (Volume flow l/h)
(iperl) 17: 0000
(iperl) 1844AE4C4225222068077A670000000413CFE20100023B0000
iperl   20222542         123.599 m3     2019-03-12 07:41.29
`

mikkel75, 2019-03-12:
Question - the lines

(iperl) 16: 3B vif (Volume flow l/h)
(iperl) 17: 0000

Does that mean the use was 0000 since last data?

weetmuts, 2019-03-12:
Nice! Unfortunately I do not know how the iperl meter calculates the flow value.

I think that the meters are usually configured to give useful data to drivebys, which
happen once per week at a random time or so. So the flow is perhaps a max or min or median
flow over the last month or so..... so if it is a new meter, that explains the zero reading.

If you are monitoring every single meter value, then you can see the flow from there instead.

weetmuts, 2019-03-12:
I added the max_flow_m3h key to the json output and the extra colums to the other output.

weetmuts, 2019-03-20:
I

closed 2019-03-20 by weetmuts