wmbusmeters/wmbusmeters #13: Add support for Tauron T1 power meters

marcinkowalczyk, 2019-03-19:
Hi, 

 In Poland Tauron (one of 4 main energy distributors) replaces old Energy Meters to new ones with wireless m-bus devices. Over web you can enable wireless mbus broadcast plus get get port and encryption key. I've captured data from my meter but unfortunatly format is not recognized. There are 3 meter manufactures but all of them support same protocol.

```
(serial /dev/ttyUSB0) received "A5E2034E4401063712048602027A000040057FE1A953DEFF6D2DCF7547BA0E6809F7345D4BF9F1F70C97F85316E524514AAD96D8B9A70E8460F5BFF6CCF3319476188CA57F5F76F616120015DFA39657E4A60000000039693C"
(im871a) timestamp 00000000
(im871a) rssi 39
(im871a) got crc16 3c69 expected 3c69
(wmbus) received telegram 86041237 C-field=44 M-field=0601 (APA) A-field-version=02 A-field-dev-type=02 (Electricity meter) Ci-field=7a (EN 13757-3 Application Layer with short Transport Layer) CC-field=00 (slow_resp) ACC=00
(wmbus) frame "4E4401063712048602027A000040057FE1A953DEFF6D2DCF7547BA0E6809F7345D4BF9F1F70C97F85316E524514AAD96D8B9A70E8460F5BFF6CCF3319476188CA57F5F76F616120015DFA39657E4A6"
(wmbus) payload "7FE1A953DEFF6D2DCF7547BA0E6809F7345D4BF9F1F70C97F85316E524514AAD96D8B9A70E8460F5BFF6CCF3319476188CA57F5F76F616120015DFA39657E4A6"
(wmbus) 00: 4e length (78 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 0106 m-field (601=APA)
(wmbus) 04: 37120486 a-field-addr (86041237)
(wmbus) 08: 02 a-field-version
(wmbus) 09: 02 a-field-type (Electricity meter)
(wmbus) 0a: 7a ci-field (EN 13757-3 Application Layer with short Transport Layer)
(wmbus) 0b: 00 acc
(wmbus) 0c: 00 status ()
(wmbus) 0d: 4005 config (encrypted AES_CBC)
(wmbus) 4E4401063712048602027A00004005
(meter) MyTapWater: for me? 86041237
(meter) MyTapWater: not for me: no media match
(im871a) telegram ignored by all configured meters!
```

Tauron provided full specs http://amiplus.tauron-dystrybucja.pl/o-amiplus/han/Documents/TAURON_AMIplus-modul_Wireless_M-Bus_public_v2.1_FIN.pdf but they are in polish only.
Meter is configured to T1 slave

TX data options

OBIS code | Description | DIF (hex)  | VIF (hex)  | Format        | Example 
---------- | ------------- | ---------- | ------------- | ---------- | -------------
0.0:1.0.0.255   | Timestamp                            | 06         | 6D             | CP48          | -
0.0:C.1.0.255   | S/N                                  | 0C         | 78             | 8 digits BCD  |  12345678
1.0:1.8.0.255   | Energy from GRID                     | 0E         | 03             | 12 digits BCD | 000000000.000 kWh
1.0:1.8.1.255   | Energy from GRID                     | 8E, 10     | 03             | 12 digits BCD | 000000000.000 kWh
1.0:1.8.2.255   | Energy from GRID                     | 8E, 20     | 03             | 12 digits BCD | 000000000.000 kWh
1.0:1.8.3.255   | Energy from GRID                     | 8E, 30     | 03             | 12 digits BCD | 000000000.000 kWh
1.0:1.8.4.255   | Energy from GRID                     | 8E, 80, 10 | 03             | 12 digits BCD | 000000000.000 kWh
1.0:2.8.0.255   | Energy to GRID                       | 0E         | 83, 3C         | 12 digits BCD | 000000000.000 kWh
1.0:2.8.1.255   | Energy to GRID                       | 8E, 10     | 83, 3C         | 12 digits BCD | 000000000.000 kWh
1.0:2.8.2.255   | Energy to GRID                       | 8E, 20     | 83, 3C         | 12 digits BCD | 000000000.000 kWh
1.0:2.8.3.255   | Energy to GRID                       | 8E, 30     | 83, 3C         | 12 digits BCD | 000000000.000 kWh
1.0:2.8.4.255   | Energy to GRID                       | 8E, 80, 10 | 83, 3C         | 12 digits BCD | 000000000.000 kWh
1.0:1.6.0.255   | Maximum power from GRID              | 1B         | 2B             | 6 digits BCD  |  000.000 kW
1.0:1.7.0.255   | Power from GRID - value on end of TX | 0B         | 2B             | 6 digits BCD  |  000.000 kW
1.0:2.7.0.255   | Power to GRID - value on end of TX   | 0B         | AB, 3C         | 6 digits BCD  |  000.000 kW
1.0:32.7.0.255  | Voltage - phase 1                    | 0A         | FD, C9, FC, 01 | 4 digits BCD  |  0000 V
1.0:52.7.0.255  | Voltage - phase 2                    | 0A         | FD, C9, FC, 02 | 4 digits BCD  |  0000 V
1.0:72.7.0.255  | Voltage - phase 3                    | 0A         | FD, C9, FC, 03 | 4 digits BCD  |  0000 V
1.0:15.35.0.255 | Current power limit                  | 0B         | AB, C8, FC, 10 | 6 digits BCD  |  000.000 kW

Depending on your tariff plan energy from grid will use different OBIS codes (table 6 rows 3-12)

Can you please let me know if I can use this project and if so how can I proceed ?

Regards
Marcin

weetmuts, 2019-03-19:
Interesting! It sounds very good that it is supported to get the key to your own meter to do your own readings.

I have added a stub for the amiplus meter. Try the latest code with:
./build/wmbusmeters --debug auto MyElectricity amiplus 86041237 YOURKEY

and post the output here (collect a couple of telegrams) or email me the output. 

If the key and my changes are correct, then we can see the content of the packet being decoded. 
When that works, then we can select how to store the content into the JSON format.

marcinkowalczyk, 2019-03-20:
Hi,

 Sorry I missed your comment earlier:

```(serial /dev/ttyUSB0) received "A5E2034E4401063712048602027A00004005556B3555E30155A21023D24D5ACA64A258992A27886E8EB8D1AF08CC3F41ECF51216902D6B32330A4064E886148C9B6A6C57AEEB7D0534BBA836DD33C69882540000000038EC9E"
(im871a) timestamp 00000000
(im871a) rssi 38
(im871a) got crc16 9eec expected 9eec
(wmbus) received telegram 86041237 C-field=44 M-field=0601 (APA) A-field-version=02 A-field-dev-type=02 (Electricity meter) Ci-field=7a (EN 13757-3 Application Layer with short Transport Layer) CC-field=00 (slow_resp) ACC=00
(wmbus) frame "4E4401063712048602027A00004005556B3555E30155A21023D24D5ACA64A258992A27886E8EB8D1AF08CC3F41ECF51216902D6B32330A4064E886148C9B6A6C57AEEB7D0534BBA836DD33C6988254"
(wmbus) payload "556B3555E30155A21023D24D5ACA64A258992A27886E8EB8D1AF08CC3F41ECF51216902D6B32330A4064E886148C9B6A6C57AEEB7D0534BBA836DD33C6988254"
(wmbus) 00: 4e length (78 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 0106 m-field (601=APA)
(wmbus) 04: 37120486 a-field-addr (86041237)
(wmbus) 08: 02 a-field-version
(wmbus) 09: 02 a-field-type (Electricity meter)
(wmbus) 0a: 7a ci-field (EN 13757-3 Application Layer with short Transport Layer)
(wmbus) 0b: 00 acc
(wmbus) 0c: 00 status ()
(wmbus) 0d: 4005 config (encrypted AES_CBC)
(wmbus) 4E4401063712048602027A00004005
(meter) MyElectricity: for me? 86041237
(meter) MyElectricity: yes for me
(amiplus) MyElectricity 86041237 (Mode5) decrypting "556B3555E30155A21023D24D5ACA64A258992A27886E8EB8D1AF08CC3F41ECF51216902D6B32330A4064E886148C9B6A6C57AEEB7D0534BBA836DD33C6988254"
(Mode5) IV 01063712048602020000000000000000
(Mode5) warning: telegram payload does not start with 2F2F (did you use the correct encryption key?)
(Mode5) decrypted "091770E4C3D59D0A0184C39852F81A8E5FE48B82DA9F581CD47F0357267D4CCCDEEED0990A180B0A31537F2852F23AB5AA10EAD444CB2A654EB009C0E41666BD"
(amiplus) log "telegram=|4E4401063712048602027A00004005|091770E4C3D59D0A0184C39852F81A8E5FE48B82DA9F581CD47F0357267D4CCCDEEED0990A180B0A31537F2852F23AB5AA10EAD444CB2A654EB009C0E41666BD|+42"
(dvparser) warning: unexpected end of data (dife expected)
(dvparser) warning: unexpected end of data (vif expected)
(dvparser) found new format "0917E4C3D59D0A01F81A8E5FE48B82DA9F581C267D0A1831532852F23AB5AA1044CB2AC0E41666BD" with hash 8129, remembering!
(dvparser) warning: cannot extract double from non-existant key "04833B"
(amiplus) 0f: 09 dif (2 digit BCD Instantaneous value)
(amiplus) 10: 17 vif (Volume 10¹ m³)
(amiplus) 11: 70
(amiplus) 12: E4 dif (32 Bit Integer/Binary Minimum value storagenr=1)
(amiplus) 13: C3 dife (subunit=1 tariff=0 storagenr=7)
(amiplus) 14: D5 dife (subunit=3 tariff=4 storagenr=167)
(amiplus) 15: 9D dife (subunit=3 tariff=20 storagenr=6823)
(amiplus) 16: 0A dife (subunit=3 tariff=20 storagenr=88743)
(amiplus) 17: 01 vif (Energy 10⁻² Wh)
(amiplus) 18: 84C39852
(amiplus) 1c: F8 dif (Selection for Readout Value during error state storagenr=1)
(amiplus) 1d: 1A dife (subunit=0 tariff=1 storagenr=21)
(amiplus) 1e: 8E vif (Energy MJ)
(amiplus) 1f: 5F vife (?)
(amiplus) 20: E4 dif (32 Bit Integer/Binary Minimum value storagenr=1)
(amiplus) 21: 8B dife (subunit=0 tariff=0 storagenr=23)
(amiplus) 22: 82 dife (subunit=0 tariff=0 storagenr=87)
(amiplus) 23: DA dife (subunit=4 tariff=16 storagenr=5207)
(amiplus) 24: 9F dife (subunit=4 tariff=80 storagenr=128087)
(amiplus) 25: 58 dife (subunit=20 tariff=336 storagenr=1176663)
(amiplus) 26: 1C vif (Mass 10¹ kg)
(amiplus) 27: D47F0357
(amiplus) 2b: 26 dif (48 Bit Integer/Binary Minimum value)
(amiplus) 2c: 7D vif (?)
(amiplus) 2d: 4CCCDEEED099
(amiplus) 33: 0A dif (4 digit BCD Instantaneous value)
(amiplus) 34: 18 vif (Mass g)
(amiplus) 35: 0B0A
(amiplus) 37: 31 dif (8 Bit Integer/Binary Value during error state)
(amiplus) 38: 53 vif (Mass kg/h)
(amiplus) 39: 7F
(amiplus) 3a: 28 dif (Selection for Readout Minimum value)
(amiplus) 3b: 52 vif (Mass 10⁻¹ kg/h)
(amiplus) 3c: F2 dif (16 Bit Integer/Binary Value during error state storagenr=1)
(amiplus) 3d: 3A dife (subunit=0 tariff=3 storagenr=21)
(amiplus) 3e: B5 vif (Power 10⁵ J/h)
(amiplus) 3f: AA vife (?)
(amiplus) 40: 10 vife (?)
(amiplus) 41: EAD4
(amiplus) 43: 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
(amiplus) 44: CB vif (Volume flow ext. cm³/s)
(amiplus) 45: 2A vife (?)
(amiplus) 46: 654EB009
(amiplus) 4a: C0 dif (No data Instantaneous value storagenr=1)
(amiplus) 4b: E4 dife (subunit=1 tariff=2 storagenr=9)
(amiplus) 4c: 16 dife (subunit=1 tariff=6 storagenr=201)
(amiplus) 4d: 66 vif (External temperature 10⁻¹ °C)
(amiplus) 4e: BD dif (variable length Value during error state)
(amiplus) 4E4401063712048602027A00004005091770E4C3D59D0A0184C39852F81A8E5FE48B82DA9F581CD47F0357267D4CCCDEEED0990A180B0A31537F2852F23AB5AA10EAD444CB2A654EB009C0E41666BD
MyElectricity   86041237         0.000 kwh       0.000 kwh      2019-03-20 11:37.56
```

weetmuts, 2019-03-20:
Great! I added extraction of the consumption and production (backward flow?) of energy and power.
Some more data from the telegram is going to be added, voltage and upper limit of consumption?
But I need to improve my difvif parser a bit first.

Human readable output:
MyElectricity	86041237	 15694.050 kwh	 0.330 kw	 7.480 kwh	 0.000 kw	2019-03-20 22:18.35

JSON output:
{"media":"electricity","meter":"amiplus","name":"MyElectricity","id":"86041237","total_energy_consumption_kwh":15694.050000,"current_power_consumption_kw":0.330000,"total_energy_production_kwh":7.480000,"current_power_production_kw":0.000000,"device_date_time":"2019-03-20 12:57","timestamp":"2019-03-20T21:20:02Z"}

Please check if it works for you!

marcinkowalczyk, 2019-03-20:
Hi,

 Just checked 

```(serial /dev/ttyUSB0) received "A5E2034E4401063712048602027A000040058065C3E96130B68F5BE7C32790A5B3E9C29FC9F80D9929BBD6390DB6940B4E7F3AA6021BFB7E5666EF87224375AC5005DA188ED62085B5A266C8207FF5CDEB9C000000003700BE"
(im871a) timestamp 00000000
(im871a) rssi 37
(im871a) got crc16 be00 expected be00
(wmbus) received telegram 86041237 C-field=44 M-field=0601 (APA) A-field-version=02 A-field-dev-type=02 (Electricity meter) Ci-field=7a (EN 13757-3 Application Layer with short Transport Layer) CC-field=00 (slow_resp) ACC=00
(wmbus) frame "4E4401063712048602027A000040058065C3E96130B68F5BE7C32790A5B3E9C29FC9F80D9929BBD6390DB6940B4E7F3AA6021BFB7E5666EF87224375AC5005DA188ED62085B5A266C8207FF5CDEB9C"
(wmbus) payload "8065C3E96130B68F5BE7C32790A5B3E9C29FC9F80D9929BBD6390DB6940B4E7F3AA6021BFB7E5666EF87224375AC5005DA188ED62085B5A266C8207FF5CDEB9C"
(wmbus) 00: 4e length (78 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 0106 m-field (601=APA)
(wmbus) 04: 37120486 a-field-addr (86041237)
(wmbus) 08: 02 a-field-version
(wmbus) 09: 02 a-field-type (Electricity meter)
(wmbus) 0a: 7a ci-field (EN 13757-3 Application Layer with short Transport Layer)
(wmbus) 0b: 00 acc
(wmbus) 0c: 00 status ()
(wmbus) 0d: 4005 config (encrypted AES_CBC)
(wmbus) 4E4401063712048602027A00004005
(meter) MyElectricity: for me? 86041237
(meter) MyElectricity: yes for me
(amiplus) MyElectricity 86041237 (Mode5) decrypting "8065C3E96130B68F5BE7C32790A5B3E9C29FC9F80D9929BBD6390DB6940B4E7F3AA6021BFB7E5666EF87224375AC5005DA188ED62085B5A266C8207FF5CDEB9C"
(Mode5) IV 01063712048602020000000000000000
(Mode5) decrypted "2F2F0E031900701500000B2B280400066D005E167423400C78371204860BABC8FC100000000E833C8074000000000BAB3C0000000AFDC9FC0139022F2F2F2F2F"
(amiplus) log "telegram=|4E4401063712048602027A00004005|2F2F0E031900701500000B2B280400066D005E167423400C78371204860BABC8FC100000000E833C8074000000000BAB3C0000000AFDC9FC0139022F2F2F2F2F|+47"
(dvparser) found new format "0E030B2B066D0C780BABC8FC100E833C0BAB3C0AFDC9FC01" with hash fae5, remembering!
(dvparser) found key 0E03 for EnergyWh and storagenr=0
(dvparser) found key 0B2B for PowerW and storagenr=0
(dvparser) found key 066D for DateTime and storagenr=0
(amiplus) 0f: 2F skip
(amiplus) 10: 2F skip
(amiplus) 11: 0E dif (12 digit BCD Instantaneous value)
(amiplus) 12: 03 vif (Energy Wh)
(amiplus) 13: * 190070150000 total energy (15700.019000 kwh)
(amiplus) 19: 0B dif (6 digit BCD Instantaneous value)
(amiplus) 1a: 2B vif (Power W)
(amiplus) 1b: * 280400 current power (0.428000 kw)
(amiplus) 1e: 06 dif (48 Bit Integer/Binary Instantaneous value)
(amiplus) 1f: 6D vif (Date and time type)
(amiplus) 20: * 005E16742340 device datetime (2019-03-20 22:30)
(amiplus) 26: 0C dif (8 digit BCD Instantaneous value)
(amiplus) 27: 78 vif (Fabrication no)
(amiplus) 28: 37120486
(amiplus) 2c: 0B dif (6 digit BCD Instantaneous value)
(amiplus) 2d: AB vif (Power W)
(amiplus) 2e: C8 vife (upper limit)
(amiplus) 2f: FC vife (additive correction constant: unit of VIF * 10^-3)
(amiplus) 30: 10 vife (?)
(amiplus) 31: 000000
(amiplus) 34: 0E dif (12 digit BCD Instantaneous value)
(amiplus) 35: 83 vif (Energy Wh)
(amiplus) 36: 3C vife (backward flow)
(amiplus) 37: * 807400000000 total energy returned (7.480000 kwh)
(amiplus) 3d: 0B dif (6 digit BCD Instantaneous value)
(amiplus) 3e: AB vif (Power W)
(amiplus) 3f: 3C vife (backward flow)
(amiplus) 40: * 000000 current power returned (0.000000 kw)
(amiplus) 43: 0A dif (4 digit BCD Instantaneous value)
(amiplus) 44: FD vif (Second extension of VIF-codes)
(amiplus) 45: C9 vife (10^0 Volts)
(amiplus) 46: FC vife (Reserved)
(amiplus) 47: 01 vife (Credit of 10^-2 of the nominal local legal currency units)
(amiplus) 48: 3902
(amiplus) 4a: 2F skip
(amiplus) 4b: 2F skip
(amiplus) 4c: 2F skip
(amiplus) 4d: 2F skip
(amiplus) 4e: 2F skip
(amiplus) 4E4401063712048602027A000040052F2F0E031900701500000B2B280400066D005E167423400C78371204860BABC8FC100000000E833C8074000000000BAB3C0000000AFDC9FC0139022F2F2F2F2F
MyElectricity   86041237         15700.019 kwh   0.428 kw        7.480 kwh       0.000 kw       2019-03-20 21:29.57
```

weetmuts, 2019-03-22:
Nice. Are the actual readings what you expect as well?

marcinkowalczyk, 2019-03-22:
Hi,

Haven't check the meter it self but they look ok. 
Thx!

majekw, 2019-03-26:
I have meter from another manufacturer (Echelon) which works with 'amiplus', but wmbusmeters issues `not for me`  warnings from `(meter)` part.

This is output from `wmbusmeters --verbose ...`
```
(wmbus) received telegram 000xxxxx C-field=44 M-field=10b6 (DEV) A-field-version=02 A-field-dev-type=37 (Unknown) Ci-field=7a (EN 13757-3 Application Layer with short Transport Layer) CC-field=00 (slow_resp) ACC=29
(meter) Licznik: probably not for me since manufacturer differs
(meter) Licznik: probably not for me since media does not match
(amiplus) Licznik 000xxxxx Licznik      000xxxxx         13432.232 kwh   0.581 kw        0.002 kwh       0.000 kw       2019-03-26 22:46.58
```

weetmuts, 2019-04-01:
Hi @majekw ! 

I have added support for having amiplus a generic meter with multiple manufacturers
and added your meter manufacturer to it. Please check if you no longer see the warning.

Also it would be very valuable to have a log telegram=|...| line from your meter so that I can add it
to the regression testing. If you pick a single such line and post it here or email it to me.
You cross out (x) your serial nr if you like, then it looks something like this:

(amiplus) log "telegram=|4E440106xxxxxxxx02027A00004005|2F2F0E031440691500000B2B850300066D00730C7423400C78371204860BABC8FC100000000E833C8074000000000BAB3C0000000AFDC9FC0136022F2F2F2F2F|+34"

FeatureExpert, 2019-04-01:
@weetmuts Works perfectly, thank you. Please find log from my meter below (DEV also):

(amiplus) log "telegram=|4E44B610xxxxxxxx02377A72004005|2F2F0C7825120300066D4067006224400E033321750900000E833C0000000000000B2B4704000BAB3C0000000AFDC9FC0142020BABC8FC100000002F2F2F2F2F|+855"

majekw, 2019-04-02:
@weetmuts , works better now, but `not form me since media does not match` remained:
```
(meter) tauron: probably not for me since media does not match
(amiplus) log "telegram=|4E44B610xxxxxxxx02377AC9004005|2F2F0C7871490200066D41430C6224400E032310491300000E833C0200000000000B2B2202000BAB3C0000000AFDC9FC0139020BABC8FC100000002F2F2F2F2F|+2203"
```

weetmuts, 2019-04-02:
Odd, the two telegrams that you posted actually seems to come from a concentrator/gateway or something. That is why the sender has device type 0x37 (Radio Converter (meter side)) and not 0x02 (electricity meter).

Also the two telegrams you posted, actually come from two different meters...
9752.133 kwh	 0.447 kw	 0.000 kwh	 0.000 kw
13491.023 kwh	 0.222 kw	 0.002 kwh	 0.000 kw

Is this what you excepted?

majekw, 2019-04-02:
I think you mixed my meter reading with one provided by FeatureExpert :-)

As for device type, it may be correct as  documentation of this project said that some meters are equipped with additional module for wm-bus communication. I suppose that it could be M-BUS to WM-BUS 'radio converter'.

From documentation:
> Komunikacja wireless M-BUS w licznikach AMI producenta NES i Mitsubishi odbywa się za
pomocą zintegrowanego w obrysie licznika modułu komunikacyjnego, wykorzystującego port
MEP (Multipurpose Extension Port) licznika, natomiast w licznikach AMI producenta APATOR
jest to moduł komunikacyjny osadzony na płytce obwodowej licznika.

Which could be translated to something like this:
> Wireless M-BUS communications in AMI meters from NES and Mitsubishi manufacturers takes place by integrated modules using MEP port inside meter. On Apator meters communication module is embedded on main board.

Keep in mind that there are also Echelon meters (as in my case) used in this project not mentioned in documentation.

Shouldn't this issue be reopened? :-)

weetmuts, 2019-04-02:
Aha! So many users of wmbusmeters that I cant keep them apart! :-) 

Very interesting! Thank you for translating the manual extracts.

Please try now. I added the 0x37 as an accepted device type for amiplus meters.

majekw, 2019-04-02:
It works now without warnings :+1: 
Shell command is also ok.

Would it be a big problem to add voltage readings from DIF 0A, VIF FD C9 FC 01?

weetmuts, 2019-04-04:
I will add the voltage reading, I just have to improve the parser again. 
The parsing is probably not quite correct for DIF 0A, VIF FD C9 FC 01
credit local currency for voltage? does not make sense... :-)

majekw, 2019-04-04:
```
(amiplus) 43: 0A dif (4 digit BCD Instantaneous value)
(amiplus) 44: FD vif (Second extension of VIF-codes)
(amiplus) 45: C9 vife (10^0 Volts)
(amiplus) 46: FC vife (Reserved)
(amiplus) 47: 01 vife (Credit of 10^-2 of the nominal local legal currency units)
(amiplus) 48: 3902
```

Yes, parsing maybe is not perfect, but 0239V is correct value:-)

closed 2019-08-25 by marcinkowalczyk

radiowy, 2021-05-16:
Hi,
I try to send to domoticz the values read by wmbusmeter. My readings are correct from the meter.

tail: /var/log/wmbusmeters/meter_readings/MyTapEl: file truncated
{"media":"electricity","meter":"amiplus","name":"MyTapEl","id":"00072738","total_energy_consumption_kwh":24594.162,"current_power_consumption_kw":0.713,"total_energy_production_kwh":0,"current_power_production_kw":0,"device_date_time":"2021-05-16 22:41","timestamp":"2021-05-6T20:41:02Z","device":"rtlwmbus[00000001]","rssi_dbm":141}

MQTT is too much for me.
My command for the MQTT broker is:

_shell=/usr/bin/mosquitto_pub -h localhost -m '{"idx":91,"svalue": MY_PROBLEM!!}' -t 'domoticz/in'_
What do I have to enter to read in the MY_PROBLEM field !! so that I can send it to domoticz? When I enter a number in parentheses, eg "765", it sends me to the numerator value 765.

I'd like to read "total_energy_consumption_kwh"

Any ideas?

Thank you