wmbusmeters/wmbusmeters #14: Suggestion: Add support for iobroker

mikkel75, 2019-03-21:
Hi,

This is a great tool - but wouldn't it be even greater if it was possible to integrate into iobroker?
I know there is a wmbus plugin for iobroker, but that is old and not supported anymore...I would so much love to see this great tool as an adapter in iobroker ;)

Any thoughts?

/M

weetmuts, 2019-03-21:
Sounds fun! 

What is the best way for a device like wmbusmeters to integrate with iobroker?

Should wmbusmeters broadcast something so that wmbusmeters can be detected by iobroker
and then be autoconfigered from the iobroker side?

Or should wmbusmeters listen for iobroker signals, and then automatically register with iobroker
when it hears that there is an iobroker on the network?

Or should you simply supply an iobroker REST endpoint to wmbusmeters on the command line?

BIBOLV, 2019-04-28:
Hi,
Try following:
1. Install MQTT Broker/Client adapter
2. Configure and start MQTT adapter
3. Add to /etc/wmbusmeters.conf 
`shell=/usr/bin/mosquitto_pub -h localhost -t wmbusmeters/$METER_ID -m "$METER_TOTAL_M3"`
4. When wmbusmeters sends data, check under Objects -> mqtt.0/wmbusmeters/[meter_id] and you should see value reported

weetmuts, 2019-05-04:
@mikkel75 did you manage to integrate with iobroker?

weetmuts, 2019-05-21:
The howto is added to the wiki. Closing this for now.

closed 2019-05-21 by weetmuts