wmbusmeters/wmbusmeters #16: apator162 - offset of total water consumption value changes over time

jupiter9595, 2019-03-27:
Hi,

Thank you very much for your work!
You wrote:
"Added support for the at-wmbus-16-2 snap on meter (apator162).
Unfortunately it uses a vendor specific protocol,
that is merely wrapped inside a wmbus telegram. Lets hope
the offset to the consumption is stable between
meters of this type." 

I use iM871A usb dongle

I've been strangling for some time to decode more then total water consumption. Without any success. Although over the year I've collected some data from 2 Apator meter. We could join our forces to decode Apator's proprietary  protocol.

Unfortunately total water consumption changes unexpectedly over the time (there might be some logic but it happens to me after 4 months of use)

Initial location (offset you use in your code)
                                                                                                                   
                                                                          Total - 281C                                              |0000802F2F0F|688|9D|6929|00|210432|E0|0101C280|0007|10106|1B00|00DC|0000|00A01A86|AF03|FFFF|FFFF|FFFF    09-06-2018

Location around 10-2018
                                                                                                                         Total - 73F0
|0000802F2F0F|D97|3B|A929|80|20043B|20|083000D5|A566|E5A01|8C34|0010|F073|00007101|AA72|0000|AA65|0000

Of course I am not aware; is it time driven, water value driven or what...

weetmuts, 2019-04-02:
We can try, but I must say that it is probably quite difficult to get it working right.

What is the result when you use wmbusmeters to read your apator162?
Do you get any good values at all?

weetmuts, 2019-04-04:
"Unfortunately total water consumption changes unexpectedly over the time (there might be some logic but it happens to me after 4 months of use)"

Was this because the storage location offset inside the message for the water value changed, or did the water value actually change?

jupiter9595, 2019-04-04:
Hi To respond to your questions:

"What is the result when you use wmbusmeters to read your apator162?
Do you get any good values at all?"  - I observe something like described by bdkacz - one value is correct one completely wrong. Your logic in a code is correct. I can try to find the issue. The data received from iM871A usb dongle is really stable. Actually I modified code from https://github.com/ffcrg/ecpiww to decode water consumption only. As static offset , same you did.

jupiter9595, 2019-04-04:
your second question:

"Was this because the storage location offset inside the message for the water value changed, or did the water value actually change?"  - storage location offset inside the message just changed one day without any known reason. Since I could look on the meter counter I could find the little endian coded value in the message.

weetmuts, 2019-04-08:
It seems like for @bdkacz, most of the problems disappeared with the latest rtl_wmbus.
But this does not mean that there are no other problems with the Apator decoding. :-)

To debug the problem, we need the telegrams when wmbusmeters report odd values from the Apator.

jupiter9595, 2019-04-09:
Let me update my rtl_wmbus as well. With iM871A I use ecpiww software. This stick always decodes correct telegrams. Water consumption is fully correct. The offset changed only once and has been stable in new position.

To push forward our investigation:

Attached are telegrams from Apator from 2 days, no water consumption.

There should be only time / date change + there might be some kind of error occurred 2nd date

[apator.txt](https://github.com/weetmuts/wmbusmeters/files/3058985/apator.txt)
In a file the following marks stand for:
  WTot - is location of Total water consumption
V - fields which change

Over midnight AE changed to B2

I've been trying to compare these with known epoc time codding - without success.

Please take a look. Its like Mensa type logic pattern ;)))

weetmuts, 2019-04-15:
Hmm those telegrams are really not changing that much.
I just added a utility to track changes in telegrams. Do this:
javac -d utils/ utils/Analyze.java
java -cp utils/ Analyze apator_trimmed.txt

Where apator_trimmed.txt contains:
2F2F0FB44AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0FC24AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0FCA4AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0FD74AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0F026AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0F156AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0F216AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0F286AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0F366AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0F3B6AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0F4A6AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0F556AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0F736AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0F8D6AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0F9E6AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0FC06AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0FCC6AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0F558AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0F5D8AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0F688AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0F8B8AE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0FC0AAE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0FC0AAE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0FCEAAE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0FD5AAE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0FE1AAE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0FEAAAE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0FF2AAE39398000043400183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0F020B239398000043410183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0F0A0B239398000043410183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0F140B239398000043410183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0F210B239398000043410183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000
2F2F0F324B239398000043410183000D5A47496303B7DF0B10E9BE00007101CEBE0000CEBE0000

weetmuts, 2019-04-15:
You will get the following output. (3) means byte three. (Byte 0 is 2f byte 1 is 2f byte 2 is 0f.)
So byte 3 changes almost every time. byte 4 sometimes, and then suddenly 3,4,5 and 11.
 (3)C2
 (3)CA
 (3)D7
 (3)026A
 (3)15
 (3)21
 (3)28
 (3)36
 (3)3B
 (3)4A
 (3)55
 (3)73
 (3)8D
 (3)9E
 (3)C0
 (3)CC
 (3)558A
 (3)5D
 (3)68
 (3)8B
 (3)C0AA

 (3)CE
 (3)D5
 (3)E1
 (3)EA
 (3)F2
 (3)020B23 (11)41
 (3)0A
 (3)14
 (3)21
 (3)324B

weetmuts, 2019-04-15:
So there does not seem to be any big offset change in where the total water is stored in the file you sent me, nor does it look like there is any actual water flow measured either?

What is the current offset where you find your water consumption? What was it before?
Does wmbusmeters decode it correctly? I.e is wmbusmeters hard-coded offset the same as yours?

weetmuts, 2019-04-26:
Ping @jupiter9595 , any update on this?

jupiter9595, 2019-05-04:
Sorry for the late reply.

"So there does not seem to be any big offset change in where the total water is stored in the file you sent me, nor does it look like there is any actual water flow measured either?" - yes, there is no water consumption in telegrams I sent. However, looks like clock data (time / date) and so other unknown data is changing. I sent it to further investigate other data in the telegram. Personally, I think it is really worth to find error codes. (like water leakage).

"What is the current offset where you find your water consumption?"
                                                                                                                          WTot                                       |0000802F2F0F|B44|AE|3939|80|000434|00|183000D5|A474|96303|B7DF|0B10|E9BE|
Current offset for this meter is flaged by WTot - BEE9 > 48 873 dm3

"What was it before?" - same as in code

"Does wmbusmeters decode it correctly? I.e is wmbusmeters hard-coded offset the same as yours?" - I had to modify your offset to decode new Water Total location.

A new meter you buy from manufacture (Apator) has offset located as in your code.

After some time or after reaching some water consumption (I can not tell why it changed) it changes to a new location (as in the telegrams I included)

weetmuts, 2019-08-12:
Hi jupiter9595! I have just pushed code that looks at a specific byte in the apator162
header, this byte might be the one that tells us which offset to use for
the total water consumption. 

Please test!

jupiter9595, 2019-08-16:
Oh, good will test.
In addition to your finding; I guess I've found [counter - days since last reset] in Apator telegram. Where to share my finding?

jupiter9595, 2019-08-16:
Here it is what I 've checked so far:

2019-08-10 11:53, 26.185, 2F2F|0F|F24AD8930E0200437C0082CD7974774A78FF2C3F0010496600007501E65400009717000004010000040100000000000000000000
2019-08-10 12:00, 26.185, 2F2F|0F|F94AD8930E0200437C00830010755F537601E3880410496600007501E65400009717000004010000040100000000000000000000
2019-08-11 17:37, 26.185, 2F2F|0F|628BF8930E0200437D00830010755F537601E3880410496600007501E65400009717000004010000040100000000000000000000
2019-08-16 16:06, 26.904, 2F2F|0F|0390B8930E020043820082CD7974EC4D78FFE43F0010186900007501E65400009717000004010000040100000000000000000000
                          
                          
                          2F|2F|0F|F2|4A|D8|93|0E|02|00|43|7C|00|82|CD|79|74|77|4A|78|FF|2C|3F|00|10|49|66|00|00|75|01|E6|54|00|00|97|17|00|00|04|01|00|00|04|01|00000000000000000000
                          2F|2F|0F|F9|4A|D8|93|0E|02|00|43|7C|00|83|00|10|75|5F|53|76|01|E3|88|04|10|49|66|00|00|75|01|E6|54|00|00|97|17|00|00|04|01|00|00|04|01|00000000000000000000
                          2F|2F|0F|62|8B|F8|93|0E|02|00|43|7D|00|83|00|10|75|5F|53|76|01|E3|88|04|10|49|66|00|00|75|01|E6|54|00|00|97|17|00|00|04|01|00|00|04|01|00000000000000000000
                          2F|2F|0F|03|90|B8|93|0E|02|00|43|82|00|82|CD|79|74|EC|4D|78|FF|E4|3F|00|10|18|69|00|00|75|01|E6|54|00|00|97|17|00|00|04|01|00|00|04|01|00000000000000000000
                           0  1  2  3  4  5  6  7  8  9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36

if (t->content[13] == 0x83) -> not always true. At least in data I 've reviewed there is also 0x82

if (t->content[13] == 0x10 -> true for all historical data I have

weetmuts, 2019-09-01:
Thanks! I added 0x82 to the apator162 code.

weetmuts, 2019-09-01:
@jupiter9595 Please share any finds right here, if you want them added to wmbusmeters! 
:-)

bdkacz, 2019-09-04:
Hi..
After update from 12.08 I have problem  with one meter - every second reading it sends '0' as value to domoticz

weetmuts, 2019-09-06:
Edit wmbusmeters.conf so loglevel=debug
then reload the configuration: sudo systemctl reload wmbusmeters.service
or restart it.

Let it run for a while to capture the fluctuating values, then send me the log
/var/log/wmbusmeters/wmbusmeters.log

Then you can switch back to normal loglevel.

bdkacz, 2019-09-06:
Log  file https://www.dropbox.com/s/9bazdi5p378yr96/wmbusmeters.log?dl=0

ID of meter that working ok - 01592846
ID of meter with fluctuating - 01637014

The rest of the meters in logfile  belong to my neighbors :)

weetmuts, 2019-09-06:
Thanks! I have a suspicion of what the problem might be. I have pushed a potential fix.
Can you try?

bdkacz, 2019-09-06:
Hi. After 6 readings always corect value,seems fixed :) Thank You :)

bdkacz, 2019-09-06:
Hi again. Can You take a look also on this log https://www.dropbox.com/s/9bazdi5p378yr96/wmbusmeters.log?dl=0 and meter 01759628 ? I get from it two values = 0 or 4286138,04 where  value on meter itself is 64,510

weetmuts, 2019-09-06:
Cool, those readings have yet another offset where the total is located. 
A simple fix makes it print properly.

The current assumption is bit 0x80->offset25    0x40->offset22    0x10->offset14.

Please try.

bdkacz, 2019-09-06:
Wow,how fast :)  I can confirm - after update the readings of all my meters are now ok :)

weetmuts, 2019-10-16:
Lets close this for now. Open a new issue if it happens again!

closed 2019-10-16 by weetmuts