wmbusmeters/wmbusmeters #17: Protocol error

JareckiB12, 2019-03-30:
Hey,
I compiled 
`git://git.osmocom.org/rtl-sdr.git `
and
`https://github.com/xaelsouth/rtl-wmbus`
The
`rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus`
command works fine and returns messages:
<pre>C1;0;1;2019-03-30 00:08:00.000;34;54;55150040;0x27442d2c4000155555055dd0450e0ec9c70a91808211311820206758e532c492410100e66c296910
T1;0;0;2019-03-30 00:08:20.000;14;37;83714654;0x2e44ffff544671837062a0009f252901e0270000000000000200000000010001000101010101020202040302040504
T1;0;0;2019-03-30 00:08:23.000;66;132;71708009;0x2e446850098070717072a0009f252904e0274b00ff08100c0e0f0d0e0d0e10070c0f0c0e0f0f0d0c0dffffffffff0b
T1;1;1;2019-03-30 00:08:28.000;19;36;83607818;0x2e446850187860837062a0009f25e001e0271400000605060404050605050505020b090a0606080905080200060709
C1;0;1;2019-03-30 00:08:31.000;17;15;67244331;0x27442d2c314324673004022298e521b9ac6b29ea03a8a2cb6544a5552904050820814041c001d274
T1;1;1;2019-03-30 00:08:45.000;32;34;83719309;0x2e446850099371837072a0009f250d06e02769000010110b1717131d1713170c0d141215141b101814181713171714
T1;1;1;2019-03-30 00:08:52.000;18;34;83714654;0x2e446850544671837062a0009f252901e0270000000000000200000000010001000101010101020202040302040504
T1;0;0;2019-03-30 00:08:54.000;67;173;83602033;0x2e446850332060837072a0009f25d904e027860000141814151a151515181aff120701090b07050a0100000b12ff14
T1;1;1;2019-03-30 00:08:54.000;17;31;71708005;0x2e446850058070717072a0009f255701e02724000005060c0b070d09090c0c070e0f0e0b070a090200010100010102
</pre>

However, calling wmbusmeters:
`wmbusmeters --debug "rtlwmbus:868.9M"`
I get an error:
<pre>(wmbusmeters) version: 0.9.2
(config) using device: rtlwmbus
(config) with: 868.9M
(config) number of meters: 0
(rtlwmbus) using command: rtl_sdr -f 868.9M -s 16000000 - 2>/dev/null | rtl_wmbus
exec background "/bin/sh"
arg "-c"
arg "rtl_sdr -f 868.9M -s 16000000 - 2>/dev/null | rtl_wmbus"
(serialcmd) opened /bin/sh
(config) using link mode: Any
No meters configured. Printing id:s of all telegrams heard!

(serialcmd) received "3A20313A2072746C5F776D6275733A206E6F7420666F756E640A"
(rtl_wmbus) protocol error in message received!
(rtl_wmbus) protocol error "3A20313A2072746C5F776D6275733A206E6F7420666F756E640A"</pre>
 regardless of whether the tuner is inserted into the port or not. What am I doing wrong?

weetmuts, 2019-04-02:
Ah, its because rtl_wmbus prints C1;0;1; and not C1;1;1; or T1;0;0 and not T1;1;1
I am not sure, but I think that means that one,or two of the two crc checks inside the packet failed.

Try if this works:
wmbusmeters "rtlwmbus:rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus | grep 1;1;1" 

to only allow the packets with good crcs to be passed to wmbusmeters

JareckiB12, 2019-04-02:
I solved the problem. I had to copy rtl_wmbus to the folder /usr/bin Adding the rtl_wmbus to PATH did not help. You must have a hardcoded path somewhere. In this case, you could log an error about the lack of access to the rtl_wmbus. Anyway, thanks for your willingness to help.

closed 2019-04-02 by JareckiB12

weetmuts, 2019-04-02:
If you run wmbusmeters as daemon service, then it will prefix /usr/bin to the binaries. Because which PATH should be used by the daemon? (See main.cc line look for case DEVICE_RTLWMBUS:)

If running from the command line, it will simply pick it from the PATH.

But you can also specify the exact command line in the config file for the daemon like this:
device=rtlwmbus:/usr/mybin/foo/rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | /usr/mybin/foo/rtl_wmbus

This way, you can place your binaries anywhere and still call them from the daemon. :-)

But checking that the binary is in place and executable is a good suggestion, I'll do that.

JareckiB12, 2019-04-02:
Everything is clear now. Thanks!

weetmuts, 2019-04-03:
I have now pushed an update that properly prints an error message to stdout or log file when wmbusmeters fails to spawn the background process.