wmbusmeters/wmbusmeters #18: apator162 + rtl_sdr

bdkacz, 2019-04-03:
Hi. I use rtl_sdr usb  dongle to listen to apator at-wmbus-16-2. I add device as :

name=WMBUS_APA_01761839
type=apator162
id=*
key=00000000000000000000000000000000

and after a while in log I get : 

(serialcmd) received "54313B313B313B323031392D30342D30332031393A30303A34322E3030
303B39373B3134383B30313631373630373B30783665343430313036303737363631303130353037
65396438376138353030363038356263323633303731333831393531326562346364383763623035
66626135353466623433663637636639363534613638656538653139343038383936626631363064
66373532653731363233383239326538616631616332303938363230636537313265653536316437
34333630323436363931356534326631313035643963363763643930383261353435303465346630
39396536356137363536623933306337336133306639346537373531323264326664663037346235
303335636661613765303035306266333032363132666161653033613737393839623B3078366534
34303130363037373636313031303530373761383530303630383562633236333037313338313935
31326562346364383766626135353466623433663637636639363534613638656538653139343038
38313630646637353265373136323338323932653861663161633230393836323032656535363164
37343336303234363639313565343266313130356439633637383261353435303465346630393965
36356137363536623933306337336133303737353132326432666466303734623530333563666161
376530303530626633326661616530336137370A
(meter) WMBUS_APA_01761839: for me?
(meter) WMBUS_APA_01761839: probably not for me since manufacturer differs
(meter) WMBUS_APA_01761839: probably not for me since media does not match
(meter) WMBUS_APA_01761839: yes for me"

but  /var/log/wmbusmeters/meter_readings is empty - why ?

weetmuts, 2019-04-04:
I improved the logging from wmbusmeters, so that the output from (serialcmd) is actually
readable. Now it is easier to see what rtl_wmbus actually printed.

I decoded the output here and looks odd. The line seems to contain two telegrams... but the second
one starts abruptly in the middle of the first. The second one is ok, but the first has
a strange ci-field and other strange stuff.

Please run again with the improved logging to see if it happens again.

This is the decoded serial data from above, as you can see there is a ;0x in the middle. Which 
I do not think it should be there.
T1;1;1;2019-04-03 19:00:42.000;97;148;01617607;0x6e440106077661010507e9d87a85006085bc2630713819512eb4cd87cb05fba554fb43f67cf9654a68ee8e19408896bf160df752e716238292e8af1ac2098620ce712ee561d743602466915e42f1105d9c67cd9082a54504e4f099e65a7656b930c73a30f94e775122d2fdf074b5035cfaa7e0050bf302612faae03a77989b;0x6e4401060776610105077a85006085bc2630713819512eb4cd87fba554fb43f67cf9654a68ee8e194088160df752e716238292e8af1ac20986202ee561d743602466915e42f1105d9c6782a54504e4f099e65a7656b930c73a30775122d2fdf074b5035cfaa7e0050bf32faae03a77

bdkacz, 2019-04-04:
Hi. After update I can get values from metters. But there is a problem - for the same id i get two much different values for example 

{"media":"water","meter":"apator162","name":"ciepla_m","id":"01637014","total_m3":929815.243000,"timestamp":"2019-04-
04T16:40:37Z"}
{"media":"water","meter":"apator162","name":"ciepla_m","id":"01637014","total_m3":6.690000,"timestamp":"2019-04-04T16
:41:00Z"}

Where :6.690000 is acorrect  value from meter itself,but value 929815.243000 is from outer space :)

weetmuts, 2019-04-04:
I think this is a result of our limited understanding of the apator162 protocol. Unfortunately the apator162 does not really comply with the wmbus protocol. It simply wraps a proprietary protocol blob inside a single wmbus telegram. Though properly marked as vendor specific, according to the wmbus protocol, but that does not really help....

So the current implementation simply has a hard-coded offset into the proprietary blob and hopes
that the data is there. Apparently this is not true for all telegrams.

If we close this issue and join forces on the other issue opened by jupiter9595, perhaps
we can make some progress on understanding the apator162 protocol.
But I am somewhat sceptical. A perhaps achievable goal would be to always
find the water meter value location, or discard those telegrams where the value is not there....

weetmuts, 2019-04-04:
If you can collect logs with --logtelegram where the bad behaviour pops up, that would be helpful.

jupiter9595, 2019-04-04:
I need to test more apator162 + rtl_sdr option. I can guarantee data your read from  iM871A usb dongle is fully stable and offset does not change every message telegram. However, yes there is offset change but probably due to some other data collection (after some time of usage). When the meter is new (just after wmbus module being paired with the meter ) the offset is as weetmuts coded.

bdkacz, 2019-04-04:
Ok. I will collect more data with --logtelegram and upload logs  after ~24h

jupiter9595, 2019-04-04:
I think from rtl_sdr you should select only T1;1;1; messages

bdkacz, 2019-04-05:
Here are the logs after 24h : https://www.dropbox.com/s/njekjm20maapob8/log.zip?dl=0

Problems that I detected :

1)  for one meter  wrong assotiation id with metter for example   once in log we have

meter) inne: for me? 01637014
(meter) inne: yes for me

but after a while 

(meter) ciepla_d: for me? 01637014
(meter) ciepla_d: yes for me

where ciepla_d i defined as id 01608150 (and values for this id is also recorded in file for ciepla_d)  so 01637014 should always be add as inne (inne is defined as "id=*" )

2)  After reading the logs, it follows that every second reading of each meter  after decoding gives the correct value

weetmuts, 2019-04-08:
Thanks @bdkacz ! I have started looking at the logs. One thing that I noticed is that the data from your compiled rtl_wmbus looks like this:
T1;1;1;2019-04-04 19:49:01.000;17;28;01766289;0x4e4401....a875d;0x4e4401068962......783732a<0A>

I.e. you got to versions of the telegram on the same line. The first 0x4e.. is probable with
embedded crcs and the second without. When I read the rtl_wmbus code
https://github.com/xaelsouth/rtl-wmbus/blob/master/t1_c1_packet_decoder.h
this can happen if the code is changed in  t1_c1_packet_decoder, and the first #if 0 is
changed to #if 1, or something.

But wmbusmeters only expect one telegram per line, with the crcs removed. So I need to
deal with this multiple versions per line as well.

Are you perhaps using an older version of rtl_wmbus? Since the latest rtl_wmbus does
not print the two by default.

bdkacz, 2019-04-08:
Ok I will update to latest version of rtl_wmbus and post results after some time of working :)

bdkacz, 2019-04-08:
After first 10 readings looks like update of rtl_wmbus fix the problem with abnormal readings - all records in  log file have correct value/

Ps can You hep with with integration your softwre  with Domoticz ?

weetmuts, 2019-04-08:
Ok, lets close this issue post, since it works for you right now with the latest rtl_wmbus. If you experience odd output again, post in tje issue that @jupiter9595  create.

closed 2019-04-08 by weetmuts