wmbusmeters/wmbusmeters #19: Send to domoticz

bdkacz, 2019-04-07:
Hi. 

Can  somebody post some info how to send meter value to domoticz via MQTT ? How  to pass device idx from domoticz etc

weetmuts, 2019-04-08:
Do you have some link to domoticz documentation? Is a device idx different from the serial nr of the meter?

bdkacz, 2019-04-08:
Hi. Here is how Domoticz can talk with MQTT  https://www.domoticz.com/wiki/MQTT#MQTT_to_Domoticz It's also possible to talk directly via json and http https://www.domoticz.com/wiki/Domoticz_API/JSON_URL's

BIBOLV, 2019-04-27:
HowTo add water meter to domoticz (did some basic tests and was working for me, please try)
Create virtual meter in domoticz

   1. Create dummy hardware and device
       1.1 Go to Setup -> Hardware
          1.1.1  Name: anything, e.g. "wmbusmeter_sensors" (this name is only used internally)
          1.1.2  Type: "Dummy (does nothing, use for virtual switches only)"
       1.2 Click on 'Add'
       1.3 Click on 'Create Virtual Sensors' from the table of hardware above
          1.3.1  Name: anything, e.g. "Water Consumption" (this name is displayed on the dashboard etc.)
          1.3.2  Sensor type: "Custom Sensor"
		  1.3.3  Axis Label: "m3"
       1.4 Click 'Ok'
   2. Get idx of relevant virtual sensors
      2.1  Go to Setup -> Devices
      2.2  Note the Idx of your virtual sensor
   3. Setup MQTT 
      3.1  Go to Setup -> Hardware
	      3.1.1  Name: anything, e.g "MQTT"
		  3.1.2  Type: MQTT Client Gateway with LAN interface
		  3.1.3  Remote Address: IP/name of MQTT server, e.g. "localhost"
		  3.1.4  Port: 1883
		  3.1.5  Everything else should be ok, so click on the Add button at the bottom. 

In /etc/wmbusmeters.conf add MQTT configuration and change idx from 2.2
`shell=/usr/bin/mosquitto_pub -h localhost -t domoticz/in -m "{\"idx\":2,\"svalue\":\"$METER_TOTAL_M3\"}"`

When data is sent by wmbusmeters check in domoticz WUI under Utility tab if created sensor is updated with new value

bdkacz, 2019-04-27:
What if I have multiple meters ? For example for hot and cold water ?

BIBOLV, 2019-04-27:
Well you can try hack/workaround:
1. Stop domoticz!!!!
2. Find location of your domoticz installation and open db - `sqlite3 domoticz.db`
3. Select from db your sensor just to verify that it is correct one (ID is idx from gui) - `SELECT * FROM DeviceStatus WHERE ID = 2;`
4. Update ID to ID of meter that you can get from wmbusmeters logs - `UPDATE DeviceStatus SET ID = 01234567 WHERE ID = 2;`
5. Select once more meter to verify changes - `SELECT * FROM DeviceStatus WHERE ID = 01234567;`
6. Exit db - `.exit`
7. Start domoticz 
8. Change wmbusmeters config to - `shell=/usr/bin/mosquitto_pub -h localhost -t domoticz/in -m "{\"idx\":$METER_ID,\"svalue\":\"$METER_TOTAL_M3\"}"`

bdkacz, 2019-04-27:
Works great ! Thank You very much ! :)

closed 2019-04-28 by weetmuts

zalewma, 2019-05-10:
Hi Guys,
For the domoticz integration and other I would see an option to add it directly to meter config which will overwrite "default" settings.
It that case /etc/wmbusmeters.conf  is for "global configuration" and can be overwritten by:
/etc/wmbusmeters.d/MeterName which contains shell="XXXXX" with proper ID's.

What do you think?

weetmuts, 2019-05-21:
@zalewma Yep, good idea. I have pushed support for overriding the shell command in wmbusmeters.conf with a meter specific shell command. Please try.

weetmuts, 2019-05-21:
@zalewma But do you have a specific use case? $METER_ID is replaced in the command line with the meters ID. But perhaps you intended to override some other ids tailored for domoticz?

majekw, 2019-05-21:
In my case overriding shell is good as I receive messages from Tauron meter and also from water meters. I just need to post it to differently constructed topics. Right now I'm calling  script with 'if..else' to differentiate it, but with  direct shell in meter section I would be able to get rid off this scripts.

zalewma, 2019-05-21:
@weetmuts  Exactly as @majekw  wrote. Different types of meters have different filds and external command can look differently.  I made a test and shell command defined from meter works fine. Thanks for the feature :)