wmbusmeters/wmbusmeters #20: Suggestion: Add support for Water Sappel/IZAR RC 868

zalewma, 2019-04-17:
Starting a new thread to enable water monitoring device . It uses  IZAR RC  868 protocol over wm-bus. It's used in Wroclaw/Poland.

Datagram example:

(serial /dev/im871a) received "A5E2031944304CF8BED640CC01A2011500137E02C45F17397158677C5D00000000678528"
(im871a) timestamp 00000000
(im871a) rssi 67
(im871a) got crc16 2885 expected 2885
Unknown ci-field a2
(wmbus) received telegram 40d6bef8 C-field=44 M-field=4c30 (SAP) A-field-version=cc A-field-dev-type=01 (Oil meter) Ci-field=a2 (Mfct specific)
(wmbus) frame "1944304CF8BED640CC01A2011500137E02C45F17397158677C5D"
(wmbus) payload "011500137E02C45F17397158677C5D"
(wmbus) 00: 19 length (25 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 304c m-field (4c30=SAP)
(wmbus) 04: f8bed640 a-field-addr (40d6bef8)
(wmbus) 08: cc a-field-version
(wmbus) 09: 01 a-field-type (Oil meter)
(wmbus) 0a: a2 ci-field (Mfct specific)
(wmbus) 1944304CF8BED640CC01A2
(meter) EnergyMeter: for me? 40d6bef8
(meter) EnergyMeter: not for me: not my id
(im871a) telegram ignored by all configured meters!

zalewma, 2019-04-17:
When I tried currently available I got invalid "Not a valid meter id"

$:/app/wmbusmeters/build# ./wmbusmeters /dev/ttyUSB1 water multical21 40d6bef8 0
Not a valid meter id "40d6bef8"
$:/app/wmbusmeters/build# ./wmbusmeters /dev/ttyUSB1 water multical21 40d6bef8 0
Not a valid meter id "40d6bef8"
$:/app/wmbusmeters/build# ./wmbusmeters /dev/ttyUSB1 water flowiq3100 40d6bef8 0
Not a valid meter id "40d6bef8"
$:/app/wmbusmeters/build# ./wmbusmeters /dev/ttyUSB1 water supercom587 40d6bef8 0
Not a valid meter id "40d6bef8"

weetmuts, 2019-04-17:
Well, it looks like your meter sends a ci-field type that means manufacturer specific (0xa2).
Also the meter seems to send using a meter id that is:  40d6bef8
but the meter id is supposed to be BCD, ie it should be only digits 0-9, not hex.
This part is never encrypted.

Unfortunately 0xa2 would mean that most of the content is probably a non-standard protocol format, not wmbus standard.  I tried to decode it, but the standard dvparser does not give meaningful results.
It might be that it is encrypted, or not.....

We can try to reverse engineer it though... :-) but we need more log telegrams to do so.
First, can you collect a longer --debug log when you know that the volume it is measuring is changing?
Second, it says that it is an oil meter (not a water meter), are you monitoring your oil meter?
Thirdly, did you use --t1 or --c1 to detect the meter?

zalewma, 2019-04-17:
Hi Fredrik, thanks for your update. That device is water meter at my home - it's named as DIEHL Metering SAPPEL with additional radio module - SAPPEL "IZAR RC 868 I R4 PL"  Looks like this: https://www.diehl.com/metering/en/diehl-metering/portfolio/system-software/3fb656d6-8150-11e8-9935-00505683653e

Based on IZAR RC documentation there is a possibility to "AES 128-Bit encryption of the data" - for that I have asked my water supplier for encryption key - we will see how it goes.

Within my device range I have a lot of datagrams from this water meter. Snippet below:

pi@raspberrypi:/app/wmbusmeters/build $ ./wmbusmeters --t1  --debug /dev/ttyUSB0
(wmbusmeters) version: 0.9.4-1-g957c47c
(config) using device: /dev/ttyUSB0
(config) number of meters: 0
(serialtty) opened /dev/ttyUSB0
(amb8465) are you there?
(serial /dev/ttyUSB0) sent "FF0B00F4"
(serialtty) closed /dev/ttyUSB0
(serialtty) closed /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(im871a) are you there?
(serial /dev/ttyUSB0) sent "A5010100"
(serial /dev/ttyUSB0) received "A58102004CA3"
(serialtty) closed /dev/ttyUSB0
(im871a) got crc16 a34c expected a34c
(serialtty) closed /dev/ttyUSB0
(im871a) detected on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(im871a) set link mode 03
(serial /dev/ttyUSB0) sent "A5010306000203300101"
(serial /dev/ttyUSB0) received "A58104009CF7"
(im871a) got crc16 f79c expected f79c
(im871a) set config completed
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial /dev/ttyUSB0) received "A5810617FF000300B32535151000010001FF0702320001010000029401"
(im871a) got crc16 0194 expected 0194
(im871a) get config completed
(im871a) config: device mode 00
(im871a) config: link mode 03
(im871a) config: wmbus c-field 00
(im871a) config: wmbus mfg id 25b3 (IMS)
(im871a) config: wmbus device id 00101535
(im871a) config: wmbus version 01
(im871a) config: wmbus device type 00
(im871a) config: radio channel 01
(im871a) config: radio power level 07
(im871a) config: radio data rate 02
(im871a) config: radio rx window 32
(im871a) config: auto power saving 00
(im871a) config: auto RSSI attachment 01
(im871a) config: auto rx timestamp attachment 01
(im871a) config: led control 00
(im871a) config: rtc control 00
(config) using link mode: T1
No meters configured. Printing id:s of all telegrams heard!

(serial /dev/ttyUSB0) received "A5E2031944304CF8BED640CC01A2411500137E907979BBBE3E3F0B18570000000067C8CA"
(im871a) timestamp 00000000
(im871a) rssi 67
(im871a) got crc16 cac8 expected cac8
Unknown ci-field a2
(wmbus) received telegram 40d6bef8 C-field=44 M-field=4c30 (SAP) A-field-version=cc A-field-dev-type=01 (Oil meter) Ci-field=a2 (Mfct specific)
(wmbus) frame "1944304CF8BED640CC01A2411500137E907979BBBE3E3F0B1857"
(wmbus) payload "411500137E907979BBBE3E3F0B1857"
(wmbus) 00: 19 length (25 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 304c m-field (4c30=SAP)
(wmbus) 04: f8bed640 a-field-addr (40d6bef8)
(wmbus) 08: cc a-field-version
(wmbus) 09: 01 a-field-type (Oil meter)
(wmbus) 0a: a2 ci-field (Mfct specific)
(wmbus) 1944304CF8BED640CC01A2
Received telegram from: 40d6bef8
[... 60 lines cut ...]
          manufacturer: (SAP) Sappel,
           device type: Oil meter
(im871a) telegram ignored by all configured meters!
(serial /dev/ttyUSB0) received "A5E2031944304C97A4C740CC01A22113001322E216A971ADE53CDE5AE3000000002F89FA"
(im871a) timestamp 00000000
(im871a) rssi 2f
(im871a) got crc16 fa89 expected fa89
Unknown ci-field a2
(wmbus) received telegram 40c7a497 C-field=44 M-field=4c30 (SAP) A-field-version=cc A-field-dev-type=01 (Oil meter) Ci-field=a2 (Mfct specific)
(wmbus) frame "1944304C97A4C740CC01A22113001322E216A971ADE53CDE5AE3"
(wmbus) payload "2113001322E216A971ADE53CDE5AE3"
(wmbus) 00: 19 length (25 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 304c m-field (4c30=SAP)
(wmbus) 04: 97a4c740 a-field-addr (40c7a497)
(wmbus) 08: cc a-field-version
(wmbus) 09: 01 a-field-type (Oil meter)
(wmbus) 0a: a2 ci-field (Mfct specific)
(wmbus) 1944304C97A4C740CC01A2
Received telegram from: 40c7a497
          manufacturer: (SAP) Sappel,
           device type: Oil meter
(im871a) telegram ignored by all configured meters!
^C(serialtty) event loop stopped!
(serialtty) closed /dev/ttyUSB0

weetmuts, 2019-04-17:
Thanks, this log contains telegrams with ids from 5 different meters. Do you have some id
printed on the phyiscal izar addon?

One meter did return though so we got:
1944304CF8BED640CC01A2 -- 011500137E02C45F17397158677C5D
1944304CF8BED640CC01A2 -- 411500137E907979BBBE3E3F0B1857

And then we have
1944304CF79EF520D401A2 -- 41380013C69D93263470714EF2C39B
1944304CF7BED640CC01A2 -- 51150013A2DD51F090AA3D860B244F
1944304C90061441CC01A2 -- 111D0013918BE679FD9D59A6BF80F1
1944304C97A4C740CC01A2 -- 2113001322E216A971ADE53CDE5AE3

Everything up to A2 is wmbus standard, but oddly used. 
Then we can see that some values are stable, that should not be the case
if those were encrypted. Assuming the encryption is done correctly.
Then we have a lot of values that change completely, but does not look completely random....

Can you figure out if the meter sends a consistent address (perhaps one of these:
40d6bef8, 20f59ef7, 40d6bef7, 41140690, 40c7a497) and/or if this address is printed
on the physical izar radio addon?

If nothing is printed, then perhaps you can see how the rssi changes for one address, 
when you move your dongle to and from the meter?

zalewma, 2019-04-17:
Hi,  address is correct and it is stable. 
Below you have 3 datagrams from one meter.  

pi@raspberrypi:/app/wmbusmeters/build $  ./wmbusmeters --t1 --debug /dev/ttyUSB0
(wmbusmeters) version: 0.9.4-1-g957c47c

(serial /dev/ttyUSB0) received "A5E2031944304CF8BED640CC01A2511500137EB95630109FEDE6D00155EBB8880068C503"
(im871a) timestamp 0088b8eb
(im871a) rssi 68
(im871a) got crc16 03c5 expected 03c5
Unknown ci-field a2
(wmbus) received telegram 40d6bef8 C-field=44 M-field=4c30 (SAP) A-field-version=cc A-field-dev-type=01 (Oil meter) Ci-field=a2 (Mfct specific)
(wmbus) frame "1944304CF8BED640CC01A2511500137EB95630109FEDE6D00155"
(wmbus) payload "511500137EB95630109FEDE6D00155"
(wmbus) 00: 19 length (25 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 304c m-field (4c30=SAP)
(wmbus) 04: f8bed640 a-field-addr (40d6bef8)
(wmbus) 08: cc a-field-version
(wmbus) 09: 01 a-field-type (Oil meter)
(wmbus) 0a: a2 ci-field (Mfct specific)
(wmbus) 1944304CF8BED640CC01A2
Received telegram from: 40d6bef8
          manufacturer: (SAP) Sappel,
           device type: Oil meter
(im871a) telegram ignored by all configured meters!

(serial /dev/ttyUSB0) received "A5E2031944304CF8BED640CC01A2611500137E8027EAEDFD998CBD2A52E6DE8C00682DC2"
(im871a) timestamp 008cdee6
(im871a) rssi 68
(im871a) got crc16 c22d expected c22d
Unknown ci-field a2
(wmbus) received telegram 40d6bef8 C-field=44 M-field=4c30 (SAP) A-field-version=cc A-field-dev-type=01 (Oil meter) Ci-field=a2 (Mfct specific)
(wmbus) frame "1944304CF8BED640CC01A2611500137E8027EAEDFD998CBD2A52"
(wmbus) payload "611500137E8027EAEDFD998CBD2A52"
(wmbus) 00: 19 length (25 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 304c m-field (4c30=SAP)
(wmbus) 04: f8bed640 a-field-addr (40d6bef8)
(wmbus) 08: cc a-field-version
(wmbus) 09: 01 a-field-type (Oil meter)
(wmbus) 0a: a2 ci-field (Mfct specific)
(wmbus) 1944304CF8BED640CC01A2
Received telegram from: 40d6bef8
          manufacturer: (SAP) Sappel,
           device type: Oil meter
(im871a) telegram ignored by all configured meters!

(serial /dev/ttyUSB0) received "A5E2031944304CF8BED640CC01A2711500137E9708A346DC4A55663350E33A910068CD4D"
(im871a) timestamp 00913ae3
(im871a) rssi 68
(im871a) got crc16 4dcd expected 4dcd
Unknown ci-field a2
(wmbus) received telegram 40d6bef8 C-field=44 M-field=4c30 (SAP) A-field-version=cc A-field-dev-type=01 (Oil meter) Ci-field=a2 (Mfct specific)
(wmbus) frame "1944304CF8BED640CC01A2711500137E9708A346DC4A55663350"
(wmbus) payload "711500137E9708A346DC4A55663350"
(wmbus) 00: 19 length (25 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 304c m-field (4c30=SAP)
(wmbus) 04: f8bed640 a-field-addr (40d6bef8)
(wmbus) 08: cc a-field-version
(wmbus) 09: 01 a-field-type (Oil meter)
(wmbus) 0a: a2 ci-field (Mfct specific)
(wmbus) 1944304CF8BED640CC01A2
Received telegram from: 40d6bef8
          manufacturer: (SAP) Sappel,
           device type: Oil meter
(im871a) telegram ignored by all configured meters!

zalewma, 2019-04-17:
On my meter there are few numbers printed:
C14SB073592
P/N: 3044296
S/N: 04127192
41513463

From another doc: 
https://fccid.io/ANATEL/02190-14-03754/Manual/4FC21C6B-DE83-4650-9FFF-F3BEE74ED500/PDF
I could read that it might contains:
- Standard index: The index transmitted by the radio is a real-time index. It takes into account
any backflow.
- Secondary index: This may be weekly, monthly (i.e. the first of each month) or annual (i.e.
January 1st of each year). This index may be replaced by a backflow-volume metering
index.
- Remaining lifetime of the battery: This is indicated in six-month intervals at each reading.
The typical lifetime is 15

This is not exactly the version which I have - but very simmilar.

zalewma, 2019-04-18:
I also check that my current value on water meter is 00318,611 m^3 but non of the values are corresponding with currently returned datagrams. Maybe it will be 

serial /dev/ttyUSB0) received "A5E2031944304CF79EF520D401A241380013C69D93263470714EF2C39B6A0E114031EC79"
(im871a) timestamp 40110e6a
(im871a) rssi 31
(im871a) got crc16 79ec expected 79ec
Unknown ci-field a2
(wmbus) received telegram 20f59ef7 C-field=44 M-field=4c30 (SAP) A-field-version=d4 A-field-dev-type=01 (Oil meter) Ci-field=a2 (Mfct specific)
(wmbus) frame "1944304CF79EF520D401A241380013C69D93263470714EF2C39B"
(wmbus) payload "41380013C69D93263470714EF2C39B"
(wmbus) 00: 19 length (25 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 304c m-field (4c30=SAP)
(wmbus) 04: f79ef520 a-field-addr (20f59ef7)
(wmbus) 08: d4 a-field-version
(wmbus) 09: 01 a-field-type (Oil meter)
(wmbus) 0a: a2 ci-field (Mfct specific)
(wmbus) 1944304CF79EF520D401A2
Received telegram from: 20f59ef7
          manufacturer: (SAP) Sappel,
           device type: Oil meter
(im871a) telegram ignored by all configured meters!
(serial /dev/ttyUSB0) received "A5E2031944304C90061441CC01A2611D001391162A85AC7862ABBECFFC3BE911403808D3"
(im871a) timestamp 4011e93b
(im871a) rssi 38
(im871a) got crc16 d308 expected d308
Unknown ci-field a2
(wmbus) received telegram 41140690 C-field=44 M-field=4c30 (SAP) A-field-version=cc A-field-dev-type=01 (Oil meter) Ci-field=a2 (Mfct specific)
(wmbus) frame "1944304C90061441CC01A2611D001391162A85AC7862ABBECFFC"
(wmbus) payload "611D001391162A85AC7862ABBECFFC"
(wmbus) 00: 19 length (25 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 304c m-field (4c30=SAP)
(wmbus) 04: 90061441 a-field-addr (41140690)
(wmbus) 08: cc a-field-version
(wmbus) 09: 01 a-field-type (Oil meter)
(wmbus) 0a: a2 ci-field (Mfct specific)
(wmbus) 1944304C90061441CC01A2
Received telegram from: 41140690
          manufacturer: (SAP) Sappel,
           device type: Oil meter
(im871a) telegram ignored by all configured meters!
(serial /dev/ttyUSB0) received "A5E2031944304CF8BED640CC01A2111500137EF2EB16BC18A281BC655F996012405B555F"
(im871a) timestamp 40126099
(im871a) rssi 5b
(im871a) got crc16 5f55 expected 5f55
Unknown ci-field a2
(wmbus) received telegram 40d6bef8 C-field=44 M-field=4c30 (SAP) A-field-version=cc A-field-dev-type=01 (Oil meter) Ci-field=a2 (Mfct specific)
(wmbus) frame "1944304CF8BED640CC01A2111500137EF2EB16BC18A281BC655F"
(wmbus) payload "111500137EF2EB16BC18A281BC655F"
(wmbus) 00: 19 length (25 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 304c m-field (4c30=SAP)
(wmbus) 04: f8bed640 a-field-addr (40d6bef8)
(wmbus) 08: cc a-field-version
(wmbus) 09: 01 a-field-type (Oil meter)
(wmbus) 0a: a2 ci-field (Mfct specific)
(wmbus) 1944304CF8BED640CC01A2
Received telegram from: 40d6bef8
          manufacturer: (SAP) Sappel,
           device type: Oil meter
(im871a) telegram ignored by all configured meters!
(serial /dev/ttyUSB0) received "A5E2031944304CFD70D640CC01A26115401362B754611A6072FAD873AD4C7B13403E906C"
(im871a) timestamp 40137b4c
(im871a) rssi 3e
(im871a) got crc16 6c90 expected 6c90
Unknown ci-field a2
(wmbus) received telegram 40d670fd C-field=44 M-field=4c30 (SAP) A-field-version=cc A-field-dev-type=01 (Oil meter) Ci-field=a2 (Mfct specific)
(wmbus) frame "1944304CFD70D640CC01A26115401362B754611A6072FAD873AD"
(wmbus) payload "6115401362B754611A6072FAD873AD"
[... 127 lines cut ...]
(wmbus) 1944304CF7BED640CC01A2
Received telegram from: 40d6bef7
          manufacturer: (SAP) Sappel,
           device type: Oil meter
(im871a) telegram ignored by all configured meters!
(serial /dev/ttyUSB0) received "A5E2031944304CF89EF520D401A2413800131AC5EEE5B45EFC2D29E6810204194048C786"
(im871a) timestamp 40190402
(im871a) rssi 48
(im871a) got crc16 86c7 expected 86c7
Unknown ci-field a2
(wmbus) received telegram 20f59ef8 C-field=44 M-field=4c30 (SAP) A-field-version=d4 A-field-dev-type=01 (Oil meter) Ci-field=a2 (Mfct specific)
(wmbus) frame "1944304CF89EF520D401A2413800131AC5EEE5B45EFC2D29E681"
(wmbus) payload "413800131AC5EEE5B45EFC2D29E681"
(wmbus) 00: 19 length (25 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 304c m-field (4c30=SAP)
(wmbus) 04: f89ef520 a-field-addr (20f59ef8)
(wmbus) 08: d4 a-field-version
(wmbus) 09: 01 a-field-type (Oil meter)
(wmbus) 0a: a2 ci-field (Mfct specific)
(wmbus) 1944304CF89EF520D401A2
Received telegram from: 20f59ef8
          manufacturer: (SAP) Sappel,
           device type: Oil meter
(im871a) telegram ignored by all configured meters!

zalewma, 2019-04-18:
One more list with longer debug from one meter only:
(wmbus) frame "1944304CF8BED640CC01A2411500137EB67979BBBE3E3F0B1857"
(wmbus) frame "1944304CF8BED640CC01A2511500137EA15630109FEDE6D00155"
(wmbus) frame "1944304CF8BED640CC01A2611500137E9827EAEDFD998CBD2A52"
(wmbus) frame "1944304CF8BED640CC01A2711500137E8F08A346DC4A55663350"
(wmbus) frame "1944304CF8BED640CC01A2011500137EEAC45F17397158677C5D"
(wmbus) frame "1944304CF8BED640CC01A2111500137EFDEB16BC18A281BC655F"
(wmbus) frame "1944304CF8BED640CC01A2211500137EC49ACC417AD6EBD14E58"
(wmbus) frame "1944304CF8BED640CC01A2311500137ED3B585EA5B05320A575A"
(wmbus) frame "1944304CF8BED640CC01A2411500137EB67979BBBE3E3F0B1857"
(wmbus) frame "1944304CF8BED640CC01A2511500137EA15630109FEDE6D00155"
(wmbus) frame "1944304CF8BED640CC01A2611500137E9827EAEDFD998CBD2A52"
(wmbus) frame "1944304CF8BED640CC01A2711500137E8F08A346DC4A55663350"
(wmbus) frame "1944304CF8BED640CC01A2011500137EEAC45F17397158677C5D"
(wmbus) frame "1944304CF8BED640CC01A2111500137EFDEB16BC18A281BC655F"
(wmbus) frame "1944304CF8BED640CC01A2211500137EC49ACC417AD6EBD14E58"
(wmbus) frame "1944304CF8BED640CC01A2311500137ED3B585EA5B05320A575A"
(wmbus) frame "1944304CF8BED640CC01A2411500137EB67979BBBE3E3F0B1857"
(wmbus) frame "1944304CF8BED640CC01A2511500137EA15630109FEDE6D00155"
(wmbus) frame "1944304CF8BED640CC01A2611500137E9827EAEDFD998CBD2A52"
(wmbus) frame "1944304CF8BED640CC01A2711500137E8F08A346DC4A55663350"
(wmbus) frame "1944304CF8BED640CC01A2011500137EEAC45F17397158677C5D"
(wmbus) frame "1944304CF8BED640CC01A2111500137EFDEB16BC18A281BC655F"
(wmbus) frame "1944304CF8BED640CC01A2211500137EC49ACC417AD6EBD14E58"
(wmbus) frame "1944304CF8BED640CC01A2311500137ED4B585EA5B05320A575A"
(wmbus) frame "1944304CF8BED640CC01A2411500137EB17979BBBE3E3F0B1857"
(wmbus) frame "1944304CF8BED640CC01A2511500137EA65630109FEDE6D00155"
(wmbus) frame "1944304CF8BED640CC01A2611500137E9E27EAEDFD998CBD2A52"
(wmbus) frame "1944304CF8BED640CC01A2711500137E8908A346DC4A55663350"
(wmbus) frame "1944304CF8BED640CC01A2011500137EECC45F17397158677C5D"
(wmbus) frame "1944304CF8BED640CC01A2111500137EFBEB16BC18A281BC655F"
(wmbus) frame "1944304CF8BED640CC01A2211500137EC29ACC417AD6EBD14E58"
(wmbus) frame "1944304CF8BED640CC01A2311500137ED5B585EA5B05320A575A"
(wmbus) frame "1944304CF8BED640CC01A2411500137EB07979BBBE3E3F0B1857"
(wmbus) frame "1944304CF8BED640CC01A2511500137EA75630109FEDE6D00155"
(wmbus) frame "1944304CF8BED640CC01A2611500137E9E27EAEDFD998CBD2A52"
(wmbus) frame "1944304CF8BED640CC01A2711500137E8908A346DC4A55663350"
(wmbus) frame "1944304CF8BED640CC01A2011500137EEFC45F17397158677C5D"
(wmbus) frame "1944304CF8BED640CC01A2111500137EF8EB16BC18A281BC655F"
(wmbus) frame "1944304CF8BED640CC01A2211500137EC09ACC417AD6EBD14E58"
(wmbus) frame "1944304CF8BED640CC01A2311500137ED7B585EA5B05320A575A"
(wmbus) frame "1944304CF8BED640CC01A2411500137EB27979BBBE3E3F0B1857"
(wmbus) frame "1944304CF8BED640CC01A2511500137EA55630109FEDE6D00155"
(wmbus) frame "1944304CF8BED640CC01A2611500137E9C27EAEDFD998CBD2A52"
(wmbus) frame "1944304CF8BED640CC01A2711500137E8B08A346DC4A55663350"
(wmbus) frame "1944304CF8BED640CC01A2011500137EEEC45F17397158677C5D"
(wmbus) frame "1944304CF8BED640CC01A2111500137EF9EB16BC18A281BC655F"
(wmbus) frame "1944304CF8BED640CC01A2211500137EC09ACC417AD6EBD14E58"
(wmbus) frame "1944304CF8BED640CC01A2311500137ED7B585EA5B05320A575A"
(wmbus) frame "1944304CF8BED640CC01A2411500137EB27979BBBE3E3F0B1857"
(wmbus) frame "1944304CF8BED640CC01A2511500137EA55630109FEDE6D00155"
(wmbus) frame "1944304CF8BED640CC01A2711500137E8B08A346DC4A55663350"
(wmbus) frame "1944304CF8BED640CC01A2011500137EEEC45F17397158677C5D"
(wmbus) frame "1944304CF8BED640CC01A2111500137EF9EB16BC18A281BC655F"
(wmbus) frame "1944304CF8BED640CC01A2211500137EC09ACC417AD6EBD14E58"
(wmbus) frame "1944304CF8BED640CC01A2311500137ED7B585EA5B05320A575A"
(wmbus) frame "1944304CF8BED640CC01A2411500137ECD7979BBBE3E3F0B1857"
(wmbus) frame "1944304CF8BED640CC01A2511500137EDB5630109FEDE6D00155"
(wmbus) frame "1944304CF8BED640CC01A2611500137EE127EAEDFD998CBD2A52"

After  1944304CF8BED640CC01A2 there is 0-7 iterator then static number 11500137 and repeated code.

weetmuts, 2019-04-18:
Are you sure you have an actual water flow flowing? So that the total consumption counter
should increase? If you put your long log into a file (sappel.txt) like this:
1944304CF8BED640CC01A2411500137EB67979BBBE3E3F0B1857
1944304CF8BED640CC01A2511500137EA15630109FEDE6D00155
.......
1944304CF8BED640CC01A2611500137EE127EAEDFD998CBD2A52

then run java -cp utils Analyze sappel.txt A2011
(it will only analyze lines which contains A2011)
you will see that only byte 16 changes (that is offset 32 in the hex text file)
then try A2111 A2211 A2311 to A2711 and it is still only byte 16.
All the other bytes are repeating statics bound to the iterator.
So all the really useful data in the log can be extracted with:
cat sappel.txt | cut -b 33-34
not enough to store actual meter flow, encrypted or not....
I think we can expect the meter to send the total water consumption
in a single telegram.....not as small bytes over several telegrams.

zalewma, 2019-04-18:
I think also that it contains only total consumption counter, which should increase. However didn't found a pattern to get this value. On analog watter meter I have "00318,611" and metering precision should be the same.  
Counter on the device might have different values and be synchronized later by water provider. 
I can see that it changing - which means it doesn't not only contains rounded consumption m^3 but also some precision after delimiter.

weetmuts, 2019-04-18:
Well, perhaps it is encrypted in CTR mode and no good nonce, or the nonce repeats after 8 telegrams.
Then byte 16 might be the least significant byte in the counter, but encrypted. If you can collect
a more data, then we might see another byte toggle (either byte 15 or 17) when you have consumed more than 256 liters of water.

zalewma, 2019-04-18:
I took all frames from debug starting with "1944304" add have currently +35k lines. Only to see if there is a general rule.

(wmbus) frame "1944304CFD70D640CC01A23115401362D5C20E1DC6EE446F0EA5"
(wmbus) frame "1944304CF8BED640CC01A2211500137EFF9DCC417AD6EBD14E58"
(wmbus) frame "1944304C90061441CC01A2611D001391603085AC7862ABBECFFC"
(wmbus) frame "1944304CF7BED640CC01A201150013A261C69F970CA138BC5947"
(wmbus) frame "1944304CC996F520D401A2013800130DA05A2DD9F8F5A63A545A"
(wmbus) frame "1944304CF89EF520D401A2713800131AFA9F3F493C884744CD86"
(wmbus) frame "1944304CF79EF520D401A261380013C6CDCDB56233D6FD44F19E"
(wmbus) frame "1944304C97A4C740CC01A241130013222EFD1C8B690DE8040CEC"
(wmbus) frame "1944304C3DBFD640CC01A20115001347FF5F9AE1E45AE8E800CC"
(wmbus) frame "1944304CFD70D640CC01A24115401362B00EF24C23D5496E41A8"
(wmbus) frame "1944304CF8BED640CC01A2311500137EE8B285EA5B05320A575A"
(wmbus) frame "1944304C8C061441CC01A2611D0013744DEB021A1D3451D190F8"
(wmbus) frame "1944304C90061441CC01A2711D001391741FCC0759B17265D6FE"
(wmbus) frame "1944304CF89EF520D401A2013800131A9F53C318D9B34A45828B"
(wmbus) frame "1944304CF7BED640CC01A211150013A276E9D63C2D72E1674045"
(wmbus) frame "1944304CC996F520D401A2113800130DB7756472D9267FE14D58"
(wmbus) frame "1944304CF79EF520D401A271380013C6DAE2FCC91205249FE89C"
(wmbus) frame "1944304C52E51341CC01A2011D001322E461160B76DE4A7F6577"
(wmbus) frame "1944304C3DBFD640CC01A21115001347E970D34AC589313319CE"
(wmbus) frame "1944304CFD70D640CC01A25115401362A721BBE7020690B558AA"
(wmbus) frame "1944304CF8BED640CC01A2411500137E8D7E79BBBE3E3F0B1857"
(wmbus) frame "1944304C8C061441CC01A2711D0013745AC44BB13CE7880A89FA"
(wmbus) frame "1944304C90061441CC01A2011D00139110D33056BC8A7F6499F3"
(wmbus) frame "1944304CF7BED640CC01A221150013A24F980CC14F068B0A6B42"
(wmbus) frame "1944304CC996F520D401A2213800130D8E04BE8FBB52158C665F"
(wmbus) frame "1944304CF89EF520D401A2113800131A887C8AB3F860939E9B89"
(wmbus) frame "1944304CF79EF520D401A201380013C6BF2E0098F73E299EA791"
(wmbus) frame "1944304C3DBFD640CC01A22115001347D30109B7A7FD5B5E32C9"
(wmbus) frame "1944304CFD70D640CC01A261154013629E50611A6072FAD873AD"
(wmbus) frame "1944304C52E51341CC01A2111D001322F34E5FA0570D93A47C75"
(wmbus) frame "1944304CF8BED640CC01A2511500137E9A5130109FEDE6D00155"
(wmbus) frame "1944304C90061441CC01A2111D00139100FC79FD9D59A6BF80F1"
(wmbus) frame "1944304CF7BED640CC01A231150013A258B7456A6ED552D17240"
(wmbus) frame "1944304CF89EF520D401A2213800131AB10D504E9A14F9F3B08E"
(wmbus) frame "1944304CF79EF520D401A211380013C6A8014933D6EDF045BE93"
(wmbus) frame "1944304C52E51341CC01A2211D001322CA3F855D3579F9C95772"
(wmbus) frame "1944304C3DBFD640CC01A23115001347C22E401C862E82852BCB"
(wmbus) frame "1944304CFD70D640CC01A27115401362897F28B141A123036AAF"
(wmbus) frame "1944304CF8BED640CC01A2611500137EA320EAEDFD998CBD2A52"
(wmbus) frame "1944304C90061441CC01A2211D001391388DA300FF2DCCD2ABF6"
(wmbus) frame "1944304CF7BED640CC01A241150013A23D7BB93B8BEE5FD03D4D"
(wmbus) frame "1944304CF89EF520D401A2313800131AA62219E5BBC72028A98C"
(wmbus) frame "1944304C97A4C740CC01A2011300132272403A27EE428F6868E6"
(wmbus) frame "1944304CF79EF520D401A221380013C6917093CEB4999A289594"
(wmbus) frame "1944304C3DBFD640CC01A24115001347A7E2BC4D63158F8464C6"
(wmbus) frame "1944304CF8BED640CC01A2711500137EB40FA346DC4A55663350"
(wmbus) frame "1944304CFD70D640CC01A20115401362ECB3D4E0A49A2E0225A2"
(wmbus) frame "1944304C8C061441CC01A2211D001374115624B69A7B36BDF4F2"
(wmbus) frame "1944304CF7BED640CC01A251150013A22A54F090AA3D860B244F"
(wmbus) frame "1944304C90061441CC01A2311D0013912CA2EAABDEFE1509B2F4"
(wmbus) frame "1944304CC996F520D401A2513800130DEBC842DE5E69188D2952"
(wmbus) frame "1944304CF89EF520D401A2413800131AC3EEE5B45EFC2D29E681"
(wmbus) frame "1944304C97A4C740CC01A21113001322656F738CCF9156B371E4"
(wmbus) frame "1944304CF79EF520D401A231380013C6865FDA65954A43F38C96"
(wmbus) frame "1944304CF8BED640CC01A2011500137ED1C35F17397158677C5D"
(wmbus) frame "1944304C3DBFD640CC01A25115001347B0CDF5E642C6565F7DC4"
(wmbus) frame "1944304CFD70D640CC01A21115401362FB9C9D4B8549F7D93CA0"
(wmbus) frame "1944304C8C061441CC01A2311D00137406796D1DBBA8EF66EDF0"
(wmbus) frame "1944304CF7BED640CC01A261150013A213252A6DC849EC660F48"
(wmbus) frame "1944304C90061441CC01A2411D001391486E16FA3BC51808FDF9"
(wmbus) frame "1944304CF89EF520D401A2513800131AD4C1AC1F7F2FF4F2FF83"
(wmbus) frame "1944304CF79EF520D401A241380013C6E393263470714EF2C39B"
(wmbus) frame "1944304C3DBFD640CC01A2611500134789BC2F1B20B23C3256C3"
(wmbus) frame "1944304C52E51341CC01A2511D001322AFF3790CD042F4C8187F"
(wmbus) frame "1944304CF8BED640CC01A2111500137EC6EC16BC18A281BC655F"
(wmbus) frame "1944304CFD70D640CC01A22115401362C2ED47B6E73D9DB417A7"
(wmbus) frame "1944304CF7BED640CC01A271150013A2040A63C6E99A35BD164A"
(wmbus) frame "1944304CC996F520D401A2713800130DC596D1881DCEAB3B1B57"
[... 470 lines cut ...]
(wmbus) frame "1944304CF7BED640CC01A271150013A2000063C6E99A35BD164A"
(wmbus) frame "1944304C90061441CC01A2511D0013912F5B5F511A16C1D3E4FB"
(wmbus) frame "1944304CF89EF520D401A2713800131AFC9F3F493C884744CD86"
(wmbus) frame "1944304C3DBFD640CC01A22115001347BB0C09B7A7FD5B5E32C9"
(wmbus) frame "1944304CF79EF520D401A231380013C6F85FDA65954A43F38C96"
(wmbus) frame "1944304CF8BED640CC01A2611500137EEF27EAEDFD998CBD2A52"
(wmbus) frame "1944304CFD70D640CC01A231154013622BF90E1DC6EE446F0EA5"
(wmbus) frame "1944304CF7BED640CC01A201150013A265CC9F970CA138BC5947"
(wmbus) frame "1944304C90061441CC01A2611D001391162A85AC7862ABBECFFC"
[... cut ...]
(wmbus) frame "1944304C3DBFD640CC01A24115001347C9EFBC4D63158F8464C6"
(wmbus) frame "1944304CF8BED640CC01A2011500137E9DC45F17397158677C5D"
(wmbus) frame "1944304CFD70D640CC01A25115401362591ABBE7020690B558AA"
(wmbus) frame "1944304CF7BED640CC01A221150013A24B920CC14F068B0A6B42"
(wmbus) frame "1944304CF89EF520D401A2213800131AB70D504E9A14F9F3B08E"
(wmbus) frame "1944304CF79EF520D401A261380013C6B3CDB56233D6FD44F19E"
(wmbus) frame "1944304CFD70D640CC01A26115401362606B611A6072FAD873AD"
(wmbus) frame "1944304CF8BED640CC01A2111500137E8AEB16BC18A281BC655F"

zalewma, 2019-04-18:
When I removed from it similar parts I could get 4 characters:
pi@raspberrypi:/var/log/wmbusmeters $ cat  wmbusmeters.log | grep frame| grep 1944304CF8BED640CC01A2011500137| sed -e "s:5F17397158677C5D::;s:1944304CF8BED640CC01A2011500137E::"| uniq
(wmbus) frame "D1C3"
(wmbus) frame "02C4"
(wmbus) frame "97C4"
(wmbus) frame "9AC4"
(wmbus) frame "9DC4"
(wmbus) frame "9CC4"
(wmbus) frame "80C4"
(wmbus) frame "83C4"
(wmbus) frame "82C4"
(wmbus) frame "84C4"
(wmbus) frame "87C4"
(wmbus) frame "89C4"
(wmbus) frame "88C4"
(wmbus) frame "8BC4"
(wmbus) frame "8DC4"
(wmbus) frame "8FC4"
(wmbus) frame "8EC4"
(wmbus) frame "B1C4"
(wmbus) frame "B3C4"
(wmbus) frame "B4C4"
(wmbus) frame "BAC4"
(wmbus) frame "BDC4"
(wmbus) frame "BFC4"
(wmbus) frame "BEC4"
(wmbus) frame "A5C4"
(wmbus) frame "A4C4"
(wmbus) frame "A9C4"
(wmbus) frame "A8C4"
(wmbus) frame "ABC4"
(wmbus) frame "AAC4"
(wmbus) frame "ADC4"
(wmbus) frame "ACC4"
(wmbus) frame "AFC4"
(wmbus) frame "AEC4"
(wmbus) frame "52C5"
(wmbus) frame "56C5"
(wmbus) frame "40C5"
(wmbus) frame "46C5"
(wmbus) frame "4BC5"
(wmbus) frame "4EC5"
(wmbus) frame "77C5"
(wmbus) frame "78C5"
(wmbus) frame "7CC5"
(wmbus) frame "7FC5"
(wmbus) frame "63C5"
(wmbus) frame "62C5"
(wmbus) frame "65C5"
(wmbus) frame "64C5"
(wmbus) frame "67C5"
(wmbus) frame "68C5"
(wmbus) frame "6CC5"
(wmbus) frame "6FC5"
(wmbus) frame "6EC5"
(wmbus) frame "13C5"
(wmbus) frame "14C5"
(wmbus) frame "19C5"
(wmbus) frame "18C5"
(wmbus) frame "1AC5"
(wmbus) frame "09C5"
(wmbus) frame "0EC5"
(wmbus) frame "33C5"
(wmbus) frame "35C5"
(wmbus) frame "37C5"
(wmbus) frame "39C5"
(wmbus) frame "3BC5"
(wmbus) frame "3AC5"
(wmbus) frame "3DC5"
(wmbus) frame "3EC5"
(wmbus) frame "21C5"
(wmbus) frame "20C5"
(wmbus) frame "23C5"
(wmbus) frame "22C5"

nestorayuso, 2019-04-25:
yes, sappel protocol is wmbus mode T1 with a non standard encryption. the encryption key is 64 bit long and almost everybody uses the same default key.

in the data field there is a randon number so frames with the same meter index will produce different encrypted frames.

in the data field there is a fixed value of 0x4B in a fixed position, so after decrypt the frame you can check if the frame is well decrypted or not.

yes, the 6 bytes address field is encoded into Alphabetic characters and digits,

weetmuts, 2019-04-26:
Thanks @nestorayuso, after searching a bit, it seems like the protocol actually has a name.
Could it be the PRIOS protocol for izar readers/meters? (Not that there is any documentation
available on that either, but it feels better if it has a name, and not just a random protocol. :-) )

@zalewma do you have some more logs, where more bytes are changing?

zalewma, 2019-04-27:
Hi @weetmuts , List below:
cat wmbusmeters.log* | grep frame| grep F89EF520D401A20| cut -c47-53| uniq | head
ACD6CC3
ACC6CC3
ACF6CC3
A9F53C3
A9953C3
A9B50C3
A8550C3
AB750C3
AA150C3
AD350C3

cat wmbusmeters.log* | grep frame| grep F89EF520D401A20| cut -c47-53| uniq | tail
AA46FC3
AD76FC3
AC76FC3
AF66FC3
AE66FC3
A116FC3
A016FC3
A306FC3
A236FC3
A446FC3

zalewma, 2019-04-27:
I found one interesting thing, on MS Azure there is image with IZAR@NET application .
For the moment I found few things. Encryption usage mode:
-- first user device individual keys.
-- If device has no individual key, try global individual keys (defaultRfKeys)
-- If none of the global individual keys work, use hydrometer specific keys (csi-internal keys)
-- (basically that is what the bitmask 0x010601 does

Now the individual keys are per product steam:
	<manufacturer code="SAP" name="Sappel">
		<device deviceKey="HYDRUS_SAP" generation="25" alarmType="hydrus" />
		<device deviceKey="CETASHEAT_SAP" generation="26" alarmType="ray" />
		<device deviceKey="CETASHEAT_III_SAP" generation="29"
			alarmType="ray" />
		<device deviceKey="SHARKY_2A_SAP" generation="2A" alarmType="sharky" />
		<device deviceKey="SHARKY_2B_SAP" generation="2B" alarmType="sharky" />
		<device deviceKey="SHARKY_2E_SAP" generation="2E" alarmType="sharky" />
		<device deviceKey="CETASHEAT_IV_SAP" generation="43" alarmType="ray" />
		<device deviceKey="SCYLARINT_7_SAP" generation="47" alarmType="sharky" />
		<device deviceKey="FLYPPER_IV_SAP" generation="49" alarmType="ray" />
		<device deviceKey="IZAR_R30_CP" generation="80" alarmType="izar" />
		<device deviceKey="IZAR_R30_DP" generation="81" alarmType="izar" />
		<device deviceKey="IZAR_R35_CP" generation="85" alarmType="izar" />
		<device deviceKey="IZAR_R35_DP" generation="86" alarmType="izar" />
		<device deviceKey="IZAR_RS_PRIOS" generation="8B" alarmType="izar" />
		<device deviceKey="IZAR_RC_IW_R4" generation="8E" alarmType="izar" />
		
		<device deviceKey="ALTAIR_V2_V3" supplier_code="C" subcode="A"
			alarmType="izar" />
		<device deviceKey="AQUARIUS" supplier_code="C" subcode="B"
			alarmType="izar" />
		<device deviceKey="AQUILA_V2_V3" supplier_code="C" subcode="C"
			alarmType="izar" />
		<device deviceKey="BONYTO" supplier_code="C" subcode="D"
			alarmType="izar" />
		<device deviceKey="ALTAIR_V3_V4_TI" supplier_code="C" subcode="E"
			alarmType="izar" />
		<device deviceKey="ALTAIR_V4_TI" supplier_code="C" subcode="F"
			alarmType="izar" />
		<device deviceKey="VEGA" supplier_code="C" subcode="G"
			alarmType="izar" />
		<device deviceKey="AQUILA_V3_TI" supplier_code="C" subcode="H"
			alarmType="izar" />
		<device deviceKey="CORONA_M" supplier_code="C" subcode="I"
			alarmType="izar" />
		<device deviceKey="AQUILA_V4" supplier_code="C" subcode="J"
			alarmType="izar" />
		<device deviceKey="WP-MFD" supplier_code="C" subcode="K"
			alarmType="izar" />
		

		<device deviceKey="ALTAIR_V4_COMPOSITE_TI" supplier_code="C"
			subcode="L" alarmType="izar" />
		<device deviceKey="ALTAIR_V4_COMPOSITE" supplier_code="C"
			subcode="M" alarmType="izar" />
		<device deviceKey="WS-MFD" supplier_code="C" subcode="N"
			alarmType="izar" />
		
		<device deviceKey="ALTAIR_V4" supplier_code="C" subcode="O"
			alarmType="izar" />
		<device deviceKey="AQUARIUS_TI" supplier_code="C" subcode="P"
			alarmType="izar" />
		<device deviceKey="AQUARIUS_COMPOSITE" supplier_code="C"
			subcode="Q" alarmType="izar" />
		<device deviceKey="AQUARIUS_COMPOSITE_TI" supplier_code="C"
			subcode="R" alarmType="izar" />
		<device deviceKey="ALTAIR_V3_MID" supplier_code="C" subcode="S"
			alarmType="izar" />
		<device deviceKey="CORONA_M_MID" supplier_code="C" subcode="T"
			alarmType="izar" />
		<device deviceKey="AQUARIUS_MID" supplier_code="C" subcode="U"
			alarmType="izar" />
		<device deviceKey="ALTAIR_V4_COMPOSITE" supplier_code="C"
			subcode="V" alarmType="izar" />

		<device deviceKey="HYDRUS_SAP_R4" supplier_code="H" subcode="S"
			alarmType="izar" />
		<device deviceKey="ALTAIR_V3_COMPOSITE" supplier_code="H"
			subcode="A" alarmType="izar" />
	</manufacturer>

zalewma, 2019-04-27:
Also there is description for izar frame mappings:
 <sensorStatusMapping sensorType="izar">
        <statusSource name="prios_frame">
            <parallel>
                <statusDefinition name="izar_is_leak" value="0x8000" severity="HIGH">
                    <category name="leakage"/>
                </statusDefinition>
                <statusDefinition name="izar_was_leak" value="0x4000" severity="MEDIUM">
                    <category name="leakage"/>
                </statusDefinition>
                <statusDefinition name="izar_blocked" value="0x2000" severity="HIGH">
                    <category name="main_hw"/>
                </statusDefinition>
                <statusDefinition name="izar_backflow" value="0x0080" severity="HIGH">
                    <category name="minor_hw"/>
                </statusDefinition>
                <statusDefinition name="izar_underflow" value="0x0040" severity="HIGH">
                    <category name="dimension"/>
                </statusDefinition>
                <statusDefinition name="izar_overflow" value="0x0020" severity="HIGH">
                    <category name="dimension"/>
                </statusDefinition>
                <statusDefinition name="izar_is_manip_ele" value="0x0008" severity="HIGH">
                    <category name="manipulation"/>
                </statusDefinition>
                <statusDefinition name="izar_was_manip_ele" value="0x0004" severity="MEDIUM">
                    <category name="manipulation"/>
                </statusDefinition>
                <statusDefinition name="izar_is_manip_mec" value="0x0002" severity="HIGH">
                    <category name="manipulation"/>
                </statusDefinition>
                <statusDefinition name="izar_was_manip_mec" value="0x0001" severity="MEDIUM">
                    <category name="manipulation"/>
                </statusDefinition>
            </parallel>
            <exclusive/>
        </statusSource>
    </sensorStatusMapping>

zalewma, 2019-04-27:
Also there are telegram examples:
{"telegrams": [
{"t":17,"tts":{"qi":{"snr":13}},"mbd":{"raw":{"d":"1944304CEFC972010000A2111F07138F791B8637330B879A27FE"},"ts":{"mss":1356983143000,"v":1,"any":3},"sid":"M0004A32BD4C3","ie":0,"did":{"ids":"PSAP024300015000"},"isi":0,"iss":0,"isf":0,"ad":{"mid":0}}},
{"t":17,"tts":{"qi":{"snr":5,"rssi":-99}},"mbd":{"raw":{"d":"1944304CDA2A9D208401A15537001323F99AC205322417770F54"},"ts":{"mss":1343815181000,"v":1,"any":3},"sid":"M0004A32BD4C3","ie":0,"did":{"ids":"PSAP0C10AA300122000"},"isi":0,"iss":0,"isf":0,"ad":{"mid":0}}},
{"t":17,"tts":{"qi":{"rssi":-68}},"mbd":{"raw":{"d":"1944304C759B25792021A2641F02130F05C80EBC7E51B1E37BAD"},"ts":{"mss":1344942174000,"v":1,"any":3},"sid":"M0004A32BD4C3","ie":0,"did":{"ids":"PSAP2212079259B75000"},"isi":0,"iss":0,"isf":0,"ad":{"mid":2}}},
{"t":50,"tts":{"qi":{"rssi":-111}},"mbd":{"raw":{"d":"1044000000000000000070F002FD170500"},"ts":{"mss":1343815199000,"v":1,"any":3},"sid":"M0004A32BD4C3","ie":0,"did":{"ids":"U@@@000000000000000"},"isi":0,"iss":0,"isf":0,"ad":{"mid":3}}},
{"t":48,"tts":{"qi":{"rssi":-99}},"mbd":{"raw":{"d":"1F44A5115100701038377A7B88000002A7182D020DFD11084861727279203033"},"ts":{"mss":1344942144000,"v":1,"any":3},"sid":"M0004A32BD4C3","ie":0,"did":{"ids":"UDME383710700051000"},"isi":0,"iss":0,"isf":0,"ad":{"mid":3}}},
{"t":17,"tts":{"qi":{"rssi":-86}},"mbd":{"raw":{"d":"1944304C759B25792021A2641F02130F05C80EBC7E51B1E37BAD"},"ts":{"mss":1341831593000,"v":1,"any":3},"sid":"M0004A32BD4C3","ie":0,"did":{"ids":"PSAP2212079259B75000"},"isi":0,"iss":0,"isf":0,"ad":{"mid":3}}},
{"t":33,"tts":{"qi":{"rssi":-110}},"mbd":{"raw":{"d":"1744A511860700000010710000000001FD0C6102FD173000"},"ts":{"mss":1344942154000,"v":1,"any":3},"sid":"M0004A32BD4C3","ie":0,"did":{"ids":"UDME860710000000000"},"isi":0,"iss":0,"isf":0,"ad":{"mid":2}}},
{"t":49,"tts":{"qi":{"rssi":-94}},"mbd":{"raw":{"d":"46442423000000087037723809281993154003C00000802F2F0374F20A000D780838333930383239310C933A709999998940FD1A0001FD670202FD7407132F2F2F2F2F2F2F2F2F"},"ts":{"mss":1343815188000,"v":1,"any":3},"sid":"M0004A32BD4C3","ie":0,"did":{"ma":{"ids":"UHYD703708000000000"},"ids":"UELS400319280938000"},"isi":0,"iss":0,"isf":0,"ad":{"mid":0}}},
{"t":49,"tts":{"qi":{"rssi":-116}},"mbd":{"raw":{"d":"4944A511950020113837725791822793154003DF7020053977360BE72F5219F134564EC82C3D47D4B6B05F741A6E4D9D97DB1750DD9E730DFD1164203130207972726148000000000000"},"ts":{"mss":1344942074000,"v":1,"any":3},"sid":"M0004A32BD4C3","ie":0,"did":{"ma":{"ids":"UDME383711200095000"},"ids":"UELS400327829157000"},"isi":0,"iss":0,"isf":0,"ad":{"mid":0}}},
{"t":49,"tts":{"qi":{"rssi":-109}},"mbd":{"raw":{"d":"2C44A5114304601238377220092819931580036188000002A7187C010DFD110D30323930383239313532323632"},"ts":{"mss":1343815173000,"v":1,"any":3},"sid":"M0004A32BD4C3","ie":0,"did":{"ma":{"ids":"UDME383712600443000"},"ids":"UELS800319280920000"},"isi":0,"iss":0,"isf":0,"ad":{"mid":3}}},
{"t":49,"tts":{"qi":{"snr":10}},"mbd":{"raw":{"d":"4144A511070050133837722009281993158003DB88000002A7181C010DFD112200000000000000000000000000000000000000000000000000000000000000000000"},"ts":{"mss":1343815187000,"v":1,"any":3},"sid":"M0004A32BD4C3","ie":0,"did":{"ma":{"ids":"UDME383713500007000"},"ids":"UELS800319280920000"},"isi":0,"iss":0,"isf":0,"ad":{"mid":2}}},
{"t":48,"tts":{"qi":{"rssi":-79}},"mbd":{"raw":{"d":"0E4424230204050200037A21800000"},"ts":{"mss":1344942274000,"v":1,"any":3},"sid":"M0004A32BD4C3","ie":0,"did":{"ids":"UHYD000302050402000"},"isi":0,"iss":0,"isf":0,"ad":{"mid":2}}},
{"t":50,"tts":{"qi":{"snr":13}},"mbd":{"raw":{"d":"10442423FFFFFFFF000E70F002FD175000"},"ts":{"mss":1343815160000,"v":1,"any":3},"sid":"M0004A32BD4C3","ie":0,"did":{"ids":"UHYD000EFFFFFFFF000"},"isi":0,"iss":0,"isf":0,"ad":{"mid":0}}},
{"t":32,"tts":{"qi":{"rssi":-110}},"mbd":{"raw":{"d":"504424230304771800007AED0042925D3F40771A4C61062B29862AB3282A78609E61542A56E18CDEEAED12FC93B8119DFB935453478AC7437ADBD6E224C9359605B16CD368BBC3264CE2BCC05D8EDE2263"},"ts":{"mss":1344942154000,"v":1,"any":3},"sid":"M0004A32BD4C3","ie":0,"did":{"ids":"UHYD030400001877000"},"isi":0,"iss":0,"isf":0,"ad":{"mid":2}}},
{"t":48,"tts":{"qi":{"snr":12}},"mbd":{"raw":{"d":"3E4424236351973525067A0050300527A5EDC07FABB7FEB913EC5DCE61D318C788B99920E8476F0FFCFDAF314B68BB0AA260844492C8CDCCE1CCD98809A5EF"},"ts":{"mss":134189159000,"v":1,"any":3},"sid":"M0004A32BD4C3","ie":0,"did":{"ids":"UHYD250635975163000"},"isi":0,"iss":0,"isf":0,"ad":{"mid":1}}},
{"t":49,"tts":{"qi":{"rssi":-86}},"mbd":{"raw":{"d":"7644242362009014640E721105113924232F04E90060059E5505D5D5D6B3108D902D57C00C3D16F935FC559E90E8FA583444D3DA0643695E6EE9EED4FA0C8FE8CE9B53899E367E74575EEB5A3DF9A280140C29446201DF22394A9D790912892F9278A3B8965787DBC08C40B2C45AFBB738BDEB4EAE6152"},"ts":{"mss":1341829797000,"v":1,"any":3},"sid":"M0004A32BD4C3","ie":0,"did":{"ma":{"ids":"UHYD640E14900062000"},"ids":"UHYD2F0439110511000"},"isi":0,"iss":0,"isf":0,"ad":{"mid":0}}},
{"t":49,"tts":{"qi":{"rssi":-89}},"mbd":{"raw":{"d":"5644242352059014640E7298740739242343041D004005FA4E1276D9681A5173593882A17000EF6BAA392FD7130532F36D7B5C9F3E8439B02AB4D11A0F1A84B711113137A69E59802CB1D344AF81E173D348F73A55DA3A"},"ts":{"mss":1341829797000,"v":1,"any":3},"sid":"M0004A32BD4C3","ie":0,"did":{"ma":{"ids":"UHYD640E14900552000"},"ids":"UHYD430439077498000"},"isi":0,"iss":0,"isf":0}},
[... cut ...]
{"t":18,"tts":{"qi":{"rssi":-75}},"mbd":{"raw":{"d":"29442423860732240079A341110013EDF5E0D526B83DB8A127678D75C5C7B54350BD20B7EC13DE829F08"},"ts":{"mss":1344942174000,"v":1,"any":3},"sid":"M0004A32BD4C3","ie":0,"did":{"ids":"UHYD860779002432000"},"isi":0,"iss":0,"isf":0,"ad":{"mid":1}}},
{"t":50,"tts":{"qi":{"rssi":-105}},"mbd":{"raw":{"d":"10442423440010938A0270F002FD170500"},"ts":{"mss":1343815200000,"v":1,"any":3},"sid":"M0004A32BD4C3","ie":0,"did":{"ids":"UHYD8A0293100044000"},"isi":0,"iss":0,"isf":0,"ad":{"mid":3}}}
]}

nestorayuso, 2019-04-29:
Meter number encoding (explanation)

1.) Description

On the IZAR R3 radio module, we have eight bytes (64
bits) in which to store the meter number:
- two bytes for the meter manufacturer ID (see Appendix 1)
- six bytes for the number itself.
As the meter numbers used by the various manufacturers
throughout the world do not have a common structure, an
encoding principle which reduces the risk of doubles to a
minimum needs to be found.
We have therefore defined an encoding system to memorize
the maximum number of characters while bearing in mind that,
if we exceed the capacities described below, there is a chance
that we will get some doubles.
We found the following principle to be the most adequate:
Starting from the right of the meter number, we memorize
firstly the digits up to the maximum allowed (see below). We
then memorize the number obtained in the least significant
part of the encoded number.
If the meter comprises alphabetic characters and there is
sufficient space left, we memorize the maximum number of
characters possible (starting from the right of the number) in
the most significant part. The characters are capitalized.
In the six bytes of the encoded number, the data is
placed in the following manner:
Description Alphabetic characters . .
.
Digits

4 bits N * 5 bits . .
.
m bits

The first four bits are reserved for a reference in a
structure table which allows us to reconstitute the meter
number without having the database with us.
A table will be defined with the most common number structures
(CCCCCCCC, LCCLLCCCCCC, etc.) for this purpose (L:Letters;
C:Digit).

2.) Capacity
In the radio frame, 48 bits are allocated to the meter number.
The coding capacities are therefore:
12 or 13 digits - 0 letters
11 digits - 1 letter
9 or 10 digits - 2 letters
8 digits - 3 letters
6 or 7 digits - 4 letters
5 digits - 5 letters
3 or 4 digits - 6 letters
2 digits - 7 letters
0 or 1 digits - 8 letters
If the meter number to be encoded does not exceed one of the
capacities described above, there will be no doubles.
As far as the SPDE standard is concerned, the number consists
of eight digits and three letters. Encoding it therefore poses
no problem. There will be no doubles.
3.) Encoding methodology
Given the available encoding capacity, there is a risk
that the 32-bit variables capacity will be exceeded. The
encoding therefore needs to be adapted using the following
principle, for example:
3.1) To encode digits with 32-bit variables
We use two 32-bit variables A and B (initialized to 0)
and complete the following loop for all selected digits:
- We multiply A and B by 10
- We add (the ASCII code of the digit – 48) to variable
A
- We add to B the 16 most significant bits of A shifted
16 to the right
- We add this value to B
- We erase the 16 most significant bits of A
Once all the digits are encoded, we simply take the six bytes in
the following order:
1
st encoded byte = bits 24 to 31 of B
2
nd encoded byte = bits 16 to 23 of B
3
rd encoded byte = bits 8 to 15 of B
4
th encoded byte = bits 0 to 7 of B
5
th encoded byte = bits 8 to 15 of A
6
th encoded byte = bits 0 to 7 of A
3.2) To encode letters with 32-bit variables
We use three 32-bit variables A, B and C (initialized to 0)
and complete the following loop:
- We shift variables A and B five to the right
- We write (ASCII code – 64) in variable C
- We shift C 27 to the left to place the five bits of the
far left letter
- We add C to A
- We take the 16 least significant bits of A which we shift
16 to the left
- We add this value to B
- We erase the 16 least significant bits of A
Once all the letters have been processed, we then carry out
the following operations once again to reserve space for the
meter number structure table pointer:
- We shift A and B four to the right
- We take the 16 least significant bits of A which we shift
16 to the left
- We add this value to B
We then obtain the final encoded number:
1
st encoded byte = 1st encoded byte + bits 24 to 31 of A
2
nd encoded byte = 2nd encoded byte + bits 16 to 23 of A
3
rd encoded byte = 3rd encoded byte + bits 24 to 31 of B
4
th encoded byte = 4th encoded byte + bits 16 to 23 of B
5
th encoded byte = 5th encoded byte + bits 8 to 15 of B
6
th encoded byte = 6th encoded byte + bits 0 to 7 of ß
4.) Examples
4 Bits for the structure reference,
44 Bits for the encoding of the letters and digits.
Example 1
SPDE structure, the first letter is stored in ‘MANUFACTURER ID’
Meter N° A01BC234567
Number taken into account : 01234567
Letter taken into account: ABC
Structure ref. = 0
Encoding of structure ref. plus letters : 0000 00001 00010 00011

Digit encoding: 0001 0010 1101 0110 1000 0111(HEX 12D687)
Final encoding: 0000 0000 1000 1000 0110 0000 0001 0010
1101 0110 1000 0111 (HEXA 00886012D687)
Example 2
Meter N° 999 999 999 999 9
Number taken into account : 9999999999999
No letter
Structure ref. = 0
Digit encoding: HEXA 09184E729FFF
Final encoding:0000 1001 0001 1000 0100 1110 0111 0010 1001 1111
1111 1111 (HEXA 09184E729FFF)
Example 3
Meter N° ABCDEFGH
Number taken into account : NONE
Letter taken into account: ABCDEFGH
Structure ref. = 0
Letters encoding: 00001 00010 00011 00100 00101 00110 00111
01000
Final encoding:0000 0000 1000 1000 0110 0100 0010 1001 1000 1110
1000 0000 (HEXA 008864298E80 )
Example 4
Meter N° A01BC2345678
Number taken into account : 012345678
Letter taken into account: BC
Structure ref. = 0
Encoding of structure ref. plus letters: 0000 00010 00011
Codage Chiffre : (HEXA BC614E)
Final encoding:(HEXA 010C00BC614E)

marcinkowalczyk, 2019-08-25:
@zalewma any luck with Sappel meter ?

zalewma, 2019-08-26:
Hi,
No luck - we got some hints, however don't work.

c-soft, 2019-08-31:
Would be great to see that work! I got same meter. @weetmuts did you have some time to look through the hints by @zalewma? Any thoughts?
@zalewma can you post some link to that app you got the info from?

skazi0, 2019-10-24:
I have a first draft version of IZAR decoder implemented. Give me a couple of days to test it and I will create PR.

marcinkowalczyk, 2019-10-24:
Great news!

c-soft, 2019-10-29:
@skazi0 how is it going? can't wait to get my hands on testing it. If you have branch I can check out for testing - let me know :)

skazi0, 2019-10-29:
@marcinkowalczyk @c-soft there you go: https://github.com/weetmuts/wmbusmeters/pull/42

weetmuts, 2019-10-30:
Looks great! Thank you @skazi0 !

What remains before you consider it to be a non-draft pull request?

skazi0, 2019-10-30:
@weetmuts I want to wait until end of month to confirm the meaning of "some other value". I suspect it is copied from the main total when new month starts. There are also some more fields which could be extracted but this could be done later as followup PRs.

weetmuts, 2019-10-30:
The current month ends today, so I expect you have confirmation tomorrow. :-)

Meters that are logged as "Work in Progress" in the README, can be changed at any time.

Meters promoted to supported should not have their existing json keys changed (without a very good reason) . More json keys can be added though.

skazi0, 2019-10-30:
@weetmuts my calendar says otherwise :o) but I think we can safely wait a couple of days to make sure what the other value is. Also some feedback from other guys would be welcome as I tested this code only on my two meters.

c-soft, 2019-10-30:
... nevermind, I got it to work using your DEFAULT_KEY from the code. Talking about security of this setup!!! 🤦‍♂ :D 

Anyway, I can confirm that I'm able to read my IZAR water reader correctly, the "other value" indeed seems like "last month readout". 

Sample number from my readout:

```
GreenhouseWater 40f426a9        358.893 m3      339.923 m3      2019-10-30 19:38.01
GreenhouseWater 40f426a9        358.893 m3      339.923 m3      2019-10-30 19:38.09
GreenhouseWater 40f426a9        358.893 m3      339.923 m3      2019-10-30 19:38.17
```

I will now add config and see if both of them will work together, although I see no reason why they should not. 

Thanks!!

skazi0, 2019-10-30:
@c-soft great to hear it works for someone else too :o)
FTR:
1. The meter ID doesn't seem to match anything written on the meter or the transmitter. I used the RSSI method to find the IDs of my meters (i.e. place the receiver near your meter, start without config in T1 listen mode to print all seen IDs and find the one with best RSSI).
2. The default keys seem to be a good starting point (i.e. don't set any key in config). If this doesn't work, ask your water provider and/or landlord for some info on the custom keys configured in your meter.

marcinkowalczyk, 2019-10-30:
@zalewma I assume you are testing within Wroclaw. Do they use encryption keys?

zalewma, 2019-10-31:
Great work @skazi0  it's working in my case also. I was able to scan and read 10 meters in my area. Example data:
MyTap   20....f7        38.412 m3       38.286 m3       2019-10-31 09:54.19
MyTap   20....f8        55.578 m3       54.431 m3       2019-10-31 09:54.19
MyTap   41....52        114.191 m3      104.726 m3      2019-10-31 09:54.21
MyTap   41....90        91.796 m3       82.513 m3       2019-10-31 09:54.22
MyTap   40....04        560.518 m3      558.995 m3      2019-10-31 09:54.22
MyTap   40....87        1309.186 m3     1298.906 m3     2019-10-31 09:54.24
MyTap   40....60        24.546 m3       18.62 m3        2019-10-31 09:54.25
MyTap   41....c7        93.902 m3       87.027 m3       2019-10-31 09:54.25
MyTap   40....14        640.308 m3      633.307 m3      2019-10-31 09:54.26
MyTap   20....f8        55.578 m3       54.431 m3       2019-10-31 09:54.28
MyTap   20....f7        38.412 m3       38.286 m3       2019-10-31 09:54.29
MyTap   41....90        91.796 m3       82.513 m3       2019-10-31 09:54.31
MyTap   40....87        1309.186 m3     1298.906 m3     2019-10-31 09:54.33
MyTap   40....60        24.546 m3       18.62 m3        2019-10-31 09:54.34

I also found my device which is correctly incrementing:
MyTap   20....f8        55.578 m3       54.431 m3       2019-10-31 09:54.19
MyTap   20....f8        55.581 m3       54.431 m3       2019-10-31 09:55.20
MyTap   20....f8        55.584 m3       54.431 m3       2019-10-31 09:55.29
MyTap   20....f8        55.586 m3       54.431 m3       2019-10-31 09:55.38
MyTap   20....f8        55.589 m3       54.431 m3       2019-10-31 09:55.47
MyTap   20....f8        55.591 m3       54.431 m3       2019-10-31 09:55.55
MyTap   20....f8        55.593 m3       54.431 m3       2019-10-31 09:56.04
MyTap   20....f8        55.594 m3       54.431 m3       2019-10-31 09:56.13
MyTap   20....f8        55.596 m3       54.431 m3       2019-10-31 09:56.30
MyTap   20....f8        55.598 m3       54.431 m3       2019-10-31 09:56.39
MyTap   20....f8        55.6 m3 54.431 m3       2019-10-31 09:56.56
MyTap   20....f8        55.602 m3       54.431 m3       2019-10-31 09:57.04
MyTap   20....f8        55.603 m3       54.431 m3       2019-10-31 09:57.14
MyTap   20....f8        55.604 m3       54.431 m3       2019-10-31 09:57.23

marcinkowalczyk, 2019-10-31:
Hi,

 Seems mine some_other_m3 value changed to total_m3 at midnight 1st Nov local time (while meter reports time in UTC)

total_m3: 6460.126
some_other_m3: 6460.123

so I think some_other_m3 is value that is responsible for calculating usage in current period.

zalewma, 2019-11-01:
Hi, I tried to add izar device as a service and sent it to domoticz however, for the moment logs are not created and probably that part is not working.

What I mean is I've created: 
/etc/wmbusmeters.d/WaterOutdor

With:
name=OutdorWaterMeter
type=izar
id=20f..ef8
key=""

This should create a dedicated file with meter data in folder 'meterfiles'.
@weetmuts do you know what is missing?

zalewma, 2019-11-01:
Yes, in Wroclaw - they don't use encryption but only encoding.

Pozdrawiam
Marcin Zalewski

weetmuts, 2019-11-01:
> This should create a dedicated file with meter data in folder 'meterfiles'.
@zalewma What does your /etc/wmbusmeters.conf look like?

You must have the setting meterfiles=/var/log/wmbusmeters/meterfiles
in the conf file. If that is there, and it still does not work, can you post the debug output, with loglevel=debug set in your conf file?

@skazi0 Should we perhaps fix the medium to water instead of oil? 
Simply add "t->a_field_device_type = 7;" to the end of processContent.

zalewma, 2019-11-01:
> You must have the setting meterfiles=/var/log/wmbusmeters/meterfiles
> in the conf file. If that is there, and it still does not work, can you post the debug output, with loglevel=debug set in your [...]
This was already set, I have other meter which creates a file a triggers it via shell command, but that one doesn't make it.

root@raspberrypi:~# ls -l /etc/wmbusmeters.d/
razem 8
-rwxr-xr-x 1 root root 1432 sie  1 21:41 EnergyMeter
-rwxr-xr-x 1 root root  727 paź 31 10:10 WaterOutdor
root@raspberrypi:~#

root@raspberrypi:~# cat /etc/wmbusmeters.conf |grep /app
meterfiles=/app/log/wmbusmeters/meter_readings
logfile=/app/log/wmbusmeters/wmbusmeters.log
root@raspberrypi:~#

root@raspberrypi:~# ls -l /app/log/wmbusmeters/meter_readings/
razem 76784
-rw-rw-rw- 1 wmbusmeters wmbusmeters 76359793 lis  1 20:39 EnergyMeter
root@raspberrypi:~#

Will attach log file.

> 
> @skazi0 Should we perhaps fix the medium to water instead of oil?
> Simply add "t->a_field_device_type = 7;" to the end of processContent.

H

zalewma, 2019-11-01:
(wmbusmeters) logging started 2019-11-01 20:41:44
(wmbusmeters) version: 0.9.15-7-g5e76799
(config) store meter files in: "/app/log/wmbusmeters/meter_readings"
(config) using device: auto
(config) number of meters: 1
(serialtty) opened /dev/im871a
(im871a) are you there?
(serial /dev/im871a) sent "A5010100"
(serial /dev/im871a) received "A58102004CA3"
(serialtty) closed /dev/im871a
(im871a) got crc16 a34c expected a34c
(im871a) detected on /dev/im871a
(serialtty) opened /dev/im871a
(config) meter amiplus link mode(s): t1
(config) all possible link modes that the meters might transmit on: t1
(config) listen link modes calculated to be: t1
(im871a) set link mode 03
(serial /dev/im871a) sent "A5010306000203300101"
(serial /dev/im871a) received "A58104009CF7"
(im871a) got crc16 f79c expected f79c
(im871a) set config completed
(im871a) get config
(serial /dev/im871a) sent "A5010500"

(wmbus) received telegram 20f59ef8 C-field=44 M-field=4c30 (SAP) A-field-version=d4 A-field-dev-type=01 (Oil meter) Ci-field=a2 (Mfct specific)
(wmbus) frame "1944304CF89EF520D401A211370013C608844933396FF045BE9A"
(wmbus) payload "11370013C608844933396FF045BE9A"
(wmbus) 00: 19 length (25 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 304c m-field (4c30=SAP)
(wmbus) 04: f89ef520 a-field-addr (20f59ef8)
(wmbus) 08: d4 a-field-version
(wmbus) 09: 01 a-field-type (Oil meter)
(wmbus) 0a: a2 ci-field (Mfct specific)
(wmbus) 1944304CF89EF520D401A2
(meter) EnergyMeter: for me? 20f59ef8
(meter) EnergyMeter: not for me: not my id
(im871a) telegram ignored by all configured meters!

skazi0, 2019-11-01:
> Hi,
> 
> Seems mine some_other_m3 value changed to total_m3 at midnight 1st Nov local time (while meter reports time in UTC)
> 
> total_m3: 6460.126
> some_other_m3: 6460.123
> 
> so I think some_other_m3 is value that is responsible for calculating usage in current period.

I renamed the field to "last_month_total_m3".
My meters seem to have incorrect time set because they updated the value before midnight (different shift for each meter).

zalewma, 2019-11-06:
> (wmbusmeters) logging started 2019-11-01 20:41:44
> (wmbusmeters) version: 0.9.15-7-g5e76799
> (config) store meter files in: "/app/log/wmbusmeters/meter_readings"
> (config) using device: auto
> (config) number of meters: 1
> (serialtty) opened /dev/im871a
> (im871a) are you there?
> (serial /dev/im871a) sent "A5010100"
> (serial /dev/im871a) received "A58102004CA3"
> (serialtty) closed /dev/im871a
> (im871a) got crc16 a34c expected a34c
> (im871a) detected on /dev/im871a
> (serialtty) opened /dev/im871a
> (config) meter amiplus link mode(s): t1
> (config) all possible link modes that the meters might transmit on: t1
> (config) listen link modes calculated to be: t1
> (im871a) set link mode 03
> (serial /dev/im871a) sent "A5010306000203300101"
> (serial /dev/im871a) received "A58104009CF7"
> (im871a) got crc16 f79c expected f79c
> (im871a) set config completed
> (im871a) get config
> (serial /dev/im871a) sent "A5010500"
> 
> (wmbus) received telegram 20f59ef8 C-field=44 M-field=4c30 (SAP) A-field-version=d4 A-field-dev-type=01 (Oil meter) Ci-field=a2 (Mfct specific)
> (wmbus) frame "1944304CF89EF520D401A211370013C608844933396FF045BE9A"
> (wmbus) payload "11370013C608844933396FF045BE9A"
> (wmbus) 00: 19 length (25 bytes)
> (wmbus) 01: 44 c-field (SND_NR)
> (wmbus) 02: 304c m-field (4c30=SAP)
> (wmbus) 04: f89ef520 a-field-addr (20f59ef8)
> (wmbus) 08: d4 a-field-version
> (wmbus) 09: 01 a-field-type (Oil meter)
> (wmbus) 0a: a2 ci-field (Mfct specific)
> (wmbus) 1944304CF89EF520D401A2
> (meter) EnergyMeter: for me? 20f59ef8
> (meter) EnergyMeter: not for me: not my id
> (im871a) telegram ignored by all configured meters!
> 
> My config file:
> root@raspberrypi:/app/log/wmbusmeters# cat /etc/wmbusmeters.d/WaterOutdor
> name=OutdorWaterMeter
> type=izar
> id=20f59ef8
> key=""

Works fine... I've corrected key format instead of 'key=""' need to be 'key='

skazi0, 2019-11-07:
@zalewma you can skip the `key=` line completely if you don't have custom keys.

weetmuts, 2019-11-15:
I think we can close this issue now and open new issues if necessary. Do you agree @skazi0 ?

skazi0, 2019-11-15:
@weetmuts I agree but maybe @zalewma should have the final word here :o)

zalewma, 2019-11-16:
Hi Guys, @weetmuts, @zalewma , it can be close for sure. Thanks for the effort !

closed 2019-11-16 by zalewma