wmbusmeters/wmbusmeters #21: wmbusmeters works from command line but not as service

inc90, 2019-04-25:
**Firstly, nice program!** 

I have the following curious issue: I can start wmbusmeters from the command line and receive telegrams from my iPerl water meter. When I start it as a service it does not receive anything.

**Setup:** 
Ubuntu 16.04 LTS with latest patches
Realtek RTL2838UHIDIR SDR
Latest rtl-sdr, rtl-wmbus, wmbusmeters from GitHub

**Configuration:**
> /etc/wmbusmeters.conf
```
loglevel=debug
device=rtlwmbus:rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus
logtelegrams=true
format=fields
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
logfile=/var/log/wmbusmeters/wmbusmeters.log
```
> /etc/wmbusmeters.d/iPerlWaterMeter
```
name=iPerlWaterMeter
type=iperl
id=20730893
key=035F3B08E6FF114F2B6C6B3xxxxxxxx
```

**Start from command line:**

> wmbusmeters --verbose --useconfig=/ --meterfiles=/ rtlwmbus:868.95M

In another window .....
> tail -f /var/log/wmbusmeters/wmbusmeters.log -n 15
(wmbusmeters) version: 0.9.4-2-g33ec7ec
(config) store meter files in: "/var/log/wmbusmeters/meter_readings"
(config) using device: rtlwmbus
(config) with: rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus
(config) number of meters: 1
(rtlwmbus) using command: rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus"
(serialcmd) opened /bin/sh
(config) using link mode: Any
(iperl) configured "iPerlWaterMeter" "iperl" "20730893" encrypted

> ls -l /var/log/wmbusmeters/meter_readings/iPerlWaterMeter
-rw-rw-r-- 1 mo mo 65 Apr 25 15:49 iPerlWaterMeter

> cat /var/log/wmbusmeters/meter_readings/iPerlWaterMeter
`iPerlWaterMeter;20730893;36.089000;0.000000;2019-04-25T13:49:24Z`

A telegram was received and decoded correctly. I also receive telegrams from other meters which are then (correctly) ignored. => Command line works!

**Start as daemon**

> sudo systemctl status wmbusmeters
```
? wmbusmeters.service - wmbusmeters service
   Loaded: loaded (/etc/systemd/system/wmbusmeters.service; disabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-04-25 08:21:29 CEST; 3h 48min ago
  Process: 22320 ExecStart=/usr/sbin/wmbusmetersd /var/run/wmbusmeters/wmbusmeters.pid (code=exited, status=0/SUCCESS)
  Process: 22315 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/run/wmbusmeters (code=exited, status=0/SUCCESS)
  Process: 22310 ExecStartPre=/bin/mkdir -p /var/run/wmbusmeters (code=exited, status=0/SUCCESS)
  Process: 22305 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
  Process: 22300 ExecStartPre=/bin/mkdir -p /var/log/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
 Main PID: 22324 (wmbusmetersd)
    Tasks: 6
   Memory: 4.8M
      CPU: 2h 22min 39.757s
   CGroup: /system.slice/wmbusmeters.service
           tq22324 /usr/sbin/wmbusmetersd /var/run/wmbusmeters/wmbusmeters.pid
           tq22326  -c rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus
           tq22327 rtl_sdr -f 868.95M -s 1.6e6 -
           mq22328 rtl_wmbus

Apr 25 08:21:29 FUTRO-400 systemd[1]: Starting wmbusmeters service...
Apr 25 08:21:29 FUTRO-400 wmbusmetersd[22320]: (wmbusmeters) started /var/run/wmbusmeters/wmbusmeters.pid
Apr 25 08:21:29 FUTRO-400 systemd[1]: Started wmbusmeters service.
```

> tail -n15 -f /var/log/wmbusmeters/wmbusmeters.log
```
(wmbusmeters) logging started 2019-04-25 08:21:29
(wmbusmeters) version: 0.9.4-2-g33ec7ec
(config) store meter files in: "/var/log/wmbusmeters/meter_readings"
(config) using device: rtlwmbus
(config) with: rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus
(config) number of meters: 1
(rtlwmbus) using command: rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus"
(serialcmd) opened /bin/sh
(config) using link mode: Any
(iperl) configured "iPerlWaterMeter" "iperl" "20730893" encrypted
(wmbusmeters) waiting for telegrams
```

No matter how long I wait, nothing is ever received. The service seems to be correctly started. When I do a ps -A all the programs are started, apparently with the correct parameters.

Question1. Any ideas why this might be happening, or how I could go about debugging it?

Question2. There seems to be a typo in main.cc, line 224
` command = prefix+"rtl_sdr -f "+freq+" -s 16000000 - 2>/dev/null | "+prefix+"rtl_wmbus";`
My dongle gives an incorrect sample rate error. I changed it to 1.6e6 and the error went away.
Is there a 0 to many i.e. 16M instead of 1.6M?

Best regards,
Muiris.

inc90, 2019-04-26:
Update: I tried starting the service as root, but it did not work either. 

Very wierdly I can start the command line version and leave it running in the background with " 
..... &" and this works perfectly!

weetmuts, 2019-04-26:
I think you simply need to specify the full path to rtl_sdr and rtl_wmbus when it is run as a daemon.
Its because the daemon process does not have a normal path when it is started.
device=rtlwmbus:/usr/bin/rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus
or wherever you placed the binaries.

As for 1.6e6, yep it does look like I made a mistake there! Thanks!

weetmuts, 2019-04-26:
Just specifying device:rtlwmbus will automatically use /usr/bin/rtl_sdr and /usr/bin/rtl_wmbus
when wmbusmeters is run as a daemon. The README writes about this, albeit short.

weetmuts, 2019-04-26:
Ok, just pushed the fix for 1.6e6, now you should be able to use plain:
device=rtlwmbus

weetmuts, 2019-05-04:
@inc90 any success?

inc90, 2019-05-08:
Not with the deamon as it was installed. It definitely runs, the processes are there, it just doesn't write any files for me. The last log entry is waiting for telegrams, after that nothing more happens. 

I changed the service as listed below i.e. I let the command line version run as a daemon. Not pretty, but it works. :-)
```

[Unit]
Description=wmbusmeters service
After=network.target
StopWhenUnneeded=true

[Service]
Type=simple
PrivateTmp=yes
User=wmbusmeters
Group=wmbusmeters
Restart=always
RestartSec=1
StartLimitInterval=10
StartLimitBurst=3
# WorkingDirectory=~

# Run ExecStartPre with root-permissions
PermissionsStartOnly=true
ExecStartPre=-/bin/mkdir -p /var/log/wmbusmeters/meter_readings
ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmbusmeters
ExecStartPre=-/bin/mkdir -p /var/run/wmbusmeters
ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/run/wmbusmeters
# ExecStart=/usr/sbin/wmbusmetersd /var/run/wmbusmeters/wmbusmeters.pid
ExecStart=/usr/bin/wmbusmeters --verbose --useconfig=/ --meterfiles=/ rtlwmbus
PIDFile=/var/run/wmbusmeters/wmbusmeters.pid

[Install]
WantedBy=multi-user.target
```

weetmuts, 2019-06-06:
Ok, I'll close this issue.  Not quite satisfactory, since we do not know why the problem occured. But if it occurs again I will try to resolve it properly.

closed 2019-06-06 by weetmuts

weetmuts, 2019-08-13:
@inc90 Hi! I think i fixed this bug. When entering daemon mode, it is not enough
to close stdin,stdout,stderr, one should then re-open stdin,stdout,stderr pointing
to /dev/null. This is fixed now, please test if you have the time. :-)