wmbusmeters/wmbusmeters #23: [Feature request] Techem Radio 3 water meters

FeatureExpert, 2019-04-27:
Hello weetmuts, thanks for your awesome work.
Could it be possible to add support for Techem Radio 3 warm/cold water meters?

> (serialcmd) received "T1;1;1;2019-04-27 07:20:05.000;53;142;62007430;0x2f446850307400627462a2069f255900b029310000000306060906030609070606050509050505050407040605070500<0A>"
> (rtlwmbus) got full frame
> Unknown ci-field a2
> (wmbus) received telegram 62007430 C-field=44 M-field=5068 (TCH) A-field-version=74 A-field-dev-type=62 (Unknown) Ci-field=a2 (Mfct specific)
> (wmbus) frame "2F446850307400627462A2069F255900B029310000000306060906030609070606050509050505050407040605070500"
> (wmbus) payload "069F255900B029310000000306060906030609070606050509050505050407040605070500"
> (wmbus) 00: 2f length (47 bytes)
> (wmbus) 01: 44 c-field (SND_NR)
> (wmbus) 02: 6850 m-field (5068=TCH)
> (wmbus) 04: 30740062 a-field-addr (62007430)
> (wmbus) 08: 74 a-field-version
> (wmbus) 09: 62 a-field-type (Unknown)
> (wmbus) 0a: a2 ci-field (Mfct specific)
> (wmbus) 2F446850307400627462A2
> Received telegram from: 62007430
>           manufacturer: (TCH) Techem Service,
>            device type: Unknown
> (rtl_wmbus) telegram ignored by all configured meters!
> 

A-field-dev-type=62 <-- warm water meter
A-field-dev-type=72 <-- cold water meter

Based on information found over the internet and my own observations, this is how I calculate current meter readings from mbus telegrams:

Payload:

>06 9F25 5900 B029 3100 00000306060906030609070606050509050505050407040605070500

9F25 - previous period end date (?)
5900 - water consumption in previous period
B029 - current date
3100 - water consumption in current period

Calculations:

5900 -> 0059h -> 89
3100 -> 0031h -> 49

Current meter reading: (89 + 49) / 10 = 13.8 m3

weetmuts, 2019-04-27:
Excellent work! Thanks! It was easy to add the new meter mkradio3 based on your work.
Please test it to make sure it works!

FeatureExpert, 2019-04-27:
Works perfectly, thank you! Now I don't have to stop wmbusmeters every night to switch from getting readings from amiplus to water meters :)

```
build/wmbusmeters --format=json rtlwmbus:868.9M test mkradio3 62007430,41564358,62607939,40757277 '' | grep -v Unknown
{"media":"warm water","meter":"mkradio3","name":"test","id":"62607939","total_m3":1.300000,"target_m3":0.900000,"timestamp":"2019-04-27T12:11:04Z"}
{"media":"warm water","meter":"mkradio3","name":"test","id":"62007430","total_m3":13.800000,"target_m3":8.900000,"timestamp":"2019-04-27T12:12:28Z"}
{"media":"cold water","meter":"mkradio3","name":"test","id":"41564358","total_m3":24.800000,"target_m3":17.100000,"timestamp":"2019-04-27T12:12:29Z"}
{"media":"cold water","meter":"mkradio3","name":"test","id":"40757277","total_m3":7.500000,"target_m3":2.000000,"timestamp":"2019-04-27T12:13:33Z"}
```

Maybe just one cosmetic thing ;)

```
--- wmbusmeters/src/meter_mkradio3.cc   2019-04-27 14:00:09.023071811 +0200
+++ wmbusmeters.tomek/src/meter_mkradio3.cc     2019-04-27 14:45:37.106221446 +0200
@@ -95,7 +95,7 @@
         return;
     }
 
-    verbose("(%s) telegram for %s %02x%02x%02x%02x\n", "supercom587",
+    verbose("(%s) telegram for %s %02x%02x%02x%02x\n", "mkradio3",
             name().c_str(),
             t->a_field_address[0], t->a_field_address[1], t->a_field_address[2],
             t->a_field_address[3]);
@@ -112,7 +112,7 @@
         t->content = t->payload;
     }
     char log_prefix[256];
-    snprintf(log_prefix, 255, "(%s) log", "supercom587");
+    snprintf(log_prefix, 255, "(%s) log", "mkradio3");
     logTelegram(log_prefix, t->parsed, t->content);
     int content_start = t->parsed.size();
     processContent(t);

```

closed 2019-04-27 by FeatureExpert

weetmuts, 2019-04-27:
Great! Pushed the fix for the typos.

sledzik1984, 2019-12-28:
[image]

Can I find ID of meter somewhere on the meter? Got many readings - none of which corresponds to reading that if physically on my meter... 
Maybe the Datamatrix code on meter? 

Any help appreciated :)

weetmuts, 2019-12-30:
@sledzik1984 My guess is that the meters id is: 60425537 
a wmbus compliant id is always 8 numerical digits.

But I have never actually seen an mkradio3 meter. :-)

weetmuts, 2019-12-30:
@sledzik1984 Since mkradio3 is an addon to your mechanical meter, the addon might
not be synced with the meter mechanical status. Try to listen to 60425537 and check
that you get updated values when you run water, and correspondingly no changes when the water stands still.

FeatureExpert, 2019-12-30:
@sledzik1984 As far as I know, labels printed on meter and transmitted meter numbers are totally different. Maybe not the smartest idea, but worked for me. Grab a laptop with rtl dongle. Set gain to lowest value (not auto!) and place it close to the meter. You should be able to see readings for your meter only.