wmbusmeters/wmbusmeters #25: Qundis Q caloric not read with IM871A-Usb

reset12, 2019-05-20:
Hello,
i like to read my Qundis Q caloric heat meters from IM871A-USB Stick.
I have installed wmbusmeters on my Ubuntu 18.04 Laptop and started with the following parameters with no success.
What time i must wait for Qundis Heat meter sends data ? or have i a Parameter problem
when i dont get data ?

wmbusmeters --debug --verbose --c1 /dev/ttyUSB0
(wmbusmeters) version: 
(config) using device: /dev/ttyUSB0
(config) number of meters: 0
(serialtty) opened /dev/ttyUSB0
(amb8465) are you there?
(serial /dev/ttyUSB0) sent "FF0B00F4"
(serialtty) closed /dev/ttyUSB0
(serialtty) closed /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(im871a) are you there?
(serial /dev/ttyUSB0) sent "A5010100"
(serial /dev/ttyUSB0) received "A58102004CA3"
(serialtty) closed /dev/ttyUSB0
(im871a) got crc16 a34c expected a34c
(serialtty) closed /dev/ttyUSB0
(im871a) detected on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(im871a) set link mode 06
(serial /dev/ttyUSB0) sent "A5010306000206300101"
(serial /dev/ttyUSB0) received "A58104009CF7"
(im871a) got crc16 f79c expected f79c
(im871a) set config completed
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial /dev/ttyUSB0) received "A5810617FF000600B32563011000010001FF070232000101000000F70A"
(im871a) got crc16 0af7 expected 0af7
(im871a) get config completed
(im871a) config: device mode 00
(im871a) config: link mode 06
(im871a) config: wmbus c-field 00
(im871a) config: wmbus mfg id 25b3 (IMS)
(im871a) config: wmbus device id 00100163
(im871a) config: wmbus version 01
(im871a) config: wmbus device type 00
(im871a) config: radio channel 01
(im871a) config: radio power level 07
(im871a) config: radio data rate 02
(im871a) config: radio rx window 32
(im871a) config: auto power saving 00
(im871a) config: auto RSSI attachment 01
(im871a) config: auto rx timestamp attachment 01
(im871a) config: led control 00
(im871a) config: rtc control 00
(config) using link mode: C1
No meters configured. Printing id:s of all telegrams heard!

weetmuts, 2019-05-21:
After searching a bit, it seems like the QCaloric can transmit C telegrams or S telegrams.
The meter can apparently be configured using some software through irda.

I have now pushed support for --s1 and --s1m, try them both to see if you receive any
telegrams from the meter.

reset12, 2019-05-21:
Hello Fredrik,
--s1 or s1m is not allowed!

wmbusmeters version: 

Usage: wmbusmeters {options} <device> ( [meter_name] [meter_type] [meter_id] [meter_key] )*

As <options> you can use:

    --addconversion=<unit>+ add conversion to these units to json and meter env variables (GJ)
    --c1 or --t1 listen to C1 or T1 messages when no meters are supplied, not needed for rtlwmbus
    --debug for a lot of information
    --exitafter=<time> exit program after time, eg 20h, 10m 5s
    --format=<hr/json/fields> for human readable, json or semicolon separated fields
    --logfile=<file> use this file instead of stdout
    --logtelegrams log the contents of the telegrams for easy replay
    --meterfiles=<dir> store meter readings in dir
    --meterfilesaction=(overwrite|append) overwrite or append to the meter readings file
    --oneshot wait for an update from each meter, then quit
    --separator=<c> change field separator to c
    --shell=<cmdline> invokes cmdline with env variables containing the latest reading
    --shellenvs list the env variables available for the meter
    --useconfig=<dir> load config files from dir/etc
    --verbose for more information

As a <device> you can use: auto
which will look for the links /dev/im87a,/dev/amb8475 and /dev/rtlsdr (the
links are automatically generated by udev if you have run the install scripts.)
and start wmbusmeters with the proper tty device or rtlwmbus background process.

As a <device> you can also use: the exact /dev/ttyUSB0 to your dongle if you do not want
to install the udev rule.

As a <device> you can also use: rtlwmbus
to spawn the background process: \"rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus\"
You can also use: rtlwmbus:868.9M to use this fq instead. Fq tuning can sometimes
be necessary. Or you can specify the entire background process command line: \"rtlwmbus:<commandline>\"

As meter quadruples you specify:
<meter_name> a mnemonic for this particular meter
<meter_type> one of the supported meters
<meter_id> an 8 digit mbus id, usually printed on the meter
<meter_key> an encryption key unique for the meter
    if the meter uses no encryption, then supply ""

Supported water meters:
Kamstrup Multical 21 (multical21)
Kamstrup flowIQ 3100 (flowiq3100)
Sontex Supercom 587 (supercom587)
Sensus iPERL (iperl)
Techem MK Radio 3 (mkradio3)

Supported heat cost allocator:
Qundis Q caloric (qcaloric)

Supported heat meters:
Techem Vario 4 (vario451)

Work in progress:
Water meter Apator at-wmbus-16-2 (apator162)
Heat meter Kamstrup Multical 302 (multical302)
Electricity meter Kamstrup Omnipower (omnipower) and Tauron Amiplus (amiplus)

Do you have a different version ?

My Version is not displayed, but i have ist installed from github at Sunday.

with regards
Bruno

weetmuts, 2019-05-21:
>My Version is not displayed, but i have ist installed from github at Sunday. 

You need to build the version on github, I just added s1 and s1m for you, and pushed the code to github just a few minutes ago. :-)

reset12, 2019-05-21:
Hello Fredrik,
i wait 3 Minutes with the new Parameter s1m and s1 with no answer:

(config) using device: /dev/ttyUSB0
(config) number of meters: 0
(serialtty) opened /dev/ttyUSB0
(amb8465) are you there?
(serial /dev/ttyUSB0) sent "FF0B00F4"
(serialtty) closed /dev/ttyUSB0
(serialtty) closed /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(im871a) are you there?
(serial /dev/ttyUSB0) sent "A5010100"
(serial /dev/ttyUSB0) received "A58102004CA3"
(serialtty) closed /dev/ttyUSB0
(im871a) got crc16 a34c expected a34c
(serialtty) closed /dev/ttyUSB0
(im871a) detected on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(im871a) set link mode 01
(serial /dev/ttyUSB0) sent "A5010306000201300101"
(serial /dev/ttyUSB0) received "A58104009CF7"
(im871a) got crc16 f79c expected f79c
(im871a) set config completed
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial /dev/ttyUSB0) received "A5810617FF000100B32563011000010001FF070232000101000000F38E"
(im871a) got crc16 8ef3 expected 8ef3
(im871a) get config completed
(im871a) config: device mode 00
(im871a) config: link mode 01
(im871a) config: wmbus c-field 00
(im871a) config: wmbus mfg id 25b3 (IMS)
(im871a) config: wmbus device id 00100163
(im871a) config: wmbus version 01
(im871a) config: wmbus device type 00
(im871a) config: radio channel 01
(im871a) config: radio power level 07
(im871a) config: radio data rate 02
(im871a) config: radio rx window 32
(im871a) config: auto power saving 00
(im871a) config: auto RSSI attachment 01
(im871a) config: auto rx timestamp attachment 01
(im871a) config: led control 00
(im871a) config: rtc control 00
(config) using link mode: S1m
No meters configured. Printing id:s of all telegrams heard!

^C(serialtty) event loop stopped!
(serialtty) closed /dev/ttyUSB0
(serialtty) closed /dev/ttyUSB0
--------------------------------------------------------
 wmbusmeters --debug --verbose --s1 /dev/ttyUSB0
(wmbusmeters) version: 
(config) using device: /dev/ttyUSB0
(config) number of meters: 0
(serialtty) opened /dev/ttyUSB0
(amb8465) are you there?
(serial /dev/ttyUSB0) sent "FF0B00F4"
(serialtty) closed /dev/ttyUSB0
(serialtty) closed /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(im871a) are you there?
(serial /dev/ttyUSB0) sent "A5010100"
(serial /dev/ttyUSB0) received "A58102004CA3"
(serialtty) closed /dev/ttyUSB0
(im871a) got crc16 a34c expected a34c
(serialtty) closed /dev/ttyUSB0
(im871a) detected on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(im871a) set link mode 00
(serial /dev/ttyUSB0) sent "A5010306000200300101"
(serial /dev/ttyUSB0) received "A58104009CF7"
(im871a) got crc16 f79c expected f79c
(im871a) set config completed
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial /dev/ttyUSB0) received "A5810617FF000000B32563011000010001FF0702320001010000009FB9"
(im871a) got crc16 b99f expected b99f
(im871a) get config completed
(im871a) config: device mode 00
(im871a) config: link mode 00
(im871a) config: wmbus c-field 00
(im871a) config: wmbus mfg id 25b3 (IMS)
(im871a) config: wmbus device id 00100163
(im871a) config: wmbus version 01
(im871a) config: wmbus device type 00
(im871a) config: radio channel 01
(im871a) config: radio power level 07
(im871a) config: radio data rate 02
(im871a) config: radio rx window 32
(im871a) config: auto power saving 00
(im871a) config: auto RSSI attachment 01
(im871a) config: auto rx timestamp attachment 01
(im871a) config: led control 00
(im871a) config: rtc control 00
(config) using link mode: S1
No meters configured. Printing id:s of all telegrams heard!

^C(serialtty) event loop stopped!
(serialtty) closed /dev/ttyUSB0
(serialtty) closed /dev/ttyUSB0

I have one Reserve Qundis here for testing with 0000 on the Display because it was never used on an heat plate. See attachment.

with regards
Bruno

reset12, 2019-05-21:
Hello Fredrik,
i am not shure, that the Qundis is sending permanent data.
I try to contact the service of Qundis, to get information about the sending intervall of the heatmeter.
When the Heatmeter is not sending, you can not find an error in your programm.
Please wait with activities, i give you a feedback what Qundis answers.

with regards
Bruno

reset12, 2019-05-21:
Hello Frederik,
The Qundis heat meter should be send every Day from 8:00-18:00 every 90 sec with the protocol
wmbus T1 long. The ID of manufactor is INE for Innotas (Company from where i get the heat meters.4

With regards
Bruno

weetmuts, 2019-05-21:
Nice, then try with --t1 and see if you get anything.

reset12, 2019-05-21:
Yeah,

now i got a Answer:
No meters configured. Printing id:s of all telegrams heard!

(serial /dev/ttyUSB0) received "A5E2037644C5250100108355087201001083C5255508010000002F2F0B6E000000426E000082016E0000C2016E000082026E0000C2026E000082036E0000C2036E000082046E0000C2046E000082056E0000C2056E000082066E0000C2066E000082076E0000C2076E000082086E0000C2086E000002FD17000000000000897E9A"
(im871a) timestamp 00000000
(im871a) rssi 89
(im871a) got crc16 9a7e expected 9a7e
(wmbus) received telegram 83100001 C-field=44 M-field=25c5 (INE) A-field-version=55 A-field-dev-type=08 (Heat Cost Allocator) Ci-field=72 (EN 13757-3 Application Layer with long Transport Layer)
(wmbus) frame "7644C5250100108355087201001083C5255508010000002F2F0B6E000000426E000082016E0000C2016E000082026E0000C2026E000082036E0000C2036E000082046E0000C2046E000082056E0000C2056E000082066E0000C2066E000082076E0000C2076E000082086E0000C2086E000002FD170000"
(wmbus) payload "01001083C5255508010000002F2F0B6E000000426E000082016E0000C2016E000082026E0000C2026E000082036E0000C2036E000082046E0000C2046E000082056E0000C2056E000082066E0000C2066E000082076E0000C2076E000082086E0000C2086E000002FD170000"
(wmbus) 00: 76 length (118 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: c525 m-field (25c5=INE)
(wmbus) 04: 01001083 a-field-addr (83100001)
(wmbus) 08: 55 a-field-version
(wmbus) 09: 08 a-field-type (Heat Cost Allocator)
(wmbus) 0a: 72 ci-field (EN 13757-3 Application Layer with long Transport Layer)
(wmbus) 7644C52501001083550872
Received telegram from: 83100001
          manufacturer: (INE) INNOTAS Elektronik, Germany
           device type: Heat Cost Allocator
(im871a) telegram ignored by all configured meters!

Where can i found the Value the meter measured value (must be 000000) like the Display ?

with regards
Bruno

weetmuts, 2019-05-21:
Thanks, it seems like you have an EurisII from Innotas (ie not a Qundis).
 https://innotas.de/produkte/wp-content/uploads/2017/08/Produktblatt-EHKV_V2.5-english.pdf

I have added support for eurisii to wmbusmeters and pushed the code.
Please try with:
wmbusmeters  /dev/ttyUSB0 MyHCA eurisii 83100001 ""
and it will probably print this, since no heat has been detected yet....
MyHCA	83100001	0 hca	2019-05-21 15:24.04

the meters probably stores the last 17 months of heat cost allocation, 
ie the value of current_consumption_hca when the month changes.
Currently the json only prints the most recent month and month 17.
But I can add all the other months as well....
wmbusmeters  --format=json /dev/ttyUSB0 MyHCA eurisii 83100001 ""
{"media":"heat cost allocation","meter":"eurisii","name":"MyHCA","id":"83100001","current_consumption_hca":0,"consumption_at_set_date_1_hca":0,"consumption_at_set_date_17_hca":0,"timestamp":"2019-05-21T13:24:43Z"}

The meter also outputs the error flags, but they are zero now as well, and we do not
know what the bits mean. You can see the decoded content when using --debug.
I like meters that conform fully to the wmbus standard. :-)

weetmuts, 2019-05-21:
Do a "make clean" after you pulled, there are some problem in the depency tracking in the makefiles.

reset12, 2019-05-21:
Hello Fredrik,
you make a great job.
MyHCA	83100001	0 hca	2019-05-21 15:53.15
MyHCA	83100001	0 hca	2019-05-21 15:55.02
MyHCA	83100001	0 hca	2019-05-21 15:56.28
MyHCA	83100001	0 hca	2019-05-21 15:58.11
MyHCA	83100001	0 hca	2019-05-21 16:01.05
MyHCA	83100001	0 hca	2019-05-21 16:02.30
MyHCA	83100001	0 hca	2019-05-21 16:03.58
MyHCA	83100001	0 hca	2019-05-21 16:05.29
MyHCA	83100001	0 hca	2019-05-21 16:07.07
MyHCA	83100001	0 hca	2019-05-21 16:08.40
MyHCA	83100001	0 hca	2019-05-21 16:10.23
MyHCA	83100001	0 hca	2019-05-21 16:11.51
MyHCA	83100001	0 hca	2019-05-21 16:13.30
MyHCA	83100001	0 hca	2019-05-21 16:14.55
MyHCA	83100001	0 hca	2019-05-21 16:16.31
MyHCA	83100001	0 hca	2019-05-21 16:18.05

Necessary is the actual value for the accounting with the tenant. Good for control are also the last 17 Month.

I want to go outside the Apartments to read the actual values. I have actual 44 meters on the appartments.

Please send your Email to reset12 (at) gmx.de i will spend some money for your work.

with regards
Bruno

weetmuts, 2019-05-21:
Great! I'll add the other months but can you ask Innotas about the meaning of the error flag bits? 

The 16 bits are all zero right now, you but probably want to understand the error conditions, if they occur, like the tamper/sabotage bit for example.... (Even though you could treat any deviation from zero as a reason to go and replace the meter.)

I do not know that much about heat cost allocators. Do one sum all the hca:s collected and then divide each meter's hca with the sum of the hcas to get a percentage of the total consumption for each meter? Then use that percentage on the cost of heating all the apartments to see the actual cost for each apartment?

reset12, 2019-05-21:
Hello Fredrik,
there are 2 possible ways to account the hca displays the used heat:
1. All hca have the same calibration and you must calculate the units displayed on each hca with a factor for the radiator to have the right value. This is because there are different radiators (size and power) in the apartments.
2. The factor of the radiators are programmed in the hca, the display show now the normalized heat usage.

I use the case 1 because i dont need to programm the hca and when one hca is in error, i can change it with my reserve hca with no problems. I like to reduce the cost to my tenant and have terminated the contract with biller.

I try to ask Innotas and hope for a answer.

with regards
Bruno

weetmuts, 2019-05-21:
Ok, I added all the data inside the telegram to the json. Worth noticing is that the current consumption
can be up to 999999, but the consumption_at_set_date can be at most 65535...odd.
Temporarily wmbusmeters prints the error flags as the string 0x0000 (for no bits set).
Please run it for a while to test the values and see that it works as you expect.

The json now looks like this:
{"media":"heat cost allocation","meter":"eurisii","name":"MyHCA","id":"83100001","current_consumption_hca":0,"consumption_at_set_date_hca":0,"consumption_at_set_date_2_hca":0,"consumption_at_set_date_3_hca":0,"consumption_at_set_date_4_hca":0,"consumption_at_set_date_5_hca":0,"consumption_at_set_date_6_hca":0,"consumption_at_set_date_7_hca":0,"consumption_at_set_date_8_hca":0,"consumption_at_set_date_9_hca":0,"consumption_at_set_date_10_hca":0,"consumption_at_set_date_11_hca":0,"consumption_at_set_date_12_hca":0,"consumption_at_set_date_13_hca":0,"consumption_at_set_date_14_hca":0,"consumption_at_set_date_15_hca":0,"consumption_at_set_date_16_hca":0,"consumption_at_set_date_17_hca":0,"error_flags":"0x0000","timestamp":"2019-05-21T17:33:26Z"}

reset12, 2019-05-21:
Hello Fredrik,
i can start testing first at tomorrow, because the hca doesnt send after 18:00 .

with regards
Bruno

weetmuts, 2019-05-22:
Ok, pushed a version of wmbusmeters where the EurisII support is theoretically complete.

What needs to be tested is that when for example a meter is unscrewed from the element, it will actually show "error_flags":"SABOTAGE" and not "error_flags":"RESET". If the meter is entirely wmbus compliant, the code should be correct. But perhaps they switched the bytes. :-)

Also the meter can send long or short telegrams. Only the short has a date in the telegram.
If you have the time to test a short telegram and publish the --debug log here, that would be valuable
to check that the date extraction works properly. And I get another regression test case for the code.
:-)

Innotas seems to be a great company since they so quickly provided the documentation for the meter and its protocol.

reset12, 2019-06-04:
Hello Fredrik,
i got the software licence from innotas but i have problem to get it running on my virtualbox.
The USB-Adapter iM871A is not offical supported, only Innotas and Amper Stick are defined in the software of innotas. All the Software and Tools of innotas only works with Windows.
The Values are not Ok by me, depends on the programmed deadline Tcount on Page 18, my hcas programmed
at deadline 12 (End of year), so after that month the values start new. This is the reason why the actual values are lower than the values i read manually on 27.12.18.
The best is to print out all values, so every user can find his values depending on individual deadlines(depending on order).
Can you send me the Init-String of iM871A-USB that i can use a Terminal programm to analyse if data is send in Windows. Innotas told me the Baud rate is 2400 Baud and i should test, if the USB-Receiver sends data in Windows.

with regards
Bruno

weetmuts, 2019-06-04:
I suggest that you just buy an Amber stick because making im871a work with their software, 
is probably impossible. The reason is that there is not just a difference in init strings but
also how the telegrams are wrapped/sent within the serial protocal between the dongle and the PC.

The init strings can be seen when you run with --debug, you can for example see:
(im871a) set link mode 06
(serial /dev/ttyUSB0) sent "A5010306000206300101"
(serial /dev/ttyUSB0) received "A58104009CF7"

The serial binary string A50103... sets the T1 listening mode for the dongle.
The response confirms this. Now the dongle will send telegrams
over the serial wire, like this. You can see the im871a wrapper A5E203,
the frame begins with 7644.... and then there is stuff at the end where im871a tells
the rssi/crc etc...

(serial /dev/ttyUSB0) received "A5E2037644C5250100108355087201001083C5255508010000002F2F0B6E000000426E000082016E0000C2016E000082026E0000C2026E000082036E0000C2036E000082046E0000C2046E000082056E0000C2056E000082066E0000C2066E000082076E0000C2076E000082086E0000C2086E000002FD17000000000000897E9A"

weetmuts, 2019-06-04:
>The best is to print out all values, so every user can find his values depending on individual >deadlines(depending on order).

Ok, I think wmbusmeters prints all the information in the telegram, is there something here I need to do to help you?

reset12, 2019-06-04:
Hello Fredrik,
i think i try to change the USB-Stick to Amber.
My major need are:
1. Wildcard like 'sudo wmbusmeters  --t1 /dev/ttyUSB0   MyHCA eurisii 83100001 ""'
                  'sudo wmbusmeters  --t1 /dev/ttyUSB0   MyHCA eurisii 831* ""'
   With this wildcard i will be able to scan all my hcas in one interval.

2. Option to Display all Parameters/Values of Option T1 Page 18 of Documentation EURISII.
   This is nessesary because my Deadline parameter is End of Year and my Accounting Date is 30.4 
   of the year. Other people my be have another values here.

do you got my package ?

with regards
Bruno

reset12, 2019-06-04:
Hello Fredrik,
i could read the Values with the Innotas Windows-Software with my IR-Optical USB-Head wich i use to
read my EHZ Smart electric meter.
I send you the Values from my test hca in the attachment.
The error flags are not setable and i will not break my hca to test it.
I think it is good that these parameters are not to change, then the consumption values can not be manipulated.

weetmuts, 2019-06-06:
I have now added support for having a wildcard at the end of the id:
This should now work (note the necessary quotes around "831*"):
sudo wmbusmeters --t1 /dev/ttyUSB0 MyHCA eurisii "831*" ""

>> 2. Option to Display all Parameters/Values of Option T1 Page 18 of Documentation EURISII. 

I believe the code does this. The output is:

{"media":"heat cost allocation","meter":"eurisii","name":"HeatMeter","id":"88018801","current_consumption_hca":112233,"consumption_at_set_date_hca":273,"set_date":"","consumption_at_s\
et_date_2_hca":529,"consumption_at_set_date_3_hca":785,"consumption_at_set_date_4_hca":1041,"consumption_at_set_date_5_hca":1297,"consumption_at_set_date_6_hca":1553,"consumption_at_s\
et_date_7_hca":1809,"consumption_at_set_date_8_hca":2065,"consumption_at_set_date_9_hca":2321,"consumption_at_set_date_10_hca":4113,"consumption_at_set_date_11_hca":4369,"consumption_\
at_set_date_12_hca":4625,"consumption_at_set_date_13_hca":4881,"consumption_at_set_date_14_hca":5137,"consumption_at_set_date_15_hca":5393,"consumption_at_set_date_16_hca":5649,"consu\
mption_at_set_date_17_hca":5905,"error_flags":"MEASUREMENT RESET","timestamp":"1111-11-11T11:11:11Z"}

or am I missing something here? Please explain.

reset12, 2019-06-07:
Hello Fredrik,
The wildcard works very well, but which parameter i must use to get all values ? 
From my question 2.

with regards
Bruno

weetmuts, 2019-06-07:
Have you tried  --format=json ?

reset12, 2019-06-14:
Hello Fredrik,
format=json is working, thanks.

I have now buyed a Amber Stick to test the Innotas software, but i dont get the Stick running.

On Linux with your Software i have the following Log:
sudo wmbusmeters --debug --verbose --format=json --meterfiles=/tmp  --t1 /dev/ttyUSB0 MyHCA eurisii "831*" ""
(wmbusmeters) version: 
(config) store meter files in: "/tmp"
(config) using device: /dev/ttyUSB0
(config) number of meters: 1
(serialtty) opened /dev/ttyUSB0
(amb8465) are you there?
(serial /dev/ttyUSB0) sent "FF0B00F4"
(serial /dev/ttyUSB0) received "FF8B0453001DE9D7"
(serialtty) closed /dev/ttyUSB0
(serialtty) closed /dev/ttyUSB0
(amb8465) detected on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(config) meter eurisii link mode(s): t1
(config) all possible link modes that the meters might transmit on: t1
(config) explicitly listening to: t1
(config) listen link modes calculated to be: t1
(amb8465) set link mode 05
(serial /dev/ttyUSB0) sent "FF040105FF000000"
(serial /dev/ttyUSB0) received "FF"
(serial /dev/ttyUSB0) received "8401007A"
(amb8465) rssi 122
(amb8465) set link mode completed
(amb8465) set link mode "0100"
(amb8465) get config
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial /dev/ttyUSB0) received "FF8A7A00780080"
(serial /dev/ttyUSB0) received "710200000000FFFFFA00FFFFFFFFFF"
(serial /dev/ttyUSB0) received "FFFFFFFFFFFFFFFFFFFFFF0C320002"
(serial /dev/ttyUSB0) received "6400FFFFFFFFFF010004000000FFFFFF"
(serial /dev/ttyUSB0) received "01440000000000000000FFFF0B0601"
(serial /dev/ttyUSB0) received "00FFFFFFFFFF00050000FFFFFFFFFFFF"
(serial /dev/ttyUSB0) received "FF0000FFFFFFFFFFFFFFFFFFFFFFFF"
(serial /dev/ttyUSB0) received "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
(serial /dev/ttyUSB0) received "FFFFFFFFFFFFFFFFFFFF6F"
(amb8465) rssi 111
(amb8465) get config completed
(amb8465) get config "7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B060100FFFFFFFFFF00050000FFFFFFFFFFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
(amb8465) config: uart 78
(amb8465) config: radio Channel ff
(amb8465) config: rssi enabled ff
(amb8465) config: mode Preselect 00
(config) listen to link modes: t1
(wmbusmeters) configured "MyHCA" "eurisii" "831*" not-encrypted

Do you have any hint, the Amber Support is checking if the mode is wrong.

with regards
Bruno

weetmuts, 2019-06-14:
Please try the latest version of wmbusmeters. I fixed a bug two days ago,
that caused t1 mode to fail with amber dongle.

reset12, 2019-06-15:
Hello Fredrik,
great Job, it works now with the Amber Stick.
Now i must get it running with the Windows Software of Innotas.
Here ist the output of Wmbusmeters with my hca for your information:
sudo wmbusmeters --debug --verbose --format=json --meterfiles=/tmp  --t1 /dev/ttyUSB0 MyHCA eurisii "831*" ""
(wmbusmeters) version: 
(config) store meter files in: "/tmp"
(config) using device: /dev/ttyUSB0
(config) number of meters: 1
(serialtty) opened /dev/ttyUSB0
(amb8465) are you there?
(serial /dev/ttyUSB0) sent "FF0B00F4"
(serial /dev/ttyUSB0) received "FF8B0453001DE9D7"
(serialtty) closed /dev/ttyUSB0
(amb8465) detected on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(config) meter eurisii link mode(s): t1
(config) all possible link modes that the meters might transmit on: t1
(config) explicitly listening to: t1
(config) listen link modes calculated to be: t1
(amb8465) set link mode 08
(serial /dev/ttyUSB0) sent "FF040108F2000000"
(serial /dev/ttyUSB0) received "FF8401007A"
(amb8465) rssi 122
(amb8465) set link mode completed
(amb8465) set link mode "0100"
(amb8465) get config
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial /dev/ttyUSB0) received "FF8A7A007800"
(serial /dev/ttyUSB0) received "80710200000000FFFFFA00FFFFFFFFFF"
(serial /dev/ttyUSB0) received "FFFFFFFFFFFFFFFFFFFFFF0C320002"
(serial /dev/ttyUSB0) received "6400FFFFFFFFFF010004000000FFFF"
(serial /dev/ttyUSB0) received "FF01440000000000000000FFFF0B0601"
(serial /dev/ttyUSB0) received "00FFFFFFFFFF00050000FFFFFFFFFF"
(serial /dev/ttyUSB0) received "FFFF0000FFFFFFFFFFFFFFFFFFFFFFFF"
(serial /dev/ttyUSB0) received "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
(serial /dev/ttyUSB0) received "FFFFFFFFFFFFFFFFFFFFFF6F"
(amb8465) rssi 111
(amb8465) get config completed
(amb8465) get config "7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B060100FFFFFFFFFF00050000FFFFFFFFFFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
(amb8465) config: uart 78
(amb8465) config: radio Channel ff
(amb8465) config: rssi enabled ff
(amb8465) config: mode Preselect 00
(config) listen to link modes: t1
(wmbusmeters) configured "MyHCA" "eurisii" "831*" not-encrypted
(serial /dev/ttyUSB0) received "7644C5"
(serial /dev/ttyUSB0) received "250100108355087201001083C52555"
(serial /dev/ttyUSB0) received "08220000002F2F0B6E000000426E0000"
(serial /dev/ttyUSB0) received "82016E0000C2016E000082026E0000"
(serial /dev/ttyUSB0) received "C2026E000082036E0000C2036E000082"
(serial /dev/ttyUSB0) received "046E0000C2046E000082056E0000C2"
(serial /dev/ttyUSB0) received "056E000082066E0000C2066E00008207"
(serial /dev/ttyUSB0) received "6E0000C2076E000082086E0000C208"
(serial /dev/ttyUSB0) received "6E000002FD170000"
(amb8465) rssi 151
(wmbus) received telegram 83100001 C-field=44 M-field=25c5 (INE) A-field-version=55 A-field-dev-type=08 (Heat Cost Allocator) Ci-field=72 (EN 13757-3 Application Layer with long Transport Layer) CC-field=00 (slow_resp) long tpl header ACC=08 SN=00000000
(wmbus) frame "7644C5250100108355087201001083C5255508220000002F2F0B6E000000426E000082016E0000C2016E000082026E0000C2026E000082036E0000C2036E000082046E0000C2046E000082056E0000C2056E000082066E0000C2066E000082076E0000C2076E000082086E0000C2086E000002FD170000"
(wmbus) payload "2F2F0B6E000000426E000082016E0000C2016E000082026E0000C2026E000082036E0000C2036E000082046E0000C2046E000082056E0000C2056E000082066E0000C2066E000082076E0000C2076E000082086E0000C2086E000002FD170000"
(wmbus) 00: 76 length (118 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: c525 m-field (25c5=INE)
(wmbus) 04: 01001083 a-field-addr (83100001)
(wmbus) 08: 55 a-field-version
(wmbus) 09: 08 a-field-type (Heat Cost Allocator)
(wmbus) 0a: 72 ci-field (EN 13757-3 Application Layer with long Transport Layer)
(wmbus) 0b: 08 acc
(wmbus) 0c: 22 status ()
(wmbus) 0d: 1083 config ()
(wmbus) 7644C5250100108355087201001083
[... 199 lines cut ...]
(eurisii) 54: * 0000 consumption at set date 13 (0.000000 hca)
(eurisii) 56: 82 dif (16 Bit Integer/Binary Instantaneous value)
(eurisii) 57: 07 dife (subunit=0 tariff=0 storagenr=14)
(eurisii) 58: 6E vif (Units for H.C.A.)
(eurisii) 59: * 0000 consumption at set date 14 (0.000000 hca)
(eurisii) 5b: C2 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
(eurisii) 5c: 07 dife (subunit=0 tariff=0 storagenr=15)
(eurisii) 5d: 6E vif (Units for H.C.A.)
(eurisii) 5e: * 0000 consumption at set date 15 (0.000000 hca)
(eurisii) 60: 82 dif (16 Bit Integer/Binary Instantaneous value)
(eurisii) 61: 08 dife (subunit=0 tariff=0 storagenr=16)
(eurisii) 62: 6E vif (Units for H.C.A.)
(eurisii) 63: * 0000 consumption at set date 16 (0.000000 hca)
(eurisii) 65: C2 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
(eurisii) 66: 08 dife (subunit=0 tariff=0 storagenr=17)
(eurisii) 67: 6E vif (Units for H.C.A.)
(eurisii) 68: * 0000 consumption at set date 17 (0.000000 hca)
(eurisii) 6a: 02 dif (16 Bit Integer/Binary Instantaneous value)
(eurisii) 6b: FD vif (Second extension of VIF-codes)
(eurisii) 6c: 17 vife (Error flags (binary))
(eurisii) 6d: * 0000 error flags (0000)
(eurisii) 7644C52501001083550872010010832F2F0B6E000000426E000082016E0000C2016E000082026E0000C2026E000082036E0000C2036E000082046E0000C2046E000082056E0000C2056E000082066E0000C2066E000082076E0000C2076E000082086E0000C2086E000002FD170000

with regards
Bruno

reset12, 2019-06-19:
Hello Fredrik,
i was yesterday at may appartments to read out the hcas with your new wildcard parameters.
I found a problem in your programm, the wildcard works well, but you write the noew found hca over the existing. I have alwas only on record in the output-file.
You must add for every hca found a record in outputfile.
I used the parameter sudo wmbusmeters  --format=json --meterfilesaction=append --meterfiles=/tm/  --t1 /dev/ttyUSB0 MyHCA eurisii "831*" ""

You can every hca found in an internal array and write it out at a specific time like 10 Minutes or write all hcas out and after user end the programm user can deletes duplicates in a second step (easy way).

I sen the Amber USB-Stick with external Antenne back, because it is not better als the IMST Stick and i dont get it running with the Innotas Software. Your Programm Init the Amberstick by yourself with the right parameters, but for the Innotas Software the Stick mit be Init with a Firmwareprogramm of Wuerth Electronics. So your Software is definitly the better solution when people does not need to chnage parameters in the hca.

I plan to install your software on a raspberry, so i can read the Data from remote. The problem i have, is to find a location in the house, where i have Power, Internet and can reach all hcas.

with regards
Bruno

with regards
Bruno

weetmuts, 2019-06-20:
ok, i added --meterfilesnaming=id which will select the meter id from the received telegrams
as the filename for meterfiles. It also supports --meterfilesnaming=name-id 
The default is --meterfilesnaming=name which is the original behaviour.

Please test!

reset12, 2019-07-01:
> as the filename for meterfFredrik,ello fles. It also supports --meterfilesnaming=name-id

Hello Fredrik,
i tested the new option --meterfilesnaming=id today and it works very well.
It save much time to get all data from the different meters.

Please can you change the string consumption_at_set_date_hca to consumption_at_set_date_hca1 for the first month. It is more easy to process the data.

with regards
Bruno

weetmuts, 2019-07-07:
Ok, I added consumption_at_set_date_1_hca in addition to consumption_at_set_date_hca
and I did the same change for the qcaloric meter.

Perhaps I will drop consumption_at_set_date_hca for both Eurisii and Qcaloric in the future, or perhaps not. :-)

reset12, 2019-07-08:
Hello Fredrik,
Thank you for the change.

with regards
Bruno

closed 2019-08-14 by weetmuts