wmbusmeters/wmbusmeters #26: Amber USB Dongle 8465-M and Apator

marcinbien, 2019-05-30:
Hi,
I have issue with reading values from Apator Water Meter (apator162).
I'm using Amber USB Dongle 8465-M.
I'm not sure if I'm doing it right.

When I'm trying to sniff all telegrams using command:
 `wmbusmeters --logtelegrams --debug --format=json --c1 auto`

I see a telegram from my meter:
`(wmbus) 6E4401061034260105077A4C006085
(serial /dev/amb8465) received "6E"
(serial /dev/amb8465) received "4401063676690105077A98006085F912"
(serial /dev/amb8465) received "320384F4AD4FF6E3B86A790A461160"
(serial /dev/amb8465) received "3E85ABD154168E13EDA6779B66E5C7EB"
(serial /dev/amb8465) received "5E43C0FD359B9E1C33FA801C94B046"
(serial /dev/amb8465) received "2620F005F1AB5B263D2BFC41D42A74"
(serial /dev/amb8465) received "5712D2F43C674E881881CDFC1603BB62"
(serial /dev/amb8465) received "C2636CD4E67122336F7544305C1C62"
(serial /dev/amb8465) received "2058"
(amb8465) rssi 219
(wmbus) received telegram 01697636 C-field=44 M-field=0601 (APA) A-field-version=05 A-field-dev-type=07 (Water meter) Ci-field=7a (EN 13757-3 Application Layer with short Transport Layer) CC-field=00 (slow_resp) short header ACC=98 
(wmbus) frame "6E4401063676690105077A98006085F912320384F4AD4FF6E3B86A790A4611603E85ABD154168E13EDA6779B66E5C7EB5E43C0FD359B9E1C33FA801C94B0462620F005F1AB5B263D2BFC41D42A745712D2F43C674E881881CDFC1603BB62C2636CD4E67122336F7544305C1C622058"
(wmbus) payload "F912320384F4AD4FF6E3B86A790A4611603E85ABD154168E13EDA6779B66E5C7EB5E43C0FD359B9E1C33FA801C94B0462620F005F1AB5B263D2BFC41D42A745712D2F43C674E881881CDFC1603BB62C2636CD4E67122336F7544305C1C622058"
(wmbus) 00: 6e length (110 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 0106 m-field (601=APA)
(wmbus) 04: 36766901 a-field-addr (01697636)
(wmbus) 08: 05 a-field-version
(wmbus) 09: 07 a-field-type (Water meter)
(wmbus) 0a: 7a ci-field (EN 13757-3 Application Layer with short Transport Layer)
(wmbus) 0b: 98 acc
(wmbus) 0c: 00 status ()
(wmbus) 0d: 6085 config (encrypted AES_CBC bidirectional)
(wmbus) 6E4401063676690105077A98006085`

But when I'm trying to read data only from my meter using command:
`# wmbusmeters --logtelegrams --debug --verbose --format=json auto kitchen apator162 01697636 00000000000000000000000000000000`

I'm getting only information that program is started and that's all 
`(wmbusmeters) version: 0.9.8
(config) using device: auto
(config) number of meters: 1
(serialtty) closed /dev/im871a
(serialtty) opened /dev/amb8465
(amb8465) are you there?
(serial /dev/amb8465) sent "FF0B00F4"
(serial /dev/amb8465) received "86FF8B042701BB54B9"
(serialtty) closed /dev/amb8465
(serialtty) closed /dev/amb8465
(amb8465) detected on /dev/amb8465
(serialtty) opened /dev/amb8465
(amb8465) set link mode 05
(serial /dev/amb8465) sent "FF040105FF000000"
(serial /dev/amb8465) received "FF840100"
(serial /dev/amb8465) received "7A"
(amb8465) rssi 122
(amb8465) set link mode completed
(amb8465) set link mode "0100"
(amb8465) get config
(serial /dev/amb8465) sent "FF0A02008077"
(serial /dev/amb8465) received "FF8A7A00780080"
(serial /dev/amb8465) received "710200000000FFFFFA00FFFFFFFFFF"
(serial /dev/amb8465) received "FFFFFFFFFFFFFFFFFFFFFF0032000214"
(serial /dev/amb8465) received "00FFFFFFFFFF010004000000FFFFFF"
(serial /dev/amb8465) received "01440000000000000000FFFF0B040100"
(serial /dev/amb8465) received "FFFFFFFFFF00030000FFFFFFFFFFFF"
(serial /dev/amb8465) received "FF0000FFFFFFFFFFFFFFFFFFFFFFFF"
(serial /dev/amb8465) received "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
(serial /dev/amb8465) received "FFFFFFFFFFFFFFFFFFFF17"
(amb8465) rssi 23
(amb8465) get config completed
(amb8465) get config "7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B040100FFFFFFFFFF00030000FFFFFFFFFFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
(amb8465) config: uart 78
(amb8465) config: radio Channel ff
(amb8465) config: rssi enabled ff
(amb8465) config: mode Preselect 00
(config) using link mode: T1
(wmbusmeters) configured "kitchen" "apator162" "01697636" encrypted
`

weetmuts, 2019-06-03:
I extracted your meter frame and put it into a simulation_apator.txt file:
telegram=6E4401063676690105077A98006085F912320384F4AD4FF6E3B86A790A4611603E85ABD154168E13EDA6779B66E5C7EB5E43C0FD359B9E1C33FA801C94B0462620F005F1AB5B263D2BFC41D42A745712D2F43C674E881881CDFC1603BB62C2636CD4E67122336F7544305C1C622058

Then ran wmbusmeters 
./build/wmbusmeters simulation_apator.txt kitchen apator162 01697636 00000000000000000000000000000000

And it properly printed:
kitchen	01697636	0.924 m3	2019-06-03 08:46.09

When you listen for a specific meter and have --debug enable, you should still
see the frames, that are then discarded, if they are not for the configured meter.

Since there are no frames in the output, perhaps the meter is not sending?
It might only send during day-hours to convserve battery?

nestorayuso, 2019-06-03:
{
  "Raw": "0x6e4401063676690105077a98006085f912320384f4ad4ff6e3b86a790a4611603e85abd154168e13eda6779b66e5c7eb5e43c0fd359b9e1c33fa801c94b0462620f005f1ab5b263d2bfc41d42a745712d2f43c674e881881cdfc1603bb62c2636cd4e67122336f7544305c1c622058",
  "Length": 110,
  "CField": "0x44",
  "CFieldString": "SND_NR",
  "MField": "0x0106",
  "MFieldCodeString": "APA",
  "MFieldLongString": "SA (Electricity), Zolkiewskiego 21/29, 87-100, Torun,Poland",
  "Id": 23688758,
  "IdString": "01697636",
  "Version": 5,
  "Device": "0x07",
  "DeviceString": "Water",
  "CiField": "0x7a",
  "HeaderKnown": true,
  "PayloadKnown": true,
  "Header": {
    "EncryptionMode": 5,
    "EncryptionModeString": "AES with CBC",
    "EncryptedBlocks": 6,
    "HopCount": 0,
    "IsAccessible": true,
    "IsBidirectionalMode": true,
    "IsSynchronous": false,
    "ReservedBit": false,
    "TelegramType": 0,
    "AccessNumber": 152,
    "StatusByte": 0,
    "ConfigField": [
      96,
      133
    ]
  },
  "Body": {
    "Raw": "0xf912320384f4ad4ff6e3b86a790a4611603e85abd154168e13eda6779b66e5c7eb5e43c0fd359b9e1c33fa801c94b0462620f005f1ab5b263d2bfc41d42a745712d2f43c674e881881cdfc1603bb62c2636cd4e67122336f7544305c1c622058",
    "DataRecords": [],
    "IsEncrypted": true,
    "DecryptionFailed": false
  },
  "AField": {
    "Id": 23688758,
    "Version": 5,
    "Device": 7
  },
  "CrcValid": false
}

weetmuts, 2019-06-03:
@nestorayuso Neat, that looks like the output from another wmbus parser.
But as you can see, that parser also can only report that the Apator sends
a single "DifFunctionString": "Manufacturer specific data" blob. :-)

What tool did you use for that?

weetmuts, 2019-06-04:
Aha! Now I see what your problem is. Your meter transmits C1 telegrams, but wmbusmeters hardcodes T1 telegrams. for this meter. Apparently the Apator can be configured to use either T1 or C1.

I'll add support for C1 and apator.

marcinbien, 2019-06-06:
Thx @weetmuts. In the meantime I figured it out also. 

Now I need to figure out why I'm not able to receive any telegrams from electricity meter. I have the Tauron Amiplus meter but but when I'm listening in the T1 mode I don't see any frames - I don't know how frequently they send information.

majekw, 2019-06-06:
@marcinbien, my Amiplus Echelon meter sends data about every 1 minute, but sometimes there are gaps for few minutes. I think I didn't see gaps longer than 10 minutes. It's random and I don't know the cause. I suspect some interference as I also receive signals from ~70 water meters around :-)
Maybe stupid question: did you enable radio transmission on Tauron website? Because it's off by default.

weetmuts, 2019-06-06:
@marcinbien Ah, now I have added proper support for listening to C1 or T1 apators....
That was ridiculously complicated! :-)  

Because:
different wmbus dongles might listen to different possible link modes.
im871a and amb8465 can only listen to 1 link mode out of C1,T1,S1,S1m ...
but rtlwmbus can listen to C1 and T1 at the same time...

a meter such as multical21 only transmits on C1
and supercom587 only T1
and apator162 C1 or T1....

then you can lock down exactly what you want to listen to using
--listento=c1,t1 if you have an rtlwmbus or just --listento=t1
if you have an im871a or amb8465. --c1 and --t1 still works.

So, what is the user interface. Simple. You specify:
apator162:t1 or apator162:c1 
to tell wmbusmeters what kind of telegrams you expect for this meter.
actually you can also do:
apator162:c1,t1  which is the default and this will only work if you use an rtlwmbus dongle.

Please try it out.

marcinbien, 2019-06-09:
@weetmuts thx everything is working fine with apator wmbus meters

marcinbien, 2019-06-09:
@majekw yes I have enabled it on tauron website. I see "Adres komunikacyjny" this is probably the meter Id and "Klucz szyfrujący" this is the AES encryption key.

I've meter "NES (Network Energy Services) Power by ECHELON" and I'm using same Amber USB Dongle 8465-M.

Log:
`
wmbusmeters --logtelegrams --debug --format=json --t1 auto energy amiplus:t1 12345678 FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
(wmbusmeters) version: 0.9.8-6-gcecf607
(config) using device: auto
(config) number of meters: 1
(serialtty) closed /dev/im871a
(serialtty) opened /dev/amb8465
(amb8465) are you there?
(serial /dev/amb8465) sent "FF0B00F4"
(serial /dev/amb8465) received "D5FF8B042701BB54B9"
(serialtty) closed /dev/amb8465
(serialtty) closed /dev/amb8465
(amb8465) detected on /dev/amb8465
(serialtty) opened /dev/amb8465
(config) meter amiplus link mode(s): t1
(config) all possible link modes that the meters might transmit on: t1
(config) explicitly listening to: t1
(config) listen link modes calculated to be: t1
(amb8465) set link mode 05
(serial /dev/amb8465) sent "FF040105FF000000"
(serial /dev/amb8465) received "FF840100"
(serial /dev/amb8465) received "7A"
(amb8465) rssi 122
(amb8465) set link mode completed
(amb8465) set link mode "0100"
(amb8465) get config
(serial /dev/amb8465) sent "FF0A02008077"
(serial /dev/amb8465) received "FF8A7A00780080"
(serial /dev/amb8465) received "710200000000FFFFFA00FFFFFFFFFFFF"
(serial /dev/amb8465) received "FFFFFFFFFFFFFFFFFFFF0032000214"
(serial /dev/amb8465) received "00FFFFFFFFFF010004000000FFFFFF"
(serial /dev/amb8465) received "01440000000000000000FFFF0B040100"
(serial /dev/amb8465) received "FFFFFFFFFF00030000FFFFFFFFFFFF"
(serial /dev/amb8465) received "FF0000FFFFFFFFFFFFFFFFFFFFFFFFFF"
(serial /dev/amb8465) received "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
(serial /dev/amb8465) received "FFFFFFFFFFFFFFFFFFFF17"
(amb8465) rssi 23
(amb8465) get config completed
(amb8465) get config "7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B040100FFFFFFFFFF00030000FFFFFFFFFFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
(amb8465) config: uart 78
(amb8465) config: radio Channel ff
(amb8465) config: rssi enabled ff
(amb8465) config: mode Preselect 00
(config) listen to link modes: t1
`

but I'm not getting any frames

weetmuts, 2019-06-09:
Ah, it could be that I misread the documentation for the dongle.

Can you change the 0x05 to 0x09 in wmbus_amb8465.cc
on line 247? If that does not work try 0x08.

It might be that 0x05 is the setting to transmit in T1, but 0x08 is receive in T1.
0x09 is interesting, perhaps it can receive both C1 and T1...

Can you try 0x09 and then run wmbusmeters --t1 auto  for a while 
to see if you hear telegrams both from your c1 apator and from your t1 amiplus?

The manual for the dongle can be found here: is https://www.ambiductor.se/nedladdning/amb-8465-8466-8568/english?download=139

weetmuts, 2019-06-11:
I have pushed a fix. Please test! You should now be able to listen to both your apator162:c1 and your amiplus:t1 at the same time.

weetmuts, 2019-06-11:
And please test:
wmbusmeters --listento=c1,t1 auto
to see that it hears both your meters!

marcinbien, 2019-06-12:
Hi, thanks for fix. Currently I'm able to listen for apator water meters on t1 but still I don't see frames from Tauron NES (Network Energy Services) Power by ECHELON electricity meter.
I think my issue with electricity meter is not related with your software - I will try catch frames using software provided by Amber (Amber Commander).

weetmuts, 2019-06-13:
Ok, lets close this issue, since you can now listen to your apator162 both using c1 and t1.

Open another one if you still have problems with the amiplus compatible meter.

closed 2019-06-13 by weetmuts