wmbusmeters/wmbusmeters #27: Timestamp 2h diff in json output

antosim, 2019-06-06:
Hi,

I just started to use the wmbusmeter tool to decrypt the messages from Multical meter. I noted that the timestamp differs 2h in the json output. The timestamp is correct when no format is given.

Include lastdebug/verbose output and date info from my server just after stopped wmbusmeter:

(serial /dev/im871a) received "A5E20323442D2C010641581B168D20DA24012B2019D4C34436FAAEB23E2EE5C4057834F561A92135C5000000A3FA"
(im871a) timestamp 0000c535
(im871a) rssi 00
(im871a) got crc16 faa3 expected faa3
(wmbus) received telegram 58410601 C-field=44 M-field=2c2d (KAM) A-field-version=1b A-field-dev-type=16 (Cold water meter) Ci-field=8d (Extended Link Layer II (8 Byte)) CC-field=20 (slow_resp sync) ell header ACC=da SN=202b0124
(wmbus) frame "23442D2C010641581B168D20DA24012B2019D4C34436FAAEB23E2EE5C4057834F561A921"
(wmbus) payload "19D4C34436FAAEB23E2EE5C4057834F561A921"
(wmbus) 00: 23 length (35 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 2d2c m-field (2c2d=KAM)
(wmbus) 04: 01064158 a-field-addr (58410601)
(wmbus) 08: 1b a-field-version
(wmbus) 09: 16 a-field-type (Cold water meter)
(wmbus) 0a: 8d ci-field (Extended Link Layer II (8 Byte))
(wmbus) 0b: 20 cc-field (slow_resp sync)
(wmbus) 0c: da acc
(wmbus) 0d: 24012b20 sn
(wmbus) 23442D2C010641581B168D20DA24012B20
(meter) MyTapWater: for me? 58410601
(meter) MyTapWater: yes for me
(multical21) MyTapWater 58410601 (Mode1) decrypting "19D4C34436FAAEB23E2EE5C4057834F561A921"
(Mode1) IV 2D2C010641581B162024012B20000000
(Mode1) decrypted first block "EF9079EB61ACEA00006B2A00004C2200"
(Mode1) IV+1 2D2C010641581B162024012B20000001
(Mode1) decrypted second block "000A15"
(Mode1) decrypted "EF9079EB61ACEA00006B2A00004C2200000A15"
(multical21) log "telegram=|23442D2C010641581B168D20DA24012B20|EF9079EB61ACEA00006B2A00004C2200000A15|+151"
(ACS) ignoring telegram since format signature hash 0x61eb is yet unknown.
(multical21) 11: ef90 payload crc
(multical21) 13: 79 frame type (short frame)
(multical21) 14: eb61 format signature
(multical21) 23442D2C010641581B168D20DA24012B20EF9079EB61
{"media":"cold water","meter":"multical21","name":"MyTapWater","id":"58410601","total_m3":0,"target_m3":0,"max_flow_m3h":0,"flow_temperature_c":127,"external_temperature_c":127,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2019-06-06T08:46:11Z"}
^C(serialtty) event loop stopped!
(serialtty) closed /dev/im871a
(serialtty) closed /dev/im871a
banshee@kerberos ~ $ date
tor  6 jun 2019 10:46:19 CEST

It seems like the timestamp in the json is UTC time, while the server time is UTC+2 = CEST, which is the correct time.

Regards
//Anders

weetmuts, 2019-06-06:
Yes, well, the timestamp is 2019-06-06T08:46:11Z which means that it is in UTC time, no daylight savings. If I understood it correctly. The idea was that the json is going to be parsed by software anyway, so the conversion back to your local timezone could be done on that side....

But, on the other hand, since the conversion is to be done by software it could just as
well print using the local time zone....perhaps. What do you think?

antosim, 2019-06-07:
I think it is better if the wmbusmeter handles the time conversion. There might be an issue with legacy users though, who will get a gap/diff, if they have noticed the time difference.

weetmuts, 2019-07-07:
Well, I finally decided, it will stay in UTC time. I have added a note to the README.
The note also explains that the human readable output is in local time.

Anyhow, thanks for asking the question! Hope wmbusmeters works for you!

closed 2019-07-07 by weetmuts