wmbusmeters/wmbusmeters #28: apator162 - identified numbers of days since last reset

jupiter9595, 2019-09-05:
Still assumption but well documented and proven for 3 different meters
apator162 proprietary protocol - numbers of days since last reset / initial setup.

Below data from 2 different meters

[M1] 2019-09-05 14:47.32       2F|2F|0F|A9|65|99|93|0E|02|08|43|F3|01|82|DA|55|76|84|65|79|08|9E|07|0A|10|87|16|01|00|71|01|87|16|01|00|87|16|01|00|A0|1A|86|AF|03|FF|FFE6E2
[M2] 2019-09-05                2F|2F|0F|CF|65|99|93|0E|02|00|43|96|00|82|CD|79|74|CE|65|79|FF|E4|4B|00|10|2E|8A|00|00|75|01|0E|85|00|00|E6|54|00|00|97|17|00|00|04|01|00000401000000000000
                           0  1  2  3  4  5  6  7  8  9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36

M1 - set up 2018
M2 - set up 2019

location 11, 12

M1 - hex 01F3 - 499 days
M2 - hex 0096 - dec 150 days

weetmuts, 2019-09-16:
I added code for this, which you can try out, its pushed to the branch ReadApatorInitDate.

It works for your telegram and for the other apator162 telegrams I have in the test data,
except one.....

Unfortunately there is an apator162 telegram from the existing test data where the number
of days becomes so many.... so the init_date becomes the year 1894.....

Try the branch and run "make test" and you will find it.
How should we handle this?

weetmuts, 2019-12-28:
I have tried this, and i seems like perhaps the offset change between telegrams.
I will close this until we get more verified data and telegram logs. Thanks for the info anyway!

closed 2019-12-28 by weetmuts