wmbusmeters/wmbusmeters #29: Add exclude to meter matching

majekw, 2019-09-10:
Hello,

it would be nice feature to exclude some ids from meter matching.
Or maybe make more specific matching and process packet using only one decoder?
Maybe something like this `id=22*,!2222222`?

My scenario: I have 2 entries in wmbusmeters.d/, one for my amiplus meter with specific id, and second with `id=*` to process all water meters (mkradio3) I could receive around.
Right now packets from energy meter are  processed by amiplus AND mkradio3, so I'm getting a lot of nonsense messages from power meter processed as water meter. Exclude energy meter from water meters could fix this :-)

weetmuts, 2019-09-16:
Ok, just pushed an implementation that does this. Please try!

majekw, 2019-09-16:
Thank you!
But something is wrong. I've got message: 
`Not a valid meter id nor a valid meter match expression "*,!00156327,!00048713"`
and all messages from water meters are dropped. My other electricity meters still works fine.
I run wmbusmeters with:
` wmbusmeters --useconfig=/`
Config for water meter in `/etc/wmbusmeters.d/woda`:
```name=woda
type=mkradio3
id=*,!00156327,!00048713
key=
```

weetmuts, 2019-09-16:
Meh, my silly bug. Fixed and pushed.

majekw, 2019-09-16:
Now it works great :+1: 
Thank you!

closed 2019-09-16 by majekw