wmbusmeters/wmbusmeters #30: How to update

Horndrup, 2019-09-24:
What is the best way to install updates?

weetmuts, 2019-09-24:
Running "sudo make install" should be safe. The only thing it should overwrite are the binaries and the manual page.

It will print "... something ...  unchanged" for the other configuration files.

It does not currently detect that it has to upgrade any important configuration file.

I suppose it should! :-) Aka room for improvement.

But you can manually compare the file that would have been installed (is visible inside the install.sh script) with the file already installed. 

Or you could just remote wmbusmeters.service and 99-wmbus-usb-serial.rules
and redo the install. Since these are the only ones that might have changed.
Though it was some time ago since they changed.

weetmuts, 2019-10-16:
Did you solve the upgrade? Can I close this issue now?

Horndrup, 2019-10-20:
Yes I solved it. Sorry for my late feedback.

closed 2019-10-20 by Horndrup