wmbusmeters/wmbusmeters #32: GWF watermeter

ahoiahoi, 2019-10-01:
Greetings
I have a https://gwf.ch/en/produkt/gwfcoder-radio-wireless-m-bus/
```
Here are a few telegrams:
(wmbus) received telegram 00611742 C-field=44 M-field=1ee6 (GWF) A-field-version=02 A-field-dev-type=0e (Bus/System component) Ci-field=72 (EN 13757-3 Application Layer with long Transport Layer) CC-field=00 (slow_resp) long tpl header ACC=07 SN=00000000
(mtk3) MyTapWater 00611742 (mtk3) log "telegram=|3644E61E42176100020E7282997515|CEE6C7D7947AB4CA442D749A1A6F7A2BDB15F369D6B9CD0C9793F38EF52F4AAF|+414"
(wmbus) received telegram 00611742 C-field=44 M-field=1ee6 (GWF) A-field-version=02 A-field-dev-type=0e (Bus/System component) Ci-field=72 (EN 13757-3 Application Layer with long Transport Layer) CC-field=00 (slow_resp) long tpl header ACC=07 SN=00000000
(mtk3) MyTapWater 00611742 (mtk3) log "telegram=|3644E61E42176100020E7282997515|42E6FE98643C41E281D190EAEEC2A69F8C027CA698D883F6D215BEF924279F17|+429"
(wmbus) received telegram 00611742 C-field=44 M-field=1ee6 (GWF) A-field-version=02 A-field-dev-type=0e (Bus/System component) Ci-field=72 (EN 13757-3 Application Layer with long Transport Layer) CC-field=00 (slow_resp) long tpl header ACC=07 SN=00000000
(mtk3) MyTapWater 00611742 (mtk3) log "telegram=|3644E61E42176100020E7282997515|17808884BB9F59579D9EA84AB4A0936350C1938DE0CE1385D0ADDB1773F31C7D|+444"
(wmbus) received telegram 00611742 C-field=44 M-field=1ee6 (GWF) A-field-version=02 A-field-dev-type=0e (Bus/System component) Ci-field=72 (EN 13757-3 Application Layer with long Transport Layer) CC-field=00 (slow_resp) long tpl header ACC=07 SN=00000000
(mtk3) MyTapWater 00611742 (mtk3) log "telegram=|3644E61E42176100020E7282997515|FADE33E99FF004A137C111E008743E8F2EBC9ECE67C5B3032060EBE92C4566F3|+459"
(wmbus) received telegram 00611742 C-field=44 M-field=1ee6 (GWF) A-field-version=02 A-field-dev-type=0e (Bus/System component) Ci-field=72 (EN 13757-3 Application Layer with long Transport Layer) CC-field=00 (slow_resp) long tpl header ACC=07 SN=00000000
(mtk3) MyTapWater 00611742 (mtk3) log "telegram=|3644E61E42176100020E7282997515|DEB0C51BE197DD4CDE39E700A2EEA13F2FD1CD6F401A55CA40D0A698E7871D7E|+474"
(wmbus) received telegram 00611742 C-field=44 M-field=1ee6 (GWF) A-field-version=02 A-field-dev-type=0e (Bus/System component) Ci-field=72 (EN 13757-3 Application Layer with long Transport Layer) CC-field=00 (slow_resp) long tpl header ACC=07 SN=00000000
(mtk3) MyTapWater 00611742 (mtk3) log "telegram=|3644E61E42176100020E7282997515|8F3DA579019245D3124DE479847F6DD69AE8AE1A65821D66B8EF4B6DB59EEAE8|+489"
(wmbus) received telegram 00611742 C-field=44 M-field=1ee6 (GWF) A-field-version=02 A-field-dev-type=0e (Bus/System component) Ci-field=72 (EN 13757-3 Application Layer with long Transport Layer) CC-field=00 (slow_resp) long tpl header ACC=07 SN=00000000
(mtk3) MyTapWater 00611742 (mtk3) log "telegram=|3644E61E42176100020E7282997515|7811790F409F8A9F4C371ACB74C8F60D8B6B6E4F63D3DD5554144A50DC51DBC9|+504"
(wmbus) received telegram 00611742 C-field=44 M-field=1ee6 (GWF) A-field-version=02 A-field-dev-type=0e (Bus/System component) Ci-field=72 (EN 13757-3 Application Layer with long Transport Layer) CC-field=00 (slow_resp) long tpl header ACC=07 SN=00000000
(mtk3) MyTapWater 00611742 (mtk3) log "telegram=|3644E61E42176100020E7282997515|27F22A2A0763CCE865AD0F70265880813D4DDA18561B2297CF815B9641726773|+519"
(wmbus) received telegram 00611742 C-field=44 M-field=1ee6 (GWF) A-field-version=02 A-field-dev-type=0e (Bus/System component) Ci-field=72 (EN 13757-3 Application Layer with long Transport Layer) CC-field=00 (slow_resp) long tpl header ACC=07 SN=00000000
(mtk3) MyTapWater 00611742 (mtk3) log "telegram=|3644E61E42176100020E7282997515|AF3249D2E196A8249A6AD03B6EF0D66951AB329061341DAB64B939910597149E|+534"
```

Is it possible to parse the content?

Thanks

weetmuts, 2019-10-02:
Yes, it looks like it should be possible! However, you need the encryption key for your meter.
Do you have it already?

ahoiahoi, 2019-10-02:
Hi, no I don't have it. How would I optain it? Is it per device or per vendor?

weetmuts, 2019-10-02:
It is a device specific key that makes sure that you cannot listen to your neighbours water consumption.

There should be no harm in you having the key for your own meter.

You have to ask your water municipality/company that provided you with the meter if you can get your key. For the purpose of home automation and optimizimg your water consumption.

A lot of water companies/municipalities do hand out the keys.

ahoiahoi, 2019-10-03:
So, the water company told me they cannot give me the key, because it is generic :(
That's it, I guess?

weetmuts, 2019-10-04:
Yes, unfortunately.

Not a very good technical solution to have a generic key for all meters.

ahoiahoi, 2019-10-04:
Alright, thank you.
I will have to recycle my rsa token ocr reader then ;(

closed 2019-10-04 by ahoiahoi

weetmuts, 2019-10-07:
But if you really need to monitor your water consumption, and want to reuse your wmbus dongle, you could buy your own meter that transmits using wmbus and then have it installed after the water municipalities meter.

If you find a nice meter, that is not supported by wmbusmeter, I can add support for it, as long as the meter conforms to the wmbus (and/or the OMS standard), ie no guarantees given for naughty meters like the apator162, techem and others that stuff proprietary stuff into the wmbus telegrams. :-)

lostcontrol, 2021-06-15:
> So, the water company told me they cannot give me the key, because it is generic :(
> That's it, I guess?

Just for the record, same meter brand and same answer from the water company: the key is the same for all meters :cry: