wmbusmeters/wmbusmeters #33: Bmeters rfm rx2 support

merbanan, 2019-10-12:
Hi, I have a rfm rx2 usb stick. It outputs raw wmbus (without crc) over the serial port. What is needed to add support for this device?

I wrote a proof of concept program to interface with the device and you only need to open the serial port and the telegrams just start to appear. It should be fairly easy to add support for this device.

https://github.com/merbanan/wmbus

weetmuts, 2019-10-13:
Yep, it should be easy to add basic support for the device.

Does it listen to t1, c1 or both at the same time?

Can you do a dmesg after you inserted the rfmrx2 dongle and post here the lines that identifies the USB device idVendor and idProduct?

merbanan, 2019-10-13:
[236161.672636] usb 3-1: new full-speed USB device number 2 using xhci_hcd
[236161.861116] usb 3-1: New USB device found, idVendor=2047, idProduct=0863
[236161.861120] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[236161.861122] usb 3-1: Product: MSP430-USB Example
[236161.861124] usb 3-1: Manufacturer: Texas Instruments
[236161.941048] cdc_acm 3-1:1.0: No union descriptor, testing for castrated device
[236161.941070] cdc_acm 3-1:1.0: ttyACM0: USB ACM device
[236161.944251] usbcore: registered new interface driver cdc_acm
[236161.944252] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters

merbanan, 2019-10-13:
I also sent you an email few weeks ago, it maybe went to the spam folder.

merbanan, 2019-10-13:
I also have 2 different thermo/hygrometers that needs support. Almost all fields are identified.

merbanan, 2019-10-13:
Regarding what it listens to it for sure supports T1. No idea if anything else is supported.

https://www.piigab.com/produkter/hardware/pi-temp/pi-temp_wl/

and

http://media1.lansensystems.com/datasheets/LAN-WMBUS-G2-TH_v2_2.pdf

can be received.

merbanan, 2019-10-14:
https://www.bmeters.com/en/bm_product/rfm-rx2/

weetmuts, 2019-10-15:
Ok, since we do not know how to properly detect nor configure an rfmrx2 usb dongle, I opted for a generic solution. If you specify a /dev/ttyXXX device and suffix it with for example :38400 it will use that number as a baud rate and just start listening. 

If we receive more info on how to use an rfmrx2 dongle, then we can add proper support for it.

Run for example:
wmbusmeters --listento=t1 /dev/ttyACM0:38400 
(of course it does not currently care about the --listento=t1, since it cannot be configured.)

Please test! I have no such dongle. :-)

merbanan, 2019-10-15:
They use a usb capable MCU and then the CC110L transceiver. The CC110L chip is most likely configured to trigger on the preamble and sync. This means it can only support a specific bitrate and frequency. So it can most likely only use the T1 mode. The gui has support for configuring sensors. So there is a configuration command protocol.

In theory the MCU could reprogram the transceiver for mode C but most likely there is no support for that.

merbanan, 2019-10-15:
HYDROLINK, HYDROCLIMA and HYDROCAL are software that can use the rfmrx2 dongle. I'll try to sniff the serialbus to see what they actually are transmitting. But most likely the receiving part is just a simple packet formater.

merbanan, 2019-10-15:
The code works but after a while the device stops returning packets. A restart fixes the issue, most likely the needs a open close cycle of the ttyACM0 devices when that is detected. I also observed this with my proof of concept code.

merbanan, 2019-10-15:
```
serial /dev/ttyACM0) received "5744B40984217016101B7A0F0800000265A60842657E08820165BA0822657908126584086265710852652D0902FB1AC70142FB1ABC018201FB1A8E0122FB1AB70112FB1AC80162FB1A6F0152FB1AC701066D3B13286E2A00"
(wmbus) received telegram 16702184 C-field=44 M-field=09b4 (BMT) A-field-version=10 A-field-dev-type=1b (Room sensor (eg temperature or humidity)) Ci-field=7a (EN 13757-3 Application Layer with short Transport Layer) CC-field=00 (slow_resp) short header ACC=0f 
(wmbus) frame "5744B40984217016101B7A0F0800000265A60842657E08820165BA0822657908126584086265710852652D0902FB1AC70142FB1ABC018201FB1A8E0122FB1AB70112FB1AC80162FB1A6F0152FB1AC701066D3B13286E2A00"
(wmbus) payload "0265A60842657E08820165BA0822657908126584086265710852652D0902FB1AC70142FB1ABC018201FB1A8E0122FB1AB70112FB1AC80162FB1A6F0152FB1AC701066D3B13286E2A00"
(wmbus) 00: 57 length (87 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: b409 m-field (9b4=BMT)
(wmbus) 04: 84217016 a-field-addr (16702184)
(wmbus) 08: 10 a-field-version
(wmbus) 09: 1b a-field-type (Room sensor (eg temperature or humidity))
(wmbus) 0a: 7a ci-field (EN 13757-3 Application Layer with short Transport Layer)
(wmbus) 0b: 0f acc
(wmbus) 0c: 08 status ()
(wmbus) 0d: 0000 config ()
(wmbus) 5744B40984217016101B7A0F080000
Received telegram from: 16702184
          manufacturer: (BMT) BMETERS, Italy
           device type: Room sensor (eg temperature or humidity)
(dummyserial) telegram ignored by all configured meters!

```

weetmuts, 2019-10-16:
Ok, I added the option --reopenafter=10m which means that it will close and reopen the usb device every 10 minutes. Can also be added to the wmbusmeters.conf.

See if this solves your problem. Perhaps it is enough to reopen every 60m or 12h?

merbanan, 2019-10-16:
I think this is a bug in the hardware that depends on the amount of transmitted telegrams. A 10m reopen will probably solve all issues.

merbanan, 2019-10-16:
Looks ok
(config) wmbusmeters close/open the wmbus dongle fd after every 600 seconds
but parameter order matters

`--reopenafter=10m --listento=t1 /dev/ttyACM0:38400`
OK

```
--listento=t1 /dev/ttyACM0:38400 --reopenafter=10m
For each meter you must supply a: name,type,id and key.
```
NOK

Runnig a test now.

merbanan, 2019-10-16:
Still stable after an hour. I'll run it some more and then close the issue if it works fine.

weetmuts, 2019-10-20:
Ok, since rfmrx2 does need some tailored configuration (--reopenafter=10m)
therefore I have added the rfmrx2 mnemonic, that will set the reopenafter=10m automatically.

I have also added an explicit rfmrx2 and added the idVendor/idDevice to the install script.
This means that the wmbusmeters daemon will start automatically when you plug in the dongle
or when your computer boots.

Remove /etc/udev/rules.d/99-wmbus-usb-serial.rules and run "make install"
to have the new udev rule installed. Now reboot or trigger a udev reload.

Now please test:
wmbusmeters --listento=t1 /dev/ttyACM0:rfmrx2
wmbusmeters --listento=t1 auto

And of course test if it starts automatically when you insert the dongle.

Thanks!

merbanan, 2019-10-27:
wmbusmeters --listento=t1 /dev/ttyACM0:rfmrx2
No meters configured. Printing id:s of all telegrams heard!

Received telegram from: 16702184
          manufacturer: (BMT) BMETERS, Italy
           device type: Room sensor (eg temperature or humidity)
Received telegram from: 16702179
          manufacturer: (BMT) BMETERS, Italy
           device type: Room sensor (eg temperature or humidity)
Received telegram from: 00014146
          manufacturer: (LAS) Lansen Systems, Sweden
           device type: Room sensor (eg temperature or humidity)

Seems to work fine, I have not tested the auto starting features.

merbanan, 2019-10-27:
Another thing, the hardware might support C1 as that is the same frequency and rate. But still untested until I get in the range of a C1 signal. I could lend you the device but I need it to verify reception for rtl_433 for the time being.

weetmuts, 2019-11-17:
Sure! I can test the device when it is available. Lets close this issue now though.

closed 2019-11-17 by weetmuts

izll, 2022-02-14:
Hi @merbanan ,
I purchased an rfm-rx2 and a Raspberry pi 4.
I installed the wmbusmeters from snap, using the default configuration, I always only get the following error message in the log:
"No wmbus device detected, waiting for a device to be plugged in."
I tried modify the config like this, but same results: device=/dev/ttyACM0:t1
How it worked for you?
Btw, how can run the snap version in command line and not as service? 
Thanks.

weetmuts, 2022-02-14:
The rfmrx2 device was removed because I do not have such a dongle to test.  Where did you buy this one?
The device needs no configuration, just listen to the serial port with the right speed. This is what worked before:

--reopenafter=10m --listento=t1 /dev/ttyACM0:rawtty:38400

weetmuts, 2022-02-14:
@merbanan Thanks but I do not need it. If it is merely a raw telegram stream, there is no point in adding an explicit driver.

@izil Actually I do not know. :-/ Perhaps you can build directly, it is very simple to do?  I am working on a proper deb.

izll, 2022-02-14:
@weetmuts Thanks, I will try it.

izll, 2022-02-14:
@weetmuts 
This is the results:
root@raspberrypi:/usr/bin# ./wmbusmeters --reopenafter=10m --listento=t1 /dev/ttyACM0:rawtty:38400
Unknown option "--reopenafter=10m"
root@raspberrypi:/usr/bin# ./wmbusmeters --listento=t1 /dev/ttyACM0:rawtty:38400
Wmbus devices found but no meters supplied. You must supply which link modes to listen to. Eg. auto:c1
root@raspberrypi:/usr/bin# ./wmbusmeters /dev/ttyACM0:38400:t1
Started config rawtty on /dev/ttyACM0 listening on t1
No meters configured. Printing id:s of all telegrams heard!

Same result with compiled version on Ubuntu 21.10 
I think it will works, but there is no meter here currently. 
I'll test it with a meter in a few days.
Thanks for the help.