wmbusmeters/wmbusmeters #34: Support pi-temp_wl/

merbanan, 2019-10-13:
{0058}
57 44 b4 09 84 21 70 16 10 1b 7a b2 08 00 00 02 
65 a0 08 42 65 8f 08 82 01 65 9f 08 22 65 89 08 
12 65 a0 08 62 65 51 08 52 65 2b 09 02 fb 1a ba 
01 42 fb 1a b0 01 82 01 fb 1a bd 01 22 fb 1a a9 
01 12 fb 1a ba 01 62 fb 1a a6 01 52 fb 1a f5 01 
06 6d 3b 3b b3 6b 2a 00

Datagram from rfm rx2 (without crc).

merbanan, 2019-10-13:
I have the meaning of all values and will update with those later. The first of temp/hygro is the direct measurement.

merbanan, 2019-10-14:
Current temperature, avg temp last hour, avg temp last day, min temp last hour, max temp last hour, min temp last day, max temp last day

Current humidity, avg humidity last hour, avg humidity last day, min humidity last hour, max humidity last hour, min humidity last day, max humidity last day

merbanan, 2019-10-14:
This brand is Bmeters. https://www.bmeters.com/en/bm_product/rfm-amb/

weetmuts, 2019-10-14:
I pushed a version. However, this is a case where we need to think about the json key naming....
Other meters use "set date" ie storage nr >= 1, for last month change. But this
meter according to the manual uses storagenr >= to indicate previous days?

weetmuts, 2019-10-14:
The json currently looks like this. But it might change.
{"media":"room sensor","meter":"rfmamb","name":"Tempoo","id":"11772288","current_temperature_c":22.08,"maximum_temperature_c":22.08,"minimum_temperature_c":21.85,"maximum_temperature_at_set_date_1_c":23.47,"minimum_temperature_at_set_date_1_c":21.29,"current_relative_humidity_rh":44.2,"minimum_relative_humidity_rh":42.5,"maximum_relative_humidity_rh":44.2,"maximum_relative_humidity_at_set_date_1_rh":50.1,"minimum_relative_humidity_at_set_date_1_rh":42.2,"temperature_at_set_date_1_c":21.91,"relative_humidity_at_set_date_1_rh":43.2,"temperature_at_set_date_2_c":22.07,"relative_humidity_at_set_date_2_rh":44.5,"device_date_time":"2019-10-11 19:59","timestamp":"2019-10-14T18:42:29Z"}

merbanan, 2019-10-14:
My understanding of mbus and oms terminology is fairly limited. Do you have any references for this type of information? I can only find reference for current value.

merbanan, 2019-10-14:
Anyway I think we can close this now.

closed 2019-10-14 by merbanan

weetmuts, 2019-10-20:
Json is updated. Average is over the last hour (_1h_) or the last day (_24h_) etc.

{"media":"room sensor","meter":"rfmamb","name":"Rummet","id":"11772288","current_temperature_c":22.08,"average_temperature_1h_c":21.91,"average_temperature_24h_c":22.07,"maximum_temperature_1h_c":22.08,"minimum_temperature_1h_c":21.85,"maximum_temperature_24h_c":23.47,"minimum_temperature_24h_c":21.29,"current_relative_humidity_rh":44.2,"average_relative_humidity_1h_rh":43.2,"average_relative_humidity_24h_rh":44.5,"minimum_relative_humidity_1h_rh":42.2,"maximum_relative_humidity_1h_rh":50.1,"maximum_relative_humidity_24h_rh":0,"minimum_relative_humidity_24h_rh":0,"device_date_time":"2019-10-11 19:59","timestamp":"1111-11-11T11:11:11Z"}