wmbusmeters/wmbusmeters #35: Support Lansen thermo/hygro-meter

merbanan, 2019-10-13:
On the air payload (without crc)

{002f}
2e 44 33 30 46 41 01 00 07 1b 7a 63 48 20 25 af 
3b 6f be fc 5f 40 b3 b1 d0 96 0d 52 1d c8 cd 53 
7a 05 8e 5a 91 6d 48 f8 00 2e 78 a4 d0 d3 63 

Encrypted part
af 3b 6f be fc 5f 40 b3 b1 d0 96 0d 52 1d c8 cd 
53 7a 05 8e 5a 91 6d 48 f8 00 2e 78 a4 d0 d3 63 
print_dec

Decrypted part
2f 2f 02 65 84 08 42 65 83 08 82 01 65 95 08 02 
fb 1a ae 01 42 fb 1a ae 01 82 01 fb 1a a9 01 2f

merbanan, 2019-10-13:
Should have 3 temp fields and 3 hygro fields. The first of both are direct measurements. Others are currently unknown.

weetmuts, 2019-10-14:
Added. This is the json output 
{"media":"room sensor","meter":"lansenth","name":"Tempoo","id":"00010203","current_temperature_c":21.8,"current_relative_humidity_rh":43,"temperature_at_set_date_1_c":21.79,"relative_humidity_at_set_date_1_rh":43,"temperature_at_set_date_2_c":21.97,"relative_humidity_at_set_date_2_rh":42.5,"timestamp":"2019-10-14T14:54:42Z"}

merbanan, 2019-10-14:
Looks good, closing issue.

closed 2019-10-14 by merbanan

merbanan, 2019-10-17:
http://media1.lansensystems.com/datasheets/LAN-WMBUS-E-CO2_1_0.pdf

"MBUS DATA Instant, Average hour, Average 24 hours."

So that is what the 3 values are.

weetmuts, 2019-10-20:
Thanks @merbanan  ! I haveUpdated the json to have better keys based on this info.

{"media":"room sensor","meter":"lansenth","name":"Tempoo","id":"00010203","current_temperature_c":21.8,"current_relative_humidity_rh":43,"average_temperature_1h_c":21.79,"average_relative_humidity_1h_rh":43,"average_temperature_24h_c":21.97,"average_relative_humidity_24h_rh":42.5,"timestamp":"1111-11-11T11:11:11Z"}