wmbusmeters/wmbusmeters #36: Add rtl_433 support for datagram reception

merbanan, 2019-10-13:
rtl_433 will soon support mbus reception.

rtl_433 -R 104 -s 1200k -f 868800k

time      : 2019-10-13 19:31:02
model     : Wireless M-Bus                         Manufacturer: BMT         ID        : 16702184
Version   : 16           Device Type: 0x1B         Device Type String: Room sensor                     Control   : 0x44
Data Length: 78          Data      : 7ab00800000265130942652809820165e50822651b0912652e09626586085265680902fb1a820142fb1a8f018201fb1ab10122fb1a7a0112fb1a9b0162fb1a8b0152fb1af101066d3b31c46c2a00
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-10-13 19:31:17
model     : Wireless M-Bus                         Manufacturer: LAS         ID        : 14146
Version   : 7            Device Type: 0x1B         Device Type String: Room sensor                     Control   : 0x44
Data Length: 37          Data      : 7ac44820250893278890146570a085119a67f75e2880c1544edbfe8589c8b6df799e1aa102
Integrity : CRC

Would be nice with an alternative for datagram reception.

weetmuts, 2019-10-15:
Does it handle T and C telegrams?

merbanan, 2019-10-15:
This branch:
https://github.com/merbanan/rtl_433/tree/feat-fsk_demod_new

with this command:
src/rtl_433 -R 104 -s 1000k -f 868800k

currently gave this output:

```
time      : 2019-10-15 22:41:28
model     : Wireless M-Bus                         Manufacturer: BMT         ID        : 16702184
Version   : 16           Device Type: 0x1B         Device Type String: Room sensor                     Control   : 0x44
Data Length: 100         Data      : 5744b40984217016101b1f2a7ae70800000265840842657e08820165a960ba0822657908126584086265710852650eae2d0902fb1ac80142fb1abc018201fb1a2b848e0122fb1ab70112fb1ac80162fb1a6f81940152fb1ac701066d3b3b276e2a000561
Integrity : CRC
```

There looks to be mode C support in the code but I have not seen anything being decoded properly of the air. Nor have I tried any samples in that mode. So it might work but my aim is of course to add the support. The code will most likely change little but another option for rtl-sdr devices can only be good. I have no idea if my demodulator performs better then the rtl_wmbus one.

weetmuts, 2019-10-16:
I have tried rtl_433 and it works very nice for receiving C1 telegram!

merbanan, 2019-10-28:
rtl_433 feat-fsk_demod_new_test branch now outputs the same data as rtl_wmbus for T1 mode.

merbanan, 2019-10-28:
rtl_433 is worse at receiving compared to rtl_wmbus and I'm not sure I will be able to fix that. So maybe a specific device driver is a not so good idea. rtl_433 has some strengths and that is running on less powerful hardware. So how about a generic standard input driver reading from a pipe? That way one can get command line rtl_433 support also. And the possibility to send the payload from far away to be processed locally. Parsing json and text would be nice, text should be fairly similar to a serial port. Or just parse the text and search for a starting header.

bboral, 2020-01-08:
For some reasons rtl_wmbus cannot read my wmbus meters while rtl_433 does it perfectly. 
It would be great to be able to use it as a feed for wmbusmeters.

merbanan, 2020-01-08:
@bboral I'll post a solution for you later this evening.

bboral, 2020-01-08:
This is great news! Thanks a lot @merbanan!

weetmuts, 2020-01-08:
Hi @bboral, does your meter transmit C1 telegrams?

bboral, 2020-01-08:
I'm not an expert but based on output I guess it is T type

 {“time” : “2020-01-08 11:22:12", “model” : “Wireless M-Bus”, “mode” : “T”, “M” : “SAP”, “id” : 41280725, “version” : 204, “type” : 1, “type_string” : “Oil”, “C” : 68, “data_length” : 26, “data” : “1944304c1fa7c740cc01a26111001322761d5ab7f7dfaea15d60”, “mic” : “CRC”}

weetmuts, 2020-01-08:
Yep, that is an izar water meter, transmitting on T1. Current consumption 1006.918 m3.
:-)

Strange that rtl_wmbus cannot receive it.

weetmuts, 2020-01-08:
By the way, @merbanan , the izar meter is non-compliant to the wmbus protocol, so the actual meter id in the telegram is actually hex: 40c7a71f , ie not bcd as specified by the standard.

So where does the id 41280725 reported by rtl_433 come from?

merbanan, 2020-01-08:
@weetmuts https://github.com/merbanan/rtl_433/blob/master/src/devices/m_bus.c#L153

weetmuts, 2020-01-08:
Ah, yes, the bcd2int does its thing, despite the input values not being proper bcd.

merbanan, 2020-01-08:
https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_json_to_rtlwmbus.py

This is the wrapper script I wrote.

bboral, 2020-01-08:
@weetmuts 
> Yep, that is an izar water meter, transmitting on T1. Current consumption 1006.918 m3.
Interesting, looks like not my meter :) i had to catch something from neighbourhood.

@merbanan I'll check that out this evening and get back :)

merbanan, 2020-01-08:
@bboral can you send me a sample signal of your device? I want to see if I can get the proper id for it.

rtl_433 -S known -s 1600k

weetmuts, 2020-01-08:
Other Izar users have said that there is so far, no known way of correlating the printed numbers on the meter with the wmbus telegram id.... also the value sent is not necessarily the same as the mechanical meters. So it might just be your meter anyway. :-)

You could listen to the wmbus telegrams and see if the consumption correlates with you, not
using water, and then using water.

bboral, 2020-01-08:
It works! :)
@weetmuts I found both my meters and the data +/- corresponds to physical numbers however 
i noticed that data are sent from two different sources or they are recognised like that my wmbusmeters:
{"media":"water","meter":"izar","name":"TestMeter","id":"2124244d","total_m3":42.392,"last_month_total_m3":42.392,"timestamp":"2020-01-08T20:18:57Z"}
{"media":"radio converter (meter side)","meter":"izar","name":"TestMeter","id":"00072642","total_m3":42.392,"last_month_total_m3":42.392,"timestamp":"2020-01-08T20:18:58Z"}
**{"**

@merbanan I'm not sure this is what you asked for but this is output from the command:
Karinas-MacBook-Air:build karina$ rtl_433 -S known -s 1600k
rtl_433 version 19.08-114-g70d2e52 branch master at 201912301710 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/karina/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

	Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
	A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 112 out of 141 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-104 108-116 119 121 124-128 131-141 ]
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sample rate set to 1600000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-01-08 21:26:58
model     : Nexus Temperature/Humidity             House Code: 145
Channel   : 1            Battery   : LOW           Temperature: 21.30 C      Humidity  : 33 %
*** Saving signal to file g001_433.92M_1600k.cu8 (1416368 samples, 2883584 bytes)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-01-08 21:27:05
model     : Nexus Temperature                      House Code: 198
Channel   : 1            Battery   : OK            Temperature: 20.90 C
*** Saving signal to file g002_433.92M_1600k.cu8 (1423750 samples, 2883584 bytes)
^Z
[1]+  Stopped                 rtl_433 -S known -s 1600k

weetmuts, 2020-01-09:
Great! Though it is quite odd that you receive the same value from two different ids.
If you can run for a while with --logtelegrams, and send me the output, then I can take look at them.

merbanan, 2020-01-09:
@bboral rtl_433 -S known -s 1600k -f 868.8M

I missed to add the frequency. After that you will get some sample files. I'd like to get one of those.

@weetmuts the idea is that if there is a bcd violation then report the hex value as id, the logic is not 100% but at least we get a proper unique id

weetmuts, 2020-01-09:
@merbanan In wmbusmeters I always convert the ids to hex-strings and print them. If it is bcd, then it looks like a number, otherwise, we see the hex string. :-) No need to translate to an actual integer value.

merbanan, 2020-01-09:
@weetmuts I have to look at the code then. I did not write it so I just accept it but now that I think of it I just need to byte swap the hex values and print it as such.

bboral, 2020-01-09:
@merbanan I hope this is what is needed :)
[g003_868.8M_1600k.cu8.txt](https://github.com/weetmuts/wmbusmeters/files/4042141/g003_868.8M_1600k.cu8.txt)
[g003_868.8M_1600k.cu8.txt](https://github.com/weetmuts/wmbusmeters/files/4042142/g003_868.8M_1600k.cu8.txt)

@weetmuts I captured 3-4 requests I'm not sure the last one was processed and written to meter readings file.
The text file includes both console and meter output
[wmbusmaster_telegrams.txt](https://github.com/weetmuts/wmbusmeters/files/4042011/wmbusmaster_telegrams.txt)

There is one more strange thing I noticed. I cannot read my meters when I'm close to them but I can read them being on the opposite side of my house. I wonder whether I actually read signal from my meters or maybe from some repeater/amplifier. There is a water pomp station nearby with some radio antenna so maybe meters are read by this antenna and then signal is propagated with more power?

weetmuts, 2020-02-02:
Looks like your telegrams says:
Foo	40c7a71f	1007.432 m3	1006.499 m3	2020-02-02 21:01.08
Foo	40c7a71f	1007.432 m3	1006.499 m3	2020-02-02 21:01.08
Foo	40c7a71f	1007.432 m3	1006.499 m3	2020-02-02 21:01.08
When decoded. Looks good.
The last telegram fails to be decoded because it is not an izar water meter, but something else,
that is encrypted using the standard AES_CBC_IV mode.

Received telegram from: 00072642
manufacturer: (DEV) Develco Products, Denmark
device type: Radio converter (meter side)

Are you using a wildcard '*' to listen to all meters?

weetmuts, 2020-08-12:
Pushed direct support for rtl_433 to wmbusmeters. I.e. you can write rtl433 instead of rtlwmbus. E.g.
wmbusmeters --listento=any rtl433

merbanan, 2020-08-19:
Can this be closed now?

weetmuts, 2020-08-19:
Yep! Support added!

closed 2020-08-19 by weetmuts