wmbusmeters/wmbusmeters #37: Bmeters water meter payload

merbanan, 2019-10-14:
Here is a payload from a water meter

4e44b409XXXXXXXX1307XXXX [7a 48 00 40 05] 948a8c3f164c34465990fdeacdf05e9dc625dbf27163eaf8f9172a74c1880ae5f6315baa3c45c5bc4d8ebd3659d0a473b27c3fae5cb1b7f7cbda557bb7865beaa28885b80251ec3ae776

I blanked out the id and crc. The CW indicates an aes encrypted payload. Do you have a set of "popular" key to try on this meter?

merbanan, 2019-10-14:
0x01-0x0F and all 0xFF should be tried

merbanan, 2019-10-14:
The key is supposed to be all 00.

weetmuts, 2019-11-17:
Can you repost a telegram in the rtl_wmbus format, ie the crc checks are removed?

merbanan, 2019-12-06:
cat g001_868.9M_1600k.cu8 | rtl_wmbus
T1;1;1;2019-12-06 18:44:10.000;23;24;18160686;0x4e44b4098606161813077af000400564157017e38ee57f9b990460cc8244939534d3fa78a08153c58554c8b26f78c995e1e39ad892ede6150123f61a84db7da277f1c0489212e3c26079e16ce024e8

merbanan, 2019-12-06:
https://github.com/merbanan/rtl_433_tests/tree/master/tests/m-bus/03

sample files

weetmuts, 2019-12-07:
I am guessing that this is a  Hydrodigit water meter?

Added the hydrodigit meter, please test. Do you think the date in the telegram is the meters own clock when it sent the telegram?

There is unfortunately a majority of data after those two standard compliant two values, that are encoded as vendor proprietary. There might be more data in there, like warning flags etc .
But without additional reverse engineering they will be unknown for the time being.

Please test!

merbanan, 2019-12-07:
I read some document about programming Bmeters devices and it contained setting the time. I got the samples from a rtl_433 user about 4 weeks ago. So I don't know much more. I guess the ticket can be closed.

weetmuts, 2019-12-28:
Meter added. Closing ticket. Thanks for the telegram!

closed 2019-12-28 by weetmuts