wmbusmeters/wmbusmeters #39: Techem sensor payload

merbanan, 2019-10-14:
Should be heat meters.

4 different device types:

```
model     : Wireless M-Bus                         Manufacturer: TCH         ID        : 21500591
Version   : 113          Device Type: 0x61         Device Type String:       Control   : 0x44          Data Length: 63
Data      : 364468509105502171614b75a000de2647b201d0ec0b0000b71680b2ae9569159978d3a9582577ca65aa67534f09f03508db54763d251666a9e160196ed7f0
Integrity : CRC
```

```
time      : @0.022837s
model     : Wireless M-Bus                         Manufacturer: TCH         ID        : 30586050
Version   : 34           Device Type: 0x43         Device Type String:       Control   : 0x44          Data Length: 63
Data      : 36446850506058302243f477a100de2616d600d0d302000117481611603da8108832c72c64544d52fd262070a465b213d559524b3583c9142938f040079406
Integrity : CRC
```

```
time      : @0.041196s
model     : Wireless M-Bus                         Manufacturer: TCH         ID        : 30717777
Version   : 105          Device Type: 0x80         Device Type String:       Control   : 0x44          Data Length: 59
Data      : 3244685077777130698091f6a011de264401e0340600390838080600fc9b000000051009120d0a1123282718161dea460f120a040000000000ada3
Integrity : CRC

```

```
model     : Wireless M-Bus                         Manufacturer: TCH         ID        : 25071981
Version   : 116          Device Type: 0x62         Device Type String:       Control   : 0x44          Data Length: 56
Data      : 2f44685081190725746203a9a206de266813e0344d000000041517125af6130c0a2924291e0f23202b262524270dfd2a231c2018061aa2d6
Integrity : CRC
```

merbanan, 2019-10-14:
Might already be supported.

weetmuts, 2019-11-17:
Can you repost the data in the rtl_wmbus format, ie crc:s removed.

merbanan, 2019-12-06:
Samples are at:

https://github.com/merbanan/rtl_433_tests/tree/mbus-techem/tests/m-bus/04

Then running this command:

rtl_433 -F json -Y 1 -s 1000k -r rtl_433_tests/tests/m-bus/04/g*_868.9M_1000k.cu8 | python ../wmbus/rtl_433_json_to_rtlwmbus.py

gives the following output, afaik the result is rtl_wmbus compliant.

```
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g001_868.9M_1000k.cu8
T1;1;1;@0.041195s.000;54;46;30717777;0x32446850777771306980a011de264401e03406003b0839080600000000051009120d0a1123282718161d0f120a040000000000
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g002_868.9M_1000k.cu8
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g003_868.9M_1000k.cu8
T1;1;1;@0.041195s.000;54;46;30718698;0x32446850988671306980a011de264e02e0340c00c008bb080a010000010e201724226021324448393317000000000000000000
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g004_868.9M_1000k.cu8
T1;1;1;@0.041194s.000;54;46;30717768;0x32446850687771306980a011de266807e034a000fa096b0b4f2b0006295c775d6d6d755c9b8f747d7c9071542f1e1701060106
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g005_868.9M_1000k.cu8
T1;1;1;@0.041193s.000;54;46;30714673;0x32446850734671306980a011de263105e03414001d09320914000000041a4c2e6465461d76cdb6736067300109000000000000
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g006_868.9M_1000k.cu8
T1;1;1;@0.041187s.000;54;46;30714676;0x32446850764671306980a011de263600e0340200ec08d808000100000000000101020002060804000001000000021109010000
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g007_868.9M_1000k.cu8
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g008_868.9M_1000k.cu8
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g009_868.9M_1000k.cu8
T1;1;1;@0.041488s.000;54;46;25153687;0x2f446850873615257462a206de26c502e034100000000401020404040305080404030403050304050404030402030401
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g010_868.9M_1000k.cu8
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g011_868.9M_1000k.cu8
T1;1;1;@0.022815s.000;54;46;30585388;0x36446850885358302243a100de26157600d036000000670300039c40ca40ad10c40d4c384d67e2703cad25183faccc008205101c604002
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g012_868.9M_1000k.cu8
T1;1;1;@0.041507s.000;54;46;25063431;0x2f446850313406257462a206de264c07e034310000000805080a09090a06090b090c0b0b0f0b0b0a0c070a090a08080a
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g013_868.9M_1000k.cu8
T1;1;1;@0.041197s.000;54;46;30717780;0x32446850807771306980a011de26e000e03401004208580801000000000205030806050713171d211b1b080f06010000000000
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g014_868.9M_1000k.cu8
T1;1;1;@0.041196s.000;54;46;30714670;0x32446850704671306980a011de26cf06e0342b00e70a0c0e2b00000000275a396c7b903983cfb9bf78773a472b000000000000
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g015_868.9M_1000k.cu8
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g016_868.9M_1000k.cu8
T1;1;1;@0.022848s.000;54;46;30586121;0x36446850216158302243a100de26994100d0000000000700000014d044285ebcd24d349d84459c6cbeadd38e082c000000000000208000
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g017_868.9M_1000k.cu8
T1;1;1;@0.041195s.000;54;46;30714669;0x32446850694671306980a011de262e06e0341f00460972091f0000000525603a73764d298ce6d28878803a010c000000000000
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g018_868.9M_1000k.cu8
T1;1;1;@0.041499s.000;54;46;25071898;0x2f446850981807257462a206de265b05e0341d0000000204040706050404050504050502030902030607060709040905
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g019_868.9M_1000k.cu8
T1;1;1;@0.041499s.000;54;46;25153680;0x2f446850803615257462a206de26d009e0347100000014141616161b101614151717190f0b1e151415110c1006040215
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g020_868.9M_1000k.cu8
T1;1;1;@0.041195s.000;54;46;30718702;0x32446850028771306980a011de260200e0340000e508d008000000000000000000000000000000000000000002000000000000
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g021_868.9M_1000k.cu8
T1;1;1;@0.041192s.000;54;46;30714674;0x32446850744671306980a011de261a00e0340000f208de08000000000000010001010000030001010000000000010b06000000
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g022_868.9M_1000k.cu8
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g023_868.9M_1000k.cu8
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g024_868.9M_1000k.cu8
T1;1;1;@0.040800s.000;54;46;30586125;0x36446850256158302243a100de26596801d051050001978c204010f2d06538d966e07492798a71ae9aaa88e75975495347112744d00006
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g025_868.9M_1000k.cu8
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g026_868.9M_1000k.cu8
T1;1;1;@0.040799s.000;54;46;30586062;0x36446850626058302243a100de266f0900d000000000070000000000000000000000000008700000020000000000000000000000000000
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g027_868.9M_1000k.cu8
[... cut ...]
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g032_868.9M_1000k.cu8
T1;1;1;@0.022837s.000;54;46;30586050;0x36446850506058302243a100de2616d600d0d302000117481611a8108832c72c64544d52fd262070a465d559524b3583c9142938f04007
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g033_868.9M_1000k.cu8
Test mode active. Reading samples from file: rtl_433_tests/tests/m-bus/04/g034_868.9M_1000k.cu8
T1;1;1;@0.040684s.000;54;46;21500591;0x36446850910550217161a000de2647b201d0ec0b0000b71680b269159978d3a9582577ca65aa67534f0908db54763d251666a9e160196e

```

merbanan, 2019-12-06:
The data should not be encrypted but I don't know for sure.

weetmuts, 2019-12-07:
Thanks, I tried with the vario451 meter and perhaps it displays the correct values.
Can you check?

The protocol from these meters are unfortunately completely non-standard...
naughty proprietary stuff wrapped inside a wmbus telegram.

merbanan, 2019-12-07:
I also got these samples from a rtl_433 user. I have no way testing this. Issue can be closed.

weetmuts, 2019-12-28:
Ok, closing for now. We cannot debug until we know the expected output.

closed 2019-12-28 by weetmuts

antonmeyer, 2021-03-27:
msgdec: 6850;25153687;62;30.06.20;15.03.21;805;105
msgdec: 6850;30717776;80;30.06.20;15.03.21;18.91;18.97;3;0
msgdec: 6850;25153718;62;30.06.20;15.03.21;1230;109
msgdec: 6850;30717764;80;30.06.20;15.03.21;21.43;25.98;1;798
msgdec: 6850;30717768;80;30.06.20;15.03.21;31.27;47.52;1854;1558
msgdec: 6850;21500591;61;30.06.20;00.12.20;54569;3583
msgdec: 6850;30716753;80;30.06.20;15.03.21;24.72;38.65;2185;1249
msgdec: 6850;12366779;80;30.06.20;15.03.21;27.12;42.46;646;708
msgdec: 6850;30714670;80;30.06.20;15.03.21;29.94;50.78;950;1183
msgdec: 6850;30714495;80;30.06.20;15.03.21;19.64;19.35;7;45
msgdec: 6850;30714496;80;30.06.20;15.03.21;27.40;37.80;1141;1098
msgdec: 6850;30717769;80;30.06.20;15.03.21;29.42;45.59;2823;2023
msgdec: 6850;30717767;80;30.06.20;15.03.21;30.06;47.02;1521;1322
msgdec: 6850;30717752;80;30.06.20;15.03.21;26.21;36.05;64;1129
msgdec: 6850;30708724;80;30.06.20;15.03.21;21.36;23.14;443;680
msgdec: 6850;30717804;80;30.06.20;15.03.21;16.87;16.65;389;353
msgdec: 6850;30718698;80;30.06.20;15.03.21;19.74;19.28;204;377