wmbusmeters/wmbusmeters #40: rtl_sdr + apator (AT-WMBUS-16-2a)

lsw271, 2019-10-25:
Hello,
I use rtl_sdr usb dongle to listen to apator at-wmbus-16-2**a**. I've got 2a version which have got another battery but i'm not sure about that its the same protocol as version 2, i hope so :) . 
I add device as :
name=MyTapWater
type=apator162
id=01478662
key=00000000000000000000000000000000

about id i'm not sure its my water meter number becouse i've got few numbers on my water meter but i found this number in logs while i'm listening wmbus, i've also got the second water meter which i can find in logs so i think its correct

while im trying 
`wmbusmeters --logtelegrams --debug --verbose --format=json rtlwmbus MyTapWater apator162:t1 01478662 00000000000000000000000000000000`

i've got:
`(wmbusmeters) version: 0.9.13
(config) using device: rtlwmbus
(config) number of meters: 1
(rtlwmbus) using command: rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus"
(serialcmd) opened /bin/sh
(config) meter apator162 link mode(s): t1
(config) all possible link modes that the meters might transmit on: t1
(config) listen link modes calculated to be: t1
(config) listen to link modes: any
(wmbusmeters) configured "MyTapWater" "apator162" "01478662" encrypted
(serialcmd) received ": 1: rtl_wmbus: Permission denied<0A>"
(serialcmd) received "Found 1 device(s):<0A>"
(serialcmd) received "  0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>"
(serialcmd) received "Using device 0: Generic RTL2832U OEM<0A>"
(serialcmd) received "usb_claim_interface error -6<0A>"
(serialcmd) received "usb_claim_interface error -6<0A>Fail<00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00>"
(serialcmd) received "ed to open rtlsdr device #0.<0A>"
(bgshell) 9985 exited with return code 126
(serialcmd) closed /bin/sh
(rtlwmbus) child process exited! Command was: "rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus"
(serialtty) event loop stopped!
(serialcmd) closed /bin/sh`

it's mean that i've got wrong key?

i also trying 
`wmbusmeters --verbose --useconfig=/ --meterfiles=/ rtlwmbus:868.95M
Not a valid meter type ""
Not a valid meter id ""`

example of listening wmbus:

T1;1;1;2019-10-25 20:10:08.000;127;119;01478662;0x4e4401066286470105077ad100408545c894edb36e83870a0cac5d3ed785600486c8b2c65e1b98b56b1d1f206a411a947a67ea342f0985603533582
72ff8abdfaf0dacef50d145991db9c873f4147d
T1;1;1;2019-10-25 
20:14:20.000;128;120;01478662;0x4e4401066286470105077ad2004085a6e48033247e61e39e8481ec330476a05657f3e4fbe1c4d1a250ed902fb451c0f76747d2d6d0c3a44b00d0fd8
72ecc94429f49c1d8ef5c72bf0084594213b460

@weetmuts could you say sth about that?

Regards 
Luke

weetmuts, 2019-10-26:
If you look at the log it says:

Found 1 device(s):<0A>" 
(serialcmd) received " 0: Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>" 
(serialcmd) received "Using device 0: Generic RTL2832U OEM<0A>"
(serialcmd) received "usb_claim_interface error -6<0A>" 
(serialcmd) received "usb_claim_interface error.

Check that the daemon is not running. (sudo systemctl stop wmbusmeters.service)
Check that rtl_sdr is not running, sometimes it gets stuck. (sudo killall -9 rtl_sdr)
Check the group ownershup and permissions on /dev/rtlsdr 
(ls -l /dev/rtlsdr, follow link: ls -l /dev/bus/usb/001/024 outputs:
crw-rw-rw- 1 root wmbusmeters 189, 23 okt 26 07:47 /dev/bus/usb/001/024)
Check that your own user is member of the wmbusmeters group. (groups)

Bump the source code to the latest in git. (This should not matter, but does not hurt.)

Now try the commandline again to see if it works.

If the command line works. Stop it. Check that rtl_sdr is not running. 
Now try to start the daemon. (sudo systemctl start wmbusmeters.service)

lsw271, 2019-10-28:
i've got:
pi@Domoticz:~ $ ls -l /dev/rtlsdr
lrwxrwxrwx 1 root root 15 paź 29 00:01 /dev/rtlsdr -> bus/usb/001/005

when i'm trying the commandline again:
pi@Domoticz:~ $ sudo wmbusmeters --logtelegrams --debug --verbose --format=json rtlwmbus MyTapWater apator162:t1 01478662 00000000000000000000000000000000
(wmbusmeters) version: 0.9.15-2-g97cb924
(config) using device: rtlwmbus
(config) number of meters: 1
(rtlwmbus) using command: rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus"
(serialcmd) opened /bin/sh
(config) meter apator162 link mode(s): t1
(config) all possible link modes that the meters might transmit on: t1
(config) listen link modes calculated to be: t1
(config) listen to link modes: any
(wmbusmeters) configured "MyTapWater" "apator162" "01478662" encrypted
(serialcmd) received ": 1: "
(serialcmd) received "rtl_wmbus: Permission denied"
(serialcmd) received "<0A>"
(serialcmd) received "Found 1 device(s):<0A>"
(serialcmd) received "  0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>"
(serialcmd) received "Using device 0: Generic RTL2832U OEM<0A>"
(serialcmd) received "Found Rafael Micro R820T tuner<0A>"
(serialcmd) received "[R82XX] PLL not locked!<0A>"
(serialcmd) received "Sampling at 1600000 S/s.<0A>"
(serialcmd) received "Tuned to 868950000 Hz.<0A>"
(serialcmd) received "Tuner gain set to automatic.<0A>"
(serialcmd) received "Reading samples in async mode...<0A>"
(serialcmd) received "Signal caught, exiting!<0A>Short write, samples lost, exiting!<0A>"
(serialcmd) received "<0A>User cancel, exiting...<0A>"
(bgshell) 1250 exited with return code 126
(serialcmd) closed /bin/sh
(rtlwmbus) child process exited! Command was: "rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus"
(serialtty) event loop stopped!
(serialcmd) closed /bin/sh

pi@Domoticz:~ $ wmbusmeters --verbose --useconfig=/ --meterfiles=/ rtlwmbus:868.9M
Not a valid meter type ""
Not a valid meter id nor a valid meter match expression ""

maybe it would be the problem with libusb1.0? i had some problems with instaling this. I think that rtl_sdr running ok, rtl_test - t results:
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
[R82XX] PLL not locked!
Sampling at 2048000 S/s.
No E4000 tuner found, aborting.

All of it I had install it with this commands:
git clone git://git.osmocom.org/rtl-sdr.git
git clone https://github.com/xaelsouth/rtl-wmbus.git
git clone https://github.com/weetmuts/wmbusmeters/.git

apt install libusb-1.0-0-dev libusb-1.0-0 dh-autoreconf

_git://git.osmocom.org/rtl-sdr.git_
cd rtl-sdr/
autoreconf -i
./configure
make
sudo make install
sudo ldconfig
ln -s /usr/local/bin/rtl_sdr /usr/bin/rtl_sdr

_https://github.com/xaelsouth/rtl-wmbus.git_
cd ../rtl-wmbus/
make
cp build/rtl_wmbus /usr/bin/

unfortunately i'm the person who need to be lead by the hand in this case :(

weetmuts, 2019-11-17:
Run "ls -l /dev/rtlsdr"
and then run "ls -l  /dev/bus/usb/001/005"
or whatever the symlink points to.

Run "groups"

and post the output to these commands. Thanks!

weetmuts, 2019-11-17:
If "wmbusmeters" is not listed when you run "groups"
then you need to add yourself to the wmbusmeters group.
Like this, assuming your login/username is lsw271.

sudo usermod -a -G wmbusmeters lsw271

then logout and login again.

weetmuts, 2019-12-18:
Did it work? Can we close this issue?

weetmuts, 2019-12-28:
Assuming problem solved. Closing.

lsw271, 2020-01-24:
Forgive me for not answering, i've got a new family member... :). 
Unfortunately, there are still problems, but I feel there is progress. I changed system (RPI > domoticz to RPI > docker > HASSio). And i'm trying to install your program on raspian buster. 

i did what you asked:

pi@raspberrypi:~ $ ls -l /dev/rtlsdr
lrwxrwxrwx 1 root root 15 Jan 24 00:06 /dev/rtlsdr -> bus/usb/001/014
pi@raspberrypi:~ $ ls -l /dev/bus/usb/001/014
crw-rw-rw- 1 root wmbusmeters 189, 13 Jan 24 00:06 /dev/bus/usb/001/014
pi@raspberrypi:~ $ groups
pi adm dialout cdrom sudo audio video plugdev games users input netdev wmbusmeters docker gpio i2c spi
pi@raspberrypi:~ $ sudo usermod -a -G wmbusmeters pi

logs:
pi@raspberrypi:~ $ tail -f /var/log/syslog
Jan 24 00:06:13 raspberrypi kernel: [ 2930.680505] usb 1-1.3: new high-speed USB device number 14 using dwc_otg
Jan 24 00:06:13 raspberrypi kernel: [ 2930.822220] usb 1-1.3: New USB device found, idVendor=0bda, idProduct=2838, bcdDevice= 1.00
Jan 24 00:06:13 raspberrypi kernel: [ 2930.822229] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan 24 00:06:13 raspberrypi kernel: [ 2930.822234] usb 1-1.3: Product: RTL2838UHIDIR
Jan 24 00:06:13 raspberrypi kernel: [ 2930.822238] usb 1-1.3: Manufacturer: Realtek
Jan 24 00:06:13 raspberrypi kernel: [ 2930.822242] usb 1-1.3: SerialNumber: 00000001
Jan 24 00:06:13 raspberrypi mtp-probe: checking bus 1, device 14: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3"
Jan 24 00:06:13 raspberrypi mtp-probe: bus: 1, device: 14 was not an MTP device
Jan 24 00:06:13 raspberrypi mtp-probe: checking bus 1, device 14: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3"
Jan 24 00:06:13 raspberrypi mtp-probe: bus: 1, device: 14 was not an MTP device

another:
pi@raspberrypi:~ $ tail -f /var/log/wmbusmeters/wmbusmeters.log
No wmbus device found! Exiting!
(wmbusmeters) logging started 2020-01-23 23:48:04
(wmbusmeters) version: 0.9.22-2-g468ba75
(config) store meter files in: "/var/log/wmbusmeters/meter_readings"
(config) using device: /dev/1-1.3
(config) number of meters: 2
(detect) "/dev/1-1.3" ""
(detect) is_tty=0 is_stdin=0 is_file=0
(detect) not a valid device file /dev/1-1.3
No wmbus device found! Exiting!

and the service:

pi@raspberrypi:~ $ sudo systemctl status  wmbusmeters@1-1.3.service
● wmbusmeters@1-1.3.service - "wmbusmeters service on 1-1.3"
   Loaded: loaded (/etc/systemd/system/wmbusmeters@.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2020-01-23 23:48:06 GMT; 28min ago
  Process: 20189 ExecStartPre=/bin/mkdir -p /var/log/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
  Process: 20196 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmbusmeters (code=exited, status=0/SUCCE
  Process: 20202 ExecStartPre=/bin/mkdir -p /var/run/wmbusmeters (code=exited, status=0/SUCCESS)
  Process: 20204 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/run/wmbusmeters (code=exited, status=0/SUCCE
  Process: 20210 ExecStart=/usr/sbin/wmbusmetersd --device=/dev/1-1.3 /var/run/wmbusmeters/wmbusmeters-1-1.3.pid (cod
 Main PID: 20211 (code=exited, status=1/FAILURE)

Jan 23 23:48:06 raspberrypi systemd[1]: wmbusmeters@1-1.3.service: Service RestartSec=1s expired, scheduling restart.
Jan 23 23:48:06 raspberrypi systemd[1]: wmbusmeters@1-1.3.service: Scheduled restart job, restart counter is at 3.
Jan 23 23:48:06 raspberrypi systemd[1]: Stopped "wmbusmeters service on 1-1.3".
Jan 23 23:48:06 raspberrypi systemd[1]: wmbusmeters@1-1.3.service: Start request repeated too quickly.
Jan 23 23:48:06 raspberrypi systemd[1]: wmbusmeters@1-1.3.service: Failed with result 'exit-code'.
Jan 23 23:48:06 raspberrypi systemd[1]: Failed to start "wmbusmeters service on 1-1.3".

when i trying to call wmbusmeters without @... 

pi@raspberrypi:~ $ sudo systemctl status  wmbusmeters.service
Unit wmbusmeters.service could not be found.

and when i try to listen the meter the next time my rtl_sdr is not detected as if i could only use it once:
pi@raspberrypi:~ $ rtl_test -t
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Using device 0: Generic RTL2832U OEM
usb_claim_interface error -6
Failed to open rtlsdr device #0.

i've added rtl_sdr device to blacklist /etc/modprobe.d becouse without that i'vee got a problem with kernel

What do you think about that @weetmuts? :) 

Regards
Luke

lsw271, 2020-01-24:
I found this https://superuser.com/questions/1206664/disable-mtp-udev-rules-for-specific-device-so-it-can-be-mount-as-a-usb-mass-stor/1206665#1206665
I will try IT tomorrow

lsw271, 2020-01-25:
@weetmuts i'dont know how to fix this :( 
i think its problem with "MTP device..." maybe of raspian buster? 

how can i remove all and reinstall from scratch?

weetmuts, 2020-01-26:
Ok, it looks like you have an rtl_sdr dongle (realtek something). This
means that wmbusmeters will invoke "rtl_sdr | rtl_wmbus" as a background process
since currently wmbusmeters cannot talk to the realtek dongle directly.

First make sure this command works:
/usr/bin/rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | /usr/bin/rtl_wmbus

(You grab rtl_wmbus from here: https://github.com/xaelsouth/rtl-wmbus)

If you get output like C1;1;1........ or T1;1;...... 
this means that you are receiving wmbus telegrams.

Then you can start wmbusmeters (stop the daemon first: sudo systemctl stop wmbusmeters.service)
like this:
wmbusmeters --listento=c1,t1 rtlwmbus
and hopefully it will start printing the types of meters you hear and their id numbers.

Also "make install" prints exactly which files it has installed, or left in place when re-installing.
Check that list, and remove those files if you want to uninstall. Sorry for the inconvenience, 
but the number of files are so small, that it is pretty easy.

weetmuts, 2020-01-26:
Oh, and congratulations to your new family member! It is a small one that needs constant attention, I presume? :-)

lsw271, 2020-01-27:
> Ok, it looks like you have an rtl_sdr dongle (realtek something). This
> means that wmbusmeters will invoke "rtl_sdr | rtl_wmbus" as a background process
> since currently wmbusmeters cannot talk to the [...]

when i'm trying
> /usr/bin/rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | /usr/bin/rtl_wmbus

nothing happened, i was waiting for maybe hour then i canceled

daemon is not working, when i'm trying start it by hand i've got the same status
pi@raspberrypi:~ $ sudo systemctl status wmbusmeters@1-1.3.service
● wmbusmeters@1-1.3.service - "wmbusmeters service on 1-1.3"
   Loaded: loaded (/etc/systemd/system/wmbusmeters@.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2020-01-26 23:58:28 GMT; 40s ago
  Process: 20839 ExecStartPre=/bin/mkdir -p /var/log/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
  Process: 20840 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
  Process: 20841 ExecStartPre=/bin/mkdir -p /var/run/wmbusmeters (code=exited, status=0/SUCCESS)
  Process: 20842 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/run/wmbusmeters (code=exited, status=0/SUCCESS)
  Process: 20843 ExecStart=/usr/sbin/wmbusmetersd --device=/dev/1-1.3 /var/run/wmbusmeters/wmbusmeters-1-1.3.pid (code=exi
 Main PID: 20844 (code=exited, status=1/FAILURE)

Jan 26 23:58:28 raspberrypi systemd[1]: wmbusmeters@1-1.3.service: Service RestartSec=1s expired, scheduling restart.
Jan 26 23:58:28 raspberrypi systemd[1]: wmbusmeters@1-1.3.service: Scheduled restart job, restart counter is at 3.
Jan 26 23:58:28 raspberrypi systemd[1]: Stopped "wmbusmeters service on 1-1.3".
Jan 26 23:58:28 raspberrypi systemd[1]: wmbusmeters@1-1.3.service: Start request repeated too quickly.
Jan 26 23:58:28 raspberrypi systemd[1]: wmbusmeters@1-1.3.service: Failed with result 'exit-code'.
Jan 26 23:58:28 raspberrypi systemd[1]: Failed to start "wmbusmeters service on 1-1.3".

i think it's a problem with rtl_sdr dongle, but few weeks ago it was ok

pi@raspberrypi:~ $ wmbusmeters --logtelegrams --debug --verbose --format=json rtlwmbus MyTapWater apator162:t1 01478662 00   000000000000000000000000000000
(wmbusmeters) version: 0.9.22-2-g468ba75
(config) using device: rtlwmbus
(config) number of meters: 1
(detect) "rtlwmbus" ""
(detect) driver: rtlwmbus
(rtlwmbus) using command: rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus"
(serialcmd) opened /bin/sh
(config) meter apator162 link mode(s): t1
(config) all possible link modes that the meters might transmit on: t1
(config) listen link modes calculated to be: t1
(wmbusmeters) configured "MyTapWater" "apator162" "01478662" encrypted
(config) listen to link modes: any
(serial) waiting for stop
(serial) received ascii Found 1 device(s):<0A>
(serial) received ascii   0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>
**(serial) received ascii Using device 0: Generic RTL2832U OEM<0A>
(serial) received ascii usb_claim_interface error -6<0A>
(serial) received ascii Failed to open rtlsdr device #0.<0A>**
(bgshell) 26322 exited with return code 2
(serial) closing non working fd=3
(serial) no devices working
(serial) stopping manager
(serialcmd) closed /bin/sh
(serial) event loop stopped!
(serial) closing 0 devices
(serial) no devices working
(serialcmd) closed /bin/sh

if its a problem with dongle i think you can't help me

lsw271, 2020-01-27:
> 
> 
> Oh, and congratulations to your new family member! It is a small one that needs constant attention, I presume? :-)

thanks :) yes, i've got demanding newborn and also 2,5years old another boy, so i have a problem with time :)

lsw271, 2020-01-28:
Maybe my information is not kept completely, but I hope something can guide you

I haven't got any files in /var/run/wmbusmeters/ its ok?

when i'm trying: pi@raspberrypi:~/rtl-wmbus $ cat samples2.bin | build/rtl_wmbus
T1;1;1;2020-01-27 22:18:28.000;104;122;71200023;0x294468502300207172f0a0009f21000000a07d2202000100000000000000000000000400030203020000
T1;1;1;2020-01-27 22:18:28.000;34;33;64700082;0x6644242382007064640e721221008492261704300050059de7c930894d0d5e5aab82e4d9eefe39d0b338f936c0db728d6786698b5c0b4b88e516611072a991fd1fbfd2a5b8bee9f9c6bf0ce56ae94e2cf7c75b250bfd8ed0f2ec0b9c5f3adc3f7cfa326857c98d

i've got new examples every second but i think its not mine

when i put the dongle into USB another logs **I get only once (first time using that command after put dongle)**:
pi@raspberrypi:~/rtl-wmbus $ wmbusmeters --logtelegrams --debug --verbose --format=json rtlwmbus MyTapWater apator162:t1 01478662 00000000000000000000000000000000
(wmbusmeters) version: 0.9.22-2-g468ba75
(config) using device: rtlwmbus
(config) number of meters: 1
(detect) "rtlwmbus" ""
(detect) driver: rtlwmbus
(rtlwmbus) using command: rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus"
(serialcmd) opened /bin/sh
(config) meter apator162 link mode(s): t1
(config) all possible link modes that the meters might transmit on: t1
(config) listen link modes calculated to be: t1
(wmbusmeters) configured "MyTapWater" "apator162" "01478662" encrypted
(config) listen to link modes: any
(serial) waiting for stop
(serial) received ascii Found 1 device(s):<0A>
(serial) received ascii   0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>
(serial) received ascii Using device 0: Generic RTL2832U OEM<0A>
(serial) received ascii Found Rafael Micro R820T tuner<0A>
(serial) received ascii [R82XX] PLL not locked!<0A>
(serial) received ascii Sampling at 1600000 S/s.<0A>
(serial) received ascii Tuned to 868950000 Hz.<0A>
(serial) received ascii Tuner gain set to automatic.<0A>
(serial) received ascii Reading samples in async mode...<0A>
(serial) received ascii Allocating 15 zero-copy buffers<0A>
(serial) received ascii Failed to allocate zero-copy buffer for transfer 8<0A>Falling back to buffers in userspace<0A>

but "zero-copy buffer" it's not sounds good, when i'm trying next time i've got logs like in previous comment
"(serial) received ascii Using device 0: Generic RTL2832U OEM<0A>
(serial) received ascii usb_claim_interface error -6<0A>
(serial) received ascii Failed to open rtlsdr device #0.<0A>"

weetmuts, 2020-01-28:
The background process with rtl_sdr can be a bit tricky when they get in a bad state:
To get a clean system do:
sudo systemctl stop wmbusmeters.service
sudo killall -9 rtl_sdr
sudo killall -9 rtl_wmbus
sudo killall -9 rwmbusmeters

Check that none of these processes are running before you try wmbusmeters again.

However it looks good that you have received the wmbus telegrams.
Now try on the same command line:

"/usr/bin/rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | /usr/bin/rtl_wmbus | wmbusmeters --t1 stdin:rtlwmbus"

To see if you get readings and perhaps you will see your meters number there.

lsw271, 2020-01-28:
I suspect that problems may also arise from the power supply. I've also got SSD drive via USB. I'll buy usb hub and then i'll connect the dongle. Currently, it happens that RPI hangs when RTL is connected.

weetmuts, 2020-02-02:
Sounds a bit like the power supply. 

Release 0.9.23 and later have better support for starting/stopping rtlsdr from a daemon.

weetmuts, 2020-02-06:
Any news? We can close this and re-open when you have the time again. :-)

lsw271, 2020-02-07:
I'll check it tonight, yesterday I received a usb hub

lsw271, 2020-02-07:
good news :) 
i've got some responses from rtl_wmbus but I need to change the frequency to 868.9 not 868.95
this one of my two water meters
```
pi@raspberrypi:~ $ rtl_sdr -f 868.9M -s 1.6e6 - | rtl_wmbus
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 1600000 S/s.
Tuned to 868900000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...
Allocating 15 zero-copy buffers
(..................)
T1;1;1;2020-02-07 22:48:34.000;77;75;01681823;0x3e4401062318680105077a570030853db580599af9cf566efc7a43cac51551ad0fc1c667f29d6bbed1c513b791091eb6845bd74437618a45fe496cb2a8d091
(..................)
```
how can I change the fq when i'm trying to listen by:
`wmbusmeters --logtelegrams --debug --verbose --format=json rtlwmbus MyTapWater apator162:t1 01478662 00000000000000000000000000000000`

maybe it was always with 868.9 fq but i don't checked this last few weeks

i removed some files and when i'm trying:
```
pi@raspberrypi:~ $ ls -l /dev/rtlsdr
ls: cannot access '/dev/rtlsdr': No such file or directory
```
how can i fix it? because i think its important  :)

when i put the dongle in logs i can see:

```
Feb  7 23:49:19 raspberrypi kernel: [867707.522846] usb 1-1.2.3: New USB device found, idVendor=0bda, idProduct=2838, bcdDevice= 1.00
Feb  7 23:49:19 raspberrypi kernel: [867707.522855] usb 1-1.2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb  7 23:49:19 raspberrypi kernel: [867707.522859] usb 1-1.2.3: Product: RTL2838UHIDIR
Feb  7 23:49:19 raspberrypi kernel: [867707.522863] usb 1-1.2.3: Manufacturer: Realtek
Feb  7 23:49:19 raspberrypi kernel: [867707.522867] usb 1-1.2.3: SerialNumber: 00000001
Feb  7 23:49:19 raspberrypi mtp-probe: checking bus 1, device 13: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2.3"
Feb  7 23:49:19 raspberrypi mtp-probe: bus: 1, device: 13 was not an MTP device
Feb  7 23:49:19 raspberrypi mtp-probe: checking bus 1, device 13: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2.3"
Feb  7 23:49:19 raspberrypi mtp-probe: bus: 1, device: 13 was not an MTP device
Feb  7 23:49:19 raspberrypi systemd[1]: Invalid unit name "wmbusmeters@/dev/rtlsdr_3.service" was escaped as "wmbusmeters@-dev-rtlsdr_3.service" (maybe you should use systemd-escape?)
Feb  7 23:49:19 raspberrypi systemd[1]: /etc/systemd/system/wmbusmeters@.service:12: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
Feb  7 23:49:19 raspberrypi systemd[1]: /etc/systemd/system/wmbusmeters@.service:22: PIDFile= references path below legacy directory /var/run/, updating /var/run/wmbusmeters/wmbusmeters--dev-rtlsdr_3.pid → /run/wmbusmeters/wmbusmeters--dev-rtlsdr_3.pid; please update the unit file accordingly.
Feb  7 23:49:19 raspberrypi systemd[1]: Invalid unit name "wmbusmeters@/dev/rtlsdr_3.service" was escaped as "wmbusmeters@-dev-rtlsdr_3.service" (maybe you should use systemd-escape?)
Feb  7 23:49:19 raspberrypi systemd[1]: Starting "wmbusmeters service on /dev/rtlsdr_3"...
Feb  7 23:49:19 raspberrypi wmbusmetersd[4290]: (wmbusmeters) started /var/run/wmbusmeters/wmbusmeters--dev-rtlsdr_3.pid
Feb  7 23:49:19 raspberrypi systemd[1]: Started "wmbusmeters service on /dev/rtlsdr_3".
Feb  7 23:49:19 raspberrypi wmbusmetersd[4291]: (config) loading meter file /etc/wmbusmeters.d/MyTapWater
Feb  7 23:49:19 raspberrypi wmbusmetersd[4291]: (config) name=MyTapWater
Feb  7 23:49:19 raspberrypi wmbusmetersd[4291]: (config) type=apator162
Feb  7 23:49:19 raspberrypi wmbusmetersd[4291]: (config) id=01478662
Feb  7 23:49:19 raspberrypi wmbusmetersd[4291]: (config) key=<notprinted>
Feb  7 23:49:19 raspberrypi wmbusmetersd[4291]: (config) loading meter file /etc/wmbusmeters.d/.MyTapWater.swp
Feb  7 23:49:19 raspberrypi wmbusmetersd[4291]: Not a valid meter type ""
Feb  7 23:49:19 raspberrypi wmbusmetersd[4291]: Not a valid meter id nor a valid meter match expression ""
Feb  7 23:49:19 raspberrypi wmbusmetersd[4291]: (config) use rtlwmbus instead of raw device /dev/rtlsdr_3
Feb  7 23:49:19 raspberrypi wmbusmetersd[4291]: (config) overriding device with "rtlwmbus"
Feb  7 23:49:19 raspberrypi wmbusmetersd[4291]: (wmbusmeters) using log file /var/log/wmbusmeters/wmbusmeters.log
Feb  7 23:49:21 raspberrypi systemd[1]: wmbusmeters@-dev-rtlsdr_3.service: Succeeded.
Feb  7 23:49:22 raspberrypi systemd[1]: wmbusmeters@-dev-rtlsdr_3.service: Service RestartSec=1s expired, scheduling restart.
Feb  7 23:49:22 raspberrypi systemd[1]: wmbusmeters@-dev-rtlsdr_3.service: Scheduled restart job, restart counter is at 1.
Feb  7 23:49:22 raspberrypi systemd[1]: Stopped "wmbusmeters service on /dev/rtlsdr_3".
Feb  7 23:49:22 raspberrypi systemd[1]: Starting "wmbusmeters service on /dev/rtlsdr_3"...
Feb  7 23:49:23 raspberrypi wmbusmetersd[4452]: (wmbusmeters) started /var/run/wmbusmeters/wmbusmeters--dev-rtlsdr_3.pid
Feb  7 23:49:23 raspberrypi wmbusmetersd[4454]: (config) loading meter file /etc/wmbusmeters.d/MyTapWater
Feb  7 23:49:23 raspberrypi systemd[1]: Started "wmbusmeters service on /dev/rtlsdr_3".
[... 12 lines cut ...]
Feb  7 23:49:26 raspberrypi systemd[1]: wmbusmeters@-dev-rtlsdr_3.service: Scheduled restart job, restart counter is at 2.
Feb  7 23:49:26 raspberrypi systemd[1]: Stopped "wmbusmeters service on /dev/rtlsdr_3".
Feb  7 23:49:26 raspberrypi systemd[1]: Starting "wmbusmeters service on /dev/rtlsdr_3"...
[... cut ...]
Feb  7 23:49:29 raspberrypi systemd[1]: wmbusmeters@-dev-rtlsdr_3.service: Scheduled restart job, restart counter is at 3.
Feb  7 23:49:29 raspberrypi systemd[1]: Stopped "wmbusmeters service on /dev/rtlsdr_3".
Feb  7 23:49:29 raspberrypi systemd[1]: wmbusmeters@-dev-rtlsdr_3.service: Start request repeated too quickly.
Feb  7 23:49:29 raspberrypi systemd[1]: wmbusmeters@-dev-rtlsdr_3.service: Failed with result 'start-limit-hit'.
Feb  7 23:49:29 raspberrypi systemd[1]: Failed to start "wmbusmeters service on /dev/rtlsdr_3".

```

lsw271, 2020-02-08:
when i'm trying by 868.9M fq i've got some errors with lib?

```
pi@raspberrypi:~ $ wmbusmeters --logtelegrams --debug --verbose --format=json rtlwmbus:868.9M MyTapWater apator162:t1 01478662 00000000000000000000000000000000
(wmbusmeters) version: 0.9.25-1-gaf98f36
(config) using device: rtlwmbus
(config) with: 868.9M
(config) number of meters: 1
(detect) "rtlwmbus" "868.9M"
(detect) driver: rtlwmbus
(rtlwmbus) using command: rtl_sdr -f 868.9M -s 1.6e6 - | rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 868.9M -s 1.6e6 - | rtl_wmbus"
(serialcmd) opened /bin/sh
(config) meter apator162 link mode(s): t1
(config) all possible link modes that the meters might transmit on: t1
(config) listen link modes calculated to be: t1
(wmbusmeters) configured "MyTapWater" "apator162" "01478662" encrypted
(config) listen to link modes: any
(serial) waiting for stop
(serial) received ascii Found 1 device(s):<0A>
(rtlwmbus) checkRTLWMBusFrame "Found 1 device(s):<0A>"
(rtlwmbus) only text
(serial) received ascii   0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>
(rtlwmbus) checkRTLWMBusFrame "  0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>"
(rtlwmbus) only text
(serial) received ascii Using device 0: Generic RTL2832U OEM<0A>
(rtlwmbus) checkRTLWMBusFrame "Using device 0: Generic RTL2832U OEM<0A>"
(rtlwmbus) only text
(serial) received ascii Found Rafael Micro R820T tuner<0A>
(rtlwmbus) checkRTLWMBusFrame "Found Rafael Micro R820T tuner<0A>"
(rtlwmbus) only text
(serial) received ascii [R82XX] PLL not locked!<0A>
(rtlwmbus) checkRTLWMBusFrame "[R82XX] PLL not locked!<0A>"
(rtlwmbus) only text
(serial) received ascii Sampling at 1600000 S/s.<0A>
(rtlwmbus) checkRTLWMBusFrame "Sampling at 1600000 S/s.<0A>"
(rtlwmbus) only text
(serial) received ascii Tuned to 868900000 Hz.<0A>
(rtlwmbus) checkRTLWMBusFrame "Tuned to 868900000 Hz.<0A>"
(rtlwmbus) only text
(serial) received ascii Tuner gain set to automatic.<0A>
(rtlwmbus) checkRTLWMBusFrame "Tuner gain set to automatic.<0A>"
(rtlwmbus) only text
(serial) received ascii Reading samples in async mode...<0A>
(rtlwmbus) checkRTLWMBusFrame "Reading samples in async mode...<0A>"
(rtlwmbus) only text
(serial) received ascii Allocating 15 zero-copy buffers<0A>
(rtlwmbus) checkRTLWMBusFrame "Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(serial) received ascii Failed to allocate zero-copy buffer for transfer 5<0A>Falling back to buffers in userspace<0A>
(rtlwmbus) checkRTLWMBusFrame "Failed to allocate zero-copy buffer for transfer 5<0A>Falling back to buffers in userspace<0A>"
(rtlwmbus) only text
(serial) received ascii Failed to submit transfer 6<0A>Please increase your allowed usbfs buffer size with the following command:<0A>echo 0 <3E> /sys/module/usbcore/parameters/usbfs_memory_mb<0A>
(rtlwmbus) checkRTLWMBusFrame "Failed to submit transfer 6<0A>Please increase your allowed usbfs buffer size with the following command:<0A>echo 0 <3E> /sys/module/usbcore/parameters/usbfs_memory_mb<0A>"
(rtlwmbus) only text
(serial) received ascii <0A>Library error -5, exiting...<0A>
(rtlwmbus) checkRTLWMBusFrame "<0A>Library error -5, exiting...<0A>"
(rtlwmbus) only text
(bgshell) 1167 exited with return code 2
(serial) closing non working fd=3
(serial) no devices working
(serial) stopping manager
(serialcmd) closed /bin/sh
(serial) event loop stopped!
(serial) closing 0 devices
(serial) no devices working
(serialcmd) closed /bin/sh
```

lsw271, 2020-02-08:
I checked my another meter by
` wmbusmeters --logtelegrams --debug --verbose --format=json rtlwmbus:868.9M garden_meter apator162:t1,c1 01681823 00000000000000000000000000000000
`
and get:
` wmbusmeters --logtelegrams --debug --verbose --format=json rtlwmbus:868.9M garden_meter apator162:t1,c1 01681823 00000000000000000000000000000000
(wmbusmeters) version: 0.9.25-1-gaf98f36
(config) using device: rtlwmbus
(config) with: 868.9M
(config) number of meters: 1
(detect) "rtlwmbus" "868.9M"
(detect) driver: rtlwmbus
(rtlwmbus) using command: rtl_sdr -f 868.9M -s 1.6e6 - | rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 868.9M -s 1.6e6 - | rtl_wmbus"
(serialcmd) opened /bin/sh
(config) meter apator162 link mode(s): c1,t1
(config) all possible link modes that the meters might transmit on: c1,t1
(config) listen link modes calculated to be: c1,t1
(wmbusmeters) configured "garden_meter" "apator162" "01681823" encrypted
(config) listen to link modes: any
(serial) waiting for stop
(serial) received ascii Found 1 device(s):<0A>
(rtlwmbus) checkRTLWMBusFrame "Found 1 device(s):<0A>"
(rtlwmbus) only text
(serial) received ascii   0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>
(rtlwmbus) checkRTLWMBusFrame "  0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>"
(rtlwmbus) only text
(serial) received ascii Using device 0: Generic RTL2832U OEM<0A>
(rtlwmbus) checkRTLWMBusFrame "Using device 0: Generic RTL2832U OEM<0A>"
(rtlwmbus) only text
(serial) received ascii Found Rafael Micro R820T tuner<0A>
(rtlwmbus) checkRTLWMBusFrame "Found Rafael Micro R820T tuner<0A>"
(rtlwmbus) only text
(serial) received ascii [R82XX] PLL not locked!<0A>
(rtlwmbus) checkRTLWMBusFrame "[R82XX] PLL not locked!<0A>"
(rtlwmbus) only text
(serial) received ascii Sampling at 1600000 S/s.<0A>
(rtlwmbus) checkRTLWMBusFrame "Sampling at 1600000 S/s.<0A>"
(rtlwmbus) only text
(serial) received ascii Tuned to 868900000 Hz.<0A>
(rtlwmbus) checkRTLWMBusFrame "Tuned to 868900000 Hz.<0A>"
(rtlwmbus) only text
(serial) received ascii Tuner gain set to automatic.<0A>
(rtlwmbus) checkRTLWMBusFrame "Tuner gain set to automatic.<0A>"
(rtlwmbus) only text
(serial) received ascii Reading samples in async mode...<0A>
(rtlwmbus) checkRTLWMBusFrame "Reading samples in async mode...<0A>"
(rtlwmbus) only text
(serial) received ascii Allocating 15 zero-copy buffers<0A>
(rtlwmbus) checkRTLWMBusFrame "Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(serial) received ascii Failed to allocate zero-copy buffer for transfer 8<0A>Falling back to buffers in userspace<0A>
(rtlwmbus) checkRTLWMBusFrame "Failed to allocate zero-copy buffer for transfer 8<0A>Falling back to buffers in userspace<0A>"
(rtlwmbus) only text
(........another meter telegrams.....)
(serial) received ascii T1;1;1;2020-02-08 22:56:42.000;123;123;01681823;0x3e4401062318680105077ae8003085016c4b0f1af2b48db4bf5f120fed6962aa31b6dfa175dcbe1d3f8d2c5096fd5b573fe00b9e09ae2722b00a4f6598a9f0<0A>
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2020-02-08 22:56:42.000;123;123;01681823;0x3e4401062318680105077ae8003085016c4b0f1af2b48db4bf5f120fed6962aa31b6dfa175dcbe1d3f8d2c5096fd5b573fe00b9e09ae2722b00a4f6598a9f0<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 63
(meter) garden_meter: for me? 01681823
(meter) garden_meter: yes for me
(meter) garden_meter 01681823 "3E4401062318680105077AE8003085016C4B0F1AF2B48DB4BF5F120FED6962AA31B6DFA175DCBE1D3F8D2C5096FD5B573FE00B9E09AE2722B00A4F6598A9F0"
(wmbus) parseDLL @0 63
(wmbus) parseELL @10 53
(wmbus) parseAFL @10 53
(wmbus) parseTPL @10 53
(TPL) decrypting "016C4B0F1AF2B48DB4BF5F120FED6962AA31B6DFA175DCBE1D3F8D2C5096FD5B573FE00B9E09AE2722B00A4F6598A9F0"
(TPL) IV 0106231868010507E8E8E8E8E8E8E8E8
(TPL) decrypted "2F2F0FF4A8C29408020843460283000A4CF3A88203931B2010AE1200007101AE1200009C120000A0DE262402FFFF0024"
(dvparser) cannot handle dif 0F ignoring rest of telegram.
(dvparser) found new format "" with hash ffff, remembering!
(telegram) DLL L=3e C=44 (from meter SND_NR) M=0601 (APA) A=01681823 VER=05 TYPE=07 (Water meter)
(telegram) TPL CI=7a ACC=e8 STS=00 CFG=8530 (AES_CBC_IV synchronous)
(apator162) log "telegram=|3E4401062318680105077AE80030852F2F|0FF4A8C29408020843460283000A4CF3A88203931B2010AE1200007101AE1200009C120000A0DE262402FFFF0024|+666"
(apator162) Guessing offset to be 23 from byte 0x83: total ae120000
(apator162) 00: 3e length (62 bytes)
(apator162) 01: 44 dll-c (from meter SND_NR)
(apator162) 02: 0106 dll-mfct (APA)
(apator162) 04: 23186801 dll-id (01681823)
(apator162) 08: 05 dll-version
(apator162) 09: 07 dll-type (Water meter)
(apator162) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(apator162) 0b: e8 tpl-acc-field
(apator162) 0c: 00 tpl-sts-field
(apator162) 0d: 3085 tpl-cfg (AES_CBC_IV synchronous )
(apator162) 0f: 2f2f decrypt check bytes
{"media":"water","meter":"apator162","name":"garden_meter","id":"01681823","total_m3":4.782,"timestamp":"2020-02-08T22:56:42Z"}
(........another meter telegrams.....)
`
### value 4,782 is correct :)

@weetmuts three questions:
1. How to change the default fq inside the wmbusmeters service / rtl_wmbus, now i'm calling "by the hand" not automatically
2. Why when I using command to listen my meter i can do this only once (next time i've got errors. Its working correctly when i re-instert the dongle (but only once request)
`(serial) received ascii usb_claim_interface error -6<0A>
(rtlwmbus) checkRTLWMBusFrame "usb_claim_interface error -6<0A>"
(rtlwmbus) only text
(serial) received ascii Failed to open rtlsdr device #0.<0A>
(rtlwmbus) checkRTLWMBusFrame "Failed to open rtlsdr device #0.<0A>"
`
3. Why its working with one meter (garden water meter) but not with another (main water meter in my house)? its the same protocol, garden meter was installed after a year than main meter

weetmuts, 2020-02-09:
>> value 4,782 is correct :)

Yay! Congratulations!

>>  How to change the default fq inside the wmbusmeters service / rtl_wmbus, now i'm calling "by the hand" not automatically

You specify device=rtlwmbus:868.9  in the wmbusmeters.conf file, see the README.

>> Why when I using command to listen my meter i can do this only once (next time i've got errors. Its working correctly when i re-instert the dongle (but only once request)

Because the rtl_sdr does not nicely die. Try "sudo killall -9 rtl_sdr" and "sudo killall -9 rtl_wmbus" to cleanup before starting wmbusmeters again. I do not know why this happens. 

>> Why its working with one meter (garden water meter) but not with another (main water meter in my house)? its the same protocol, garden meter was installed after a year than main meter

What is not working? 
If you do not receive any telegrams at all, then try closer to the meter.
Do you receive telegrams but they decode to the wrong value?
Then this can be due to the naught proprietary apator162 protocol not being fully understood.....

lsw271, 2020-02-09:
> What is not working?
> If you do not receive any telegrams at all, then try closer to the meter.
> Do you receive telegrams but they decode to the wrong value?
> Then this can be due to the naught proprietary apator162 protocol not being fully understood.....

@weetmuts look at this log, this is my main water meter

`pi@raspberrypi:~/rtl-sdr $ wmbusmeters --logtelegrams --debug --verbose --format=json rtlwmbus:868.9M MyTapWater apator162:t1 01478662 00000000000000000000000000000000
(wmbusmeters) version: 0.9.25-1-gaf98f36
(config) using device: rtlwmbus
(config) with: 868.9M
(config) number of meters: 1
(detect) "rtlwmbus" "868.9M"
(detect) driver: rtlwmbus
(rtlwmbus) using command: rtl_sdr -f 868.9M -s 1.6e6 - | rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 868.9M -s 1.6e6 - | rtl_wmbus"
(serialcmd) opened /bin/sh
(config) meter apator162 link mode(s): t1
(config) all possible link modes that the meters might transmit on: t1
(config) listen link modes calculated to be: t1
(wmbusmeters) configured "MyTapWater" "apator162" "01478662" encrypted
(config) listen to link modes: any
(serial) waiting for stop
(serial) received ascii Found 1 device(s):<0A>
(rtlwmbus) checkRTLWMBusFrame "Found 1 device(s):<0A>"
(rtlwmbus) only text
(serial) received ascii   0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>
(rtlwmbus) checkRTLWMBusFrame "  0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>"
(rtlwmbus) only text
(serial) received ascii Using device 0: Generic RTL2832U OEM<0A>
(rtlwmbus) checkRTLWMBusFrame "Using device 0: Generic RTL2832U OEM<0A>"
(rtlwmbus) only text
(serial) received ascii Found Rafael Micro R820T tuner<0A>
(rtlwmbus) checkRTLWMBusFrame "Found Rafael Micro R820T tuner<0A>"
(rtlwmbus) only text
(serial) received ascii [R82XX] PLL not locked!<0A>
(rtlwmbus) checkRTLWMBusFrame "[R82XX] PLL not locked!<0A>"
(rtlwmbus) only text
(serial) received ascii Sampling at 1600000 S/s.<0A>
(rtlwmbus) checkRTLWMBusFrame "Sampling at 1600000 S/s.<0A>"
(rtlwmbus) only text
(serial) received ascii Tuned to 868900000 Hz.<0A>
(rtlwmbus) checkRTLWMBusFrame "Tuned to 868900000 Hz.<0A>"
(rtlwmbus) only text
(serial) received ascii Tuner gain set to automatic.<0A>
(rtlwmbus) checkRTLWMBusFrame "Tuner gain set to automatic.<0A>"
(rtlwmbus) only text
(serial) received ascii Reading samples in async mode...<0A>
(rtlwmbus) checkRTLWMBusFrame "Reading samples in async mode...<0A>"
(rtlwmbus) only text
(serial) received ascii Allocating 15 zero-copy buffers<0A>
(rtlwmbus) checkRTLWMBusFrame "Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(serial) received ascii Failed to allocate zero-copy buffer for transfer 8<0A>Falling back to buffers in userspace<0A>
(rtlwmbus) checkRTLWMBusFrame "Failed to allocate zero-copy buffer for transfer 8<0A>Falling back to buffers in userspace<0A>"
(rtlwmbus) only text
(serial) received ascii T1;1;1;2020-02-08 00:21:19.000;124;117;01478662;0x4e4401066286470105077a4d004085e7b99c0451118578615eab6d8859ab416e7fce636868f4c61f85e3006c22250a4ce2c15b8bc590d6166b0d5e0257e646d1843f7cbc15e5cc05837aa8c81ba95c<0A>
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2020-02-08 00:21:19.000;124;117;01478662;0x4e4401066286470105077a4d004085e7b99c0451118578615eab6d8859ab416e7fce636868f4c61f85e3006c22250a4ce2c15b8bc590d6166b0d5e0257e646d1843f7cbc15e5cc05837aa8c81ba95c<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 79
(meter) MyTapWater: for me? 01478662
(meter) MyTapWater: yes for me
(meter) MyTapWater 01478662 "4E4401066286470105077A4D004085E7B99C0451118578615EAB6D8859AB416E7FCE636868F4C61F85E3006C22250A4CE2C15B8BC590D6166B0D5E0257E646D1843F7CBC15E5CC05837AA8C81BA95C"
(wmbus) parseDLL @0 79
(wmbus) parseELL @10 69
(wmbus) parseAFL @10 69
(wmbus) parseTPL @10 69
(TPL) decrypting "E7B99C0451118578615EAB6D8859AB416E7FCE636868F4C61F85E3006C22250A4CE2C15B8BC590D6166B0D5E0257E646D1843F7CBC15E5CC05837AA8C81BA95C"
(TPL) IV 01066286470105074D4D4D4D4D4D4D4D
(TPL) decrypted "2F2F0F3708C294100200435803848B745953486C0910000010C58702007501F1800200E5640200534A02003431020080150200D9000200A0DC939703FFFF54E1"
(dvparser) cannot handle dif 0F ignoring rest of telegram.
(dvparser) found new format "" with hash ffff, remembering!
(telegram) DLL L=4e C=44 (from meter SND_NR) M=0601 (APA) A=01478662 VER=05 TYPE=07 (Water meter)
(telegram) TPL CI=7a ACC=4d STS=00 CFG=8540 (AES_CBC_IV accessibility)
(apator162) log "telegram=|4E4401066286470105077A4D0040852F2F|0F3708C294100200435803848B745953486C0910000010C58702007501F1800200E5640200534A02003431020080150200D9000200A0DC939703FFFF54E1|+6"
(apator162) Unknown value in proprietary(unknown) apator162 protocol. Ignoring telegram. Found 0x84 expected bit 0x01, 0x10, 0x40 or 0x80 to be set.
(apator162) 00: 4e length (78 bytes)
(apator162) 01: 44 dll-c (from meter SND_NR)
(apator162) 02: 0106 dll-mfct (APA)
(apator162) 04: 62864701 dll-id (01478662)
(apator162) 08: 05 dll-version
(apator162) 09: 07 dll-type (Water meter)
(apator162) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(apator162) 0b: 4d tpl-acc-field
(apator162) 0c: 00 tpl-sts-field
(apator162) 0d: 4085 tpl-cfg (AES_CBC_IV accessibility )
(apator162) 0f: 2f2f decrypt check bytes
{"media":"water","meter":"apator162","name":"MyTapWater","id":"01478662","total_m3":0,"timestamp":"2020-02-08T00:21:19Z"}
`
**(apator162) Unknown value in proprietary(unknown) apator162 protocol. Ignoring telegram. Found 0x84 expected bit 0x01, 0x10, 0x40 or 0x80 to be set.**

lsw271, 2020-02-12:
@weetmuts how can I help you to decode this part?
`(apator162) Unknown value in proprietary(unknown) apator162 protocol. Ignoring telegram. Found 0x84 expected bit 0x01, 0x10, 0x40 or 0x80 to be set.`

anonymous-one, 2020-02-16:
Sorry to chime in out of nowhere without any help, but...

a) i too have a demanding 1.25 year old ;)

b) Coincidentally TODAY I was getting around to getting my water meters on my HA setup and was about to order a 2nd meter and have my hydraulic guy install it... long story short... but my water company has 2 of these in my home (one for garden one for primary)... i just ordered my sdr usb adapter, will be here on Tuesday... I will dig into this on Tuesday night.

c) The devices on my water meter are AT-WMBUS-16-2a

d) My ID numbers are clearly visible on the sticker (7 digit). Only checked one, guessing the 2nd one is the same.

e) They were installed from what I remember about 2-3 years ago.

f) Pan z Polski? These devices are popular here in Poland.

weetmuts, 2020-02-16:
Hi @lsw271  !
In simulations/simulation_t1.txt
you have several telegrams from apator162 meters.
Our guess so far is that bit in a certain byte indicates where the 
water consumption is located. But it seems like there is something more.

Check all the telegrams in the simulations file and compare with your
telegram to see if you can figure out the rule for when the meter value location changes.

Check the source to meter_apator162 for the bit testing code.

I will get back to the apator162 eventually, but right now I am busy with the CUL driver.
:-)

weetmuts, 2020-02-16:
Hi @anonymous-one !

If you are lucky, it might work out of the box.  Do you have an rtlsdr dongle or something else?
First listen to telegrams, to see your own meter.
wmbusmeters --listento=t1,c1 auto 
(Apator162 can send on either t1 or c1 depending on its config.)
Then when you know you can hear your own meter.
Do
wmbusmeters auto MyWater apator162 12345678 "00000000000000000000000000000000"

These meters seems to often use a key with all zeroes.

anonymous-one, 2020-02-16:
Hi weetmuts.

First of all, major thanks for all the work here. Saved me a bit of money and having 'yet another' device on my wifi doing nothing besides listening to water meter pulses.

Will let you know either way (works / doesnt) on Tuesday.

This is the adapter I just ordered:

https://www.amazon.de/gp/product/B01HA642SW/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

Coincidentally I also have some decent 868mhz antennas left over from when I was getting my z-wave mesh up and running.

Also my water meters are nice and close to where I will be sniffing from, I have a RPI on each of the floors of my home... Garage included. So the primary one is about 5M away from the RPI, the garden one is about 20M away (behind 1 wall)...

Last but not least, about 2 decades (yikes!) of *nix experience as well as about that in various softdev (mainly on linux as well) behind me... So any issues I post will be regarding getting the packets decrypted...

Once again, big thanks!

anonymous-one, 2020-02-16:
PS : Only time will tell if the key here is 0000... but I give it a higher chance then not because they did they whole neighborhood (guessing, 150-200? homes...) in one go... and well... hoping they decided to go lazy and not re-key each of the transmitters... Then again who knows maybe they have an option to give the device a new key on install.

lsw271, 2020-02-17:
> Hi @lsw271 !
> In simulations/simulation_t1.txt
> you have several telegrams from apator162 meters.
> Our guess so far is that bit in a certain byte indicates where the
> water consumption is located. But [...]

Thanks! After few telegrams and looking visually on my meter i think i've found my water consumption:

telegram=|4E4401066286470105077AEF0040852F2F|0F07112294100200436103848B745953486C0910000010**81920200**7501F1800200E5640200534A02003431020080150200D9000200A0DC939703FFFFA434|+541    ///// "2020-02-16T23:37:55Z"
**81920200** = 029281 = 168577 (168.577) this is my current water meter level 

tomorrow i'll do more tests, but i dont understand why the program can't decode that, meter readings are in the same place in telegram as readings of my garden meter

lsw271, 2020-02-17:
> Sorry to chime in out of nowhere without any help, but...
>
> a) i too have a demanding 1.25 year old ;)
>
> b) Coincidentally TODAY I was getting around to getting my water meters on my HA setup and was [...]

Hi @anonymous-one,
you've got similar hardware configuration (meter and child :D) 
good luck with the wmbusmeters configuration ;)

tak jestem z Polski :)

anonymous-one, 2020-02-17:
witaj!

moj polski jest taki sobie, cale zycie zagranica... ostatnie pare lat w polsce...

for everyone elses benefit, i will speak in english... i am waiting for my sdr usb to arrive, but, if i was in the same situation as you, i would try:

your error log shows:

(apator162) Unknown value in proprietary(unknown) apator162 protocol. Ignoring telegram. Found 0x84 expected bit 0x01, 0x10, 0x40 or 0x80 to be set.

edit file:

https://github.com/weetmuts/wmbusmeters/blob/master/src/meter_apator162.cc

replace:

```
uchar guess = content[11];
if ((guess & 0x83) == 0x83)
{
    o = 23;
}
```

with:

```
uchar guess = content[11];
if ((guess & 0x84) == 0x84)
{
    o = 23;
}
else
if ((guess & 0x83) == 0x83)
{
    o = 23;
}
```

if this does not work / help... maybe weetmuts can share how to convert the offset bytes. if you are impatient, try (50?) down, and keep trying ;)

anonymous-one, 2020-02-17:
note : because you say they are in the same location as your garden meter... i would give it a high chance that o = 23 (in 0x84) will work.

anonymous-one, 2020-02-17:
@lsw271 

2 questions for you:

1) how did you convert : 81920200 = 029281 = 168577 (168.577)? im not sure i understand.

2) just out of curiosity, how often does the meter send a report?

lsw271, 2020-02-17:
> @lsw271
> 
> 2 questions for you:
> 
> 1. how did you convert : 81920200 = 029281 = 168577 (168.577)? im not sure i understand.
> 2. just out of curiosity, how often does the meter send a report?

I used this to convert https://www.rapidtables.com/convert/number/hex-to-decimal.html
81 92 02 00 from backward and read from left 00 02 92 81 (numer is compatibile so i think its ok ;) like in my second water meter. 

I don't remember on which hours it starts more frequent (maybe at 8 AM) but it sends up to 30seconds (at day) and another time (at night) maybe its a few minutes between telegrams

lsw271, 2020-02-17:
> witaj!
> 
> moj polski jest taki sobie, cale zycie za granica... ostatnie pare lat w polsce...
> 
> for everyone elses benefit, i will speak in english... i am waiting for my sdr usb to arrive, but, if i was in the same situation as you, i would try:
> 
> your error log shows:
> 
> (apator162) Unknown value in proprietary(unknown) apator162 protocol. Ignoring telegram. Found 0x84 expected bit 0x01, 0x10, 0x40 or 0x80 to be set.
> 
> edit file and recompile / reinstall:
> 
> https://github.com/weetmuts/wmbusmeters/blob/master/src/meter_apator162.cc
> 
> replace:
> 
> ```
> uchar guess = content[11];
> if ((guess & 0x83) == 0x83)
> {
>     o = 23;
> }
> ```
> 
> with:
> 
> ```
> uchar guess = content[11];
> if ((guess & 0x84) == 0x84)
> {
>     o = 23;
> }
> else
> if ((guess & 0x83) == 0x83)
> {
>     o = 23;
> }
> ```
> 
> if this does not work / help... maybe weetmuts can share how to convert the offset bytes. if you are impatient, try (50?) down, and keep trying ;)

I think your Polish is better than my English :D 
I will try your proposition tonight ;)

lsw271, 2020-02-17:
@weetmuts, @anonymous-one  i've added this:

```
uchar guess = content[11];
if ((guess & 0x84) == 0x84)
{
    o = 23;
}
```
and now i've got right meter readings :D

@weetmuts i think you could add it to source code

thanks you very much!

closed 2020-02-17 by lsw271

weetmuts, 2020-02-18:
Thanks I have added 0x84 to the github master, and another workaround.
Please test! So that it does not break for you.

Unfortunately we are still guessing, what these bytes mean. :-/

anonymous-one, 2020-02-18:
Once again, HUGE HUGE HUGE thanks for this project.

I have a very comprehensive HA setup... 3-4 years in the making and getting the water meters online has been on the to do for EVER. We are getting bills that equal about a 6 person family. There are 2.5 (2 + small boy) of us here. We need to get this sorted out.

---

Works 100%.

Ironically, I have the same setup as lsw271.

Meter 1 - Primary
Meter 2 - Garden

In my setup, the SAME exact situation as him... 0x84 being sent on the primary.

With the 0x84 patch, all works :D:D:D

THANK YOU!!!!!!!

lsw271, 2020-02-20:
> Thanks I have added 0x84 to the github master, and another workaround.
> Please test! So that it does not break for you.
> 
> Unfortunately we are still guessing, what these bytes mean. :-/

@weetmuts everything looks ok, values are correct :) 
thank you! ;)