wmbusmeters/wmbusmeters #41: Segmentation fault

merbanan, 2019-10-28:
During a debug run I started getting strange values eventually ending in a crash. Imo the receiver just returned garbage for whatever reason but that shouldn't lead to a crash. Maybe the last payload can reproduce the crash.

`(serial /dev/ttyACM0) received "2E44333046410100071B7A12482025B08ECFA413404DED8B0E93D694ED3599390FC0D45F8024E89830D77EB9840BF4"
(wmbus) received telegram 12082865 C-field=ec M-field=2207 (HPG) A-field-version=65 A-field-dev-type=2c (Reserved for environmental meter) Ci-field=08 (?)
(wmbus) frame "65EC072265280812652C086265DE075265160802FB1A7C0142FB1A81018201FB1AB40122FB1A7F0112FB1A830162FB1A9E0152FB1ACB01066D3B31D67A2A002E44333046410100071B7A12482025B08ECFA413404DED8B0E93D694ED3599390FC0D45F8024E8"
(wmbus) payload "6265DE075265160802FB1A7C0142FB1A81018201FB1AB40122FB1A7F0112FB1A830162FB1A9E0152FB1ACB01066D3B31D67A2A002E44333046410100071B7A12482025B08ECFA413404DED8B0E93D694ED3599390FC0D45F8024E8"
(wmbus) 00: 65 length (101 bytes)
(wmbus) 01: ec c-field (?)
(wmbus) 02: 0722 m-field (2207=HPG)
(wmbus) 04: 65280812 a-field-addr (12082865)
(wmbus) 08: 65 a-field-version
(wmbus) 09: 2c a-field-type (Reserved for environmental meter)
(wmbus) 0a: 08 ci-field (?)
(wmbus) 65EC072265280812652C08
Received telegram from: 12082865
          manufacturer: (HPG) Unknown
           device type: Reserved for environmental meter
(rawtty) telegram ignored by all configured meters!
(serial /dev/ttyACM0) received "5744B40984217016101B7A440800000265290842652B08820165EC072265280812652C086265DE075265160802FB1A7C0142FB1A81018201FB1AB40122FB1A7F0112FB1A830162FB1A9E0152FB1ACB01066D3B31D67A2A00"
(serial /dev/ttyACM0) received "5744B40979217016101B7A4E0800000265BE084265B30882016554082265B0081265B7086265F7075265E70802FB1A390142FB1A3C018201FB1A850122FB1A370112FB1A3F0162FB1A5C0152FB1AC401066D3B1D0D7B2A00"
(wmbus) received telegram f40b84b9 C-field=30 M-field=7ed7 (_VW) A-field-version=57 A-field-dev-type=44 (Unknown) Ci-field=b4 (Mfct specific)
(wmbus) frame "9830D77EB9840BF45744B40984217016101B7A440800000265290842652B08820165EC072265280812652C086265DE075265160802FB1A7C0142FB1A81018201FB1AB40122FB1A7F0112FB1A830162FB1A9E0152FB1ACB01066D3B31D67A2A005744B40979217016101B7A4E0800000265BE084265B30882016554082265B0081265B7086265F7075265E70802FB1A390142FB1A3C018201FB"
(wmbus) payload "0984217016101B7A440800000265290842652B08820165EC072265280812652C086265DE075265160802FB1A7C0142FB1A81018201FB1AB40122FB1A7F0112FB1A830162FB1A9E0152FB1ACB01066D3B31D67A2A005744B40979217016101B7A4E0800000265BE084265B30882016554082265B0081265B7086265F7075265E70802FB1A390142FB1A3C018201FB"
(wmbus) 00: 98 length (152 bytes)
(wmbus) 01: 30 c-field (?)
(wmbus) 02: d77e m-field (7ed7=_VW)
(wmbus) 04: b9840bf4 a-field-addr (f40b84b9)
(wmbus) 08: 57 a-field-version
(wmbus) 09: 44 a-field-type (Unknown)
(wmbus) 0a: b4 ci-field (Mfct specific)
(wmbus) 9830D77EB9840BF45744B4
Received telegram from: f40b84b9
          manufacturer: (_VW) Unknown
           device type: Unknown
(rawtty) telegram ignored by all configured meters!
(serial /dev/ttyACM0) received "5744B40984217016101B7A450800000265290842652B08820165EC072265280812652C086265DE075265160802FB1A7C0142FB1A81018201FB1AB40122FB1A7F0112FB1A830162FB1A9E0152FB1ACB01066D3B31D67A2A00"
(wmbus) received telegram 01371afb C-field=85 M-field=2201 (HPA) A-field-version=12 A-field-dev-type=fb (Unknown) Ci-field=1a (?)
(wmbus) frame "1A850122FB1A370112FB1A3F0162FB1A5C0152FB1AC401066D3B1D"
(wmbus) payload "3F0162FB1A5C0152FB1AC401066D3B1D"
(wmbus) 00: 1a length (26 bytes)
(wmbus) 01: 85 c-field (?)
(wmbus) 02: 0122 m-field (2201=HPA)
(wmbus) 04: fb1a3701 a-field-addr (01371afb)
(wmbus) 08: 12 a-field-version
(wmbus) 09: fb a-field-type (Unknown)
(wmbus) 0a: 1a ci-field (?)
(wmbus) 1A850122FB1A370112FB1A
Received telegram from: 01371afb
          manufacturer: (HPA) Unknown
           device type: Unknown
(rawtty) telegram ignored by all configured meters!
(serial /dev/ttyACM0) received "5744B40984217016101B7A460800000265290842652B08820165EC072265280812652C086265DE075265160802FB1A7C0142FB1A81018201FB1AB40122FB1A7F0112FB1A830162FB1A9E0152FB1ACB01066D3B31D67A2A00"
(wmbus) received telegram 09b44457 C-field=7b M-field=002a (@AJ) A-field-version=84 A-field-dev-type=21 (Valve (gas or water)) Ci-field=70 (Application error from device without Transport Layer)
(wmbus) frame "0D7B2A005744B40984217016101B"
(wmbus) payload "16101B"
(wmbus) 00: 0d length (13 bytes)
(wmbus) 01: 7b c-field (?)
(wmbus) 02: 2a00 m-field (2a=@AJ)
(wmbus) 04: 5744b409 a-field-addr (09b44457)
(wmbus) 08: 84 a-field-version
(wmbus) 09: 21 a-field-type (Valve (gas or water))
(wmbus) 0a: 70 ci-field (Application error from device without Transport Layer)
(wmbus) 0D7B2A005744B409842170
Received telegram from: 09b44457
          manufacturer: (@AJ) Unknown
           device type: Valve (gas or water)
(rawtty) telegram ignored by all configured meters!
(serial /dev/ttyACM0) received "5744B40979217016101B7A500800000265BE084265B30882016554082265B0081265B7086265F7075265E70802FB1A390142FB1A3C018201FB1A850122FB1A370112FB1A3F0162FB1A5C0152FB1AC401066D3B1D0D7B2A00"
(wmbus) received telegram 29650200 C-field=45 M-field=0008 (@@H) A-field-version=08 A-field-dev-type=42 (Unknown) Ci-field=65 (Reserved for OBIS-based Data sent by the Readout device to the meter with short Transport Layer)
(wmbus) frame "7A450800000265290842652B08820165EC072265280812652C086265DE075265160802FB1A7C0142FB1A81018201FB1AB40122FB1A7F0112FB1A830162FB1A9E0152FB1ACB01066D3B31D67A2A005744B40984217016101B7A460800000265290842652B08820165EC072265280812652C086265DE075265160802"
(wmbus) payload "2B08820165EC072265280812652C086265DE075265160802FB1A7C0142FB1A81018201FB1AB40122FB1A7F0112FB1A830162FB1A9E0152FB1ACB01066D3B31D67A2A005744B40984217016101B7A460800000265290842652B08820165EC072265280812652C086265DE075265160802"
(wmbus) 00: 7a length (122 bytes)
(wmbus) 01: 45 c-field (?)
(wmbus) 02: 0800 m-field (08=@@H)
(wmbus) 04: 00026529 a-field-addr (29650200)
(wmbus) 08: 08 a-field-version
(wmbus) 09: 42 a-field-type (Unknown)
(wmbus) 0a: 65 ci-field (Reserved for OBIS-based Data sent by the Readout device to the meter with short Transport Layer)
(wmbus) 7A45080000026529084265
Received telegram from: 29650200
          manufacturer: (@@H) Unknown
           device type: Unknown
(rawtty) telegram ignored by all configured meters!
(serial /dev/ttyACM0) received "2E44333046410100071B7A13482025FE3EFDBBD0BB74EACA9EEA0C1D11BB055BFC5921B5772A6AABA04186E541FB6B"
(serial /dev/ttyACM0) received "5744B40979217016101B7A510800000265BE084265B30882016554082265B0081265B7086265F7075265E70802FB1A390142FB1A3C018201FB1A850122FB1A370112FB1A3F0162FB1A5C0152FB1AC401066D3B1D0D7B2A00"
(wmbus) received telegram 811afb42 C-field=1a M-field=017c (@K\) A-field-version=01 A-field-dev-type=82 (Unknown) Ci-field=01 (?)
(wmbus) frame "FB1A7C0142FB1A81018201FB1AB40122FB1A7F0112FB1A830162FB1A9E0152FB1ACB01066D3B31D67A2A005744B40979217016101B7A500800000265BE084265B30882016554082265B0081265B7086265F7075265E70802FB1A390142FB1A3C018201FB1A850122FB1A370112FB1A3F0162FB1A5C0152FB1AC401066D3B1D0D7B2A002E44333046410100071B7A13482025FE3EFDBBD0BB74EACA9EEA0C1D11BB055BFC5921B5772A6AABA04186E541FB6B5744B40979217016101B7A510800000265BE084265B30882016554082265B0081265B7086265F7075265E70802FB1A390142FB1A3C018201FB1A850122FB1A370112FB1A3F0162FB1A5C"
(wmbus) payload "FB1AB40122FB1A7F0112FB1A830162FB1A9E0152FB1ACB01066D3B31D67A2A005744B40979217016101B7A500800000265BE084265B30882016554082265B0081265B7086265F7075265E70802FB1A390142FB1A3C018201FB1A850122FB1A370112FB1A3F0162FB1A5C0152FB1AC401066D3B1D0D7B2A002E44333046410100071B7A13482025FE3EFDBBD0BB74EACA9EEA0C1D11BB055BFC5921B5772A6AABA04186E541FB6B5744B40979217016101B7A510800000265BE084265B30882016554082265B0081265B7086265F7075265E70802FB1A390142FB1A3C018201FB1A850122FB1A370112FB1A3F0162FB1A5C"
(wmbus) 00: fb length (251 bytes)
(wmbus) 01: 1a c-field (?)
(wmbus) 02: 7c01 m-field (17c=@K\)
(wmbus) 04: 42fb1a81 a-field-addr (811afb42)
(wmbus) 08: 01 a-field-version
(wmbus) 09: 82 a-field-type (Unknown)
(wmbus) 0a: 01 ci-field (?)
(wmbus) FB1A7C0142FB1A81018201
Received telegram from: 811afb42
          manufacturer: (@K\) Unknown
           device type: Unknown
(rawtty) telegram ignored by all configured meters!
(serial /dev/ttyACM0) received "5744B40984217016101B7A470800000265290842652B08820165EC072265280812652C086265DE075265160802FB1A7C0142FB1A81018201FB1AB40122FB1A7F0112FB1A830162FB1A9E0152FB1ACB01066D3B31D67A2A00"
Segmentation fault
`

weetmuts, 2019-10-28:
It gets out of sync. 

I have added code to detect this and try to resync. Worst case, it will throw away some telegrams.
Check verbose/debug log for "out of sync" or "no sensible telegram found" to see how often
this happens.

The previous fuzzing only fuzzed the difvif parser. Now I have added support
for fuzzing the whole telegram, and for sure, the afl-fuzz finds ~10 disting crashes.

I will fix these.

merbanan, 2019-10-28:
I guess this can be closed now.

weetmuts, 2019-10-29:
Well, run it for a while to see how often it gets out of sync.

weetmuts, 2019-11-03:
I have now pushed 0.9.17 which among many other small fixes also is afl_fuzz resistant on the raw telegram level. At least for a couple of minutes. So it is much less likely to get a segmentation fault now.

Will continue to fuzz it to find potential holes.

closed 2019-11-17 by weetmuts