wmbusmeters/wmbusmeters #43: WMBUS-D1-TC support

merbanan, 2019-11-01:
Another day another device. So this device apparently listens to wmbus transmissions and outputs them on /dev/ttyUSB0 at 115200. When running the device it outputs an encapsulated payload. 

A telegram that was decoded by the Bmeters device:

5744B40938227016101B7ADA00000002653D0842653308820165A60722652E0812653D08626583075265C20702FB1A740142FB1A70018201FB1A690122FB1A6E0112FB1A740162FB1A640152FB1A6D01066D3B09B7612B00

was decoded into this by the device:

7e 06 5944b40938227016101b7ada00000002653d0842653308820165a60722652e0812653d08626583075265c20702fb1a740142fb1a70018201fb1a690122fb1a6e0112fb1a740162fb1a640152fb1a6d01066d3b09b7612b00
d7b5b7 7e

The encapsulation starts with 7e 06, then compared to the Bmeters payload the value after is 2 larger. At the end 3 extra bytes are present and then it all ends with 7e. By chance it sometimes decode properly. To support this device one must most likely buffer the data until it is completely received and then send it for decode.

./wmbusmeters --debug --listento=t1 /dev/ttyUSB0:115200
(wmbusmeters) version: 0.9.15-4-g04d8662 with local changes
(config) using device: /dev/ttyUSB0
(config) with: 115200
(config) number of meters: 0
(rawtty) found /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(config) all possible link modes that the meters might transmit on: none
(config) explicitly listening to: t1
(config) listen link modes calculated to be: t1
(config) listen to link modes: t1
No meters configured. Printing id:s of all telegrams heard!

(serial /dev/ttyUSB0) received "007E"
(wmbus_rawtty) no sensible telegram found, clearing buffer.
(rawtty) protocol error in message received!
(rawtty) protocol error ""
(serial /dev/ttyUSB0) received "7E065944B40908227016101B7A870000000265490842654408820165A8072265410812654808626584075265CA0702FB1A720142FB1A75018201FB1A6C0122FB1A740112FB1A770162FB1A680152FB1A7001066D3B13B7612B00C71A357E"
(wmbus_rawtty) no sensible telegram found, clearing buffer.
(rawtty) protocol error in message received!
(rawtty) protocol error ""
(serial /dev/ttyUSB0) received "7E065944B40938227016101B7AE000"
(wmbus_rawtty) no sensible telegram found, clearing buffer.
(rawtty) protocol error in message received!
(rawtty) protocol error ""
(serial /dev/ttyUSB0) received "000002653C0842653308820165A60722652E0812653D08626583075265C20702FB1A750142FB1A70018201FB1A690122FB1A6E0112FB1A740162FB1A640152FB1A6D01066D3B13B7612B00DA849B7E"
(wmbus_rawtty) no sensible telegram found, clearing buffer.
(rawtty) protocol error in message received!
(rawtty) protocol error ""
(serial /dev/ttyUSB0) received "7E065944B40984"
(wmbus_rawtty) no sensible telegram found, clearing buffer.
(rawtty) protocol error in message received!
(rawtty) protocol error ""
(serial /dev/ttyUSB0) received "217016101B7A4608000002653608426529088201654E082265210812652E08626526085265800802FB1A7A0142FB1A7A018201FB1A580122FB1A770112FB1A7D5D0162FB1A490152FB1A7001066D3B27887F2A00D3601F7E"
(wmbus_rawtty) no sensible telegram found, clearing buffer.
(rawtty) protocol error in message received!
(rawtty) protocol error ""
(serial /dev/ttyUSB0) received "7E06"
(wmbus_rawtty) no sensible telegram found, clearing buffer.
(rawtty) protocol error in message received!
(rawtty) protocol error ""
(serial /dev/ttyUSB0) received "5944B40979217016101B7A500800000265BE0842659E088201652A08226590081265B5086265F2075265ED0802FB1A480142FB1A47018201FB1A300122FB1A440112FB1A4D0162FB1A170152FB1A6901066D3B13977F2A00DD4BAA7E"
(wmbus) received telegram 16702179 C-field=44 M-field=09b4 (BMT) A-field-version=10 A-field-dev-type=1b (Room sensor (eg temperature or humidity)) Ci-field=7a (EN 13757-3 Application Layer with short Transport Layer) CC-field=00 (slow_resp) short header ACC=50 
(wmbus) frame "5944B40979217016101B7A500800000265BE0842659E088201652A08226590081265B5086265F2075265ED0802FB1A480142FB1A47018201FB1A300122FB1A440112FB1A4D0162FB1A170152FB1A6901066D3B13977F2A00DD4B"
(wmbus) payload "0265BE0842659E088201652A08226590081265B5086265F2075265ED0802FB1A480142FB1A47018201FB1A300122FB1A440112FB1A4D0162FB1A170152FB1A6901066D3B13977F2A00DD4B"
(wmbus) 00: 59 length (89 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: b409 m-field (9b4=BMT)
(wmbus) 04: 79217016 a-field-addr (16702179)
(wmbus) 08: 10 a-field-version
(wmbus) 09: 1b a-field-type (Room sensor (eg temperature or humidity))
(wmbus) 0a: 7a ci-field (EN 13757-3 Application Layer with short Transport Layer)
(wmbus) 0b: 50 acc
(wmbus) 0c: 08 status ()
(wmbus) 0d: 0000 config ()
(wmbus) 5944B40979217016101B7A50080000
Received telegram from: 16702179
          manufacturer: (BMT) BMETERS, Italy
           device type: Room sensor (eg temperature or humidity)
(rawtty) telegram ignored by all configured meters!
(wmbus_rawtty) no sensible telegram found, clearing buffer.
(rawtty) protocol error in message received!
(rawtty) protocol error ""
(serial /dev/ttyUSB0) received "7E065944B40976217016101B7A8B0800000265410842655008820165610722654C0812655608626556075265790702FB1A6C0142FB1A71018201FB1A6E0122FB1A6E0112FB1A740162FB1A6A0152FB1A7101066D3B13B7612B00CE24597E"
(wmbus_rawtty) no sensible telegram found, clearing buffer.
(rawtty) protocol error in message received!
(rawtty) protocol error ""
(serial /dev/ttyUSB0) received "7E065944B40912227016101B7A890000000265470842654008820165A00722653D081265430862657D5E075265C00702FB1A6C0142FB1A71018201FB1A670122FB1A6F0112FB1A720162FB1A620152FB1A6B01066D3B13B7612B00C9A68F7E"
(wmbus_rawtty) no sensible telegram found, clearing buffer.
(rawtty) protocol error in message received!
(rawtty) protocol error ""
^C(serialtty) event loop stopped!
(serialtty) closed /dev/ttyUSB0

weetmuts, 2019-11-02:
Ok, should be easy to add. Do you have a link to the company that produces this device and/or some tech spec pdf or similar?

weetmuts, 2019-11-02:
Do you have the dmesg output with vendor and device id, so that we can auto-generate a udev link for it?

merbanan, 2019-11-04:
lsusb -v:

Bus 002 Device 014: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0403 Future Technology Devices International, Ltd
  idProduct          0x6001 FT232 USB-Serial (UART) IC
  bcdDevice            6.00
  iManufacturer           1 FTDI
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 TTL232RG-VREG3V3

dmesg:
[ 8470.720107] ftdi_sio 2-1.4.4:1.0: FTDI USB Serial Device converter detected
[ 8470.720165] usb 2-1.4.4: Detected FT232RL
[ 8470.720567] usb 2-1.4.4: FTDI USB Serial Device converter now attached to ttyUSB0

merbanan, 2019-11-04:
Some of the extra bytes at the end forms a RSSI value. Can a receiver device add a json value to the output? If that is the case I'll try to figure out the formatting otherwise I'll skip it.

merbanan, 2019-11-04:
The device manufacturer is Lansen Systems

weetmuts, 2019-12-28:
Ok, pushed support for the d1tc device. It cannot be autodetected right now, both because we do not know how to talk to it (if it even responds) and because its usb vendor/product device conflicts with another dongle. Also we do not know which modes it is listening on, assuming all....

So for the moment try it with the d1tc suffix to identify the dongle:

wmbusmeters --listento=t1 /dev/ttyUSB0:d1tc

Please test!

weetmuts, 2020-02-02:
Does it work as it should? Any way we could get some documentation on this dongle?

merbanan, 2020-02-04:
I no longer have the device. It worked as it should when I tested it.

closed 2020-02-04 by merbanan