wmbusmeters/wmbusmeters #44: Multical302 warning: unknown frame with correct AES Key and no data

ben3k1v, 2019-11-19:
Hi Fredrik,

i try to get the data from my new Kamstrup Multical 302, i have the AES key but getting warnings an no Data. Is this a frame problem or maybe the wrong AES-Key? There are some decrypted blocks in the log.

**Config File**
```
name=Heat
type=multical302
id=67880556
key=0870C7A5438E42B6B5D1...
```

**Debug Log**

```
wmbusmeters --debug --verbose /dev/ttyUSB0:im871a multical302 multical302:c1 67880556 "0870C7A5..."

(wmbusmeters) version: 0.9.18
(config) using device: /dev/ttyUSB0
(config) with: im871a
(config) number of meters: 1
(detect) "/dev/ttyUSB0" "im871a"
(detect) is_tty=1 is_stdin=0 is_file=0
(im871a) on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(config) meter multical302 link mode(s): c1
(config) all possible link modes that the meters might transmit on: c1
(config) listen link modes calculated to be: c1
(wmbusmeters) configured "multical302" "multical302" "67880556" encrypted
(im871a) set link mode 06
(serial /dev/ttyUSB0) sent "A5010306000206300101"
(serial) received binary A58104009CF7
(im871a) got crc16 f79c expected f79c
(im871a) set config completed
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial) received binary A5810617FF000600B32558181000010001FF070232000101000077D9DB
(im871a) got crc16 dbd9 expected dbd9
(im871a) get config completed
(im871a) config: device mode 00
(im871a) config: link mode 06
(im871a) config: wmbus c-field 00
(im871a) config: wmbus mfg id 25b3 (IMS)
(im871a) config: wmbus device id 00101858
(im871a) config: wmbus version 01
(im871a) config: wmbus device type 00
(im871a) config: radio channel 01
(im871a) config: radio power level 07
(im871a) config: radio data rate 02
(im871a) config: radio rx window 32
(im871a) config: auto power saving 00
(im871a) config: auto RSSI attachment 01
(im871a) config: auto rx timestamp attachment 01
(im871a) config: led control 00
(im871a) config: rtc control 00
(config) listen to link modes: c1
(serial) waiting for stop

(serial) received binary A5E20325442D2C5605886730048D2046405C40206914B1DD04DBC608C9DB54B00CBA81FAF673221768EBCD000000DFF5
(im871a) timestamp 0000cdeb
(im871a) rssi 00
(im871a) got crc16 f5df expected f5df
(wmbus) received telegram 67880556 C-field=44 M-field=2c2d (KAM) A-field-version=30 A-field-dev-type=04 (Heat meter) Ci-field=8d (Extended Link Layer II (8 Byte)) CC-field=20 (slow_resp sync) ell header ACC=46 SN=20405c40
(wmbus) frame "25442D2C5605886730048D2046405C40206914B1DD04DBC608C9DB54B00CBA81FAF673221768"
(wmbus) payload "6914B1DD04DBC608C9DB54B00CBA81FAF673221768"
(wmbus) 00: 25 length (37 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 2d2c m-field (2c2d=KAM)
(wmbus) 04: 56058867 a-field-addr (67880556)
(wmbus) 08: 30 a-field-version
(wmbus) 09: 04 a-field-type (Heat meter)
(wmbus) 0a: 8d ci-field (Extended Link Layer II (8 Byte))
(wmbus) 0b: 20 cc-field (slow_resp sync)
(wmbus) 0c: 46 acc
(wmbus) 0d: 405c4020 sn
(wmbus) 25442D2C5605886730048D2046405C4020
(meter) multical302: for me? 67880556
(meter) multical302: yes for me
(multical302) multical302 67880556 (Mode1) decrypting "6914B1DD04DBC608C9DB54B00CBA81FAF673221768"
(Mode1) IV 2D2C56058867300420405C4020000000
(Mode1) decrypted first block "8EDDC982F3A62F39A5B5520FC4835965"
(Mode1) IV+1 2D2C56058867300420405C4020000001
(Mode1) decrypted second block "48CF2FF7C1"
(Mode1) decrypted "8EDDC982F3A62F39A5B5520FC483596548CF2FF7C1"
(multical302) log "telegram=|25442D2C5605886730048D2046405C4020|8EDDC982F3A62F39A5B5520FC483596548CF2FF7C1|+2"
(multical302) warning: unknown frame c9 (did you use the correct encryption key?)
(multical302) 11: 8edd payload crc
(multical302) 13: c9 frame type (?)
(multical302) 25442D2C5605886730048D2046405C40208EDDC9
multical302     67880556        0 kWh   0 m3    0 kW    2019-11-19 09:40.11

(serial) received binary A5E20325442D2C5605886730048D2047415C4020956F8BA42EB7F42FCDBA180CED8C2C8E0CC349390028820000003340
(im871a) timestamp 00008228
(im871a) rssi 00
(im871a) got crc16 4033 expected 4033
(wmbus) received telegram 67880556 C-field=44 M-field=2c2d (KAM) A-field-version=30 A-field-dev-type=04 (Heat meter) Ci-field=8d (Extended Link Layer II (8 Byte)) CC-field=20 (slow_resp sync) ell header ACC=47 SN=20405c41
(wmbus) frame "25442D2C5605886730048D2047415C4020956F8BA42EB7F42FCDBA180CED8C2C8E0CC3493900"
(wmbus) payload "956F8BA42EB7F42FCDBA180CED8C2C8E0CC3493900"
(wmbus) 00: 25 length (37 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 2d2c m-field (2c2d=KAM)
(wmbus) 04: 56058867 a-field-addr (67880556)
(wmbus) 08: 30 a-field-version
(wmbus) 09: 04 a-field-type (Heat meter)
(wmbus) 0a: 8d ci-field (Extended Link Layer II (8 Byte))
(wmbus) 0b: 20 cc-field (slow_resp sync)
(wmbus) 0c: 47 acc
(wmbus) 0d: 415c4020 sn
(wmbus) 25442D2C5605886730048D2047415C4020
(meter) multical302: for me? 67880556
(meter) multical302: yes for me
(multical302) multical302 67880556 (Mode1) decrypting "956F8BA42EB7F42FCDBA180CED8C2C8E0CC3493900"
(Mode1) IV 2D2C56058867300420415C4020000000
(Mode1) decrypted first block "145B6326F26C8F1730762D2A2DEACA8B"
(Mode1) IV+1 2D2C56058867300420415C4020000001
(Mode1) decrypted second block "F0B04C998B"
(Mode1) decrypted "145B6326F26C8F1730762D2A2DEACA8BF0B04C998B"
(multical302) log "telegram=|25442D2C5605886730048D2047415C4020|145B6326F26C8F1730762D2A2DEACA8BF0B04C998B|+18"
(multical302) warning: unknown frame 63 (did you use the correct encryption key?)
(multical302) 11: 145b payload crc
(multical302) 13: 63 frame type (?)
(multical302) 25442D2C5605886730048D2047415C4020145B63
multical302     67880556        0 kWh   0 m3    0 kW    2019-11-19 09:40.27
```

Thanks very much!
Ben

weetmuts, 2019-11-19:
Hi Ben! The multical302 code is "work in progress" ie have no examples of working telegrams and no-one has probably tested this code for quite some time.

If you can email me the decryption key (oehrstroem@gmail) then I can take a look given the
telegrams in the log you sent me.

Normally, this kind of error simply means that it is the wrong encryption key.
But since it is a "work in progress" meter, I cannot say for sure.

ben3k1v, 2019-11-19:
Hi,

thanks for the quick response! I've sent you the details.

Thanks!
Ben

ben3k1v, 2019-11-19:
Hi Fredrik,

I will send you this week the T1 debug file for the multical 302.
I get all important current Data in C1 Mode:
```
(multical302) 1c: 2c0000 	total power (44)
(multical302) 1f: 630000 	total volume (99)
(multical302) 29: 1300 		current power (1900)
44 kWh	99 m3	1900 kW
```
**Datasheet Info:**
```
Current data in C1 Mode.
Heat energy E1
Volume V1
Power
Info code
```
Regards
Ben

closed 2019-11-19 by ben3k1v