wmbusmeters/wmbusmeters #45: Adding support for CUL Devices

chester4444, 2019-11-19:
Hi, im using an nanoCUL stick to receive wmbus frames from my Multical21 meter,
i have the correct AES key - tested with the FHEM platform.

Now i want to switch to your program:

frames are looking like this (7 shorter and 1 longer frame, every 16 secs):
`bY25442D2C170935741B168D209622C841224F8B33D377F4560BB4B332CB9469E50AA16CAC529683FA
bY2C442D2C170935741B168D209723C84122A806AA39352EAF7B218DF0B2E9134E1731D365BB5DDC370C72951AB48021
`
Where should i start to add support for this type of frames?

weetmuts, 2019-11-19:
Sounds fun! 

A quick comparison between the cul output and the frames in simulation_c1.txt shows:
```
  |23442D2C998734761B168D208870F81821|09EA79EDA869F57100F8180000F41800000318
bY|25442D2C170935741B168D209622C841224F8B33D377F4560BB4B332CB9469E50AA16CAC52968|3FA
  |2A442D2C998734761B168D208971F81821|542F7802FF2071000413F81800004413F4180000615B05616717
bY|2C442D2C170935741B168D209723C84122A806AA39352EAF7B218DF0B2E9134E1731D365BB5DDC370C72951AB48|021
```

It looks like the cul sends ascii bytes over the serial port?
Each telegram starts with bY followed by a telegram that includes the 2 byte CRC.
(Which is why the frames in simulation_c1 are 2 bytes shorter. The im871a/amber sticks automatically strip the CRC.) Then there seems to be some wierd 3 nibble ending?

If cul sends ascii, then I suggest you make a copy of wmbus_rtlwmbus.cc and call it wmbus_cul.cc
changing the class name to WMBusCUL etc. Add DEVICE_CUL to LIST_OF_WMBUS_DEVICES and recursivle add all missing cases in switches etc.

Now in wmbus_cul.cc since this is a tty, change the openCUL (was openRTLWMBUS) to be like for example openIM871a, that reads from a tty, and not a process.

Now change checkCULFrame (was checkRTLWMBUSFrame) to detect the bY frame start to space? Or however it encapsulates its frames. :-)

Then you need to find the location of the CRC in the telegram.  Perhaps check it, but this ought
to have been done by the hardware cul. And then remove it and reduce the length byte with 2. 
(in the examples above 0x25 to 0x23, 0x2c to 0x2a)

Then the CUL driver should work for the rest of wmbusmeters.

Are there any available technical documents for this receiver?
Can you send commands to it, to configure it?

//Fredrik

weetmuts, 2019-11-19:
If you look for the freely available OMS-Spec_Vol2_AnnexN_A032.pdf page 4/47
then it seems like the CRC is located on byte index 11 and 12. 
(The length byte 0x25, is located on byte index 1.)
Perhaps this is the CRC that is left inside the message...
As the message gets longer, there will be more crc bytes.

chester4444, 2019-11-19:
Thanks, sounds straigth forward (i started with the wrong one, im817a).

The CUL device is accepting commands: like **_V_** -> answering with the firmware version string
also you have to initialize it after startup with **_brc_** and _**X21**_  to receive c1 frames.

more information is found here: [http://culfw.de/culfw.html](url)
and here [http://culfw.de/commandref.html](url)

weetmuts, 2019-11-19:
Neat, then it should also be possible to properly detect the existence of the CUL device.
By sending some command and checking the response.

Look in: detectWMBusDeviceSetting
and extend detectImstOrAmber to detectImstOrAmberCul.

weetmuts, 2019-11-19:
The initialization could be done in setLinkMode for example.
Can it listen to both C1 and T1 telegrams?

chester4444, 2019-11-19:
mmh, I dont know about C1 / T1, maybe.

chester4444, 2019-11-19:
Well, long packets are decoding right:
original and new frame:
```
bY2C442D2C170935741B168D20A7A0FB4222DDC1A235DDB5C7F31A607A5FC9EAD5B7A3DE81B2E2B64E535288E7D180FF
  2a442d2c170935741b168d20a7a0fb4222ddc1a235ddb5c7f31a607a5fc9ead5b7a3de81b2e2b64e535288
```
results in
```
(wmbus) received telegram 74350917 C-field=44 M-field=2c2d (KAM) A-field-version=1b A-field-dev-type=16 (Cold water meter) Ci-field=8d (Extended Link Layer II (8 Byte)) CC-field=20 (slow_resp sync) ell header ACC=a7 SN=2242fba0
(multical21) MyTapWater 74350917 MyTapWater     74350917        811.887 m3      803.568 m3      0 m3/h  13 °C   22 °C   OK      2019-11-19 22:14.07
```
but the shorter are firing an error: hash yet unknown
```
bY25442D2C170935741B168D20ABB0FB42228630E0D26187CE6BB8E50BC205B4EAFF7697C9FACA80FF
  23442d2c170935741b168d20abb0fb42228630e0d26187ce6bb8e50bc205b4eaff7697c9

(wmbus) received telegram 74350917 C-field=44 M-field=2c2d (KAM) A-field-version=1b A-field-dev-type=16 (Cold water meter) Ci-field=8d (Extended Link Layer II (8 Byte)) CC-field=20 (slow_resp sync) ell header ACC=ab SN=2242fbb0
(multical21) MyTapWater 74350917 (multical21) ignoring telegram since format signature hash 0xd2f7 is yet unknown.
MyTapWater      74350917        0 m3    0 m3    0 m3/h  127 °C  127 °C  OK      2019-11-19 22:15.10

```

weetmuts, 2019-11-20:
Excellent work!

It is not an error. :-) This is how it should be.

Short telegrams have left out the dif/vif information, ie the information
about what the content bytes mean. There is a small hash instead
which is the hash of the dif/vif information.
The short telegrams cannot be decoded until a long telegram has arrived.

The long telegram contains all the dif/vif data (and the content bytes), a hash is calculated over the dif/vif data and the hash is remembered to mean that particular dif/vif data, so that future short telegrams can be decoded.

wmbusmeters actually have some hashes -> difvif data hard-coded.
Check in meter_multical21.cc for format_signature. 
You could add yours there if  you like, after you have verified that the short telegrams
gets properly decoded after receiving a long telegram.

chester4444, 2019-11-20:
> It is not an error. :-) This is how it should be.

:-) ok, i see, thanks for clarification. i will check the hardcoded stuff.

Currently its all a bit quick&dirty, so a solid autodetect is still missing
initialzing the stick is still missing, and for sure many more things.

> Can it listen to both C1 and T1 telegrams?

yes the CUL device can receive C and T telegtrams simultaneosly
C telegrams are starting with _**bY**_ and T telegrams with **_b_**.
so the _checkCULFrame_ function c/should handle this, but i just have
the C telegram meter.

How should a device like stdin:cul (would be cool for testing) work?
start the program on the command line and paste frames into the terminal,
or should i start wmbusmeters and pipe the frames in?
Both didnt work for me, wmbusmeters ignored the data

chester4444, 2019-11-20:
> Check in meter_multical21.cc for format_signature.
> You could add yours there if you like,

Done - after activating DEBUG_PARSER in dvparser.cc i easily found
the right div/vif info and now short telegrams are instantly decoded right.

BTW: you really did a good job on this whole project, **_thumbs up_**

weetmuts, 2019-11-21:
Awesome! When can I see the pull-request? :-)

> How should a device like stdin:cul (would be cool for testing) work?

There is currently fuzzing support for piping raw telegrams into stdin.
I want to add fuzzing of the data sent by different dongles. For example:
im871a, amb8465, rtlwmbus and cul, all have different ways of framing
the telegrams. I want to fuzz the decoding of that frame....even though 
the frame decoding is much simpler than decoding the wmbus telegram.

However, I need to add support for mocking of the dongle, when fuzzing.
So that when running with a mocked im871a, the mock will respond
properly to the link setting and other querying done by wmbusmeters.

I have not yet implemented these mocks, but my intention is to do so.
When those dongle specific mocks are in place, then we should
be able to pipe into stdin:cul.

Note that stdin:rtlwmbus works already, since rtlwmbus does not need
a mock, there is nothing to set, nor query.

chester4444, 2019-11-21:
> Awesome! When can I see the pull-request? :-)

never did things like this before :-), but yes for sure - i need a crash course in git.
In my additions/modifications there are a many things left, to make it bullet proof.

- [x] autodetect is working (with my setup)
- [x] CUL device initialisation is working (c1 mode)
- [x] decoding c1 telegrams is working

BTW: I intend to use _wmbusmeters_ to observe my Multical21 and send 
the infos via MQTT messages to my smarthomeNG based system.
-> drawing useful charts, check for leak/burst info messages etc,

weetmuts, 2019-11-23:
Thank you very much for the pull request! 

When you have gotten the smarthomeNG integration working, feel free to contribyte to the wiki:
https://github.com/weetmuts/wmbusmeterswiki

Where I try to collect HOWTOs for different integrations.

weetmuts, 2020-10-26:
Hi @chester4444 !

I am trying to get my nanoCUL working with wmbusmeters after the major upgrade of the detection code.
However I cannot active C-mode, the source for culfw 1.67 clearly does not support it.
(no WMBUS_CMODE found anywhere, only WMBUS_TMODE, WMBUS_SMODE)

Since you have been running with C-mode, where is the culfw source with support for C-mode? :-)

Many thanks!

chester4444, 2020-10-26:
Hi @weetmuts 

i used this one: [culfw-code-r568-trunk.zip](https://sourceforge.net/code-snapshots/svn/c/cu/culfw/code/culfw-code-r568-trunk.zip) 
(here is the complete sourceforge project [https://sourceforge.net/projects/culfw/](https://sourceforge.net/projects/culfw/) )

Be sure to activate MBUS in the corresponding _boards.h_ with _#define HAS_MBUS_ since it is deactivated because it is using to much resources. I also deactivated unused stuff like ASKSIN, SOMFY, INTERTECHNO, ... to free resources for MBUS.

Good Luck!

weetmuts, 2020-10-26:
Thanks! I managed to compile/install wmbus with C support!
I improved the detection code of the dongle in wmbusmeters as well. 
Please test if you have the time, the code is in the master branch.

Unfortunately I have only received a single telegram when I was 2 meters away from my meter.
Do you have good reception over some distance with the nanoCUL?

closed 2020-11-15 by weetmuts

michipi85, 2020-11-17:
Hello weetmuts, i have a V1.67 nanoCUL868 that I want to use.
this is integrated under 
`lrwxrwxrwx 1 root root 13 Nov 17 21:50 usb-SHA_CUL868-if00 -> ../../ttyACM1`
when I perform a scan the following occurs
`wmbusmeters auto:c1
Started auto cul on /dev/ttyACM1 listening on c1
(cul) setting link mode(s) c1 is not supported for this cul device!`

Is it really not supoorted or am I doing something wrong?

[image]

Thanks for your help

weetmuts, 2020-11-18:
The cul devices must be loaded with the latest wmbus code. Check my question earlier in this thread and chester4444:s answer.

michipi85, 2020-11-25:
Thank you. 
I had to use a different [firmware](https://www.smarthome-agentur.de/wp-content/uploads/2020/11/CUL_V3_MBUS-1.zip). Got a NanoCUL with Atmega32u4.

The water meter was found, but the driver is missing.
`(meter) watermeter: meter detection did not match the selected driver izar! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|1944A511780779194820A1|21170013355F8EDB2D03C6912B1E37|+52331`

Can you help me?

weetmuts, 2020-11-25:
Ok, I pushed a fix. The stupid izar meters just continue to surprise, why should it report itself
as "Breaker (electricity)" when its so easy admit that it is a "Water meter".......

michipi85, 2020-11-25:
Hi Fredrik, 
thanks for the fast support.
i just tried it again, there is still no driver displayed and false type. had wmbusmeters completely new installed
```
Received telegram from: 19790778
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Breaker (electricity) (0x20)
                   ver: 0x48
                device: cul
                  rssi: 0 dBm
                driver: unknown!
```

weetmuts, 2020-11-26:
Do a make clean; make
Then check: ./build/wmbusmeters --version
it should say:
wmbusmeters: 1.0.3-12-g2b39daa
2b39daab39b5744eac5d32950e12e1232e4923f5

michipi85, 2020-11-26:
Perfect!
i had to do "git pull" before.

````
Received telegram from: 19790778
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Breaker (electricity) (0x20)
                   ver: 0x48
                device: cul
                  rssi: 0 dBm
                driver: izar
````