wmbusmeters/wmbusmeters #46: add Engelmann heatmeter sensorstar2

erniberni, 2019-11-19:
Hi, 
I got a Engelmann heat-meter, product is SensoStar2. I got a reading with an optical head, that I stored in a file. The telegram has valid data, I can read the device with libmbus. I get this output from wmbusmters

> ./programme/wmbusmeters/build/wmbusmeters --listento=c1 --debug engelmann.bin
> (wmbusmeters) version: 0.9.18
> (config) using device: engelmann.bin
> (config) number of meters: 0
> (detect) "engelmann.bin" ""
> (detect) is_tty=0 is_stdin=0 is_file=1
> (serial) override with devicefile: engelmann.bin
> (rawtty) on engelmann.bin
> (serialfile) reading from file engelmann.bin
> (config) all possible link modes that the meters might transmit on: none
> (config) explicitly listening to: c1
> (config) listen link modes calculated to be: c1
> No meters configured. Printing id:s of all telegrams heard!
> 
> (config) listen to link modes: c1
> (serial) waiting for stop
> (serial) no more data on fd=3
> (serial) received binary 6893936808007275404810C51400043127000004786BF99F00046D040F632B0415E02000004415E0200000840115E02000000406A02100004406A0210000840106A021000084100600000000C410060000000084110600000000426C5F2C026C7F2C043B00000000143B08010000042B00000000142B39240000025B1800025F17000461380000000223810C01FD17000490280B0000005616
> (serial) no devices working
> (serial) stopping manager
> (serialtty) WOHO? closed engelmann.bin -1
> (wmbus_rawtty) no sensible telegram found, clearing buffer.
> (rawtty) protocol error in message received!
> (rawtty) protocol error ""
> (serial) event loop stopped!
> (serial) closing 0 devices
> 

With libmbus I get this result

> <MBusData>
> 
>     <SlaveInformation>
>         <Id>10484075</Id>
>         <Manufacturer>EFE</Manufacturer>
>         <Version>0</Version>
>         <ProductName>Engelmann SensoStar 2</ProductName>
>         <Medium>Heat: Outlet</Medium>
>         <AccessNumber>51</AccessNumber>
>         <Status>27</Status>
>         <Signature>0000</Signature>
>     </SlaveInformation>
> 
>     <DataRecord id="0">
>         <Function>Instantaneous value</Function>
>         <Unit>Fabrication number</Unit>
>         <Value>10484075</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="1">
>         <Function>Instantaneous value</Function>
>         <Unit>Time Point (time &amp; date)</Unit>
>         <Value>2019-11-03T15:05:00</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="2">
>         <Function>Instantaneous value</Function>
>         <Unit>Volume (1e-1  m^3)</Unit>
>         <Value>8416</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="3">
>         <Function>Instantaneous value</Function>
>         <Unit>Volume (1e-1  m^3)</Unit>
>         <Value>8416</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="4">
>         <Function>Instantaneous value</Function>
>         <Unit>Volume (1e-1  m^3)</Unit>
>         <Value>8416</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="5">
>         <Function>Instantaneous value</Function>
>         <Unit>Energy (kWh)</Unit>
>         <Value>8608</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="6">
>         <Function>Instantaneous value</Function>
>         <Unit>Energy (kWh)</Unit>
>         <Value>8608</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="7">
>         <Function>Instantaneous value</Function>
>         <Unit>Energy (kWh)</Unit>
>         <Value>8608</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="8">
>         <Function>Instantaneous value</Function>
>         <Unit>Energy (kWh)</Unit>
>         <Value>0</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="9">
>         <Function>Instantaneous value</Function>
>         <Unit>Energy (kWh)</Unit>
>         <Value>0</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="10">
>         <Function>Instantaneous value</Function>
>         <Unit>Energy (kWh)</Unit>
>         <Value>0</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="11">
>         <Function>Instantaneous value</Function>
>         <Unit>Time Point (date)</Unit>
>         <Value>2018-12-31</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="12">
>         <Function>Instantaneous value</Function>
>         <Unit>Time Point (date)</Unit>
>         <Value>2019-12-31</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="13">
>         <Function>Instantaneous value</Function>
>         <Unit>Volume flow (m m^3/h)</Unit>
>         <Value>0</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="14">
>         <Function>Maximum value</Function>
>         <Unit>Volume flow (m m^3/h)</Unit>
>         <Value>264</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="15">
>         <Function>Instantaneous value</Function>
>         <Unit>Power (W)</Unit>
>         <Value>0</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="16">
>         <Function>Maximum value</Function>
>         <Unit>Power (W)</Unit>
>         <Value>9273</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="17">
>         <Function>Instantaneous value</Function>
>         <Unit>Flow temperature (deg C)</Unit>
>         <Value>24</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="18">
>         <Function>Instantaneous value</Function>
>         <Unit>Return temperature (deg C)</Unit>
>         <Value>23</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="19">
>         <Function>Instantaneous value</Function>
>         <Unit>Temperature Difference (1e-2  deg C)</Unit>
>         <Value>56</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="20">
>         <Function>Instantaneous value</Function>
>         <Unit>On time (days)</Unit>
>         <Value>3201</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="21">
>         <Function>Instantaneous value</Function>
>         <Unit>Error flags</Unit>
>         <Value>0</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
>     <DataRecord id="22">
>         <Function>Instantaneous value</Function>
>         <Unit>Volume (my m^3)</Unit>
>         <Value>11</Value>
>         <Timestamp>2019-11-03T14:54:18</Timestamp>
>     </DataRecord>
> 
> </MBusData>

Is it possible to add this device?

weetmuts, 2019-11-20:
Yes, it should be possible. The reason why the decoding does not work is because
wmbusmeters right now expects a wmbus frame, ie not an mbus frame.
There are some important differences.

However, I would love to add support for mbus frames as well. 

Do you want wmbusmeters to read directly from the /dev/ttyX that happens to be 
the output from the optical head?

erniberni, 2019-11-20:
Hi, sounds great !
At the moment I just need wmbusmeters to translate the frame from a file. 
But if you want to directly control an optical head via ttyUSBx you also have to implement an optical activation sequence. This is about 2.1s long to send a 0x55. 
But this activation sequence should be optional, because without it the MBus could be read directly.

merbanan, 2020-02-04:
I have pulse meters and power meters connected via mbus-tcp. If you want to implement support for that just send me a message and will give you access to a vm.

weetmuts, 2022-01-13:
Hello @erniberni ! Better late than never... ;-)

Now you can decode mbus framed meters.  Build the latest wmbusmeters on git and run:

./build/wmbusmeters --format=json 6893936808007275404810C51400043127000004786BF99F00046D040F632B0415E02000004415E0200000840115E02000000406A02100004406A0210000840106A021000084100600000000C410060000000084110600000000426C5F2C026C7F2C043B00000000143B08010000042B00000000142B39240000025B1800025F17000461380000000223810C01FD17000490280B0000005616 MySenso auto 10484075 NOKEY
and it will print:

{"media":"heat","meter":"sensostar","name":"Foof","id":"10484075","meter_timestamp":"2019-11-03 15:04","total_kwh":8608,"total_water_m3":841.6,"current_status":"OK","timestamp":"2022-01-13T09:41:51Z"}

There is a lot of data in the telegram not yet added to the json, so we might want to add that data as welll. Run with --debug to see everything.

weetmuts, 2022-04-30:
The support for wired mbus is now sort of working. Still some bugs. The development is tracked in this issue.
https://github.com/weetmuts/wmbusmeters/issues/435

Closing this issue!

closed 2022-04-30 by weetmuts