wmbusmeters/wmbusmeters #48: multical302 volume unit issue

mbursa, 2019-11-20:
There seems to be an issue with volume units in multical302 meter.

I am getting the following data: {"media":"heat","meter":"multical302","name":"67947613","id":"67947613","total_k
wh":305,"total_volume_m3":1830,"current_kw":0,"timestamp":"2019-11-20T18:29:15Z"
}
The display of the device shows 18.30 m3. The value reported by wmbusmeters thus should be divided by 100.0
I have the meter for two weeks now and its counters had started from zero, so volume around 18 m3 is much more sensible than 1830 m3.

weetmuts, 2019-11-26:
Can you provide a few log telegram lines from your log when you have enable --debug or --logtelegrams. They look something like this:

(multical302) log "telegram=|4E44.....00|+0" 

Thanks!

weetmuts, 2019-11-26:
I got some other logs and could add proper support for Mulitcal302. 

Note that the output will be zero until you have received a full telegram.
After the full telegram has been received, then it is possible to decode the compressed
telegrams. I believe every 8th telegram is a full telegram.

Please test and verify that this. If it works properly, then we can add some
hardcoded hashes->formats for multical302 so that it can immediately start
decoding the compressed frames.

mbursa, 2019-11-26:
telegrams from meter no. 1:
|25442D2C4576946730048D20B791A20B20|D37E79E7F194310500000000002A0000FFFF0A0000|+1
|25442D2C4576946730048D20C0C0A20B20|923F79E7F1694E0500000000002A0000FFFF090000|+19
|25442D2C4576946730048D20DF40A30B20|1EE979E7F17E880500000000002C0000FFFF0C0000|+148
|25442D2C4576946730048D20E351A30B20|A87279E7F10CB60500000000002C0000FFFF0B0000|+4
|25442D2C4576946730048D20E452A30B20|A87279E7F10CB60500000000002C0000FFFF0B0000|+21
|25442D2C4576946730048D20E552A30B20|A87279E7F10CB60500000000002C0000FFFF0B0000|+37

telegrams from meter no. 2:
|25442D2C1376946730048D20BB31A70B20|154D79E7F1887F8E0100000000CA0900FFFF010000|+14
|25442D2C1376946730048D20BC31A70B20|154D79E7F1887F8E0100000000CA0900FFFF010000|+30
|25442D2C1376946730048D20BD32A70B20|154D79E7F1887F8E0100000000CA0900FFFF010000|+46
|25442D2C1376946730048D20BE32A70B20|154D79E7F1887F8E0100000000CA0900FFFF010000|+62
|25442D2C1376946730048D20BF40A70B20|154D79E7F1887F8E0100000000CA0900FFFF010000|+78

meter no 1. readings decode to 45 m3 volume, but I believe it should be 0.45 m3
meter no 2. readings decode to 2506 m3 volume, but I believe it should be 25.06 m3
both are in C1 mode

I am not sure what I should do about your second suggestion. There are short and long frames being received. But even if a short frame is received first after daemon has started (without any prior long frame), it is decoded.
Short frame:
telegram=|25442D2C1376946730048D20E9F1A70B20|154D79E7F1887F8E0100000000CA0900FFFF010000|+4
{"total_kwh":398,"total_volume_m3":2506,"current_kw":0,"timestamp":"2019-11-26T19:44:06Z"}
Long frame:
telegram=|2E442D2C1376946730048D20EA00A80B20|E1347803068E010043060000000314CA0900426CFFFF022D010001FF2100|+20
{"total_kwh":398,"total_volume_m3":2506,"current_kw":100,"timestamp":"2019-11-26T19:44:22Z"}

So the difference between short and long frame seem to be in current_kw only. And right, every 8th is long.

weetmuts, 2019-11-26:
Ah, you need to pull the latest version from github, compile and install.
Now if I run wmbusmeters on your long telegram:
telegram=|2E442D2C1376946730048D20EA00A80B20|E1347803068E010043060000000314CA0900426CFFFF022D010001FF2100|+20

I get:
{"media":"heat","meter":"multical302","name":"Heater","id":"67947613",
"total_energy_consumption_kwh":398,
"current_power_consumption_kw":0.1,
"total_volume_m3":25.06,
"at_date":"2127-15-31 00:00",
"total_energy_consumption_at_date_kwh":0,
"current_status":"","timestamp":"2019-11-26T20:08:27Z"}

weetmuts, 2019-11-26:
Ehum ,the at_date looks odd.... have to fix that. :-)

Den tis 26 nov. 2019 kl 21:12 skrev Fredrik Öhrström <oehrstroem@gmail.com>:

weetmuts, 2019-11-26:
Or maybe your meter has an odd date set?

Den tis 26 nov. 2019 kl 21:13 skrev Fredrik Öhrström <oehrstroem@gmail.com>:

mbursa, 2019-11-26:
Recompiled and now indeed, zeros are reported until a long telegram is received and since that apparently correct numbers are reported (with the exception of 'at_date' field).
short: {"media":"heat","meter":"multical302","name":"","id":"67947613","total_energy_consumption_kwh":0,"current_power_consumption_kw":0,"total_volume_m3":0,"at_date":"","total_energy_consumption_at_date_kwh":0,"current_status":"","timestamp":"2019-11-26T20:33:07Z"}
long: {"media":"heat","meter":"multical302","name":"","id":"67947613","total_energy_consumption_kwh":399,"current_power_consumption_kw":0.2,"total_volume_m3":25.07,"at_date":"2127-15-31 00:00","total_energy_consumption_at_date_kwh":0,"current_status":"","timestamp":"2019-11-26T20:33:23Z"}
short: {"media":"heat","meter":"multical302","name":"","id":"67947613","total_energy_consumption_kwh":399,"current_power_consumption_kw":0.2,"total_volume_m3":25.07,"at_date":"2127-15-31 00:00","total_energy_consumption_at_date_kwh":0,"current_status":"","timestamp":"2019-11-26T20:33:39Z"}

I am not sure if correct date is set on the meters, but there is no month 15 in the calendar.

weetmuts, 2019-11-26:
Thank, I have now added the format signature. So if you pull,build,install
it should print proper values immediately for short telegrams.

Perhaps your meter is so new that it has not yet been running for a month?
The target date is set as 0xffff in the telegram, which indicates that it
is set so on purpose.

mbursa, 2019-11-26:
Yes, it works. The meters are running for ~2 weeks only. So I will check the dates later. Perhaps we can close this...
Many thanks for your help and changes!

weetmuts, 2019-12-01:
Yes, lets close now. Open a new issue if there are any problems detected.

closed 2019-12-01 by weetmuts