wmbusmeters/wmbusmeters #51: Decoding PRIOS data failed. Ignoring Telegram

bbradioactive, 2019-11-25:
Hi, on the wmbusmeters.log file I get three lines repeating.

(meter) name: probably not for me since manufacturer differs
(meter) name: probably not for me since media does not match
(izar) Decoding PRIOS data failed. Ignoring telegram. 

However there is a log file created and updated on the meter_readings 

What could these problems be?

weetmuts, 2019-11-25:
It would seem that you have specified a meter number/id that is transmitting and received, but which is not an izar meter. If you can post or email me the --debug log output, then I can take a better look at what kind of meter it is.

weetmuts, 2019-11-25:
The meter_readings file is always created when a telegram is received with the desired number/id, but alas if the telegram fails decoding, then the previous values wil be written into the readings fle.
Values are 0 to start with.

bbradioactive, 2019-11-25:
How can I email you the logs? @weetmuts . I am using a meter made by Diehl water meter called Hydrus meter. Doesnt it use izar?

weetmuts, 2019-11-25:
Use oehrstroem@gmail.com 
I think Hydrus is different from the Izar water meter clipon. .

weetmuts, 2019-11-26:
Thanks for the telegrams, I have now added support for the Hydrus water meter.
Please test!

weetmuts, 2019-12-04:
Tested and working.

closed 2019-12-04 by weetmuts