wmbusmeters/wmbusmeters #53: Multical21 | Only receive total consumption and target consumption

gitlass, 2019-12-06:
I have a IMST871a-dongle and the water meter is a multical21(in sweden). I only receive the total consumption and target consumption values. 

I see this information in the debug:
ignoring telegram since format signature hash 0xdd34 is yet unknown

Am i doing it wrong? The total consumption value I get is correct.

Here comes the debug:
(wmbusmeters) version: 0.9.19-5-gb8bafeb
(config) using device: auto
(config) number of meters: 1
(detect) "auto" ""
(detect) driver: auto
(serialtty) opened /dev/im871a
(serial) received binary 
(im871a) are you there?
(serial /dev/im871a) sent "A5010100"
(serial) received binary A58102004CA3
(serialtty) closed /dev/im871a
(im871a) got crc16 a34c expected a34c
(im871a) on /dev/im871a
(serialtty) opened /dev/im871a
(config) meter multical21 link mode(s): c1
(config) all possible link modes that the meters might transmit on: c1
(config) listen link modes calculated to be: c1
(wmbusmeters) configured "MyTapDapper" "multical21" "63407299" encrypted
(im871a) set link mode 06
(serial /dev/im871a) sent "A5010306000206300101"
(serial) received binary A58104009CF7
(im871a) got crc16 f79c expected f79c
(im871a) set config completed
(im871a) get config
(serial /dev/im871a) sent "A5010500"
(serial) received binary A5810617FF000600B32500191000010001FF0702320001010000E0E21E
(im871a) got crc16 1ee2 expected 1ee2
(im871a) get config completed
(im871a) config: device mode 00
(im871a) config: link mode 06
(im871a) config: wmbus c-field 00
(im871a) config: wmbus mfg id 25b3 (IMS)
(im871a) config: wmbus device id 00101900
(im871a) config: wmbus version 01
(im871a) config: wmbus device type 00
(im871a) config: radio channel 01
(im871a) config: radio power level 07
(im871a) config: radio data rate 02
(im871a) config: radio rx window 32
(im871a) config: auto power saving 00
(im871a) config: auto RSSI attachment 01
(im871a) config: auto rx timestamp attachment 01
(im871a) config: led control 00
(im871a) config: rtc control 00
(config) listen to link modes: c1
(serial) waiting for stop
(serial) received binary A5E20321442D2C997240631B168D20D741524723A80788478C04E0E10912179C5C9D67FAD6DA28000000389F
(im871a) timestamp 000028da
(im871a) rssi 00
(im871a) got crc16 9f38 expected 9f38
(wmbus) received telegram 63407299 C-field=44 M-field=2c2d (KAM) A-field-version=1b A-field-dev-type=16 (Cold water meter) Ci-field=8d (Extended Link Layer II (8 Byte)) CC-field=20 (slow_resp sync) ell header ACC=d7 SN=23475241
(wmbus) frame "21442D2C997240631B168D20D741524723A80788478C04E0E10912179C5C9D67FAD6"
(wmbus) payload "A80788478C04E0E10912179C5C9D67FAD6"
(wmbus) 00: 21 length (33 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 2d2c m-field (2c2d=KAM)
(wmbus) 04: 99724063 a-field-addr (63407299)
(wmbus) 08: 1b a-field-version
(wmbus) 09: 16 a-field-type (Cold water meter)
(wmbus) 0a: 8d ci-field (Extended Link Layer II (8 Byte))
(wmbus) 0b: 20 cc-field (slow_resp sync)
(wmbus) 0c: d7 acc
[... 127 lines cut ...]
(dvparser) match? type=instantaneous vif=FlowTemperature and storagenr=-1
(dvparser) looking for type=unknown vif=ExternalTemperature storagenr=-1
(dvparser) match? type=instantaneous vif=ExternalTemperature and storagenr=-1
(dvparser) match? type=instantaneous vif=ExternalTemperature and storagenr=-1
(dvparser) match? type=instantaneous vif=ExternalTemperature and storagenr=-1
(multical21) 11: b5e2 payload crc
(multical21) 13: 78 frame type (long frame)
(multical21) 14: 02 dif (16 Bit Integer/Binary Instantaneous value)
(multical21) 15: FF vif (Vendor extension)
(multical21) 16: 20 vife (per second)
(multical21) 17: * 0000 info codes (OK)
(multical21) 19: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical21) 1a: 13 vif (Volume l)
(multical21) 1b: * 53CE0800 total consumption (577.107000 m3)
(multical21) 1f: 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
(multical21) 20: 13 vif (Volume l)
(multical21) 21: * FFC70800 target consumption (575.487000 m3)
(multical21) 24442D2C997240631B168D20DA70524723B5E27802FF200000041353CE08004413FFC70800
{"media":"cold water","meter":"multical21","name":"MyTapDapper","id":"63407299","total_m3":577.107,"target_m3":575.487,"max_flow_m3h":0,"flow_temperature_c":127,"external_temperature_c":127,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2019-12-06T11:58:16Z"}
^C(serial) stopping manager
(serial) event loop stopped!
(serial) closing 1 devices
(serial) no devices working
(serial) no devices working
(serialtty) closed /dev/im871a

weetmuts, 2019-12-06:
Your meter does not transmit the temperature nor the max flow. Some Multical21 meters simply do not report the temperature, probably because they do not have this ability. 

I will probably make it so, that the json does not contain the temperature keys, if there is no temperature reported from the meter.

The " format signature hash 0xdd34 is yet unknown" is just information that wmbusmeters
has received a compressed telegram, but it cannot be decoded until you have a received 
a full not-compressed telegram. When the full telegram is received then, wmbusmeters
knows what the hash is supposed to match. So, you might have to wait for a few
compressed telegrams has arrived until you get a long telegram and then start getting real data.

I will improve the logging and the README to explain this.

I do however add hardcoded hashes to wmbusmeters, and I just added yours,
so that wmbusmeters can start decoding the compressed telegrams immediately
if it is a meter configuration type that I have seen before.

I just added your hash as well. :-)

gitlass, 2019-12-09:
Thank you for your help and a great application!

closed 2019-12-09 by gitlass