wmbusmeters/wmbusmeters #54: Support for new IZAR water meter (RC 868 I R4)

ben3k1v, 2019-12-13:
Hi Fredrik,

I just got a new IZAR water meter. The following entries can now be found in the log file.

`(meter) WaterIndoor: probably not for me since manufacturer differs`
`(meter) WaterIndoor: probably not for me since media does not match`

All values look good.
`{"media":"water","meter":"izar","name":"WaterIndoor","id":"12345678","total_m3":0.237,"last_month_total_m3":0,"timestamp":"2019-12-13T09:49:46Z"}`

I will send you the debug log by mail.

Thanks very much!
Ben

weetmuts, 2019-12-13:
Thanks! It seems like Diehl Metering bought Sappel which produced the Izar meters.
Since it actually gives the right values, it still follows the izar proprietary protocol.

But, they change the media type from "oil" to "hot water" which probably makes sense, is this a hot water meter?

And the manufacturer changed from Sappel to Diehl Metering, which also makes sense.

And the telegram is 30% longer, more proprietary data.

It will be easy to add the new(correct?) media type and manufacturer to the izar driver.

ben3k1v, 2019-12-13:
Hi,

 

the water meter is cold, but the IZAR can be used for both types. It is only a module which will be connected on top of any water meter you like.

 

 <https://www.diehl.com/metering/en/portfolio/software-system-components/software-system-components-products/software-system-components-product/izar-rc-i-r4/72505/> https://www.diehl.com/metering/en/portfolio/software-system-components/software-system-components-products/software-system-components-product/izar-rc-i-r4/72505/

 

Thanks!

Ben

weetmuts, 2020-02-02:
Ok, I added the new manufacturer and the new media types 0x15 Hot water
and 0x66 Unknown? Are all your meters in the log water meters? 
Any differences in hot cold water?

ben3k1v, 2020-02-03:
Hi,
Yes, I think all DME meters are water meters. I only have one water meter with the ID (23840778).
No, there are no differences between the cold and the hot meters. This is only a radio module, which can be attached to cold and hot meters.
Ben

ben3k1v, 2020-02-03:
ok, i did the update and installed it. However, I now only receive the data from my water meter, the heat meter is no longer received, it runs now 2h.

`(serial) received ascii C1;1;1;2020-02-03 10:11:59.000;123;120;67880556;0x27442d2c5605886730048d209d001a5b20d2ff62daeecff5a7a57a38fb5cc2a6e12f12dcd7b6<0A>
(rtlwmbus) got full frame
(wmbus) parser expected more data! (904)
(wmbus) parser expected more data! (904)
(wmbus) telegram ignored by all configured meters!`

I sent the log by email.
Thank you
Ben

weetmuts, 2020-02-03:
Can you install my fork of rtl wmbus? Its weetmuts/rtl-wmbus

Its got a minor fix, that previously did not affect wmbusmeters, but now is important since wmbusmeters does check the payload crc.

I have made a pull request to the original rtlwmbus.

ben3k1v, 2020-02-03:
ok, i installed your rtl-wmbus but it is still not working.
full Log by mail.

Ben

weetmuts, 2020-02-06:
Updated wmbusmeters is getting more and more picky on the incoming data telegrams.
The length byte is not adjusted in the telegram sent from rtl_wmbus even though the crc bytes in the data link layer have been removed. Create a pull request for rtl_wmbus https://github.com/xaelsouth/rtl-wmbus/pull/7

But also created a workaround inside wmbusmeters so it is not so picky for rtl_wmbus. :-)

Seems to work. Closing this.

closed 2020-02-06 by weetmuts