wmbusmeters/wmbusmeters #58: BMT

merbanan, 2020-01-16:
Minimum Temperature and Maximum Temperature is most likely incorrect.

```
static char* oms_temp_el[3][4] = {
{"Temperature","Average Temperature 1h","Average Temperature 24h","Error [0][4]", },
{"Maximum Temperature 1h","Maximum Temperature 24h","Error [1][3]","Error [1][4]",},
{"Minimum Temperature 1h","Minimum Temperature 24h","Error [2][3]","Error [2][4]",}
};
```
I use this order. Instantaneous min and max cant exist. You need a duration.

[0][0]
[0][1]
[0][2]
[2][0]
[1][0]
[2][1]
[1][1]

This is the [ff][sn] index ordering I get.

weetmuts, 2020-01-19:
Hmm, I need some more info to understand what the problem is. Help! :-)

merbanan, 2020-01-20:
It was me how was confused. Even though my own confusion I apparently implemented the decoding properly in rtl_433 now that I look at the code.

closed 2020-01-20 by merbanan