wmbusmeters/wmbusmeters #59: Support for multiple connected dongles

mbursa, 2020-01-19:
Instead of using a repeater, I am trying to use two im871a dongles connected to two USB ports - each dongle then goes to a different floor.

The issue is how to run two instances of wmbusmeters each attached to a different device ... a daemon mode cannot be used obviously, so I thought of running two instances from the command line. One way is to have two distinct config folders and run each instance against its own config. There are two disadvantages: 1/ configs are identical except the device name, 2/ I have to put the device name in the config. The latter point is more serious as the device name may be changing (not always .e.g. /dev/ttyUSB0).

My best idea is that wmbusmeters, if used with --useconfig option, could still check the command line for the device name that would override the device in the config file; something like `./wmbusmeters --useconfig=/ /dev/ttyUSB0:im871a`. In this way I could make a udev rule that would start wmbusmeters passing it the device name of the dongle that has been plugged in.

Even more elegantly, this could work in the daemon mode too, but it would require more changes. Apart the device name given on the command line, the pid file should somehow include the device name too, so that multiple daemons could co-exist.

What do you think?

weetmuts, 2020-01-19:
I have been thinking about adding support for multiple concurrent dongles from the config files.
Theoretically wmbusmeters could just detect all of the dongles and start listening on them.
However, that would mean to add the usb detection of proper vendor ids etc to wmbusmeters,
now I rely on a udev rule to do that. Might not be that hard though. But then we need
to add config support for configuring the dongles differently, one might be listening
on C1, another T1 and perhaps n1a so it gets complicated. 

So I have postponed that work until somebody actually needs multiple dongles. :-)

Your suggestion is good, so lets add support for combining:
 --device=/dev/ttyUSB0:im871a 
--listento=c1
with 
--useconfig=/ and the daemon wmbusmetersd.
thereby overriding the settings for device and listento in the config file.

Your use cases would be:
./wmbusmeters --useconfig=/ --device=/dev/ttyUSB0:im871a 
./wmbusmeters --useconfig=/ --device=/dev/ttyUSB1:im871a 

./wmbusmetersd --device=/dev/ttyUSB0:im871a /var/run/wmbusmeters/wmbusmeters_floor1.pid
./wmbusmetersd --device=/dev/ttyUSB1:im871a /var/run/wmbusmeters/wmbusmeters_floor2.pid

mbursa, 2020-01-19:
OK, how can I help with it?

weetmuts, 2020-01-19:
Ok, pushed a fix. This command now works for me:

/build/wmbusmeters --useconfig=tests/config6 --device=simulations/simulation_shell.txt --listento=t1

And it will override the config file device and listento variables.

I have implemented but not tested:
wmbusmetersd --device=simulations/simulation_shell.txt thepidfile

Please test.

merbanan, 2020-01-20:
As a workaround you could mix the telegrams in MQTT or something.

weetmuts, 2020-02-02:
I have updated the code a bit, because I noticed I broke the auto-start of rtlsdr/rtlwmbus dongles with the previous fix.

The new udev rules creates numbered symlinks like: /dev/im87a1_0 /dev/amb8465_3 or /dev/rtlsdr_1

Please test!

weetmuts, 2020-02-02:
Is there perhaps a way to send a message to a daemon from udev action?

The wmbusmeters source code has no problems with running multiple dongles
at the same time, if we want that. The problem is my lack of knowledge for udev -> daemon communication.

weetmuts, 2020-02-17:
After a few problems, I have now added more detailed --debug logging when
wmbusmeters looks for the device. Other than that, it seems to work
nicely having more than one dongle in the computer at the same time.
There will be one daemon per dongle.

Closing this for now.

closed 2020-02-17 by weetmuts