wmbusmeters/wmbusmeters #61: apator162 - new offset again ?

bdkacz, 2020-01-19:
Hi.

Can You look at this logile   [https://www.dropbox.com/s/9bazdi5p378yr96/wmbusmeters.log?dl=0](url) and meter 1759628 ? I start getting "stupid" values where on meter is 82,426

bdkacz, 2020-01-19:
More data to analyze [https://www.dropbox.com/s/e4w1n2hs928u14v/wmbusmeters2.log?dl=0](url)

bdkacz, 2020-01-20:
More data to analyze  [https://www.dropbox.com/s/6cocqp0duul3wv7/wmbusmeters3.log?dl=0](url)

bdkacz, 2020-01-30:
hi. 

Any progress about fix ?

weetmuts, 2020-02-04:
I just pushed a fix. Please test.

bdkacz, 2020-02-04:
Fixed Thank You :)

bdkacz, 2020-02-04:
One fixed,another broken :(  meters 01608150 and 01637014 every second read start decode "stupid" (much higher) values  https://www.dropbox.com/s/9bazdi5p378yr96/wmbusmeters.log?dl=0

bdkacz, 2020-02-04:
Maybe it is possible  to add function to compare between  current and last read and when last read is much lower than current - ignore current ?

weetmuts, 2020-02-05:
Well, we could add support to write a start value and its date in the conf file for the meter.
total_m3=123123
timestamp=2020-02-05

and then the meter driver would try the different known offsets
and compare to the 123123 value and date. Only use the offset
that makes sense.

The alternative is to try to figure out more of the proprietary apator
protocol, which I do not really have time to do. I want to focus
on the wmbus standard compliant meters. :-)

bdkacz, 2020-02-05:
Option with start value looks ok. But there is question what next  ? For example I set start value to 50,and what means "make sense" ?  +10 to start value   ? And what if the meter itself after month have  value 60 ? Manual edit of config file for example once a month and correct start value ? IMHO better is to overwrite the start value by software itself if new value vs old value  is <10 :)

weetmuts, 2020-02-18:
Ok, I have a silly workaround, some more guessing. Pushed to master please try.

bdkacz, 2020-02-18:
Hi. Since update (2 hours) seems working ok, but I will test longer and let you know about results

bdkacz, 2020-02-18:
It seems that so far the patch is working ok

bdkacz, 2020-02-22:
After few days everyhing is ok, so I will close this issue,and open another if I will find  any others problems in future :)

closed 2020-02-22 by bdkacz

alalons, 2021-02-10:
Hello,

Here you say you were able  to collect metrona telemetría star datagrams.
Wich frequency and mode did you use?
I would like to try.

> Hi.
> 
> Can You look at this logile [https://www.dropbox.com/s/9bazdi5p378yr96/wmbusmeters.log?dl=0](url) and meter [1759628](tel:1759628) ? I start getting "stupid" values where on meter is 82,426
> 
> BTW in logfile I found heat meters. I have this heat cost allocators http://www.metrona.pl/telmetric-star.html but I don't know the protocol and/or AES password