wmbusmeters/wmbusmeters #62: Can't capture any WMBUS telegram/frame

superfino, 2020-01-23:
Hello,
I'm trying to use wmbusmeters to control my Diehl Hydrus watermeter which is  installed in a sinkhole and it's not easy to take the readings. I've a CUL which is able to receive t1 type with my fhem home automation system. Unfortunately fhem can't interpret it. So i've setup a new Debian sytem with wmbusmeters and my CUL but can't capture any data. Normally 
wmbusmeters --listento=t1 /dev/ttyUSB0:cul should capture any data but nothing happens. I've made many attemts to get it running but some messages i cant understand.
"ot a valid meter id nor a valid meter match expression "00260003
"ot a valid meter type "hydrus
Nothing helpful in the wmbusmeters.log file. I think there is something important that i miss but the to messages are very annoying since the Hydrus meter seems to be supported.  Need some ideas how to continue. thx in advance...  Jo

superfino, 2020-01-26:
Some more detailed information i gathered with --debug --verbose. The console shows the following:

 $ swmbusmeters --listento=t1 --debug --verbose auto
(wmbusmeters) version: 0.9.22-2-g468ba75
(config) using device: auto
(config) number of meters: 0
(detect) "auto" ""
(detect) driver: auto
(serialtty) opened /dev/amb8465
(serial) received binary
(amb8465) are you there?
(serial /dev/amb8465) sent "FF0B00F4"
(serial) received binary
(serialtty) closed /dev/amb8465
(serialtty) opened /dev/ttyUSB0
(serial /dev/ttyUSB0) sent "2D0D0A"
(serial) received binary 3F202820697320756E6B6E6F776E2920557365206F6E65206F66204220622043206520462047204B206C204D20522054207420562057205820780D0A
(cul) are you there?
(serial /dev/ttyUSB0) sent "560A0D"
(serial) received binary 5620312E3637206E616E6F43554C3836380D0A
CUL answered: V 1.67 nanoCUL868
(serialtty) closed /dev/ttyUSB0
(cul) on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(config) all possible link modes that the meters might transmit on: none

The Logfiles shows me:

(wmbusmeters) shutting down
(wmbusmeters) logging started 2020-01-26 13:04:26
No wmbus device found! Exiting!
(wmbusmeters) logging started 2020-01-26 13:04:29
Device /dev/ttyUSB0 is already in use and locked.
Device /dev/ttyUSB0 is already in use and locked.
Device /dev/ttyUSB0 is already in use and locked.
No wmbus device found! Exiting!
(wmbusmeters) logging started 2020-01-26 13:04:31
Device /dev/ttyUSB0 is already in use and locked.
Device /dev/ttyUSB0 is already in use and locked.
Device /dev/ttyUSB0 is already in use and locked.
No wmbus device found! Exiting!

weetmuts, 2020-01-26:
If you have run "make install" then probably the daemon has taken the USB device already.

Do:
sudo systemctl stop wmbusmeters.service
to stop it. Then retry.

weetmuts, 2020-01-26:
Can you post here what dmesg prints after you insert the CUL dongle?
It could also be that auto picks up the wrong ttyUSB device.

superfino, 2020-01-26:
Thank you for your replay,
with the explicit device statement the device isn't locked anymore and the logfile is complete emty. but unfortunately no data is captured although everything seems to be O.K. now. dmesg shows:
dmesg | grep -i usb
[    0.141500] usbcore: registered new interface driver usbfs
[    0.141643] usbcore: registered new interface driver hub
[    0.141864] usbcore: registered new device driver usb
[    0.432865] usbcore: registered new interface driver smsc95xx
[    0.466791] dwc_otg 20980000.usb: base=(ptrval)
[    0.920664] dwc_otg 20980000.usb: DWC OTG Controller
[    0.923300] dwc_otg 20980000.usb: new USB bus registered, assigned bus number 1
[    0.926077] dwc_otg 20980000.usb: irq 56, io mem 0x00000000
[    0.934198] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 4.19
[    0.939305] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.942039] usb usb1: Product: DWC OTG Controller
[    0.944793] usb usb1: Manufacturer: Linux 4.19.66+ dwc_otg_hcd
[    0.947556] usb usb1: SerialNumber: 20980000.usb
[    0.951349] hub 1-0:1.0: USB hub found
[    0.958438] usbcore: registered new interface driver usb-storage
[    0.992100] usbcore: registered new interface driver usbhid
[    0.995094] usbhid: USB HID core driver
[    2.288423] usb 1-1: new high-speed USB device number 2 using dwc_otg
[    2.573954] usb 1-1: New USB device found, idVendor=0424, idProduct=9512, bcdDevice= 2.00
[    2.588649] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    2.600899] hub 1-1:1.0: USB hub found
[    2.933479] usb 1-1.1: new high-speed USB device number 3 using dwc_otg
[    3.073957] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00, bcdDevice= 2.00
[    3.089123] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    3.209646] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-20980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:e1:fd:fd
[    3.333484] usb 1-1.2: new high-speed USB device number 4 using dwc_otg
[    3.475580] usb 1-1.2: New USB device found, idVendor=7392, idProduct=7811, bcdDevice= 2.00
[    3.490884] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    3.501804] usb 1-1.2: Product: 802.11n WLAN Adapter
[    3.510223] usb 1-1.2: Manufacturer: Realtek
[    3.517883] usb 1-1.2: SerialNumber: 00e04c000001
[    3.628572] usb 1-1.3: new full-speed USB device number 5 using dwc_otg
[    3.788255] usb 1-1.3: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[    3.802917] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    3.813530] usb 1-1.3: Product: NANO CUL
[    3.820626] usb 1-1.3: Manufacturer: SHK
[    3.827598] usb 1-1.3: SerialNumber: 868
[   12.801590] usbcore: registered new interface driver usbserial_generic
[   12.801731] usbserial: USB Serial support registered for generic
[   12.992953] usbcore: registered new interface driver ftdi_sio
[   12.993109] usbserial: USB Serial support registered for FTDI USB Serial Device
[   13.463655] ftdi_sio 1-1.3:1.0: FTDI USB Serial Device converter detected
[   13.463988] usb 1-1.3: Detected FT232RL
[   13.498250] usb 1-1.3: FTDI USB Serial Device converter now attached to ttyUSB0
[   19.945519] usbcore: registered new interface driver rtl8192cu
[   54.069424] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[   54.173580] usb 1-1.3: reset full-speed USB device number 5 using dwc_otg
[   54.315095] ftdi_sio 1-1.3:1.0: FTDI USB Serial Device converter detected
[   54.318433] usb 1-1.3: Detected FT232RL
[   54.319949] usb 1-1.3: FTDI USB Serial Device converter now attached to ttyUSB0
[   69.163601] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[   69.263512] usb 1-1.3: reset full-speed USB device number 5 using dwc_otg
[   69.406240] ftdi_sio 1-1.3:1.0: FTDI USB Serial Device converter detected
[   69.409462] usb 1-1.3: Detected FT232RL
[   69.410997] usb 1-1.3: FTDI USB Serial Device converter now attached to ttyUSB0

and that's what my console shows:
wmbusmeters --listento=t1 --debug --verbose /dev/ttyUSB0
(wmbusmeters) version: 0.9.22-2-g468ba75
(config) using device: /dev/ttyUSB0
(config) number of meters: 0
(detect) "/dev/ttyUSB0" ""
(detect) is_tty=1 is_stdin=0 is_file=0
(serialtty) opened /dev/ttyUSB0
(serial) received binary
(amb8465) are you there?
(serial /dev/ttyUSB0) sent "FF0B00F4"
(serial) received binary
(serialtty) closed /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(serial) received binary
(im871a) are you there?
(serial /dev/ttyUSB0) sent "A5010100"
(serial) received binary
(serialtty) closed /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(serial /dev/ttyUSB0) sent "2D0D0A"
(serial) received binary 3F2028F8F8787A20697320756E6B6E6F776E2920557365206F6E65206F66204220622043206520462047204B206C204D20522054207420562057205820780D0A
(cul) are you there?
(serial /dev/ttyUSB0) sent "560A0D"
(serial) received binary 5620312E3637206E616E6F43554C3836380D0A
CUL answered: V 1.67 nanoCUL868
(serialtty) closed /dev/ttyUSB0
(cul) on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(config) all possible link modes that the meters might transmit on: none
(config) explicitly listening to: t1
(config) listen link modes calculated to be: t1
No meters configured. Printing id:s of all telegrams heard!

(cul) setLinkModes
(serial /dev/ttyUSB0) sent "6272630A0D"
(serial) received binary 4F46460D0A
(cul) error in received message.
(serial /dev/ttyUSB0) sent "5830310A0D"
(config) listen to link modes: any
(serial) waiting for stop

weetmuts, 2020-01-26:
Now it looks good. Can you get really close to the meter in the sinkhole to check if the radio gets out of it? And post a --debug log when it receives a telegram.

weetmuts, 2020-01-26:
Though "(cul) error in received message." looks a bit odd.

superfino, 2020-01-26:
This is a short snipped from my fhem log, many t1 telegrams. But nothing with wmbusmeters.

2020-01-26 18:12:37 Global global UNDEFINED WMBUS_TCH_45035637_118_240 WMBUS b294468503756034576F0DF3CA080DE2691270108912702004D01040523CA153A811A6BCC1B6BCC180FCE000FE58013BB84::-86.5
2020-01-26 18:12:47 Global global UNDEFINED WMBUS_TCH_45035636_118_240 WMBUS b294468503656034576F01EB4A080DE2698270008982700004D000105723A1887CE007CB2008DC3009ED4000FE580125B81::-62.5
2020-01-26 18:12:51 Global global UNDEFINED WMBUS_TCH_45035632_118_240 WMBUS b294468503256034576F05F3BA000DE2636280108362800004D000005687B006BA1007CB2008DC3009ED4000FE50096BA80::-73.5
2020-01-26 18:12:52 Global global UNDEFINED WMBUS_TCH_45035626_118_240 WMBUS b294468502656034576F025EDA000DE2637280308372800004D0000057079006BA1007CB2008DC3009ED4000FE50096BA81::-74
2020-01-26 18:13:17 Global global UNDEFINED WMBUS_TCH_45035635_118_240 WMBUS b294468503556034576F06149A000DE2637280008372800004D0202051E282AF4BF1BF4BF008DC3009ED4000FE500E52B80::-50
2020-01-26 18:14:15 Global global UNDEFINED WMBUS_TCH_45035634_118_240 WMBUS b294468503456034576F0A0C1A000DE2637280008372800004D000205E819006BA1007CB2008DC3009ED4000FE50096BA80::-85.5
2020-01-26 18:14:39 Global global UNDEFINED WMBUS_TCH_45035627_118_240 WMBUS b294468502756034576F0E465A000DE2639280208392800004D000105DE00006BA1007CB2008DC3009ED4000FE50096BA81::-79
2020-01-26 18:14:53 Global global UNDEFINED WMBUS_TCH_45035633_118_240 WMBUS b294468503356034576F09EB3A000DE2627280108272800004D0000057A41006BA1007CB2008DC3009ED4000FE580ECD680::-77
2020-01-26 18:15:09 Global global UNDEFINED WMBUS_TCH_45035630_118_240 WMBUS b294468503056034576F0E14EA000DE2638280408382800004D000105284E006BA1007CB2008DC3009ED4000FE50096BA80::-70
2020-01-26 18:15:37 Global global UNDEFINED WMBUS_TCH_45035629_118_240 WMBUS b294468502956034576F09881A000DE2635280008352800004D000005E284006BA1007CB2008DC3009ED4000FE50096BA84::-83

So i'm clueless what went wrong here. Is there any chance to further debug the problem?

superfino, 2020-01-27:
Hello,  this is the requested --debug log fom overnight. For me it's difficult to interpret the log entries but mybe it's helpful for you.

(wmbusmeters) version: 0.9.22-2-g468ba75
(config) using device: /dev/ttyUSB0
(config) number of meters: 0
(detect) "/dev/ttyUSB0" ""
(detect) is_tty=1 is_stdin=0 is_file=0
(serialtty) opened /dev/ttyUSB0
(serial) received binary 
(amb8465) are you there?
(serial /dev/ttyUSB0) sent "FF0B00F4"
(serial) received binary 
(serialtty) closed /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(serial) received binary 
(im871a) are you there?
(serial /dev/ttyUSB0) sent "A5010100"
(serial) received binary 
(serialtty) closed /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(serial /dev/ttyUSB0) sent "2D0D0A"
(serial) received binary 3F202820697320756E6B6E6F776E2920557365206F6E65206F66204220622043206520462047204B206C204D20522054207420562057205820780D0A
(cul) are you there?
(serial /dev/ttyUSB0) sent "560A0D"
(serial) received binary 5620312E3637206E616E6F43554C3836380D0A
CUL answered: V 1.67 nanoCUL868
(serialtty) closed /dev/ttyUSB0
(cul) on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(config) all possible link modes that the meters might transmit on: none
(config) explicitly listening to: t1
(config) listen link modes calculated to be: t1
No meters configured. Printing id:s of all telegrams heard!

(cul) setLinkModes
(serial /dev/ttyUSB0) sent "6272630A0D"
(serial) received binary 4F46460D0A
(cul) error in received message.
(serial /dev/ttyUSB0) sent "5830310A0D"
(config) listen to link modes: any
(serial) waiting for stop
(serial) received binary 46314231423032333930340D0A
(cul) error in received message.
(serial) received binary 46314231423032333930340D0A
(cul) error in received message.
(serial) received binary 46314231423032333930340D0A
(cul) error in received message.
(serial) received binary 46314231423032
(serial) received binary 333930340D0A
(cul) error in received message.
(serial) stopping manager
(serial) event loop stopped!
(serial) closing 1 devices
(serial) no devices working
(serial) no devices working
(serialtty) closed /dev/ttyUSB0

weetmuts, 2020-01-29:
It seems like your CUL device response F1B1B023904 (hexified again in the log as 46314231423032333930340D0A) is not understood by wmbus_cul.cc 

I will take a look next week, when I have more time.
Or if you like, you can try to debug the CUL driver for that particular message.

weetmuts, 2020-01-29:
I thinkt @chester4444 added the cul support. Do you have any idea Chester?

chester4444, 2020-01-29:
Since i only have a C1-mode meter device here to play with, 
i had no idea how to implement T1-mode.

a quick look into in wmbus_cul.cc:
` if (data[1] != 'Y') {`
`    verbose("(cul) T1 telegrams currently not supported\n");`
`    return ErrorInFrame;`
`  }`

chester4444, 2020-01-29:
`(serial /dev/ttyUSB0) sent "6272630A0D"`
6272630A0D translated from HEX: _**brc\n\r**_
here wmbusmeters wants switch to C-mode

`(serial) received binary 4F46460D0A`
nanoCUL answers with **_OFF\n\r_**
this means your nanoCUL isnt compiled for C-mode (obviously)

(serial /dev/ttyUSB0) sent "5830310A0D"
wmbusmeters tries to activate receiver with _**X01\n\r**_

Just to explain, why it can't work out of the box

superfino, 2020-01-29:
Indeed, my Cul device is compiled for T1 mode only, that's why i'm use "--listento=t1 ".  The init string for my CUL should be "x21" "brt". So it seems to me that the CUL init string is hardcoded and "--listento=t1 " has no effect?

weetmuts, 2020-01-29:
It seems like we now can improve the cul driver. 

First we need to detect which modes are supported by the cul dongle.
Someone has a query response for this?
This should be done when we initialize the connection to the dongle.

Then the setting code should detect if it can set the desired mode 
based on the available mode(s) in the dongle.

superfino, 2020-01-29:
Regarding my CUL, it supports T-mode to be initialized wit "brt" and S-mode to be initalized with "brs".
"X21" mentioned in my last update was bullshit, my foult. Any further data neede?

chester4444, 2020-01-29:
As far is i can see in the [cul reference](culfw.de/commandref.html), there is no query command for supported protocols.

but we could check the response for the asked mode:

- **_brs_** should give **_SMODE_**
- **_brt_** should give **_TMODE_**
- **_brc_** should give **_CMODE_**

if the protocol isnt supported it results in **_OFF_**

**_X21_** means that received packets include a trailing RSSI info,
which we would ignore anyway, so _**X01**_ is a better choice 

I will/can help improving the code. Where should we start?

superfino, 2020-01-29:
Yes, i can confirm chester4444 response assumption, my last update was on the fly, sorry.

weetmuts, 2020-02-02:
I created a branch ImproveCUL with a few additions to the CUL driver.
It should now be able to set brc,brt or brs and recognize the answer CMODE,TMODE or SMODE.
If OFF or anything else is received it will print an error and stop.

Please test. I do not have a CUL device myself.

What is the expected response from X01?

superfino, 2020-02-02:
Hi, just made a quick shot this evening after coming home late with no success. The log shows some strange results but i'll have a deeper lock at this tomorrow. BTW 'X01' gives no answer.

(wmbusmeters) version: 0.9.22-2-g468ba75
(config) using device: /dev/ttyUSB0
(config) number of meters: 0
(detect) "/dev/ttyUSB0" ""
(detect) is_tty=1 is_stdin=0 is_file=0
(serialtty) opened /dev/ttyUSB0
(serial) received binary 
(amb8465) are you there?
(serial /dev/ttyUSB0) sent "FF0B00F4"
(serial) received binary 
(serialtty) closed /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(serial) received binary 
(im871a) are you there?
(serial /dev/ttyUSB0) sent "A5010100"
(serial) received binary 
(serialtty) closed /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(serial /dev/ttyUSB0) sent "2D0D0A"
(serial) received binary 
(serialtty) closed /dev/ttyUSB0
No wmbus device found! Exiting!

weetmuts, 2020-02-03:
Hmm, when I build the branch I get the version number 0.9.23-1-g9ed100c
Please try the device name to force the cul device: /dev/ttyUSB0:cul

superfino, 2020-02-03:
Hello, 
started from scratch today 
1. git clone -b ImproveCUL https://github.com/weetmuts/wmbusmeters.git
2. git branch -l   > *ImproveCUL
3. make & make install
Now running "wmbusmeters --listento=t1 --debug --logfile=/home/pi/mylog --logtelegrams --verbose /dev/ttyUSB0:cul"     end immediately with the followig result:

(wmbusmeters) version: 0.9.23-1-g9ed100c
(config) using device: /dev/ttyUSB0
(config) with: cul
(config) number of meters: 0
(detect) "/dev/ttyUSB0" "cul"
(detect) is_tty=1 is_stdin=0 is_file=0
(cul) on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(config) all possible link modes that the meters might transmit on: none
(config) explicitly listening to: t1
(config) listen link modes calculated to be: t1
No meters configured. Printing id:s of all telegrams heard!

(serial /dev/ttyUSB0) sent "6272740A0D"
(cul) set link mode t
(serial /dev/ttyUSB0) sent "6272740A0D"
(serial) received binary 
(cul) received ""
(cul) setting link mode(s) t1 is not supported for this cul device!
(serial) stopping manager

It seems wmbusmeters is missig the response TMODE and stops working. I checked again with Putty
and my CUL responds to brt with TMODE.

weetmuts, 2020-02-03:
Ah, yes. I was a bit too quick in writing the code. I have now pushed a better version.
Please test and debug. I am programming a bit in the dark here. :-)

superfino, 2020-02-03:
Bingo!!! With your reworked code i can now receive T1 telegrams with my CUL. Special thanks for your nice work. In this short timeframe i got only TCH M=5068 telegrams. These are heatmeters. I have not yet figured out when my hydrus watermeter is 'On Air' because they are not sending permanetly. Now i need a bit patience to wait for the hydrus telegram. I'll keep you informed.

weetmuts, 2020-02-04:
Cool, I merged the fix into master and deleted the branch.

chester4444, 2020-02-04:
Hi, great job, but i am testing here c-mode and i had some troubles.
I changed a few things - finally it's now working for me.

- i removed in [wmbus_cull.cc](https://github.com/weetmuts/wmbusmeters/blob/590a94e870e8b247b9202abca0fc175278faf10d/src/wmbus_cul.cc#L303)
`    if (data.size() < 10) return ErrorInFrame;`
because the expected answer "CMODE" is shorter than 10 and was therefore discarded.

- in _setLinkModes()_
the _**brc**_ command is sent twice, which should be sent only once, i removed it.
- in _setLinkModes()_ i changed this sequence 
`    bool sent = serial()->send(msg);`
`    sent_command_ = string(&msg[0], &msg[3]);`
`    received_response_ = "";`
so that calling send(msg) is at the end, because at least sent_command_ must be set prior.

weetmuts, 2020-02-04:
Thanks @chester4444 ! Fix pushed. Please check master.

chester4444, 2020-02-04:
almost perfect - now adding the mutex initialization in the WMBusCUL constructor and all should be fine
`sem_init(&command_wait_, 0, 0);`

weetmuts, 2020-02-04:
Thanks @chester4444 ! The sem_init is added. :-)

weetmuts, 2020-02-06:
Please test the 0.9.24 release and confirm that it looks ok for you @chester4444  and  @superfino.

Then we can close this.

superfino, 2020-02-08:
Hi, got version: 0.9.25-1-gaf98f36 yesterday and it seems eyerything is O.K. regarding T1 with my CUL. I've not been able up to now to fetch anything else than Techem telegram's since my Hydrus Meter has not yet started to send telegrams, but i'm in good hope that this will work too. Again thank you for your affords and best wishes....  Jo

chester4444, 2020-02-08:
Same here, (_0.9.25-1-gaf98f36_) is working like charm with my nanoCUL (in c1 mode).
Thanks, and yes ... 
> Then we can close this.

weetmuts, 2020-02-17:
The wmbus_cul.cc now properly supports both C and T modes. 
S is yet unknown.... have not seen such a telegram.

closed 2020-02-17 by weetmuts