wmbusmeters/wmbusmeters #63: Trying to get docker container launced

MTrab, 2020-01-26:
Docker documentation is missing, and when I try to spin up a Docker container it just quits without anything logged.

What env. vars are neccesary?
Where to define which dongle is used and what device to use?
Where to define decryption key?

weetmuts, 2020-01-28:
Good question, I am not a docker expert. :-) @BIBOLV can you help me here?

weetmuts, 2020-01-28:
# wmbusmeters
The program receives and decodes C1,T1 or S1 telegrams
(using the wireless mbus protocol) to acquire
utility meter readings. The readings can then be published using
MQTT, curled to a REST api, inserted into a database or stored in a log file.

[FAQ/WIKI/MANUAL pages](https://github.com/weetmuts/wmbusmeters)

## Running the wmbusmeters container
### Pre-requisite

Before running the command that creates the wmbusmeters docker container, you must add udev rules to create usb device symlink in order to have persistent link to device across host restarts and reconnect usb device:
```
cat <<EOF > /etc/udev/rules.d/99-wmbus-usb-serial.rules
SUBSYSTEM=="tty", ATTRS{idVendor}=="10c4", ATTRS{idProduct}=="ea60", SYMLINK+="im871a", MODE="0660", GROUP="docker"
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", SYMLINK+="amb8465", MODE="0660", GROUP="docker"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0bda", ATTRS{idProduct}=="2838", SYMLINK+="rtlsdr", MODE="0660", GROUP="docker"
SUBSYSTEM=="usb", ATTRS{idVendor}=="2047", ATTRS{idProduct}=="0863", SYMLINK+="rfmrx2", MODE="0660", GROUP="docker"
EOF
```
### Command line container download and run

```
docker run -d \
    --name=wmbusmeters \
    --restart=always \
    -v /etc/localtime:/etc/localtime:ro \
    -v /opt/wmbusmeters:/wmbusmeters_data \
    --device=/dev/im871a \
    weetmuts/wmbusmeters 
```

### Command line Options
| Parameter | Description |
| ------------ | ------------- |
| --name=wmbusmeters | Names the container "wmbusmeters". |
| --restart=always | Start the container when Docker starts (i.e. on boot/reboot). |
| -v /etc/localtime:/etc/localtime:ro | Ensure the container has the correct local time (alternatively, use the TZ environment variable, see below). |
| -v /opt/wmbusmeters:/wmbusmeters_data | Bind mount /opt/wmbusmeters (or the directory of your choice) into the container for persistent storage. |
| --device=/dev/im871a | Pass the device at ttyUSB0 into the container for use by wmbusmeters (you may need to change path to your device ). |

If you are using docker-compose.yml file you may also copy/paste the following into your existing docker-compose.yml, modifying the options as required (omit the version and services lines as your docker-compose.yml will already contain these).
```
version: "2"
services:
  wmbusmeters:
    image: weetmuts/wmbusmeters
    container_name: wmbusmeters
    restart: always
    volumes:
      - /opt/wmbusmeters:/wmbusmeters_data
      - /etc/localtime:/etc/localtime:ro
    devices:
      - /dev/im871a

```
Then, you can do docker-compose pull to pull the latest weetmuts/wmbusmeters image, docker-compose up -d to start the wmbusmetersz container service, and docker-compose down to stop the wmbusmeters service and delete the container. Note that these commands will also pull, start, and stop any other services defined in docker-compose.yml.

### Issues / Contributing

Please raise any issues with this container at its [GitHub repo](https://github.com/weetmuts/wmbusmeters)

weetmuts, 2020-01-28:
I did set the docker readme to the above text, however it reverted back to the default github README.

BIBOLV, 2020-01-28:
@MTrab please refer to readme @weetmuts posted above. On successful image start you should see config file in persistent storage and create meter file with encryption key and other parameters in wmbusmeters.d directory as described in generic wmbusmeters README.

MTrab, 2020-01-28:
So far so good.
Now I just have this in the log:

`You are not in the same group as the device CUL`

MTrab, 2020-01-28:
Think I got it now - forgot to set chmod :) But no data.

BIBOLV, 2020-01-28:
I suggest to check couple of things:
1. Does device reports currently (I have devices that reports only in day-time)
2. Remove any configs from wmbusmeters.d directory and restart docker image, then all received telegrams should be reported in log file.
3. Recheck config that all is correct, like link mode, etc. Try rising log level, maybe it will give some hints.
If all fails, try compiling wmbusmeters on host device and check that it works.

MTrab, 2020-01-29:
Perhaps it's a range issue.
Nothing have been logged over the night.

How do I raise loglevel in the Docker? I see the loglevel value in the conf file, but I see nothing in documentation on what valid values are.
Setting it to debug and verbose is doing nothing for the logging - no new lines are added.

BIBOLV, 2020-01-29:
One more question - did you use udev rules for device alias creation or did you pass /dev/ttyUSB0 directly to docker image?

MTrab, 2020-01-29:
Right now I'm using the udev rules.
Have tried using both /dev/ttyUSB0 and currently using /dev/amb8465

MTrab, 2020-01-29:
Apparently I did something wrong. Heres the initial debug output when restarting the docker container.

```
(wmbusmeters) version: 0.9.21-1-geed4b17
(config) store meter files in: "/wmbusmeters_data/logs/meter_readings"
(config) using device: auto
(config) number of meters: 0
(detect) "auto" ""
(detect) driver: auto
(serialtty) opened /dev/amb8465
(serial) received binary 
(amb8465) are you there?
(serial /dev/amb8465) sent "FF0B00F4"
(serial) received binary FF8B042701C8F967
(serialtty) closed /dev/amb8465
(amb8465) on /dev/amb8465
(serialtty) opened /dev/amb8465
(config) all possible link modes that the meters might transmit on: none
(config) explicitly listening to: c1
(config) listen link modes calculated to be: c1
No meters configured. Printing id:s of all telegrams heard!

(amb8465) set link mode 0e
(serial /dev/amb8465) sent "FF04010EF4000000"
(serial) received binary FF840100
(serial) received binary 7A
(amb8465) rssi 122
(amb8465) set link mode completed
(amb8465) set link mode response "0100"
(amb8465) get config
(serial /dev/amb8465) sent "FF0A02008077"
(serial) received binary FF8A7A0078
(serial) received binary 0080710200000000FFFFFA00FFFFFF
(serial) received binary FFFFFFFFFFFFFFFFFFFFFFFFFF003200
(serial) received binary 021400FFFFFFFFFF010004000000FF
(serial) received binary FFFF01440000000000000000FFFF01
(serial) received binary 060100FFFFFFFFFF000C0000FFFFFFFF
(serial) received binary FFFFFF0000FFFFFFFFFFFFFFFFFFFF
(serial) received binary FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
(serial) received binary FFFFFFFFFFFFFFFFFFFFFFFF10
(amb8465) rssi 16
(amb8465) get config completed
(amb8465) get config response "7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF01060100FFFFFFFFFF000C0000FFFFFFFFFFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
(amb8465) config: uart 78
(amb8465) config: radio Channel ff
(amb8465) config: rssi enabled ff
(amb8465) config: mode Preselect 00
(config) listen to link modes: c1
(serial) waiting for stop
```

weetmuts, 2020-01-29:
It looks like it is working now! Have you received any telegrams as well?

BIBOLV, 2020-01-29:
If udev rules are not used and device is linked to container directly then in config `device=auto` should be changed to `device=/dev/ttyUSB0:amb8465` for example.
But log you posted looks ok, so if you still do not receive anything I would suggest to compile wmbusmeters on host and check if it works then.

MTrab, 2020-01-29:
I don't have the oppurtunity to compile on this exact host (Synology NAS) - but will try a Linux hos later to day, which is closer to the water meter.

And no, still nothing received from the Multical 21 meter.

weetmuts, 2020-01-29:
Well, the communication with the Amber dongle definitely works.

(amb8465) are you there?
(serial /dev/amb8465) sent "FF0B00F4"
(serial) received binary FF8B042701C8F967
etc etc

so if you cannot hear any telegrams, then its not due to the docker configuration but the lack of radio waves. :-)

MTrab, 2020-01-29:
The last issue was down to range of the water meter - moved the dongle closer to the meter and now I have data.

closed 2020-01-29 by MTrab