wmbusmeters/wmbusmeters #64: Support for WMB13U-868 Dongle

KeenanGeorge03, 2020-01-28:
Hi i have a WMB13U-868 Dongle & would like to understand the procedure to support this dongle.

merbanan, 2020-01-28:
@KeenanGeorge03 can you supply usb info about this device? And what log does your kernel give when you plug it in? Do you have some software for it?

merbanan, 2020-01-28:
It looks like It should be possible to use the generic serial interface.

merbanan, 2020-01-28:
Where can you but this dongle? And can you post a picture of its internals?

KeenanGeorge03, 2020-01-28:
Hi @merbanan thanks for your help when i plug the device into the raspberry pi i get the following:

Jan 28 21:56:16 raspberrypi kernel: [100514.786190] usb 1-1.1: New USB device found, idVendor=067b, idProduct=2303, bcdDevice= 4.00
Jan 28 21:56:16 raspberrypi kernel: [100514.786206] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Jan 28 21:56:16 raspberrypi kernel: [100514.786218] usb 1-1.1: Product: USB-Serial Controller D
Jan 28 21:56:16 raspberrypi kernel: [100514.786230] usb 1-1.1: Manufacturer: Prolific Technology Inc.
Jan 28 21:56:16 raspberrypi kernel: [100514.788895] pl2303 1-1.1:1.0: pl2303 converter detected
Jan 28 21:56:16 raspberrypi kernel: [100514.797990] usb 1-1.1: pl2303 converter now attached to ttyUSB0
Jan 28 21:56:16 raspberrypi mtp-probe: checking bus 1, device 11: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1"
Jan 28 21:56:16 raspberrypi mtp-probe: bus: 1, device: 11 was not an MTP device
Jan 28 21:56:16 raspberrypi mtp-probe: checking bus 1, device 11: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1"
Jan 28 21:56:16 raspberrypi mtp-probe: bus: 1, device: 11 was not an MTP device

Please find datasheet link below

http://static.abstore.pl/design/accounts/soyter/img/friendcom/wmb13u-868_usb-adapter_datasheet.pdf

http://www.friendcom.cn/solutions/list_17_cid_11.html

Link to buy

https://www.alibaba.com/product-detail/WMB13U-868-Wireless-MBUS-USB-Adapter_60603649412.html

[image]

BIBOLV, 2020-01-28:
[http://www.friendcom.cn/uploadfiles/datasheet/WMbus/FC-704  User Manua.pdf](This) could help for serial communication implementation.

merbanan, 2020-01-28:
@BIBOLV that helped. The default baud rate is 19200.

Try with ./wmbusmeters --debug --listento=t1 /dev/ttyUSB0:19200 and see if you can receive some telegrams.

KeenanGeorge03, 2020-01-29:
Thanks its working.

weetmuts, 2020-01-29:
@KeenanGeorge03 That is great! Can you please collect a --debug log of it starting up and capturing a telegram and send it to me? I would like to officially add it to wmbusmeters.

KeenanGeorge03, 2020-01-29:
Hi @weetmuts 

Please let me know if the below helps 

pi@raspberrypi:/etc/wmbusmeters.d $ wmbusmeters --debug --listento=c1 /dev/ttyUSB0:19200 58465287 multical21:c1 58465287 EE7022CD0C46BBA0170C7FA483BC2234
(wmbusmeters) version: 0.9.20-3-g7fa9f10
(config) using device: /dev/ttyUSB0
(config) with: 19200
(config) number of meters: 1
(detect) "/dev/ttyUSB0" "19200"
(detect) is_tty=1 is_stdin=0 is_file=0
(rawtty) on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(config) meter multical21 link mode(s): c1
(config) all possible link modes that the meters might transmit on: c1
(config) explicitly listening to: c1
(config) listen link modes calculated to be: c1
(wmbusmeters) configured "58465287" "multical21" "58465287" encrypted
(config) listen to link modes: c1
(serial) waiting for stop
(serial) received binary 24
(serial) received binary 44
(serial) received binary 2D
(serial) received binary 2C
(serial) received binary 87
(serial) received binary 52
(serial) received binary 46
(serial) received binary 58
(serial) received binary 1B
(serial) received binary 16
(serial) received binary 8D
(serial) received binary 20
(serial) received binary 15
(serial) received binary 4C
(serial) received binary 7A
(serial) received binary 75
(serial) received binary 20
(serial) received binary 42
(serial) received binary 7B
(serial) received binary 4B
(serial) received binary BD
(serial) received binary 38
(serial) received binary 5B
(serial) received binary 43
(serial) received binary 05
(serial) received binary 1E
(serial) received binary FE
(serial) received binary 9E
(serial) received binary E0
(serial) received binary D2
(serial) received binary E4
(serial) received binary 16
(serial) received binary 78
(serial) received binary 37
(serial) received binary 9B
(serial) received binary B5
(serial) received binary 61
(wmbus) received telegram 58465287 C-field=44 M-field=2c2d (KAM) A-field-version=1b A-field-dev-type=16 (Cold water meter) Ci-field=8d (Extended Link Layer II (                                     8 Byte)) CC-field=20 (slow_resp sync) ell header ACC=15 SN=20757a4c
(wmbus) frame "24442D2C875246581B168D20154C7A7520427B4BBD385B43051EFE9EE0D2E41678379BB561"
(wmbus) payload "427B4BBD385B43051EFE9EE0D2E41678379BB561"
(wmbus) 00: 24 length (36 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 2d2c m-field (2c2d=KAM)
(wmbus) 04: 87524658 a-field-addr (58465287)
(wmbus) 08: 1b a-field-version
(wmbus) 09: 16 a-field-type (Cold water meter)
(wmbus) 0a: 8d ci-field (Extended Link Layer II (8 Byte))
(wmbus) 0b: 20 cc-field (slow_resp sync)
(wmbus) 0c: 15 acc
(wmbus) 0d: 4c7a7520 sn (encrypted session=12 time=481188)
(wmbus) 24442D2C875246581B168D20154C7A7520
(meter) 58465287: for me? 58465287
(meter) 58465287: yes for me
(multical21) 58465287 58465287 (Mode1) decrypting "427B4BBD385B43051EFE9EE0D2E41                                     678379BB561"
(Mode1) IV 2D2C875246581B16204C7A7520000000
(Mode1) decrypted first block "C4C37934DD0B316020B6120000430000"
(Mode1) IV+1 2D2C875246581B16204C7A7520000001
(Mode1) decrypted second block "002A3973"
(Mode1) decrypted "C4C37934DD0B316020B6120000430000002A3973"
(multical21) log "telegram=|24442D2C875246581B168D20154C7A7520|C4C37934DD0B31602                                     0B6120000430000002A3973|+80"
(multical21) using hard coded format for hash dd34
(dvparser) using format "02FF2004134413"
(dvparser) looking for type=instantaneous vif=Volume storagenr=0
(dvparser) match? type=instantaneous vif=Volume and storagenr=0
(dvparser) match? type=instantaneous vif=Volume and storagenr=0
(dvparser) found key 0413 for type=instantaneous vif=Volume storagenr=0
(dvparser) looking for type=unknown vif=Volume storagenr=1
(dvparser) match? type=instantaneous vif=Volume and storagenr=1
(dvparser) match? type=instantaneous vif=Volume and storagenr=1
(dvparser) match? type=instantaneous vif=Volume and storagenr=1
(dvparser) found key 4413 for type=instantaneous vif=Volume storagenr=1
(dvparser) looking for type=unknown vif=VolumeFlow storagenr=-1
(dvparser) match? type=instantaneous vif=VolumeFlow and storagenr=-1
(dvparser) match? type=instantaneous vif=VolumeFlow and storagenr=-1
(dvparser) match? type=instantaneous vif=VolumeFlow and storagenr=-1
(dvparser) looking for type=unknown vif=FlowTemperature storagenr=-1
(dvparser) match? type=instantaneous vif=FlowTemperature and storagenr=-1
(dvparser) match? type=instantaneous vif=FlowTemperature and storagenr=-1
(dvparser) match? type=instantaneous vif=FlowTemperature and storagenr=-1
(dvparser) looking for type=unknown vif=ExternalTemperature storagenr=-1
(dvparser) match? type=instantaneous vif=ExternalTemperature and storagenr=-1
(dvparser) match? type=instantaneous vif=ExternalTemperature and storagenr=-1
(dvparser) match? type=instantaneous vif=ExternalTemperature and storagenr=-1
(multical21) 11: c4c3 payload crc
(multical21) 13: 79 frame type (short frame)
(multical21) 14: 34dd format signature
(multical21) 16: 0b31 data crc
(multical21) 18: * 6020 info codes ((dry 15-21 days) (burst 1-8 hours))
(multical21) 1a: * B6120000 total consumption (4.790000 m3)
(multical21) 1e: * 43000000 target consumption (0.067000 m3)
(multical21) 24442D2C875246581B168D20154C7A7520C4C37934DD0B316020B61200004300000                                     0
58465287        58465287        4.79 m3 0.067 m3        0 m3/h  127 °C  127 °C (                                     dry 15-21 days) (burst 1-8 hours)       2020-01-29 07:47.01

KeenanGeorge03, 2020-02-19:
Hi

How do we configure the meter file for kamstrup multical 21 meters.

All are encrypted & we have the keys for all of them. Its about 20 meters
all communicating to the same raspberry pi.

Regards
Keenan George

weetmuts, 2020-02-19:
You create the necessary /etc/wmbusmeters.d/Meter-xxxxxxxx files.
One for each meter if they have unique keys. If they all share the same key,
then you can have one meter file that matches multiple ids.

If you have KEM-files then you can try the python script under util to import them.

KeenanGeorge03, 2020-02-19:
Hi

Thanks

Regards
Keenan George

KeenanGeorge03, 2020-03-19:
Hi 

Thanks i see you added support for the WMB13u-868 dongle.

I have the following issue when i run the following command.

 wmbusmeters --debug --useconfig=/ --listento=t1 /dev/ttyUSB0:19200

i get no error however the log says the following.

No wmbus device found! Exiting!

Can you please point me in the right direction.

weetmuts, 2020-03-19:
Can you post the full log? 

It is not strictly supported yet, it just happens to work, since it sends
raw wmbus telegrams on the tty. So /dev/ttyUSB0:19200 just works.

KeenanGeorge03, 2020-03-20:
Hi

We fixed the issue by updating the config file with the below:

[image: image.png]

However i get the below response. We have uploaded all the meter files.

[image: image.png]

I'm not sure what I'm doing wrong.

If we plug in the iM871a in then everything works fine.

Regards
Keenan George

weetmuts, 2020-03-20:
Hmm, I can only see [image: image.png] in your post, not the actual image.

KeenanGeorge03, 2020-03-20:
Hi

Please find images attached.

Regards
Keenan George

KeenanGeorge03, 2020-03-20:
Hi

Please find attached.

Regards
Keenan George

weetmuts, 2020-03-20:
Sorry no images. You can email me directly on oehrstroem@gmail.com

weetmuts, 2020-03-28:
After debugging this dongle for some time, it seems like the usb2serial driver prolific
is a bit suspect. See https://ubuntuforums.org/showthread.php?t=2396842

The result is that you plug the dongle in, and it works for a bit, but when you restart wmbusmeters
the dongle stops responding, and the only way to get it working again, is to unplug/replug it again.

merbanan, 2020-03-28:
I think I can confim this. I have what I think is a pl2303 based usb2serial device that I have to replug from time to time to get it working. It could be related to what usb port it is connected to or the power supply or cosmic radiation or etc.

weetmuts, 2020-04-02:
The code in master can listen to a wmb13u (/dev/ttyUSB0:wm13u), but you have to configure the dongle using its own windows tool. The data interface format should be the one where RSSI and CRC are appended to the telegram. The wmb13u driver removes the last three bytes of the telegram.

Sadly, it is not possible to consistently write to this dongle from linux. Closing this issue now.

closed 2020-04-02 by weetmuts