wmbusmeters/wmbusmeters #66: expose of vendor data

chrisquin, 2020-01-30:
I added a meter file from the meter at my home. My meter is a derivate of a qcaloric meter and this sends some vendor information which contains the `set_date` (in my case 31.12.2019). I can't add the vendor hex string to the json output. How can i add this?

Do you need some more information?

chrisquin, 2020-01-30:
to give an example (this is one of my water meters):

```
complete message:
A5E20349446532376455121607780C13074710000DFF5F2F00018F807F3100C113FFFF7F2C741610007F2C741610002501230153014E013901E8009F000B013A0131012F013C01046D00139E2100000000485573
                            |> volume <||>                   Vendor                                                                         <||> Date   <||>???   <| CRC

vendor-part:
0DFF5F2F00018F807F3100C113FFFF7F2C741610007F2C741610002501230153014E013901E8009F000B013A0131012F013C01
                              <   ><     ><   ><     >
                              |key        |repeating, not sure why
                                   |set date value of 31. dez 2019
set date value:
4161000 <-- i need this
```

i thought i might can use `hasKey` function to find `7FC7`  ... but it won't work. I'm not used to C++ and therefore i am calling a python script with the JSON-format output (i use the shell to call it)

weetmuts, 2020-02-01:
Hmm, am not sure what you mean. 

If you can grab a --debug log where you run for example:
wmbusmeters --listento=t1,c1 --debug auto
and post the log here, or email it to me.

Then I can take a look at the telegrams directly.

chrisquin, 2020-02-04:
I wan't to extract the set date value(31.12.2019) in the vendor extension, because it is not a regular wmbus section. I identified the key `7F2C7` and the value `4161000` in the vendor extension. The easiest way in my case is to handle it in my python script.

```
(landisWater) 11: 0D dif (variable length Instantaneous value)
(landisWater) 12: FF vif (Vendor extension)
(landisWater) 13: 5F vife (duration of limit exceed last upper  is 3)
(landisWater) 14: 2F varlen=47
(landisWater) 15: 00018F807F3100C113FFFF7F2C741610007F2C741610002501230153014E013901E8009F000B013A0131012F013C01
```

This is the log of the telegram i posted:

```
(serial) received binary A5E20349446532376455121607780C13074710000DFF5F2F00018F807F3100C113FFFF7F2C741610007F2C741610002501230153014E013901E8009F000B013A0131012F013C01046D00139E2100000000485573
(im871a) timestamp 00000000
(im871a) rssi 48
(im871a) got crc16 7355 expected 7355
(wmbus) received telegram 12556437 C-field=44 M-field=3265 (LSE) A-field-version=16 A-field-dev-type=07 (Water meter) Ci-field=78 (EN 13757-3 Application Layer (no tplh))
(wmbus) frame "49446532376455121607780C13074710000DFF5F2F00018F807F3100C113FFFF7F2C741610007F2C741610002501230153014E013901E8009F000B013A0131012F013C01046D00139E21"
(wmbus) payload "0C13074710000DFF5F2F00018F807F3100C113FFFF7F2C741610007F2C741610002501230153014E013901E8009F000B013A0131012F013C01046D00139E21"
(wmbus) 00: 49 length (73 bytes)
(wmbus) 01: 44 c-field (SND_NR)
(wmbus) 02: 6532 m-field (3265=LSE)
(wmbus) 04: 37645512 a-field-addr (12556437)
(wmbus) 08: 16 a-field-version
(wmbus) 09: 07 a-field-type (Water meter)
(wmbus) 0a: 78 ci-field (EN 13757-3 Application Layer (no tplh))
(wmbus) 4944653237645512160778
(meter) myHome: for me? 12556437
(meter) myHome: not for me: not my id
(meter) myWaters: for me? 12556437
(meter) myWaters: yes for me
(landisWater) myWaters 12556437 (landisWater) log "telegram=|4944653237645512160778|0C13074710000DFF5F2F00018F807F3100C113FFFF7F2C741610007F2C741610002501230153014E013901E8009F000B013A0131012F013C01046D00139E21|+1435"
(dvparser) looking for type=unknown vif=Volume storagenr=0
(dvparser) match? type=instantaneous vif=Volume and storagenr=0
(dvparser) match? type=instantaneous vif=Volume and storagenr=0
(dvparser) found key 0C13 for type=instantaneous vif=Volume storagenr=0
(landisWater) 0b: 0C dif (8 digit BCD Instantaneous value)
(landisWater) 0c: 13 vif (Volume l)
(landisWater) 0d: * 07471000 total consumption (104.707000 m3)
(landisWater) 11: 0D dif (variable length Instantaneous value)
(landisWater) 12: FF vif (Vendor extension)
(landisWater) 13: 5F vife (duration of limit exceed last upper  is 3)
(landisWater) 14: 2F varlen=47
(landisWater) 15: 00018F807F3100C113FFFF7F2C741610007F2C741610002501230153014E013901E8009F000B013A0131012F013C01
(landisWater) 44: 04 dif (32 Bit Integer/Binary Instantaneous value)
(landisWater) 45: 6D vif (Date and time type)
(landisWater) 46: 00139E21
(landisWater) 49446532376455121607780C13074710000DFF5F2F00018F807F3100C113FFFF7F2C741610007F2C741610002501230153014E013901E8009F000B013A0131012F013C01046D00139E21
(shell) exec "/bin/sh"
(shell) arg "-c"
(shell) arg "python3 configs/saveMeters.py --json="$METER_JSON" --influx_url=192.168.1.6 --influx_table=meters --file_ids=configs/ids.json"
(shell) env "METER_JSON={"media":"water","meter":"landisWater","name":"myWaters","id":"12556437","total_m3":104.707,"timestamp":"2020-01-30T18:01:02Z"}"
(shell) env "METER_TYPE=landisWater"
(shell) env "METER_NAME=myWaters"
(shell) env "METER_ID=12556437"
(shell) env "METER_TOTAL_M3=104.707"
(shell) env "METER_TIMESTAMP=2020-01-30T18:01:02Z"
(shell) waiting for child 165 to complete.
(shell) /bin/sh: return code 0
```

my config:

```
listento=s1
device=/dev/ttyUSB0
loglevel=debug
logfile=configs/meter.log
format=json
shell=python3 configs/saveMeters.py --json="$METER_JSON" --influx_url=192.168.1.6 --influx_table=meters --file_ids=configs/ids.json
```

merbanan, 2020-02-04:
@chrisquin you need to add the missing data parsing, either decode the data properly or add export for the raw bytes.

weetmuts, 2020-02-04:
Is this a qcaloric 100% compatible meter? Or should we create a new meter for it?
Do you have a name for it?

As @merbanan says, the best is to add the necessary decode into meter_qcaloric processContent.
You can for example add this:
    string tmp;
    extractDVstring(&t->values, "0DFF5F", &offset, &tmp);
    if (tmp.length() > 0) { // This will only trigger for those telegrams where 0DFF5F is found.
        vector<uchar> bin;
        hex2bin(tmp, &bin);
        // Now you can extract the binary data from and prepare some value
        // that can be printed in the json.
    }

chrisquin, 2020-02-04:
@weetmuts

1. thanks, that hint is what i needed.

2. This is my meter: https://qundis.com/products/heat-cost-allocators/ ... but ... this isn't what is really inside. The meter uses the manufacture-code `MANUFACTURER_LSE` and it is configured in S-mode. The meter can transmit in S- and C-mode, but the `meter_qcaloric.cc` is bounded to C-mode and the manufacture-code `MANUFACTURER_QDS`. It seems that my meter isn't sending all the data like the qcaloric does. (same problem with the water-meters)

3. I really like your effort and your repo! I added my own meter files and want to use docker ... i ended up to bouil my own Dockerfile because i had to find a solution where to premerge my files before make...

chrisquin, 2020-02-04:
i am using influxdb and grafana to display my measurements:

[image]

weetmuts, 2020-02-05:
The meter seems different enough to have its own meter_xxxx source file.
Please go ahead and try to add it and create a pull request, otherwise I can do it.  
Remember to add the telegrams to simulations_s1.txt and add test_t1_meters.sh
This will be the first S1 mode meter.

chrisquin, 2020-02-07:
ok, i will do it but can't say when i am finished

weetmuts, 2021-08-08:
Hi @chrisquin !
I examined your telegram that you posted above and the water meter is now supported by wmbusmeters, it will auto-detect the driver lse_07_17.

However this issue also talks about an HCA meter? But I cannot find any telegrams here with that content?

weetmuts, 2021-08-08:
Can you test the latest wmbusmeters, I think that there are whe... drivers that could be the ones that support your HCA.

weetmuts, 2021-11-28:
Closing this for now.

closed 2021-11-28 by weetmuts