wmbusmeters/wmbusmeters #67: Trying to rewrite to ESP8266

MTrab, 2020-02-01:
I'm trying to "rewrite" to function on ESP8266, but one main issue I have is, I cannot get any data thru.

Looking thru your code, it seems like you initialize with 0xFF and then you should be able to retrieve the serial with 0x0B - if I understand this correctly.

But nothing happens.
I have tried attaching my module (AMB8426-M) directly to a FTDI dongle, and sending the commands using RealTerm - no data either.

Using the same setup (FTDI and AMB8426-M) is working as supposed with your software.
Have I missed some initializing command?

BIBOLV, 2020-02-01:
Do you use Amber USB and if so are you sure that module is being powered keeping in mind that ESP operates at 3.3V?

MTrab, 2020-02-01:
As written - I use a AMB8426-M.

I have gotten a few steps closer, I got some data entering the ESP.
Now I only need to figure out how to parse the data.

weetmuts, 2020-02-02:
Interesting!
As soon as you get a serial connection with the amber module, the protocol ought to be identical....

MTrab, 2020-02-02:
I battle a bit inconsistent data - and then I need to see if I can find out how I read the protocol.
Have tried looking thru your source code, i have helped a little bit, but I'm not sure the Arduino codebase is powerfull enough to run it all - might need to cut it down pretty much to the bare essentials.

weetmuts, 2020-02-02:
Yes, wmbusmeters is getting more and more generic, which unfortunately means bigger and bigger. 
The wmbus standard is huge! Wmbusmeters merely implements a small part of it.

MTrab, 2020-02-02:
I'm aware of that :) Just trying to get the bigger picture for Amber<->Multical 21 here :D Including how you decode model, serial and find the AES encrypted part.

BIBOLV, 2020-02-02:
Idea is great, when I was getting in to this whole wmbus metering deal, I would loved if there would be device with integrated ESP and wmbus chip in one device. Maybe it would be faster, easier to have ESP just to relay wmbus message to mqtt without any processing and have wmbusmeters on other end to process received messages over mqtt?

MTrab, 2020-02-02:
My end goal is to have the ESP deliver the decrypted data from my Multical meter - wether it should be MQTT, websocket or anything else, I have not yet decided.
Eventually all data would be present in Home Assistant.

weetmuts, 2020-02-05:
I think that @BIBOLV :s idea of just relaying the received data using tcp/udp is a nice idea.
Then I can add a listening tcp/udp port in wmbusmeters that it can send to.
Then we can help you run wmbusmeters on the same computer you are running Home Assistant on.
Then wmbusmeters uses mqtt to HASS, as described here: https://github.com/weetmuts/wmbusmeterswiki/blob/master/HASS.md

MTrab, 2020-02-05:
Could be a step one. My end goal still is to have it all on the ESP, but until I "crack the nut" a solution could be to just forward the raw data to wmbusmeters.

weetmuts, 2020-02-06:
You could take a look at: https://github.com/tobiasrask/wmbus-client

This client is a minimal multical21 client, much more suitable for creating something that will fit in an ESP8266.

Good luck! And let us know when you have something fun to show!

closed 2020-02-06 by weetmuts