wmbusmeters/wmbusmeters #68: Can't get it to work on kali + rtlsdr

basti85, 2020-02-01:
Hi,

I've been trying the last few days, but to no avail :(

Qutput of testinternals --verbose 
(abbreviated, just to show *something happens at least)

`Test nr 1......
(dvparser) cannot handle dif 0F ignoring rest of telegram.
(dvparser) found new format "0B138B8200933E0DFD10" with hash e0b9, remembering!

Test nr 2......
(dvparser) found new format "82046C" with hash 80ed, remembering!

Test nr 3......
(dvparser) found new format "0C13426C4C1382046C8C04138D04931E046D03FD6C82206C0BFD0F8C40798310FD3182106C8110FD6102FD6602FD17" with hash 88ae, remembering!
Test nr 4......
(dvparser) found new format "426C" with hash 8c86, remembering!`

Output of sudo wmbusmeters --listento=c1 --debug --verbose rtlwmbus

`(wmbusmeters) version: 0.9.22-12-g7f7fb46
(config) using device: rtlwmbus
(config) number of meters: 0
(detect) "rtlwmbus" ""
(detect) driver: rtlwmbus
(rtlwmbus) using command: rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus"
(serialcmd) opened /bin/sh
(config) all possible link modes that the meters might transmit on: none
(config) explicitly listening to: c1
(config) listen link modes calculated to be: c1
No meters configured. Printing id:s of all telegrams heard!

(config) listen to link modes: any
(serial) waiting for stop
(serial) received ascii /bin/sh: 1: rtl_wmbus: not found<0A>
(serial) received ascii Found 1 device(s):<0A>
(serial) received ascii   0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A>
(serial) received ascii <0A>
(rtlwmbus) error in received message.
(serial) received ascii Using device 0: Generic RTL2832U OEM<0A>
(serial) received ascii Detached kernel driver<0A>
(serial) received ascii Found Rafael Micro R820T tuner<0A>
(serial) received ascii [R82XX] PLL not locked!<0A>
(serial) received ascii Sampling at 1600000 S/s.<0A>
(serial) received ascii Tuned to 868950000 Hz.<0A>
(serial) received ascii Tuner gain set to automatic.<0A>
(serial) received ascii Reading samples in async mode...<0A>
(serial) received ascii Signal caught, exiting!<0A>
(serial) received ascii Short write, samples lost, exiting!<0A>
(serial) received ascii <0A>User cancel, exiting...<0A>
(serial) received ascii Reattached kernel driver<0A>
(bgshell) 8073 exited with return code 127
(serial) closing non working fd=3
(serial) no devices working
(serial) stopping manager
(serialcmd) closed /bin/sh
(serial) event loop stopped!
(serial) closing 0 devices
(serial) no devices working
(serialcmd) closed /bin/sh
`
What do I do now? The device is working with other softwares like rtl_wmbus or tfrec. But both are not usable for my meters (BRUNATA-METRONA Star, Star C and water meter). They are not on the list for wmbusmeters, either, but I can only work on that once I get the basics running...

Thanks for your help, basti

weetmuts, 2020-02-02:
Check that /usr/bin/rtl_wmbus exists! 
That is the default place wmbusmeters looks for the binary. :-)

weetmuts, 2020-02-03:
Also download and build the latest version of wmbusmeters.

weetmuts, 2020-02-06:
There is a new version today. Please download, build and install and test. :-)

weetmuts, 2020-02-12:
Did it work? Can we close this?

weetmuts, 2020-02-17:
Try with the latest version of wmbusmeters.

But it seems like you simply does not have /usr/bin/rtl_wmbus installed.

closed 2020-02-17 by weetmuts