wmbusmeters/wmbusmeters #69: wmbusmeters as daemon on rpi4

Herreropl, 2020-02-02:
Hi,

I have problem with installing your software and running as a deamon on my rpi4 buster.  During installation it seems that everythink is ok:
`~/wmbusmeters $ sudo make install
Building 0.9.22-12-g7f7fb46
binaries: installed /usr/bin/wmbusmeters and /usr/sbin/wmbusmetersd
man page: installed /usr/share/man/man1/wmbusmeters.1.gz
user: wmbusmeters unmodified
conf file: /etc/logrotate.d/wmbusmeters unchanged
conf file: /etc/wmbusmeters.conf unchanged
conf dir: /etc/wmbusmeters.d unchanged
systemd: removing /etc/systemd/system/wmbusmeters@.service
systemd: backup stored here: /root/old.wmbusmeters@.service.backup
systemd: installed /etc/systemd/system/wmbusmeters@.service
udev: removing /etc/udev/rules.d/99-wmbus-usb-serial.rules
udev: backup stored here: /root/old.wmbusmeters-wmbus-usb-serial.rules.backup
udev: installed /etc/udev/rules.d/99-wmbus-usb-serial.rules
`
But when I'm trying to start/stop deamon I'm receiving:

`sudo systemctl status wmbusmeters
Unit wmbusmeters.service could not be found.
`

I think that configuration is ok, because when I'm runnig command:
`sudo wmbusmeters --useconfig=/`
 
I'm receiving:
`sudo tail -f /var/log/wmbusmeters/meter_readings/AMIplus
{"media":"radio converter (meter side)","meter":"amiplus","name":"AMIplus","id":"00138696","total_energy_consumption_kwh":11514.879,"current_power_consumption_kw":0.403,"total_energy_production_kwh":0.003,"current_power_production_kw":0,"device_date_time":"2020-02-02 17:33","timestamp":"2020-02-02T16:32:58Z"}`

Thank you for any help

BIBOLV, 2020-02-02:
Try running `sudo systemctl daemon-reload` and then again `sudo systemctl status wmbusmeters`

Herreropl, 2020-02-02:
unfortunatelly, the same results:

`herrero@starserver:~/wmbusmeters $ sudo systemctl daemon-reload
[sudo] password for herrero:
herrero@starserver:~/wmbusmeters $ sudo systemctl status wmbusmeters
Unit wmbusmeters.service could not be found.
`

weetmuts, 2020-02-02:
The name of the service just changed. Update to the latest version of wmbusmeters.
Which dongle are you using? im871a, amb8465 or an rtlsdr? 

Assuming it is an rtlsdr dongle, look for /dev/rtlsdr_0 or some other number,
then you can do "sudo systemctrl restart wmbusmeters@rtlsdr_0.service"
depending on what number the device received.

Herreropl, 2020-02-02:
OK, one step forward:
`herrero@starserver:~/wmbusmeters $ sudo systemctl start wmbusmeters@rtlsdr
herrero@starserver:~/wmbusmeters $ sudo systemctl status wmbusmeters@rtlsdr
● wmbusmeters@rtlsdr.service - "wmbusmeters service on rtlsdr"
   Loaded: loaded (/etc/systemd/system/wmbusmeters@.service; disabled; vendor pr
   Active: failed (Result: exit-code) since Sun 2020-02-02 18:33:05 CET; 2s ago
  Process: 21956 ExecStartPre=/bin/mkdir -p /var/log/wmbusmeters/meter_readings
  Process: 21957 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmb
  Process: 21958 ExecStartPre=/bin/mkdir -p /var/run/wmbusmeters (code=exited, s
  Process: 21959 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/run/wmb
  Process: 21960 ExecStart=/usr/sbin/wmbusmetersd --device=/dev/rtlsdr /var/run/
 Main PID: 21961 (code=exited, status=1/FAILURE)

Feb 02 18:33:04 starserver systemd[1]: Starting "wmbusmeters service on rtlsdr".
Feb 02 18:33:04 starserver wmbusmetersd[21960]: (wmbusmeters) started /var/run/w
Feb 02 18:33:04 starserver systemd[1]: Started "wmbusmeters service on rtlsdr".
Feb 02 18:33:04 starserver systemd[1]: Stopping "wmbusmeters service on rtlsdr".
Feb 02 18:33:05 starserver systemd[1]: wmbusmeters@rtlsdr.service: Main process
Feb 02 18:33:05 starserver systemd[1]: wmbusmeters@rtlsdr.service: Failed with r
Feb 02 18:33:05 starserver systemd[1]: Stopped "wmbusmeters service on rtlsdr".

herrero@starserver:~/wmbusmeters $ sudo tail -f /var/log/wmbusmeters/
meter_readings/       wmbusmeters.log       wmbusmeters.log.1.gz
herrero@starserver:~/wmbusmeters $ sudo tail -f /var/log/wmbusmeters/wmbusmeters.log
Could not open file "/var/log/wmbusmeters/meter_readings/AMIplus" for writing!
(wmbusmeters) logging started 2020-02-02 18:33:04
No wmbus device found! Exiting!
^X^C
herrero@starserver:~/wmbusmeters $ sudo tail -f /var/log/wmbusmeters/
meter_readings/       wmbusmeters.log       wmbusmeters.log.1.gz
herrero@starserver:~/wmbusmeters $ sudo tail -f /var/log/wmbusmeters/meter_readings/AMIplus
{"media":"radio converter (meter side)","meter":"amiplus","name":"AMIplus","id":"00138696","total_energy_consumption_kwh":11514.879,"current_power_consumption_kw":0.403,"total_energy_production_kwh":0.003,"current_power_production_kw":0,"device_date_time":"2020-02-02 17:33","timestamp":"2020-02-02T16:32:58Z"}
^C
`

weetmuts, 2020-02-02:
Since your log says: wmbusmeters@rtlsdr
and not wmbusmeters@rtlsdr_0 or some other number,
then this means that your are not running the latest wmbusmeters.

Please install the latest. :-)

weetmuts, 2020-02-02:
You should see something like this in your syslog:
Feb  2 18:48:40 raspberrypi systemd[1]: Starting "wmbusmeters service on /dev/amb8465_0"...
Feb  2 18:48:41 raspberrypi wmbusmetersd[502]: (wmbusmeters) started /var/run/wmbusmeters/wmbusmeters--dev-amb8465_0.pid
Feb  2 18:48:41 raspberrypi systemd[1]: Started "wmbusmeters service on /dev/amb8465_0".
Feb  2 18:48:41 raspberrypi wmbusmetersd[506]: (config) loading meter file /etc/wmbusmeters.d/RoneVatten
Feb  2 18:48:41 raspberrypi wmbusmetersd[506]: (config) name=Vatten
Feb  2 18:48:41 raspberrypi wmbusmetersd[506]: (config) type=multical21
Feb  2 18:48:41 raspberrypi wmbusmetersd[506]: (config) id=76348799
Feb  2 18:48:41 raspberrypi wmbusmetersd[506]: (config) key=<notprinted>
Feb  2 18:48:41 raspberrypi wmbusmetersd[506]: (config) overriding device with "/dev/amb8465_0"
Feb  2 18:48:41 raspberrypi wmbusmetersd[506]: (wmbusmeters) using log file /var/log/wmbusmeters/wmbusmeters.log

Herreropl, 2020-02-02:
Ok, I have already updated wmbusmeters and now I have:
` sudo systemctl status wmbusmeters@rtlsdr1
[sudo] password for herrero:
● wmbusmeters@rtlsdr1.service - "wmbusmeters service on rtlsdr1"
   Loaded: loaded (/etc/systemd/system/wmbusmeters@.service; disabled; vendor pr
   Active: inactive (dead)

Feb 02 22:33:17 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:
Feb 02 22:33:17 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:
...skipping...
● wmbusmeters@rtlsdr1.service - "wmbusmeters service on rtlsdr1"
   Loaded: loaded (/etc/systemd/system/wmbusmeters@.service; disabled; vendor pr
   Active: inactive (dead)

Feb 02 22:33:17 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:
Feb 02 22:33:17 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:
~

herrero@starserver:~ $ sudo tail -100 /var/log/syslog
Feb  2 22:33:17 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:12: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
Feb  2 22:33:17 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:22: PIDFile= references path below legacy directory /var/run/, updating /var/run/wmbusmeters/wmbusmeters-rtlsdr1.pid → /run/wmbusmeters/wmbusmeters-rtlsdr1.pid; please update the unit file accordingly.
herrero@starserver:~ $ sudo tail -100 /var/log/wmbusmeters/wmbusmeters.log
Could not open file "/var/log/wmbusmeters/meter_readings/AMIplus" for writing!
(wmbusmeters) logging started 2020-02-02 22:31:35
(wmbusmeters) waiting for telegrams
(wmbusmeters) shutting down
(wmbusmeters) logging started 2020-02-02 22:31:36
(wmbusmeters) waiting for telegrams
(wmbusmeters) shutting down
(wmbusmeters) logging started 2020-02-02 22:31:38
(wmbusmeters) waiting for telegrams
(wmbusmeters) shutting down
`

weetmuts, 2020-02-03:
Have you run "make install" ?
Can you check 
ls -l /var/log | grep wmbusmeters
and 
ls -l /var/log/wmbusmeters
and
ls -l /var/log/wmbusmeters/meter_readings

Herreropl, 2020-02-03:
Of course, I did it yesterday and again today. This is the output, that you ask:

`herrero@starserver:~/wmbusmeters $ sudo make install
Building 0.9.23-4-g9ed8a40
binaries: installed /usr/bin/wmbusmeters and /usr/sbin/wmbusmetersd
man page: installed /usr/share/man/man1/wmbusmeters.1.gz
user: wmbusmeters unmodified
conf file: /etc/logrotate.d/wmbusmeters unchanged
conf file: /etc/wmbusmeters.conf unchanged
conf dir: /etc/wmbusmeters.d unchanged
systemd: removing /etc/systemd/system/wmbusmeters@.service
systemd: backup stored here: /root/old.wmbusmeters@.service.backup
systemd: installed /etc/systemd/system/wmbusmeters@.service
udev: removing /etc/udev/rules.d/99-wmbus-usb-serial.rules
udev: backup stored here: /root/old.wmbusmeters-wmbus-usb-serial.rules.backup
udev: installed /etc/udev/rules.d/99-wmbus-usb-serial.rules

herrero@starserver:~/wmbusmeters $ sudo systemctl status wmbusmeters@rtlsdr1    
● wmbusmeters@rtlsdr1.service - "wmbusmeters service on rtlsdr1"
   Loaded: loaded (/etc/systemd/system/wmbusmeters@.service; disabled; vendor pr
   Active: inactive (dead)

Feb 02 22:33:17 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:
Feb 02 22:33:17 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:
Feb 02 23:27:11 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:
Feb 02 23:27:11 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:
Feb 02 23:27:45 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:
Feb 02 23:27:45 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:
Feb 02 23:29:40 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:
Feb 02 23:29:40 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:
Feb 03 19:45:55 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:
Feb 03 19:45:55 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:

herrero@starserver:~/wmbusmeters $

herrero@starserver:~/wmbusmeters $ ls -l /var/log | grep wmbusmeters
drwxr-xr-x 3 wmbusmeters wmbusmeters     100 Feb  2 00:00 wmbusmeters

herrero@starserver:~/wmbusmeters $ ls -l /var/log/wmbusmeters
total 8
drwxr-xr-x 2 wmbusmeters wmbusmeters  60 Feb  2 17:32 meter_readings
-rw-rw-rw- 1 wmbusmeters wmbusmeters 927 Feb  2 22:31 wmbusmeters.log
-rw-rw-rw- 1 wmbusmeters wmbusmeters 130 Feb  1 18:35 wmbusmeters.log.1.gz

herrero@starserver:~/wmbusmeters $ ls -l /var/log/wmbusmeters/meter_readings
total 4
-rw-r--r-- 1 wmbusmeters wmbusmeters 311 Feb  3 15:11 AMIplus

herrero@starserver:~/wmbusmeters $ sudo systemctl status wmbusmeters@rtlsdr1    
● wmbusmeters@rtlsdr1.service - "wmbusmeters service on rtlsdr1"
   Loaded: loaded (/etc/systemd/system/wmbusmeters@.service; disabled; vendor pr
   Active: failed (Result: exit-code) since Mon 2020-02-03 19:48:35 CET; 5s ago
  Process: 25564 ExecStartPre=/bin/mkdir -p /var/log/wmbusmeters/meter_readings
  Process: 25565 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmb
  Process: 25567 ExecStartPre=/bin/mkdir -p /var/run/wmbusmeters (code=exited, s
  Process: 25569 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/run/wmb
  Process: 25570 ExecStart=/usr/sbin/wmbusmetersd --device=rtlsdr1 /var/run/wmbu
 Main PID: 25571 (code=exited, status=1/FAILURE)

Feb 03 19:48:35 starserver systemd[1]: Starting "wmbusmeters service on rtlsdr1"
Feb 03 19:48:35 starserver wmbusmetersd[25570]: (wmbusmeters) started /var/run/w
Feb 03 19:48:35 starserver systemd[1]: Started "wmbusmeters service on rtlsdr1".
Feb 03 19:48:35 starserver systemd[1]: Stopping "wmbusmeters service on rtlsdr1"
Feb 03 19:48:35 starserver systemd[1]: wmbusmeters@rtlsdr1.service: Main process
Feb 03 19:48:35 starserver systemd[1]: wmbusmeters@rtlsdr1.service: Failed with
Feb 03 19:48:35 starserver systemd[1]: Stopped "wmbusmeters service on rtlsdr1".
`

weetmuts, 2020-02-03:
Thanks, but this is so weird. Can you post the contents of your udev rule, like this:
cat /etc/udev/rules.d/99-wmbus-usb-serial.rules 
and your config:
cat /etc/wmbusmeters.conf

Herreropl, 2020-02-03:
Ok, another output:

`herrero@starserver:~ $ cat /etc/udev/rules.d/99-wmbus-usb-serial.rules
SUBSYSTEM=="tty", ATTRS{idVendor}=="10c4", ATTRS{idProduct}=="ea60",SYMLINK+="im871a_%n",MODE="0660", GROUP="wmbusmeters",TAG+="systemd",ENV{SYSTEMD_WANTS}="wmbusmeters@/dev/im871a_%n.service"
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001",SYMLINK+="amb8465_%n",MODE="0660", GROUP="wmbusmeters",TAG+="systemd",ENV{SYSTEMD_WANTS}="wmbusmeters@/dev/amb8465_%n.service"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0bda", ATTRS{idProduct}=="2838",SYMLINK+="rtlsdr_%n",MODE="0660", GROUP="wmbusmeters",TAG+="systemd",ENV{SYSTEMD_WANTS}="wmbusmeters@/dev/rtlsdr_%n.service"
SUBSYSTEM=="usb", ATTRS{idVendor}=="2047", ATTRS{idProduct}=="0863",SYMLINK+="rfmrx2_%n",MODE="0660", GROUP="wmbusmeters",TAG+="systemd",ENV{SYSTEMD_WANTS}="wmbusmeters@/dev/rfmrx2_%n.service"

herrero@starserver:~ $ cat /etc/wmbusmeters.conf
loglevel=normal
device=rtlwmbus
logtelegrams=false
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
logfile=/var/log/wmbusmeters/wmbusmeters.log`

And wmbusmeters.log after sudo systemctl start wmbusmeters@rtlsdr1

`(wmbusmeters) logging started 2020-02-03 22:38:38
No wmbus device found! Exiting!
`

Is so wired that command below works well.

` sudo wmbusmeters --useconfig=/`

Herreropl, 2020-02-03:
One more test with log verbose level. Below wmbusmeters.log.
1. case (trying to start deamon)
`(wmbusmeters) logging started 2020-02-03 23:04:40
(config) store meter files in: "/var/log/wmbusmeters/meter_readings"
(config) using device: rtlsdr1
(config) number of meters: 1
No wmbus device found! Exiting!
`
2. case ( sudo wmbusmeters --useconfig=/)
`(config) store meter files in: "/var/log/wmbusmeters/meter_readings"
(config) using device: rtlwmbus
(config) number of meters: 1
(rtlwmbus) using command: rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus
(serialcmd) opened /bin/sh
(wmbusmeters) configured "AMIplus" "amiplus" "00138696" encrypted
(config) listen to link modes: any
(telegram) DLL L=4e C=44 (from meter SND_NR) M=10b6 (DEV) A=00138696 VER=02 TYPE=37 (Radio converter (meter side))
(telegram) TPL CI=7a ACC=30 STS=00 CFG=0540 (AES_CBC_IV accessibility)
(telegram) DLL L=4e C=44 (from meter SND_NR) M=10b6 (DEV) A=00138696 VER=02 TYPE=37 (Radio converter (meter side))
(telegram) TPL CI=7a ACC=31 STS=00 CFG=0540 (AES_CBC_IV accessibility)
(serial) event loop stopped!
`

weetmuts, 2020-02-04:
Do you manually start the daemon? Could that be the reason that it tries to use /dev/rtlsdr1 ?

SUBSYSTEM=="usb", ATTRS{idVendor}=="0bda", ATTRS{idProduct}=="2838",SYMLINK+="rtlsdr_%n",MODE="0660", GROUP="wmbusmeters",TAG+="systemd",ENV{SYSTEMD_WANTS}="wmbusmeters@/dev/rtlsdr_%n.service"

The udev rule and wmbusmeters expect the device to be /dev/rtlsdr_1

Can you try just unplugging and inserting the rtlsdr dongle again?

What does device=... say in your wmbusmeters.conf?

If you stop your daemon with systemctl and then try to run wmbusmeters manually.

wmbusmeters --listento=t1,c1 auto

and

wmbusmeters --listento=t1,c1 rtlwmbus

What happens?

Herreropl, 2020-02-04:
OK, I have restarted rpi without donge, and after restart I connected it again. Below output:
```
herrero@starserver:~ $ cat /etc/wmbusmeters.conf
loglevel=normal
device=rtlwmbus
logtelegrams=false
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
logfile=/var/log/wmbusmeters/wmbusmeters.log
```

```
herrero@starserver:~ $ wmbusmeters --listento=t1,c1 auto
You are not in the same group as the device /dev/rtlsdr_1
herrero@starserver:~ $ wmbusmeters --listento=t1,c1 rtlwmbus
No meters configured. Printing id:s of all telegrams heard!`
```

weetmuts, 2020-02-04:
Add yourself to the wmbusmeters group to be able to read the rtlsdr_1 device.
sudo usermod -a -G wmbusmeters herrero

Then try again.

Herreropl, 2020-02-04:
Ok, now is the same group, but:
`herrero@starserver:~ $ wmbusmeters --listento=t1,c1 auto
No wmbus device found! Exiting!
herrero@starserver:~ $ wmbusmeters --listento=t1,c1 rtlwmbus
No meters configured. Printing id:s of all telegrams heard!
`

weetmuts, 2020-02-04:
what does this say? 
ls -l /dev/rtlsdr*

Herreropl, 2020-02-04:
`lrwxrwxrwx 1 root root 15 Feb  4 07:59 /dev/rtlsdr_1 -> bus/usb/001/006
`

weetmuts, 2020-02-04:
And for this?
ls -l /dev/bus/usb/001/006 

If you let wmbusmeters run for a while (when Printing id:s of all telegrams heard!)
do you receive any telegrams?

What happens if you write device=rtlwmbus in your wmbusmeters.conf
file and unplug/insert the dongle again?

Herreropl, 2020-02-04:
Now I have rtlwmbus in my wmbusmeters.conf.

New output:

```
herrero@starserver:~ $ wmbusmeters --listento=t1,c1 auto
No wmbus device found! Exiting!
herrero@starserver:~ $ wmbusmeters --listento=t1,c1 rtlwmbus
No meters configured. Printing id:s of all telegrams heard!

Received telegram from: 00138696
          manufacturer: (DEV) Develco Products, Denmark
           device type: Radio converter (meter side)

ls -l /dev/bus/usb/001/006
crw-rw-rw- 1 root wmbusmeters 189, 5 Feb  4 12:12 /dev/bus/usb/001/006

```

Herreropl, 2020-02-04:
If I run command  sudo wmbusmeters --useconfig=/ with device=auto (in wmbusmeters.conf ) in log file appears information:
`You are not in the same group as the device /dev/rtlsdr_1`

For device=rtlwmbus everything work ok. 

I have found somethink else in systemctl status:

```
herrero@starserver:~ $ sudo systemctl status wmbusmeters@rtlwmbus
● wmbusmeters@rtlwmbus.service - "wmbusmeters service on rtlwmbus"
   Loaded: loaded (/etc/systemd/system/wmbusmeters@.service; disabled; vendor preset: enabled)
   Active: inactive (dead)

Feb 04 23:07:44 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:12: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
Feb 04 23:07:44 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:22: PIDFile= references path below legacy directory /var/run/, updating /var/run/wmbus
Feb 04 23:07:44 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:12: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
Feb 04 23:07:44 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:22: PIDFile= references path below legacy directory /var/run/, updating /var/run/wmbus
Feb 04 23:07:44 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:12: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
Feb 04 23:07:44 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:22: PIDFile= references path below legacy directory /var/run/, updating /var/run/wmbus
Feb 04 23:07:44 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:12: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
Feb 04 23:07:44 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:22: PIDFile= references path below legacy directory /var/run/, updating /var/run/wmbus
Feb 04 23:19:58 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:12: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
Feb 04 23:19:58 starserver systemd[1]: /etc/systemd/system/wmbusmeters@.service:22: PIDFile= references path below legacy directory /var/run/, updating /var/run/wmbus

herrero@starserver:~ $ sudo systemctl status wmbusmeters@rtlsdr1
● wmbusmeters@rtlsdr1.service - "wmbusmeters service on rtlsdr1"
   Loaded: loaded (/etc/systemd/system/wmbusmeters@.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2020-02-04 23:07:50 CET; 12min ago
  Process: 5928 ExecStartPre=/bin/mkdir -p /var/log/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
  Process: 5929 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
  Process: 5930 ExecStartPre=/bin/mkdir -p /var/run/wmbusmeters (code=exited, status=0/SUCCESS)
  Process: 5931 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/run/wmbusmeters (code=exited, status=0/SUCCESS)
  Process: 5932 ExecStart=/usr/sbin/wmbusmetersd --device=rtlsdr1 /var/run/wmbusmeters/wmbusmeters-rtlsdr1.pid (code=exited, status=0/SUCCESS)
 Main PID: 5933 (code=exited, status=1/FAILURE)

Feb 04 23:07:49 starserver systemd[1]: Starting "wmbusmeters service on rtlsdr1"...
Feb 04 23:07:50 starserver wmbusmetersd[5932]: (wmbusmeters) started /var/run/wmbusmeters/wmbusmeters-rtlsdr1.pid
Feb 04 23:07:50 starserver systemd[1]: Started "wmbusmeters service on rtlsdr1".
Feb 04 23:07:50 starserver systemd[1]: Stopping "wmbusmeters service on rtlsdr1"...
Feb 04 23:07:50 starserver systemd[1]: wmbusmeters@rtlsdr1.service: Main process exited, code=exited, status=1/FAILURE
Feb 04 23:07:50 starserver systemd[1]: wmbusmeters@rtlsdr1.service: Failed with result 'exit-code'.
Feb 04 23:07:50 starserver systemd[1]: Stopped "wmbusmeters service on rtlsdr1".
```

Herreropl, 2020-02-05:
I have installed the last version from your git and now I have different errors:

```
herrero@starserver:~ $ sudo systemctl status wmbusmeters@rtlsdr1
● wmbusmeters@rtlsdr1.service - "wmbusmeters service on rtlsdr1"
   Loaded: loaded (/etc/systemd/system/wmbusmeters@.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2020-02-05 15:11:34 CET; 1min 47s ago
  Process: 3354 ExecStartPre=/bin/mkdir -p /var/log/wmbusmeters/meter_readings (code=exited, status=217/USER)
  Process: 3355 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmbusmeters (code=exited, status=217/USER)

Feb 05 15:11:34 starserver systemd[1]: wmbusmeters@rtlsdr1.service: Service RestartSec=1s expired, scheduling restart.
Feb 05 15:11:34 starserver systemd[1]: wmbusmeters@rtlsdr1.service: Scheduled restart job, restart counter is at 3.
Feb 05 15:11:34 starserver systemd[1]: Stopped "wmbusmeters service on rtlsdr1".
Feb 05 15:11:34 starserver systemd[1]: wmbusmeters@rtlsdr1.service: Start request repeated too quickly.
Feb 05 15:11:34 starserver systemd[1]: wmbusmeters@rtlsdr1.service: Failed with result 'exit-code'.
Feb 05 15:11:34 starserver systemd[1]: Failed to start "wmbusmeters service on rtlsdr1".

```

weetmuts, 2020-02-05:
You cannot start the daemon on rtlsdr1 since that device should not exist....

How about if you empty out your wmbusmeters.log.
Unplug the dongle and re-insert it.
Email me the wmbusmeters.log and the /var/log/syslog.

I can help you more efficiently over email.

weetmuts, 2020-02-06:
I found a bug in the auto detection of wmbus dongles. A fix is now pushed. 

Please test again to see if this was related.

weetmuts, 2020-02-17:
Problem solved. The binary /usr/bin/rtl_sdr did not exist, it was located in /usr/local/bin
The daemon expects the binaries to be /usr/bin/rtl_sdr and /usr/bin/rtl_wbus.

An informative error message has been added to wmbusmeters that will 
let the user know that the binaries are missing when trying to run as a daemon.

closed 2020-02-17 by weetmuts