wmbusmeters/wmbusmeters #71: Feature request: Decrypt Mode 7

cinemarene, 2020-02-05:
Can you please implement Mode 7 decryption? I need it for an Energy Meter.

Here is one Hex request i got: 5A445A141920000002378C2028900F002C25EE240000C0AD312C29B1478072439115005A140102280030071003047A8FA108BC95FB6920336E483B33212C76886AC3D39D334FC0AF1CC9D3DFCA7750D4E1D0A1FEB0AEAD1012F02C5D0000

If it helps, the SmartHome solution ioBroker has it already implemented:
https://github.com/ISFH/ioBroker.wmbus/blob/4fce68c57f915840c787f1d3df1194aae5afc56e/lib/wmbus_decoder.js

Here is the documentation of it: https://oms-group.org/fileadmin/files/download4all/specification/Vol2/4.0.2/OMS-Spec_Vol2_Primary_v402.pdf
how to do.
Page 52 => 9.2.3

weetmuts, 2020-02-05:
Actually it is already implemented. :-) Please test!

cinemarene, 2020-02-05:
Thanks for the quick feedback. I did not know that because I tested it yesterday and it did not work. I couldn't find anything in the code either. Maybe I did something wrong. I will test it again today and tell you about it.

weetmuts, 2020-02-05:
Email me the key and I can test your telegram.

I am sure that there are things that still can go wrong.

closed 2020-02-05 by weetmuts